Koenkk/zigbee2mqtt #23889: [New device support]: Marmitek Link SE is wrong detected as Tuya TS0207_water_leak_detector

maschulze, 2024-09-07:
### Link

https://marmitek.com/en-de/products/link-se-zigbee-repeater-mesh-network-usb-power-zigbee-3-1

### Database entry

{"id":20,"type":"Router","ieeeAddr":"0x0cae5ffffe8736e9","nwkAddr":51795,"manufId":4098,"manufName":"_TZ3000_wmlc9p9z","powerSource":"Mains (single phase)","modelId":"TS0207","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":8,"inClusterList":[0,10,3],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0005","65506":22,"modelId":"TS0207","manufacturerName":"_TZ3000_wmlc9p9z","stackVersion":0,"dateCode":"","appVersion":64}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0xbc026efffe2a4336","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1725715299561}

### Zigbee2MQTT version

1.40.1

### Comments

The device is detected (HomeAssistant) as

Device type                  Router
Zigbee Model              TS0207
Zigbee Manufacturer   _TZ3000_wmlc9p9z
Description                  Water leak detector
Support status             Supported
IEEE Address                0x0cae5ffffe8736e9
Network address         0xCA53 / 51795
Manufacturer              Tuya
Model                         TS0207_water_leak_detector

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0207'],
    model: 'TS0207',
    vendor: '_TZ3000_wmlc9p9z',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

The device should be detected as router.

bcrosnier, 2024-09-09:
Got the same product and identification. Is this a new repeater or something?

Possibly related to https://github.com/Koenkk/zigbee2mqtt/issues/19402 or https://github.com/Koenkk/zigbee2mqtt/issues/19196 .

Note that the device *is* detected as a router (on the "About" tab, the "Device Type" is "Router"), it just doesn't expose anything of value as far as I'm aware.

maschulze, 2024-09-10:
The device is detected as a router but it exposes properties like a water leak detector:
[image]
[image]

May be it is relatet to #19196 but with a new **Zigbee Manufacturer**:

_TZ3000_wmlc9p9z

Robnex, 2024-09-28:
Some in my ZigbeeDocker:

[image]
[image]

duk64, 2024-10-25:
The same device, the same issue. It works fine!

hamphh73, 2024-10-26:
Same device, same issue. But in your pictures the device is online, mine is disabled and I and I didn't know how to activate it. In my log are following error messages:
`z2m: Failed to configure 'Repeater OG', attempt 1 (Error: ZCL command 0x0cae5ffffe8726b2/1 genPowerCfg.configReport([{"attribute":"batteryPercentageRemaining","minimumReportInterval":3600,"maximumReportInterval":65000,"reportableChange":0}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 13063 - 1 - 25 - 1 - 7 after 10000ms) at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:59:23) at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:640:47) at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:672:28) at /app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:422:20 at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20) at ZStackAdapter.sendZclFrameToEndpoint (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:420:16) at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:82:20) at Endpoint.sendRequest (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:325:20) at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:951:28) at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:688:9))`

maschulze, 2024-11-01:
Fixed with #7901

closed 2025-01-09 by github-actions[bot]