Koenkk/zigbee2mqtt #23895: z2m: Failed to configure  Develco (Frient 94430, Cavius 2103 Notes)

Haraade, 2024-09-07:
### What happened?

[2024-09-07 17:15:33] error: 	z2m: Failed to configure 'Fire Reporter 1', attempt 1 (Error: Bind 0x0015bc0031021028/35 ssIasWd from '0xe0798dfffed3c026/1' failed (Delivery failed for {"profileId":0,"clusterId":33,"sourceEndpoint":0,"destinationEndpoint":0,"options":4416,"groupId":0,"sequence":116}) at EmberOneWaitress.deliveryFailedFor (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:101:31) at EmberAdapter.onMessageSent (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:480:42) at Ezsp.emit (node:events:517:28) at Ezsp.ezspMessageSentHandler (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:4982:14) at Ezsp.callbackDispatch (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:738:26) at Ezsp.onAshFrame (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:398:22) at UartAsh.emit (node:events:517:28) at Immediate.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/ember/uart/ash.ts:1169:45) at processImmediate (node:internal/timers:476:21)) [2024-09-07 17:17:03] error: 	z2m: Failed to configure 'Fire Reporter 2', attempt 1 (Error: Bind 0x0015bc0031021d68/35 genBasic from '0xe0798dfffed3c026/1' failed ([ZDO] Failed response for '6774' cluster '32801' Status 'TABLE_FULL'.) at EmberOneWaitress.resolveZDO (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:139:35) at EmberAdapter.onZDOResponse (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:556:30) at Ezsp.emit (node:events:517:28) at Ezsp.ezspIncomingMessageHandler (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:5249:18) at Ezsp.callbackDispatch (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:782:26) at Ezsp.onAshFrame (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:398:22) at UartAsh.emit (node:events:517:28) at Immediate.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/ember/uart/ash.ts:1169:45) at processImmediate (node:internal/timers:476:21)) [2024-09-07 17:18:08] error: 	z2m: Failed to configure 'Fire Reporter 3', attempt 1 (Error: Bind 0x0015bc0031020ca3/35 genBasic from '0xe0798dfffed3c026/1' failed ([ZDO] Failed response for '29262' cluster '32801' Status 'TABLE_FULL'.) at EmberOneWaitress.resolveZDO (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:139:35) at EmberAdapter.onZDOResponse (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:556:30) at Ezsp.emit (node:events:517:28) at Ezsp.ezspIncomingMessageHandler (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:5249:18) at Ezsp.callbackDispatch (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:782:26) at Ezsp.onAshFrame (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:398:22) at UartAsh.emit (node:events:517:28) at Immediate.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/ember/uart/ash.ts:1169:45) at processImmediate (node:internal/timers:476:21)) [2024-09-07 17:20:34] error: 	z2m: Failed to configure 'Fire Reporter 1', attempt 2 (Error: Bind 0x0015bc0031021028/35 genBasic from '0xe0798dfffed3c026/1' failed ([ZDO] Failed response for '56331' cluster '32801' Status 'TABLE_FULL'.) at EmberOneWaitress.resolveZDO (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:139:35) at EmberAdapter.onZDOResponse (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:556:30) at Ezsp.emit (node:events:517:28) at Ezsp.ezspIncomingMessageHandler (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:5249:18) at Ezsp.callbackDispatch (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:782:26) at Ezsp.onAshFrame (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:398:22) at UartAsh.emit (node:events:517:28) at Immediate.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/ember/uart/ash.ts:1169:45) at processImmediate (node:internal/timers:476:21)) [2024-09-07 17:22:03] error: 	z2m: Failed to configure 'Fire Reporter 2', attempt 2 (Error: Bind 0x0015bc0031021d68/35 genBasic from '0xe0798dfffed3c026/1' failed ([ZDO] Failed response for '6774' cluster '32801' Status 'TABLE_FULL'.) at EmberOneWaitress.resolveZDO (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:139:35) at EmberAdapter.onZDOResponse (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:556:30) at Ezsp.emit (node:events:517:28) at Ezsp.ezspIncomingMessageHandler (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:5249:18) at Ezsp.callbackDispatch (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:782:26) at Ezsp.onAshFrame (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:398:22) at UartAsh.emit (node:events:517:28) at Immediate.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/ember/uart/ash.ts:1169:45) at processImmediate (node:internal/timers:476:21)) [2024-09-07 17:23:13] error: 	z2m: Failed to configure 'Fire Reporter 3', attempt 2 (Error: Bind 0x0015bc0031020ca3/35 genBasic from '0xe0798dfffed3c026/1' failed ([ZDO] Failed response for '29262' cluster '32801' Status 'TABLE_FULL'.) at EmberOneWaitress.resolveZDO (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:139:35) at EmberAdapter.onZDOResponse (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:556:30) at Ezsp.emit (node:events:517:28) at Ezsp.ezspIncomingMessageHandler (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:5249:18) at Ezsp.callbackDispatch (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:782:26) at Ezsp.onAshFrame (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:398:22) at UartAsh.emit (node:events:517:28) at Immediate.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/ember/uart/ash.ts:1169:45) at processImmediate (node:internal/timers:476:21)) [2024-09-07 17:25:28] error: 	z2m: Failed to configure 'Fire Reporter 1', attempt 3 (Error: Bind 0x0015bc0031021028/35 genBasic from '0xe0798dfffed3c026/1' failed ([ZDO] Failed response for '56331' cluster '32801' Status 'TABLE_FULL'.) at EmberOneWaitress.resolveZDO (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:139:35) at EmberAdapter.onZDOResponse (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:556:30) at Ezsp.emit (node:events:517:28) at Ezsp.ezspIncomingMessageHandler (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:5249:18) at Ezsp.callbackDispatch (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:782:26) at Ezsp.onAshFrame (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:398:22) at UartAsh.emit (node:events:517:28) at Immediate.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/ember/uart/ash.ts:1169:45) at processImmediate (node:internal/timers:476:21)) [2024-09-07 17:27:05] error: 	z2m: Failed to configure 'Fire Reporter 2', attempt 3 (Error: Bind 0x0015bc0031021d68/35 genBasic from '0xe0798dfffed3c026/1' failed ([ZDO] Failed response for '6774' cluster '32801' Status 'TABLE_FULL'.) at EmberOneWaitress.resolveZDO (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:139:35) at EmberAdapter.onZDOResponse (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:556:30) at Ezsp.emit (node:events:517:28) at Ezsp.ezspIncomingMessageHandler (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:5249:18) at Ezsp.callbackDispatch (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:782:26) at Ezsp.onAshFrame (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:398:22) at UartAsh.emit (node:events:517:28) at Immediate.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/ember/uart/ash.ts:1169:45) at processImmediate (node:internal/timers:476:21)) [2024-09-07 17:28:12] error: 	z2m: Failed to configure 'Fire Reporter 3', attempt 3 (Error: Bind 0x0015bc0031020ca3/35 genBasic from '0xe0798dfffed3c026/1' failed ([ZDO] Failed response for '29262' cluster '32801' Status 'TABLE_FULL'.) at EmberOneWaitress.resolveZDO (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:139:35) at EmberAdapter.onZDOResponse (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:556:30) at Ezsp.emit (node:events:517:28) at E [... cut ...] .callbackDispatch (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:782:26) at Ezsp.onAshFrame (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:398:22) at UartAsh.emit (node:events:517:28) at Immediate.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/ember/uart/ash.ts:1169:45) at processImmediate (node:internal/timers:476:21))

### Zigbee2MQTT version

1.40.1

### Adapter firmware version

Gecko 4.4.4

### Adapter

Ember

### Setup

Add-on Homeassistant

Koenkk, 2024-09-08:
Could you check if the issue is fixed with the following [external converter](https://gist.github.com/Koenkk/24f66d186243c46855c1092ed3714960): 

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, 
- in the frontend, check if the description of this device now ends with `CUSTOM` (this indicates the external converter has been loaded correctly)
- re-pair the device
- check if issue is fixed and that all functions of the device work.

Haraade, 2024-09-09:
Thank you for your feedback. 
This installation is far away from where I am now, so I'll have to do this when I get there. I reach the installation but can't re-pair the device now. 

Condition of a fire detector: 

{ "battery": 80, 
"battery_low": false, 
"linkquality": 216, 
"max_duration": 5, 
"restore_reports": true, 
"smoke": false, 
"supervision_reports": true, 
"test": false, 
"update": { "installed_version": 262152, 
"latest_version": 262152, "state": "idle" }, 
"Voltage": 2900 } 

Seems to work, but some of the sensors have unknown condition.

[image]

closed 2024-09-10 by Koenkk