Koenkk/zigbee2mqtt #23899: Errors in log with Ember and release 1.40.1

dan73800, 2024-09-08:
### What happened?

I have ugraded to 1.40.1 with my Sonoff zbdongle-e ember 7.4. firmware, and I have noticed errors in logs when starting

`> zigbee2mqtt@1.40.1 start
> node index.js

Starting Zigbee2MQTT without watchdog.
[2024-09-08 10:05:24] [31merror[39m: 	zh:ember:uart:ash: Received frame with CRC error
[2024-09-08 10:05:24] [31merror[39m: 	zh:ember:uart:ash: Received ERROR from adapter while connecting, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
[2024-09-08 10:05:43] [31merror[39m: 	z2m: Failed to configure 'Lixee Garage', attempt 1 (Error: ZCL command 0x00158d0005d2a667/1 haElectricalMeasurement.configReport([{"attribute":"totalApparentPower","minimumReportInterval":60,"maximumReportInterval":900,"reportableChange":1}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"manufacturerCode":null,"writeUndiv":false}) failed ({"target":31565,"apsFrame":{"profileId":260,"clusterId":2820,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":11},"zclSequence":25,"commandIdentifier":7} timed out after 10000ms)
    at Object.start (/var/www/html/plugins/z2m/resources/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:198:23)
    at EmberOneWaitress.startWaitingFor (/var/www/html/plugins/z2m/resources/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:222:50)
    at /var/www/html/plugins/z2m/resources/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:2526:55
    at runNextTicks (node:internal/process/task_queues:60:5)
    at processImmediate (node:internal/timers:447:9)
    at Queue.execute (/var/www/html/plugins/z2m/resources/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20)
    at Request.send (/var/www/html/plugins/z2m/resources/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:82:20)
    at Endpoint.zclCommand (/var/www/html/plugins/z2m/resources/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:911:28)
    at Endpoint.configureReporting (/var/www/html/plugins/z2m/resources/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:648:9)
    at async Promise.allSettled (index 20))`

### What did you expect to happen?

Should I ignore theses warnings or can I do anything to remove them ?

### How to reproduce it (minimal and precise)

Just restart Z2M

### Zigbee2MQTT version

1.40.1

### Adapter firmware version

7.4.3 [GA]

### Adapter

Sonoff ZDongle-E EmberZNet

### Setup

JEEDOM

### Debug log

> zigbee2mqtt@1.40.1 start
> node index.js

Starting Zigbee2MQTT without watchdog.
[2024-09-08 10:05:24] [31merror[39m: 	zh:ember:uart:ash: Received frame with CRC error
[2024-09-08 10:05:24] [31merror[39m: 	zh:ember:uart:ash: Received ERROR from adapter while connecting, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
[2024-09-08 10:05:43] [31merror[39m: 	z2m: Failed to configure 'Lixee Garage', attempt 1 (Error: ZCL command 0x00158d0005d2a667/1 haElectricalMeasurement.configReport([{"attribute":"totalApparentPower","minimumReportInterval":60,"maximumReportInterval":900,"reportableChange":1}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"manufacturerCode":null,"writeUndiv":false}) failed ({"target":31565,"apsFrame":{"profileId":260,"clusterId":2820,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":11},"zclSequence":25,"commandIdentifier":7} timed out after 10000ms)
    at Object.start (/var/www/html/plugins/z2m/resources/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:198:23)
    at EmberOneWaitress.startWaitingFor (/var/www/html/plugins/z2m/resources/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:222:50)
    at /var/www/html/plugins/z2m/resources/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:2526:55
    at runNextTicks (node:internal/process/task_queues:60:5)
    at processImmediate (node:internal/timers:447:9)
    at Queue.execute (/var/www/html/plugins/z2m/resources/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20)
    at Request.send (/var/www/html/plugins/z2m/resources/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:82:20)
    at Endpoint.zclCommand (/var/www/html/plugins/z2m/resources/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:911:28)
    at Endpoint.configureReporting (/var/www/html/plugins/z2m/resources/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:648:9)
    at async Promise.allSettled (index 20))
[2024-09-08 10:06:02] [31merror[39m: 	z2m: No converter available for 'current' ("")
[2024-09-08 10:06:03] [31merror[39m: 	z2m: No converter available for 'power' ("")
[2024-09-08 10:06:03] [31merror[39m: 	z2m: No converter available for 'energy' ("")
[2024-09-08 10:06:03] [31merror[39m: 	z2m: No converter available for 'voltage' ("")
[2024-09-08 10:06:03] [31merror[39m: 	z2m: No converter available for 'get' 'child_lock' ()
[... 233 lines cut ...]
[2024-09-08 10:10:07] [31merror[39m: 	z2m: No converter available for 'rms_voltage' ("")
[2024-09-08 10:10:07] [31merror[39m: 	z2m: No converter available for 'rms_voltage_ph_b' ("")
[2024-09-08 10:10:07] [31merror[39m: 	z2m: No converter available for 'rms_voltage_ph_c' ("")
[2024-09-08 10:10:07] [31merror[39m: 	z2m: No converter available for 'software_revision' ("")
[2024-09-08 10:10:07] [31merror[39m: 	z2m: No converter available for 'warn_d_p_s' ("")
[2024-09-08 10:10:07] [31merror[39m: 	z2m: No converter available for 'schedule_h_p_h_c' ("")
[2024-09-08 10:10:07] [31merror[39m: 	z2m: No converter available for 'rms_current' ("")
[2024-09-08 10:10:07] [31merror[39m: 	z2m: No converter available for 'rms_current_max' ("")
[2024-09-08 10:10:07] [31merror[39m: 	z2m: No converter available for 'available_power' ("")
[2024-09-08 10:10:07] [31merror[39m: 	z2m: No converter available for 'MOTDETAT' ("")
[2024-09-08 10:10:07] [31merror[39m: 	z2m: No converter available for 'current_tarif' ("")
[2024-09-08 10:10:07] [31merror[39m: 	z2m: No converter available for 'active_register_tier_delivered' ("")
[2024-09-08 10:10:07] [31merror[39m: 	z2m: No converter available for 'get' 'child_lock' ()
[2024-09-08 10:10:07] [31merror[39m: 	z2m: No converter available for 'power' ("")
[2024-09-08 10:10:07] [31merror[39m: 	z2m: No converter available for 'energy' ("")
[2024-09-08 10:10:07] [31merror[39m: 	z2m: No converter available for 'current' ("")
[2024-09-08 10:10:07] [31merror[39m: 	z2m: No converter available for 'voltage' ("")
[2024-09-08 10:10:07] [31merror[39m: 	z2m: No converter available for 'current' ("")
[2024-09-08 10:10:07] [31merror[39m: 	z2m: No converter available for 'power' ("")
[2024-09-08 10:10:07] [31merror[39m: 	z2m: No converter available for 'energy' ("")
[2024-09-08 10:10:07] [31merror[39m: 	z2m: No converter available for 'voltage' ("")
[2024-09-08 10:10:07] [31merror[39m: 	z2m: No converter available for 'current' ("")
[2024-09-08 10:10:07] [31merror[39m: 	z2m: No converter available for 'power' ("")
[2024-09-08 10:10:07] [31merror[39m: 	z2m: No converter available for 'energy' ("")
[2024-09-08 10:10:07] [31merror[39m: 	z2m: No converter available for 'voltage' ("")

Nerivec, 2024-09-14:
CRC errors and/or ACK timeouts right on startup can happen (especially in virtualized environments). There is a retry mechanism in place to bypass these.

As for the LiXee failing to configure, that's another story...

dan73800, 2024-09-22:
What is this mecanism you are talking about ?
For the Lixee, I have also noticed that some informations where no more read. I had to force delete of Lixee in Z2M , restart, and re-inclure it to make it work as expected.

closed 2024-12-19 by github-actions[bot]