Koenkk/zigbee2mqtt #23900: TS011F_plug_3 electrical measurements aren't pulled anymore

lied, 2024-09-08:
### What happened?

Since 2024-08-29 all of my TS011F_plug_3 plugs doesn't update their electrical measurements. I guess I did the Zigbee2MQTT v1.39.1 on my Home Assistant at that time - not sure exactly, as of course my backup is also broken for that time. ;-)

Now I'm running the latest **z2m v1.40.1**

I have 2 Tuya TS011F_plug_1 which report their electric power and energy just fine. Other vendors also report fine.

My 15 TS011F_plug_3 all doesn't report actual electrical measurements. They can be switched on/off and the state of the switch is reported. Some of my TS011F_plug_3 report a power consumption of e.g. 10 Watts, but it's stale and doesn't update.
I tried setting pulling for every 30 seconds, but that didn't change anything.

Manual pulling via Dev console is working:
[image]

### What did you expect to happen?

The default pulling every 60 seconds of my Tuya TS011F_plug_3 should work again.

### How to reproduce it (minimal and precise)

Wait for 60 seconds to trigger the auto update pulling of the plugs.

### Zigbee2MQTT version

1.40.1

### Adapter firmware version

20220219

### Adapter

Electrolama zig-a-zig-ah! (zzh!)

### Setup

Addon on HomeAssistant running in VM

### Debug log

```
[2024-09-08 11:02:44] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Plug_OG_Arbeitszimmer_PC', payload '{"child_lock":"UNLOCK","current":0,"energy":553.51,"indicator_mode":"off/on","last_seen":"2024-09-08T11:02:44+02:00","linkquality":222,"power":78,"power_outage_memory":"on","state":"ON","update":{"installed_version":69,"latest_version":69,"state":"idle"},"update_available":false,"voltage":241}'
```

Koenkk, 2024-09-09:
Could you provide the debug log starting z2m + 5 minutes after that?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

MiguelCosta, 2024-09-09:
@lied check this comment to mitigated the issue:

https://github.com/Koenkk/zigbee2mqtt/issues/23887#issuecomment-2339144924

lied, 2024-09-09:
> @lied <https://github.com/lied> check this comment to mitigated the issue:
>
> #23887 (comment)

That didn't work for me, already tried.

MiguelCosta, 2024-09-09:
@lied open your configuration.yaml file I check if you have any value like this:

[image]

lied, 2024-09-12:
> Could you provide the debug log starting z2m + 5 minutes after that?
> 
> See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

Hi,

sorry for the late reply. I had some other, non home automation related topics to take care of.
I migrated some of my ~25 Tuya Plugs to ZHA, to have power metrics.

To verify the test I learned in a brand new plug, which wasn't attached to any zigbee network at all. And now, while I was testing and trying to verify the issue, I cannot reproduce it

Here are the logs, which look completly normal now:
```
info 2024-09-12 19:52:11z2m:mqtt: MQTT publish: topic 'homeassistant/binary_sensor/0xf082c0fffebbbd51/update_available/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/plug_og_arbeitszimmer_debug/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xf082c0fffebbbd51"],"manufacturer":"Tuya","model":"Smart plug (with power monitoring by polling) (TS011F_plug_3)","name":"plug_og_arbeitszimmer_debug","via_device":"zigbee2mqtt_bridge_0x00124b00219ecd32"},"device_class":"update","enabled_by_default":false,"entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/plug_og_arbeitszimmer_debug","name":null,"object_id":"plug_og_arbeitszimmer_debug_update_available","origin":{"name":"Zigbee2MQTT","sw":"1.40.1","url":"https://www.zigbee2mqtt.io"},"payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/plug_og_arbeitszimmer_debug","unique_id":"0xf082c0fffebbbd51_update_available_zigbee2mqtt","value_template":"{{ value_json['update']['state'] == \"available\" }}"}'
info 2024-09-12 19:52:11z2m:mqtt: MQTT publish: topic 'homeassistant/update/0xf082c0fffebbbd51/update/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/plug_og_arbeitszimmer_debug/availability"}],"availability_mode":"all","command_topic":"zigbee2mqtt/bridge/request/device/ota_update/update","device":{"identifiers":["zigbee2mqtt_0xf082c0fffebbbd51"],"manufacturer":"Tuya","model":"Smart plug (with power monitoring by polling) (TS011F_plug_3)","name":"plug_og_arbeitszimmer_debug","via_device":"zigbee2mqtt_bridge_0x00124b00219ecd32"},"device_class":"firmware","entity_category":"config","entity_picture":"https://github.com/Koenkk/zigbee2mqtt/raw/master/images/logo.png","json_attributes_template":"{\"in_progress\": {{ iif(value_json['update']['state'] == 'updating', 'true', 'false') }} }","json_attributes_topic":"zigbee2mqtt/plug_og_arbeitszimmer_debug","latest_version_template":"{{ value_json['update']['latest_version'] }}","latest_version_topic":"zigbee2mqtt/plug_og_arbeitszimmer_debug","name":null,"object_id":"plug_og_arbeitszimmer_debug","origin":{"name":"Zigbee2MQTT","sw":"1.40.1","url":"https://www.zigbee2mqtt.io"},"payload_install":"{\"id\": \"0xf082c0fffebbbd51\"}","state_topic":"zigbee2mqtt/plug_og_arbeitszimmer_debug","unique_id":"0xf082c0fffebbbd51_update_zigbee2mqtt","value_template":"{{ value_json['update']['installed_version'] }}"}'
info 2024-09-12 19:52:33z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/plug_og_arbeitszimmer_debug', payload '{"child_lock":"UNLOCK","current":0,"elapsed":54593,"energy":0,"indicator_mode":"off/on","last_seen":"2024-09-12T19:52:33+02:00","linkquality":178,"power":0,"power_outage_memory":"off","state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":235}'
info 2024-09-12 19:53:01z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/plug_og_arbeitszimmer_debug', payload '{"child_lock":"UNLOCK","current":0,"elapsed":28727,"energy":0,"indicator_mode":"off/on","last_seen":"2024-09-12T19:53:01+02:00","linkquality":182,"power":0,"power_outage_memory":"off","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":235}'
info 2024-09-12 19:53:11z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/plug_og_arbeitszimmer_debug', payload '{"child_lock":"UNLOCK","current":0,"elapsed":9689,"energy":0,"indicator_mode":"off/on","last_seen":"2024-09-12T19:53:11+02:00","linkquality":138,"power":0,"power_outage_memory":"off","state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":235}'
info 2024-09-12 19:53:29z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/plug_og_arbeitszimmer_debug', payload '{"child_lock":"UNLOCK","current":0,"elapsed":17689,"energy":0,"indicator_mode":"off/on","last_seen":"2024-09-12T19:53:29+02:00","linkquality":142,"power":0,"power_outage_memory":"off","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":235}'
info 2024-09-12 19:53:31z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/plug_og_arbeitszimmer_debug', payload '{"child_lock":"UNLOCK","current":0,"energy":0,"indicator_mode":"off/on","last_seen":"2024-09-12T19:53:31+02:00","linkquality":131,"power":0,"power_outage_memory":"off","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":235}'
info 2024-09-12 19:53:33z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/plug_og_arbeitszimmer_debug', payload '{"child_lock":"UNLOCK","current":0.13,"elapsed":3955,"energy":0,"indicator_mode":"off/on","last_seen":"2024-09-12T19:53:33+02:00","linkquality":138,"power":15,"power_outage_memory":"off","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":2 [... cut ...]
info 2024-09-12 19:57:33z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/plug_og_arbeitszimmer_debug', payload '{"child_lock":"UNLOCK","current":0.15,"elapsed":57309,"energy":0,"indicator_mode":"off/on","last_seen":"2024-09-12T19:57:33+02:00","linkquality":149,"power":17,"power_outage_memory":"off","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":236}'
```

Could it be the case that my zigbee network (with about 170 zigbee devices) was just overloaded? That wouldn't explain why I could manually read the current value though.

As I can't reproduce this, it doesn't make sense to keep this issue it open.

closed 2024-09-23 by lied