Koenkk/zigbee2mqtt #23901: Z2M fails to start after: Updated Zigbee2MQTT to version 1.40.1

afkdk, 2024-09-08:
### What happened?

I simply upgraded and Z2M devices were not available.
When I check, the X2M dis not start - CPU/... flashes briefly when started in HA but stops <2 sec.
On a black screen, HA displays 502: Bad Gateway.

NOTHING was changed - and Z2M had worked flawlessly up till this upgrade - it was (AFAIK set up by default values by HA/Z2M directives) - and MQTT is working for other integrations...

And no hints in the reference web page seems relevant 🥴

HA has of course been restarted plenty of times - just-in-case something needed a restart...

-----------

enabled: false
master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
options: "-d -d"
log: false

-----------

[2024-09-08 11:33:41] debug: 	zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":"HIDDEN","panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[25]},"serialPort":{"path":"tcp://10.100.100.223:6638"},"adapter":{"disableLED":false,"concurrent":null,"delay":null,"transmitPower":9},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json"}'
[2024-09-08 11:33:41] info: 	zh:zstack:znp: Opening TCP socket with 10.100.100.223:6638
[2024-09-08 11:33:44] info: 	zh:zstack:znp: Socket error
[2024-09-08 11:33:44] error: 	z2m: Error while starting zigbee-herdsman
[2024-09-08 11:33:44] error: 	z2m: Failed to start zigbee
[2024-09-08 11:33:44] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-09-08 11:33:44] error: 	z2m: Exiting...
[2024-09-08 11:33:44] error: 	z2m: Error: Error while opening socket
    at Socket.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:178:24)
    at Socket.emit (node:events:529:35)
    at emitErrorNT (node:internal/streams/destroy:151:8)
    at emitErrorCloseNT (node:internal/streams/destroy:116:3)
    at processTicksAndRejections (node:internal/process/task_queues:82:21)
/app/node_modules/winston/node_modules/readable-stream/lib/_stream_writable.js:264
  var er = new ERR_STREAM_WRITE_AFTER_END();
           ^
Error: write after end
    at writeAfterEnd (/app/node_modules/winston/node_modules/readable-stream/lib/_stream_writable.js:264:12)
    at DerivedLogger.Writable.write (/app/node_modules/winston/node_modules/readable-stream/lib/_stream_writable.js:300:21)
    at DerivedLogger.log (/app/node_modules/winston/lib/winston/logger.js:231:12)
    at Logger.log (/app/lib/util/logger.ts:189:25)
    at Logger.info (/app/lib/util/logger.ts:202:14)
    at Znp.onPortClose (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:108:16)
    at Object.onceWrapper (node:events:632:26)
    at Socket.emit (node:events:517:28)
    at TCP.<anonymous> (node:net:350:12)
[11:33:44] INFO: Preparing to start...
[11:33:44] INFO: Socat not enabled
[11:33:45] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2024-09-08 11:33:46] info: 	z2m: Logging to console, file (filename: log.log)
[2024-09-08 11:33:46] debug: 	z2m: Loaded state from file /config/zigbee2mqtt/state.json
[2024-09-08 11:33:46] info: 	z2m: Starting Zigbee2MQTT version 1.40.1 (commit #unknown)
[2024-09-08 11:33:46] info: 	z2m: Starting zigbee-herdsman (0.57.3)
[2024-09-08 11:33:46] debug: 	z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":6754,\"extendedPanID\":[221,221,221,221,221,221,221,221],\"channelList\":[25],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/config/zigbee2mqtt/database.db\",\"databaseBackupPath\":\"/config/zigbee2mqtt/database.db.backup\",\"backupPath\":\"/config/zigbee2mqtt/coordinator_backup.json\",\"serialPort\":{\"path\":\"tcp://10.100.100.223:6638\"},\"adapter\":{\"concurrent\":null,\"delay\":null,\"disableLED\":false,\"transmitPower\":9}}"'
[2024-09-08 11:33:46] debug: 	zh:adapter: Failed to validate path: 'Error: spawn udevadm ENOENT'
[2024-09-08 11:33:46] debug: 	zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":"HIDDEN","panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[25]},"serialPort":{"path":"tcp://10.100.100.223:6638"},"adapter":{"disableLED":false,"concurrent":null,"delay":null,"transmitPower":9},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json"}'
[2024-09-08 11:33:46] info: 	zh:zstack:znp: Opening TCP socket with 10.100.100.223:6638
[2024-09-08 11:33:47] info: 	zh:zstack:znp: Socket error
[2024-09-08 11:33:47] error: 	z2m: Error while starting zigbee-herdsman
[2024-09-08 11:33:47] error: 	z2m: Failed to start zigbee
[2024-09-08 11:33:47] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-09-08 11:33:47] error: 	z2m: Exiting...
[2024-09-08 11:33:47] error: 	z2m: Error: Error while opening socket
    at Socket.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:178:24)
    at Socket.emit (node:events:529:35)
    at emitErrorNT (node:internal/streams/destroy:151:8)
    at emitErrorCloseNT (node:internal/streams/destroy:116:3)
[... 153 lines cut ...]
[2024-09-08 11:55:57] info: 	zh:zstack:znp: Socket error
[2024-09-08 11:55:57] error: 	z2m: Error while starting zigbee-herdsman
[2024-09-08 11:55:57] error: 	z2m: Failed to start zigbee
[2024-09-08 11:55:57] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-09-08 11:55:57] error: 	z2m: Exiting...
[2024-09-08 11:55:57] error: 	z2m: Error: Error while opening socket
    at Socket.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:178:24)
    at Socket.emit (node:events:529:35)
    at emitErrorNT (node:internal/streams/destroy:151:8)
    at emitErrorCloseNT (node:internal/streams/destroy:116:3)
    at processTicksAndRejections (node:internal/process/task_queues:82:21)

/app/node_modules/winston/node_modules/readable-stream/lib/_stream_writable.js:264
  var er = new ERR_STREAM_WRITE_AFTER_END();
           ^
Error: write after end
    at writeAfterEnd (/app/node_modules/winston/node_modules/readable-stream/lib/_stream_writable.js:264:12)
    at DerivedLogger.Writable.write (/app/node_modules/winston/node_modules/readable-stream/lib/_stream_writable.js:300:21)
    at DerivedLogger.log (/app/node_modules/winston/lib/winston/logger.js:231:12)
    at Logger.log (/app/lib/util/logger.ts:189:25)
    at Logger.info (/app/lib/util/logger.ts:202:14)
    at Znp.onPortClose (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:108:16)
    at Object.onceWrapper (node:events:632:26)
    at Socket.emit (node:events:517:28)
    at TCP.<anonymous> (node:net:350:12)

sven-borden, 2024-09-08:
Similar issue on my side :(

harvindhillon, 2024-09-08:
Did you try resetting the zigbee dongle itself?

Shutdown z2m. Restart dongle, once restarted see if it can be reachable. Then start z2m

From the error it looks like it cant connect to the dongle.

afkdk, 2024-09-08:
Thanks for your time investigating this problem 👍

I had already tried this but I have now retried it by disconnecting the PoE
unit, waiting 5+ sec, and then repowering...  And then Start Z2M again - No
change, still red light after a short time...

afkdk, 2024-09-08:
> If you are using a slzb-06 coordinator via POE, in its settings, make sure
> that you only have Ethernet mode enabled. I had this same problem yesterday
> and it was solved. Now I have a problem with panId.

I am using the TubesZB PoE coordinator - and I have just installed a HA
PyCLI tool for gathering some more information for possibly i
identifikation of the bug culprit.
The nice color coding of the text has vanished by the copy-paste so I may
try some better media if necessary for interpretation...

----------

s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service banner: starting

-----------------------------------------------------------
 Add-on: TubesZB Zigpy-CLI Tools
 TubesZB Zigpy-CLI Toolsadd-on
-----------------------------------------------------------
 Add-on version: 0.2.3.4
 You are running the latest version of this add-on.
 System: Home Assistant OS 13.1  (amd64 / qemux86-64)
 Home Assistant Core: 2024.9.1
 Home Assistant Supervisor: 2024.08.0
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service banner successfully started
s6-rc: info: service zigpy-cli: starting
[01:47:39] INFO: Directoty Exists
[01:47:40] INFO: Running zigpy - info on /dev/ttyS3
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/serial/serialposix.py",
line 398, in _reconfigure_port
    orig_attr = termios.tcgetattr(self.fd)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^
termios.error: (5, 'I/O error')

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/zigpy", line 8, in <module>
    sys.exit(cli())
             ^^^^^
  File "/usr/local/lib/python3.11/site-packages/click/core.py", line
1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/click/core.py", line
1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/click/core.py", line
1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/click/core.py", line
1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/click/core.py", line
1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/click/core.py", line
783, in invoke
[... 36 lines cut ...]
  File "/usr/local/lib/python3.11/site-packages/serial/__init__.py",
line 90, in serial_for_url
    instance.open()
  File "/usr/local/lib/python3.11/site-packages/serial/serialposix.py",
line 332, in open
    self._reconfigure_port(force_update=True)
  File "/usr/local/lib/python3.11/site-packages/serial/serialposix.py",
line 401, in _reconfigure_port
    raise SerialException("Could not configure port: {}".format(msg))
serial.serialutil.SerialException: Could not configure port: (5, 'I/O error')
[01:47:40] INFO: zigpy-cli-up script exited with code 1
s6-rc: warning: unable to start service zigpy-cli: command exited 1
s6-rc: info: service banner: stopping
/run/s6/basedir/scripts/rc.init: warning: s6-rc failed to properly
bring all the services up! Check your logs (in
/run/uncaught-logs/current if you have in-container logging) for more
information.
/run/s6/basedir/scripts/rc.init: fatal: stopping the container.
s6-rc: info: service banner successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

afkdk, 2024-09-08:
BTW., I just noticed that it refered to HA 2024.9.1 - it also failed in the
last 2024.8.x version.
i just updated to version 2024.9.1 in case that this Z2M problem may
originate in some unintentional bindings to this newer HA version...

jkerekes99, 2024-09-09:
Reporting same issue as the OP.
Updated 1.40.1-1 and now Z2M won’t start.

Also using a TubeZB.

Haven’t had a single issue for years until today.

harvindhillon, 2024-09-09:
Did you try reaching out to TubeZ ?

Seems like the serial connection to the zigbee radio is opening

afkdk, 2024-09-09:
Link til Tubes coordinator integration issue:

[https://github.com/tube0013/tube_gateways/issues/205](url)

Irix0, 2024-09-12:
I think my issue #23950 is related to this.

afkdk, 2024-09-13:
Agree - some similar start-up issues seems to appear in the list...

Most surprisingly, the Z2M sprung into action the other day and worked for 1+ day - so GREAT to see it all back LIVE and working 😃   I worked flawlessly without any errors - and I could get to the TubesZB web interface again...
I happened much to my surprise after a "total power cut", i.e., I have some PV battery installed so all power had to be cut to the house - and surprisingly, the Z2M was back up and running - despite that I have tried shutdown and power-cuts of both the coordinator and HA.  Likely, the startup has in some way been "just right" - like [https://github.com/Koenkk/zigbee2mqtt/issues/23950](url) and others have reported...

But as expected (as M.in computer science I know that computers work deterministic so eventually I expected to fail again), this Morning at 7AM, it suddenly failed off again 😥

Now it is back to the "failed state" - somehow the Z2M <-> TubesZB interaction locks each other up and only be conincidens, it may work again (for a short time...)

Later today, the power has to be cut again and I may be lucky that circumstances are right again so the start timing is _just _right__ so I have some more time of enjoying the great integration _as long as it works_ 🤞

afkdk, 2024-10-10:
Now it seems that I almost with certainty can circumvent this problem by 1) switching off-wait-on PoE switch (powering TubesZB ESP), 2) restart Z2M by restarting HA.
So my guess is some intricate tcp timing/queue issue locks up the network between HA and the TubesZP dongle (the remaining ports on switch and other HA network comm work fine when this happens.)

closed 2024-10-10 by afkdk