Koenkk/zigbee2mqtt #23907: Kwikset 914

kuradi, 2024-09-08:
### What happened?

recently purchased 2 kwikset 914 locks (https://www.kwikset.com/products/detail/914-smartcode-traditional-electronic-deadbolt-with-zigbee-technology?variant=914trl-zb-15). today I screwed them to the door and also tried to connect them to my zigbee network.
both come online nicely but...
the first of them comes as Kwikset 66492-001 (https://www.zigbee2mqtt.io/devices/66492-001.html#kwikset-66492-001) and the second as Weiser 9GED18000-009 (https://www.zigbee2mqtt.io/devices/66492-001.html#kwikset-66492-001). io/devices/9GED18000-009.html#weiser-9ged18000-009)
both work, but for example, the first one always set the lock_state to "not_fully_locked" and the battery is always 50% (despite the fact that I just put new batteries). and except for the fact that, in my opinion, both should be attached rather than the Kwikset 99140-031 (https://www.zigbee2mqtt.io/devices/99140-031.html)
help!

### Zigbee2MQTT version

1.40.0

### Adapter firmware version

20230507

### Adapter

tubeszb poe

### Setup

separate pi

kuradi, 2024-09-20:
any help here?

kuradi, 2024-10-05:
@Koenkk how much do you have to sponsor to get a faster response? :)

Koenkk, 2024-10-06:
I think Weiser and Kiwkset are the same branch? Could you share the data/database.db entry of both devices?

kuradi, 2024-10-07:
> I think Weiser and Kiwkset are the same branch? Could you share the data/database.db entry of both devices?

here we go:

{"id":59,"type":"EndDevice","ieeeAddr":"0x002446fffd062e27","nwkAddr":1053,"manufId":4242,"manufName":"Kwikset","powerSource":"Battery","modelId":"SMARTCODE_DEADBOLT_10","epList":[2],"endpoints":{"2":{"profId":260,"epId":2,"devId":10,"inClusterList":[0,1,3,4,5,9,32,257,1026,2821,64957],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{}},"closuresDoorLock":{"attributes":{"lockState":1,"numOfPinUsersSupported":30}},"genPollCtrl":{"attributes":{"checkinInterval":0}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":180}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0x00124b0022981e2b","endpointID":1},{"cluster":257,"type":"endpoint","deviceIeeeAddress":"0x00124b0022981e2b","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0022981e2b","endpointID":1}],"configuredReportings":[{"cluster":257,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0},{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":65000,"repChange":0}],"meta":{}}},"appVersion":52,"stackVersion":2,"hwVersion":69,"zclVersion":1,"interviewCompleted":true,"meta":{"configured":1507761327},"lastSeen":1728286531776,"checkinInterval":0}

{"id":60,"type":"EndDevice","ieeeAddr":"0x002446fffd066be9","nwkAddr":46591,"manufId":4242,"manufName":"Kwikset","powerSource":"Battery","modelId":"Smartcode","epList":[2],"endpoints":{"2":{"profId":260,"epId":2,"devId":10,"inClusterList":[0,1,3,4,5,9,32,257,1026,2821,64957],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"Smartcode","manufacturerName":"Kwikset","powerSource":3,"zclVersion":1,"appVersion":52,"stackVersion":2,"hwVersion":5}},"genPollCtrl":{"attributes":{"checkinInterval":0}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":100}},"closuresDoorLock":{"attributes":{"lockState":0,"numOfPinUsersSupported":100}}},"binds":[],"configuredReportings":[{"cluster":257,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0},{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":65000,"repChange":0}],"meta":{}}},"appVersion":52,"stackVersion":2,"hwVersion":5,"zclVersion":1,"interviewCompleted":true,"meta":{"configured":1507761327},"lastSeen":1728270992673,"checkinInterval":0}

kuradi, 2024-10-10:
but maybe i should ask to add new model at all? when im looking at ebay when im bought them then i see this info:

Brand Kwikset
MPN 99140-014
UPC 0883351544863
Model SmartCode 914
eBay Product ID (ePID) 15051856492

Koenkk, 2024-10-10:
All models are already supported, the picture is wrong because the modelID is used by multiple devices (probably the Zigbee module inside is the same)

kuradi, 2024-10-11:
> All models are already supported, the picture is wrong because the modelID is used by multiple devices (probably the Zigbee module inside is the same)

as you see my model is 99140-014 but when i looking at https://www.zigbee2mqtt.io/supported-devices/#v=Kwikset there is 99140-002, 99140-031 etc but not 99140-014... or they are the same by your oppinion?

kuradi, 2024-10-12:
> They are all slightly different, but the 99140-014 seems to be the same as 66492-001 and 9GED18000-009

but still... i have bought 2 identical locks from here https://www.ebay.com/p/15051856492?iid=195943048893
I would not like to believe that I have been sent 2 different ones, visually they are also identical, the packaging was identical, etc., etc. I also sent a follow-up inquiry to the seller just in case. there is no answer yet.
but I rather think that maybe this model has small differences compared to those that 2zm already supports. how can i make sure it is/isn't?
in addition, the lock that comes out as Kwikset 66492-001 shows such a picture in the ui
[image]

kuradi, 2024-10-12:
@Koenkk i just unscrewed covers and as i see it those boards are also identical.
[image]

Koenkk, 2024-10-13:
The software on them seems to be different (hwVersion 65 vs 5, see the database.db entries)

kuradi, 2024-10-13:
> 66492-001

is it something i can update?

kuradi, 2024-10-13:
> The software on them seems to be different (hwVersion 69 vs 5, see the database.db entries)

is it something i can update?

kuradi, 2024-10-16:
got reply from kwikset: "The firmware is in the pre-installed Zigbee chip. The Zigbee chips are not sold separately; there is no way to determine the authenticity, firmware levels, or functional condition of the chipset you are inserting into your lock. Also, we cannot guarantee that you can upgrade your lock by adding a Z-Wave or Zigbee card. Although the chip physically fits in the lock, it may or may not have any Z-Wave or Zigbee function; worse yet, it can damage the lock if there's any anomaly on the daughterboard. Lastly, by doing so, you are changing the lock, which voids the warranty on the lock. Since the Zigbee system is third-party, you'll need to contact the manufacturer for clarification on their system's available software updates and the available locations for their application. "

closed 2024-12-24 by github-actions[bot]