Koenkk/zigbee2mqtt #23909: Frontend doesn't update state on GiEX (GX02) when valve closes or when hard button is pressed.

dohco-gh, 2024-09-08:
### What happened?

Changed status to ON in frontend, after 11 seconds (set in frontend) the water valve closes but frontend doesn't change status to OFF.
The frontend doesn't change status if hard button on water valve is pressed either.
[image]

### What did you expect to happen?

The frontend should update status to off after valve is closed.
The frontend should update status if hard button is pressed.

### How to reproduce it (minimal and precise)

Set valve to close after X seconds and wait for valve to close.

### Zigbee2MQTT version

1.40.1

### Adapter firmware version

7.4.3 [GA]

### Adapter

EmberZNet

### Setup

Add-on to supervised HomeAssistant

### Debug log

debug 2024-09-08 12:26:26zh:controller:endpoint: ZCL command 0xa4c138d8589c546e/1 manuSpecificTuya.defaultRsp({"cmdId":2,"statusCode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"transactionSequenceNumber":170,"writeUndiv":false})
debug 2024-09-08 12:26:26z2m: Received Zigbee message from '0xa4c138d8589c546e', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[45,45,58,45,45,58,45,45],"type":"Buffer"},"datatype":3,"dp":102}],"seq":14080}' from endpoint 1 with groupID 0
info 2024-09-08 12:26:26z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c138d8589c546e', payload '{"battery":80,"cycle_irrigation_interval":5,"cycle_irrigation_num_times":0,"irrigation_end_time":"--:--:--","irrigation_start_time":"12:26:06","irrigation_target":11,"last_irrigation_duration":"00:00:11","linkquality":184,"mode":"duration","state":"OFF","water_consumed":0}'
debug 2024-09-08 12:26:26zh:controller:endpoint: ZCL command 0xa4c138d8589c546e/1 manuSpecificTuya.defaultRsp({"cmdId":2,"statusCode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"transactionSequenceNumber":171,"writeUndiv":false})
debug 2024-09-08 12:26:26z2m: Received Zigbee message from '0xa4c138d8589c546e', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":111}],"seq":14336}' from endpoint 1 with groupID 0
info 2024-09-08 12:26:26z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c138d8589c546e', payload '{"battery":80,"cycle_irrigation_interval":5,"cycle_irrigation_num_times":0,"irrigation_end_time":"--:--:--","irrigation_start_time":"12:26:06","irrigation_target":11,"last_irrigation_duration":"00:00:11","linkquality":184,"mode":"duration","state":"OFF","water_consumed":0}'
debug 2024-09-08 12:26:26zh:controller:endpoint: ZCL command 0xa4c138d8589c546e/1 manuSpecificTuya.defaultRsp({"cmdId":2,"statusCode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"transactionSequenceNumber":172,"writeUndiv":false})
debug 2024-09-08 12:26:26z2m: Received Zigbee message from '0xa4c138d8589c546e', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":2}],"seq":14592}' from endpoint 1 with groupID 0
info 2024-09-08 12:26:26z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c138d8589c546e', payload '{"battery":80,"cycle_irrigation_interval":5,"cycle_irrigation_num_times":0,"irrigation_end_time":"--:--:--","irrigation_start_time":"12:26:06","irrigation_target":11,"last_irrigation_duration":"00:00:11","linkquality":180,"mode":"duration","state":"ON","water_consumed":0}'
debug 2024-09-08 12:26:26zh:controller:endpoint: ZCL command 0xa4c138d8589c546e/1 manuSpecificTuya.defaultRsp({"cmdId":2,"statusCode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"transactionSequenceNumber":173,"writeUndiv":false})
debug 2024-09-08 12:26:26z2m: Received Zigbee message from '0xa4c138d8589c546e', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[49,50,58,50,54,58,50,55],"type":"Buffer"},"datatype":3,"dp":101}],"seq":14848}' from endpoint 1 with groupID 0
info 2024-09-08 12:26:26z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c138d8589c546e', payload '{"battery":80,"cycle_irrigation_interval":5,"cycle_irrigation_num_times":0,"irrigation_end_time":"--:--:--","irrigation_start_time":"12:26:27","irrigation_target":11,"last_irrigation_duration":"00:00:11","linkquality":188,"mode":"duration","state":"ON","water_consumed":0}'
debug 2024-09-08 12:26:44zh:ember:ezsp: ezspIncomingRouteRecordHandler(): callback called with: [source=5504], [sourceEui=0xa4c138d8589c546e], [lastHopLqi=132], [lastHopRssi=-67], [relayCount=3], [relayList=38560,3652,6657]
debug 2024-09-08 12:26:45zh:controller:endpoint: ZCL command 0xa4c138d8589c546e/1 manuSpecificTuya.defaultRsp({"cmdId":2,"statusCode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"transactionSequenceNumber":174,"writeUndiv":false})
debug 2024-09-08 12:26:45z2m: Received Zigbee message from '0xa4c138d8589c546e', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":106}],"seq":15104}' from endpoint 1 with groupID 0
debug 2024-09-08 12:26:45zh:controller:endpoint: ZCL command 0xa4c138d8589c546e/1 manuSpecificTuya.defaultRsp({"cmdId":2,"statusCode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"transactionSequenceNumber":175,"writeUndiv":false})
debug 2024-09-08 12:26:45z2m: Received Zigbee message from '0xa4c138d8589c546e', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":103}],"seq":15360}' from endpoint 1 with groupID 0
info 2024-09-08 12:26:45z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c138d8589c546e', payload '{"battery":80,"cycle_irrigation_interval":5,"cycle_irrigation_num_times":0,"irrigation_end_time":"--:--:--","irrigation_start_time":"12:26:27","irrigation_target":11,"last_irrigation_duration":"00:00:11","linkquality":132,"mode":"duration","state":"ON","water_consumed":0}'
debug 2024-09-08 12:26:45zh:controller:endpoint: ZCL command 0xa4c138d8589c546e/1 manuSpecificTuya.defaultRsp({"cmdId":2,"statusCode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"transactionSequenceNumber":176,"writeUndiv":false})
debug 2024-09-08 12:26:45z2m: Received Zigbee message from '0xa4c138d8589c546e', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":2}],"seq":15616}' from endpoint 1 with groupID 0
info 2024-09-08 12:26:45z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c138d8589c546e', payload '{"battery":80,"cycle_irrigation_interval":5,"cycle_irrigation_num_times":0,"irrigation_end_time":"--:--:--","irrigation_start_time":"12:26:27","irrigation_target":11,"last_irrigation_duration":"00:00:11","linkquality":136,"mode":"duration","state":"OFF","water_consumed":0}'
debug 2024-09-08 12:26:45zh:controller:endpoint: ZCL command 0xa4c138d8589c546e/1 manuSpecificTuya.defaultRsp({"cmdId":2,"statusCode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"transactionSequenceNumber":177,"writeUndiv":false})
debug 2024-09-08 12:26:45z2m: Received Zigbee message from '0xa4c138d8589c546e', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[49,50,58,50,54,58,52,53],"type":"Buffer"},"datatype":3,"dp":102}],"seq":15872}' from endpoint 1 with groupID 0
info 2024-09-08 12:26:45z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c138d8589c546e', payload '{"battery":80,"cycle_irrigation_interval":5,"cycle_irrigation_num_times":0,"irrigation_end_time":"12:26:45","irrigation_start_time":"12:26:27","irrigation_target":11,"last_irrigation_duration":"00:00:11","linkquality":132,"mode":"duration","state":"OFF","water_consumed":0}'
debug 2024-09-08 12:26:45zh:controller:endpoint: ZCL command 0xa4c138d8589c546e/1 manuSpecificTuya.defaultRsp({"cmdId":2,"statusCode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"transactionSequenceNumber":178, [... cut ...]
debug 2024-09-08 12:26:45z2m: Received Zigbee message from '0xa4c138d8589c546e', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[48,48,58,48,48,58,49,56,44,48],"type":"Buffer"},"datatype":3,"dp":114}],"seq":16128}' from endpoint 1 with groupID 0
info 2024-09-08 12:26:45z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c138d8589c546e', payload '{"battery":80,"cycle_irrigation_interval":5,"cycle_irrigation_num_times":0,"irrigation_end_time":"12:26:45","irrigation_start_time":"12:26:27","irrigation_target":11,"last_irrigation_duration":"00:00:18","linkquality":132,"mode":"duration","state":"OFF","water_consumed":0}'

closed 2024-12-19 by github-actions[bot]

Corrado-possieri, 2025-01-30:
I have the same problem. How did you manage to solve it?

dohco-gh, 2025-01-30:
It didnt got solved. Issue was closed without action.