Koenkk/zigbee2mqtt #23910: [Feature request]: Availability feature improvements for physically unpowered devices

Azimthz, 2024-09-08:
### Is your feature request related to a problem? Please describe

Here is my use case, I have some ceiling hue ligths that are operated with a traditional switch, meaning that when I turn them off they totally lose power.  I need very low interval for the availability feature to capture as soon as possible that they are off. 
The problem is that at the moment will keep stressing the zigbee network with pings, pulluting the logs and the UI, when I already know they will not become availabe again until I turn then on again. 

### Describe the solution you'd like

A possible solution is to create a flag at device level, that when that flag is enable, the availability feature will operate as normal until it gets n consecutive failed pings (to avoid false negatives), from that point on, it will not ping the device again until the device actively broadcast to the network again from it's end. 
This way the ping stressing on the network will be greatly reduced while maintaining a quick response time when the device becomes unavailable.

### Describe alternatives you've considered

At the moment and as workaround I have a relatively high avaliablity interval, much higher than desired to avoid creating stress on the network.

### Additional context

Thank you very much for this wonderful piece of sofware really appreciated!

IvanHACS, 2024-12-18:
Backing up this Feature Request.
E.g. Bulbs for additional lighting, which do not act as routers/range extenders in the network, do not need to be powered on all the time and pinged on regular basis.

LaurentChardin, 2024-12-29:
Quick question : i know this is not answering your issue but did you consider to actually change your traditionnal switch and actually rewire it to always be powered, and just use the Hue Switch Module instead ? 

Having your Hue Bulbs on will make your zigbee network more reliables as they are routers. and the Hue Switch module is quite simple to install and will usually fit behind your traditional switch (that you can use with it)

toine512, 2025-01-02:
Maybe not pinging at all is a bit extreme. Let's say 2 pings a day so that an hypothetical obscure device never annoucing itself coming online would still be detected at some point even without an active state change.

Azimthz, 2025-03-01:
> Quick question : i know this is not answering your issue but did you consider to actually change your traditionnal switch and actually rewire it to always be powered, and just use the Hue Switch [...]
 Hi @LaurentChardin thanks for the advice, appreciate it. In my current configuration some of the light are already operating this way (not through a Hue Switch module but a Shelly but same concept). Not all of them needed to be powered on constantly and I tought that been able to have that flexibitliy will be a great addition to this addon. Again thanks for the advice.

Nerivec, 2025-04-09:
The above-mentioned pull request should alleviate some of the issues described here. However, I have to make a few notes:

> Bulbs for additional lighting, which do not act as routers/range extenders in the network

As soon as a "router"-type device (which is pretty much any device not battery-powered) is online and paired to a network, it _will_ act as a router. Nearby devices may automatically switch to it depending on link quality. That means, after taking that router offline, said devices will be forced to find new parents, and routes throughout the network will shuffle.
This, at best, can cause delays, at worst, it can seriously affect network reliability (depending on how often that scenario happens).
"Router"-type devices in ZigBee **should not be taken offline regularly** once paired to a network.
While issues may not be visible in strong networks (adapt quickly), they will be more pronounced when lesser quality devices (firmware) are involved.

Nerivec, 2025-05-08:
@Azimthz any feedback on the new availability improvements from the May release? 😉

Vebryn, 2025-08-07:
Huge thanks ! I will check this feature soon !