Koenkk/zigbee2mqtt #23911: Incorrect device support for Innr RB 178 T

magicus, 2024-09-08:
### What happened?

My new Innr RB 178 T is detected by Z2M, but the device description is wrong, so it looks like I could control the RGB color of the lamp. In fact, I can only change the color temperature of white light.

This is the external description from Z2M.

```
const {deviceEndpoints, light} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['RB 178 T'],
    model: 'RB 178 T',
    vendor: 'innr',
    description: 'Automatically generated definition',
    extend: [deviceEndpoints({"endpoints":{"1":1,"2":2}}), light({"colorTemp":{"range":[200,454]}})],
    meta: {"multiEndpoint":true},
};

module.exports = definition;
```

While I can't say I fully understand this, my guess is that [line 345 in innr.js](https://github.com/Koenkk/zigbee-herdsman-converters/blob/b1234398df5bb37d6e7d1bfed7a77ca13aedcaf7/src/devices/innr.ts#L345) should read:
```
  extend: [light({colorTemp: {range: [153, 555]}, turnsOffAtBrightness1: true})],
```
instead of (as now)
```
  extend: [light({colorTemp: {range: [153, 555]}, color: {applyRedFix: true}, turnsOffAtBrightness1: true})],
```
  

### What did you expect to happen?

No RGB controls showing, since they are not supported by the light.

### How to reproduce it (minimal and precise)

Connect a, Innr RB 178 T to Z2M. Notice that RGB controls show up. Verify that they do not change the light's color.

### Zigbee2MQTT version

1.40.1

### Adapter firmware version

0x26580700

### Adapter

ConBee II

### Setup

Hass OS add-on

Koenkk, 2024-09-09:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-09-09 by Koenkk

magicus, 2024-09-09:
Nice, thank you for the quick fix! ❤️