Koenkk/zigbee2mqtt #23913: [New device support]: Tuya TS0225 Human presence sensor

hhalibo, 2024-09-09:
### Link

http://www.leapmmw.com/

### Database entry

{"id":2,"type":"Router","ieeeAddr":"0xa4c1385406d5e8f6","nwkAddr":4964,"manufId":4417,"manufName":"_TZ321C_fkzihax8","powerSource":"Mains (single phase)","modelId":"TS0225","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,4,5,1280],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"65534":0,"modelId":"TS0225","manufacturerName":"_TZ321C_fkzihax8","powerSource":1,"zclVersion":3,"appVersion":65,"stackVersion":1,"hwVersion":1,"dateCode":""}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b0024cb3e7a","zoneState":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":1,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1401274594},"lastSeen":1725843436923}

### Zigbee2MQTT version

1.35.3

### Comments

It is a 5.8GHz radar Human presence sensor.

### External definition

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const {Enum} = require('zigbee-herdsman-converters/lib/tuya');

const tzDatapoints = {  
    ...tuya.tz.datapoints,  
    key: Array.isArray(tuya.tz.datapoints.key)   
        ? [...tuya.tz.datapoints.key, 'debug_mode', 'debug_distance', 'debug_countdown', 'scene_preset', 'distance_report_mode']  
        : ['debug_mode', 'debug_distance', 'debug_countdown', 'scene_preset', 'distance_report_mode']  
};  

const definition = [
    {
	fingerprint: tuya.fingerprint('TS0225', ['_TZ321C_fkzihax8','_TZ321C_4slreunp']),
	model: 'MTD085-ZB',        
	vendor: 'LeapMMW',
	whiteLabel: [
		tuya.whitelabel('LeapMMW', 'MTD095-ZB', 'Human presence sensor', ['_TZ321C_4slreunp']),
	],
    description: 'Human presence sensor',
    fromZigbee: [tuya.fz.datapoints, fz.ias_occupancy_alarm_1, fz.ias_occupancy_alarm_1_report],
    toZigbee: [tuya.tz.datapoints, tzDatapoints],
	configure: async (device, coordinatorEndpoint) => {
		await tuya.configureMagicPacket(device, coordinatorEndpoint);
	},
	exposes: [			
		e.occupancy(), e.illuminance_lux(),
		e.numeric('target_distance', ea.STATE).withDescription('Distance to target').withUnit('m'),
		e.numeric('radar_sensitivity', ea.STATE_SET).withValueMin(10).withValueMax(100).withValueStep(10).withUnit('%')
			.withDescription('Detection threshold for the strength of object energy'),
		e.numeric('detection_range', ea.STATE_SET).withValueMin(0).withValueMax(8).withValueStep(0.1).withUnit('m')
			.withDescription('Maximum distance detected by the sensor'),
		e.numeric('shield_range', ea.STATE_SET).withValueMin(0).withValueMax(8).withValueStep(0.1).withUnit('m')
			.withDescription('Nearest distance detected by the sensor'),
		e.numeric('entry_sensitivity', ea.STATE_SET).withValueMin(10).withValueMax(100).withValueStep(10).withUnit('%')
			.withDescription('Sensitivity threshold triggered for the first time when the target enters the detection range'),
		e.numeric('entry_distance', ea.STATE_SET).withValueMin(0).withValueMax(8).withValueStep(0.1).withUnit('m')
			.withDescription('Dectection distance when unoccupied'),
		e.numeric('entry_filter_time', ea.STATE_SET).withValueMin(0).withValueMax(0.5).withValueStep(0.05).withUnit('s')
			.withDescription('Sensitivity threshold triggered for the first time when the target enters the detection range '),
		e.numeric('departure_delay', ea.STATE_SET).withValueMin(5).withValueMax(7200).withValueStep(1).withUnit('s')
			.withDescription('Confirmation time after the target disappears'),
		e.numeric('block_time', ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(0.1).withUnit('s')
			.withDescription('Time for the target to be detected again after switching from manned(occupy) to unmanned(unoccupy) mode'),
		e.enum('status_indication', ea.STATE_SET, ['OFF', 'ON'])
			.withDescription('Indicator light will turn on when human presence is detected'),
		e.enum('sensor', ea.STATE_SET, ['on', 'occupied', 'unoccupied'])
			.withDescription(`The radar sensor can be set in three states: on, occupied and unoccupied. For example, if set to occupied, ` +
				`it will continue to maintain presence regardless of whether someone is present or not. If set to unoccupied, the unoccupied ` +
				`state will be maintained permanently.`),			
		e.enum('scene_preset', ea.STATE_SET, ['Custom', 'Toilet', 'Kitchen', 'Hallway', 'Bedroom', 'Livingroom', 'Meetingroom', 'Factory default'])
			.withDescription(`Presets`),
		e.enum('distance_report_mode', ea.STATE_SET, ['Normal', 'Occupancy detection'])
			.withDescription('Indicator light will turn on when human presence is detected'),			
		e.enum('debug_mode', ea.STATE_SET, ['OFF', 'ON'])
			.withDescription(`In debug mode, radar will report more information, can be used to identify interference`),
		e.numeric('debug_distance', ea.STATE).withDescription('Real time distance to target').withUnit('m'),
		e.numeric('debug_countdown', ea.STATE).withDescription('Time before the target disappears').withUnit('s'),

	],
	meta: {            
		tuyaSendCommand: 'sendData',
		tuyaDatapoints: [
//			[1, 'presence', tuya.valueConverter.trueFalse1],
			[101, 'entry_sensitivity', tuya.valueConverter.raw],
			[102, 'entry_distance', tuya.valueConverter.divideBy100],
			[103, 'departure_delay', tuya.valueConverter.raw],
			[104, 'entry_filter_time', tuya.valueConverter.divideBy100],
			[105, 'block_time', tuya.valueConverter.divideBy10],
			[107, 'illuminance_lux', tuya.valueConverter.divideBy10],
			[108, 'debug_mode', tuya.valueConverterBasic.lookup({'OFF': tuya.enum(0), 'ON': tuya.enum(1)})],
			[109, 'debug_distance', tuya.valueConverter.divideBy100],
			[110, 'debug_countdown', tuya.valueConverter.raw],
			[111, 'scene_preset', tuya.valueConverterBasic.lookup({'Custom': tuya.enum(0), 'Toilet': tuya.enum(1), 'Kitchen': tuya.enum(2), 'Hallway': tuya.enum(3), 'Bedroom': tuya.enum(4), 'Livingroom': tuya.enum(5), 'Meetingroom': tuya.enum(6), 'Default': tuya.enum(7)})],
			[112, 'sensor', tuya.valueConverterBasic.lookup({
				'on': tuya.enum(0), 'occupied': tuya.enum(1), 'unoccupied': tuya.enum(2)})],
			[113, 'cline', tuya.valueConverter.raw],
			[114, 'status_indication', tuya.valueConverterBasic.lookup({'OFF': tuya.enum(0), 'ON': tuya.enum(1)})],
			[115, 'radar_sensitivity', tuya.valueConverter.raw],
			[116, 'shield_range', tuya.valueConverter.divideBy100],
			[117, 'detection_range', tuya.valueConverter.divideBy100],
			[118, 'equipment_status', tuya.valueConverter.raw],
			[119, 'target_distance', tuya.valueConverter.divideBy100],
			[120, 'distance_report_mode', tuya.valueConverterBasic.lookup({'Normal': tuya.enum(0), 'Occupancy detection': tuya.enum(1)})],
		
		],
	},
}];

module.exports = definition;
```

### What does/doesn't work with the external definition?

I wrote the external converter，it works well。

Koenkk, 2024-09-09:
Could you provide a PR by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/tuya.ts)?

Trrw, 2024-11-11:
Hello, I have the same device _TZ321C_fkzihax8 how can I help so that the device is supported?

closed 2025-01-02 by Koenkk

MaxW95, 2025-01-19:
I bought a similar device of same model "TS0225", but its currently not supported:

    modelZigbee:TS0225
    type:Router
    nwk:336
    manuf id:4417
    manufacturer:_TZ321C_fkzihax8
    power:Mains (single phase)
    app version:65
    hard version:1
    zcl version:3
    stack version:1
    date code:
    interviewed:false
    configured:false

    modelZigbee:TS0225
    type:Router
    nwk:336
    manuf id:4417
    manufacturer:_TZ321C_fkzihax8
    power:Mains (single phase)
    app version:65
    hard version:1
    zcl version:3
    stack version:1
    date code:
    interviewed:false
    configured:false
    endpoint:1
    profile:260
    input clusters:genBasic (0)
    genGroups (4)
    genScenes (5)
    ssIasZone (1280)
    output clusters:genTime (10)
    genOta (25)
    endpoint:242

thomphi2, 2025-01-28:
Same device, not supported in ZHA.