Koenkk/zigbee2mqtt #23916: Not supported device and issue with adding it - Aeotec Range Extender Zi - Model ZGA001-C15 (EU)

1mfaasj, 2024-09-09:
### What happened?

I can't add the device (as mentioned in the title) in Zigbee2MQTT because its not supported.
Also when I click on the device, development console,  I get an error directly "Hello, you've found a bug. Congratulations!"
but that's also blocking me from adding support for this device... I'm stuck there.

### What did you expect to happen?

a supported device in Zigbee2MQTT or at least the possibility to request support for the device.

### How to reproduce it (minimal and precise)

I put the device into paring mode and Zigbee2MQTT starts interviewing it but stops everytime at a 'not supported' status.

### Zigbee2MQTT version

1.40.1

### Adapter firmware version

20221226

### Adapter

SLZB-06

### Setup

docker

### Debug log

there are no logs except a message that the device is not supported.
see the attachments as attached.
[image]
[image]
[image]

When I click on "Click here, and save this file" nothing happens though!

chris-1243, 2024-09-11:
Could you add this converter to Zigbee2mqtt. Check here how to: 
https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_2-creating-the-external-definition

When you external converter is added, restart Z2M and see if it is recognized correctly.

```
const {} = require('zigbee-herdsman-converters/lib/modernExtend');
// Add the lines below
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['ZGA001'],
    model: 'ZGA001',
    vendor: 'AEOTEC',
    description: 'Aeotec Range Extender Zi (EU)',
    extend: [],
    fromZigbee: [fz.linkquality_from_basic],
    toZigbee: [],
    exposes: [],
};

module.exports = definition;
```

flamme-demon, 2024-09-11:
> Could you add this converter to Zigbee2mqtt. Check here how to: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_2-creating-the-external-definition
> 
> When your external converter is added, restart Z2M and see if it is recognized correctly.
> 
> ```
> const {} = require('zigbee-herdsman-converters/lib/modernExtend');
> // Add the lines below
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const reporting = require('zigbee-herdsman-converters/lib/reporting');
> const ota = require('zigbee-herdsman-converters/lib/ota');
> const utils = require('zigbee-herdsman-converters/lib/utils');
> const globalStore = require('zigbee-herdsman-converters/lib/store');
> const e = exposes.presets;
> const ea = exposes.access;
> 
> const definition = {
>     zigbeeModel: ['ZGA001'],
>     model: 'ZGA001',
>     vendor: 'AEOTEC',
>     description: 'Aeotec Range Extender Zi-C15 (EU)',
>     extend: [],
>     fromZigbee: [fz.linkquality_from_basic],
>     toZigbee: [],
>     exposes: [],
> };
> 
> module.exports = definition;
> ```

Hello Chris,

Thanks for your quick feedback.
When we go to the developer console on this component, we arrive on the bug page from where the issues create

chris-1243, 2024-09-11:
In the folder where `configuration.yaml` is located, create a file and let's call `ZGA001.js`. Copy/paste the converter provided in my previous comment in the new file.

Stop Z2M;

Modify `configuration.yaml` like this:
```
advanced:
  log_level: debug
external_converters:
  - ZGA001.js
``` 

Start Z2M and see if the device is recognized. Please provide the output of the debug log

flamme-demon, 2024-09-11:
> In the folder where `configuration.yaml` is located, create a file and let's call `ZGA001.js`. Copy/paste the converter provided in my previous comment in the new file.
>
> Stop Z2M;
>
> Modify [...]

Hello,

Thank you
Here is the log you requested.
The device is: 0x8c65a3fffe525907

The ZGA001 file doesn't seem to be doing the job. However, according to the log, it is loaded correctly.

[log(1).log](https://github.com/user-attachments/files/16966754/log.1.log)

chris-1243, 2024-09-11:
Could you try to remove and re-pair your device ? The file is loaded correctly. It seems the device was not correctly interviewed

https://www.zigbee2mqtt.io/guide/faq/#interview-fails

flamme-demon, 2024-09-11:
> Could you try to remove and re-pair your device ? The file is loaded correctly. It seems the device was not correctly interviewed
> 
> https://www.zigbee2mqtt.io/guide/faq/#interview-fails

I did it 4x before sending you the logs. I didn't want to send you something badly done.

chris-1243, 2024-09-11:
While pairing, have you followed this ?

https://www.zigbee2mqtt.io/devices/WG001.html

Your device should be supported

1mfaasj, 2024-09-12:
Somehow I've got it working, not sure what the fix now actually was.
What I did: Put the device VERY close to the coordinator and then pressed the button ONCE.
The manual describes that you should hold the button for 10 seconds but that didn't bring the device into paring mode.
But at the same time I also added the converter like @chris-1243 described and restarted MQTT afterwards.
Now it's visible in the gui:
[image]

closed 2024-09-12 by 1mfaasj

chris-1243, 2024-09-12:
@1mfaasj 

What worked is to pair zhe device really close to the coordinator.
My converter is not working as the zigbee model is not the ine published by the device. A log was needed to have this information.

Glad to see you have it paired correctly and working

flamme-demon, 2024-09-12:
> @1mfaasj
> 
> What worked is to pair zhe device really close to the coordinator. My converter is not working as the zigbee model is not the ine published by the device. A log was needed to have this information.
> 
> Glad to see you have it paired correctly and working

Ah yes, you have to pair it really close. It was on the other side of the room (about 3m and I couldn't reach it).
Here's the log
[log(2).log](https://github.com/user-attachments/files/16976396/log.2.log)

I'm going to test without the JS too.

chris-1243, 2024-09-12:
You have done the solution. The pairing must be done really close to the coordinator as explained in the documentation.

I thought first your device was not recognised but it was a pairing problem after reading your logs. It is well supported and working as far as I can see.

flamme-demon, 2024-09-12:
I need your JS file for this to work.
Would you like me to take a log without the JS file and be very close to the coordinator?

chris-1243, 2024-09-12:
Yes please. Stop Z2M, remove my converter from `configuration.yaml` and start Z2M. Your device should be still working. Try to pair it close to the coordinator as well.

Everything should work. My converter does not help as the device ID I set is not the good one

flamme-demon, 2024-09-12:
> Yes please. Stop Z2M, remove my converter from `configuration.yaml` and start Z2M. Your device should be still working. Try to pair it close to the coordinator as well.
> 
> Everything should work. My converter does not help as the device ID I set is not the good one

Without the JS file, it always remains in error.
[log1(1).log](https://github.com/user-attachments/files/16984056/log1.1.log)
[image]

Whereas his position is next to the coordinator and it's a fixed position.
The last log and the 3rd try.
Reboot Z2M + pairing.
As far as I'm concerned, there's a problem.

chris-1243, 2024-09-12:
When pairing was ok

```
[2024-09-12 10:58:31] debug: 	zh:controller: Received payload: clusterID=0, address=37993, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=255, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":13,"commandIdentifier":1},"payload":[{"attrId":5,"status":0,"dataType":66,"attrData":"WG001-Z01"}],"command":{"ID":1,"name":"readRsp","parameters":[{"name":"attrId","type":33},{"name":"status","type":32},{"name":"dataType","type":32,"conditions":[{"type":"statusEquals","value":0}]},{"name":"attrData","type":1000,"conditions":[{"type":"statusEquals","value":0}]}]}}
[2024-09-12 10:58:31] debug: 	zh:controller:endpoint: ZCL command 0x8c65a3fffe525907/1 genBasic.defaultRsp({"cmdId":1,"statusCode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"reservedBits":0,"transactionSequenceNumber":13,"writeUndiv":false})
[2024-09-12 10:58:31] debug: 	z2m: Received Zigbee message from '0x8c65a3fffe525907', type 'readResponse', cluster 'genBasic', data '{"modelId":"WG001-Z01"}' from endpoint 1 with groupID 0
[2024-09-12 10:58:31] debug: 	z2m: Skipping message, still interviewing
[2024-09-12 10:58:31] debug: 	zh:ember: ~~~> [ZCL to=37993 apsFrame={"profileId":260,"clusterId":0,"sourceEndpoint":1,"destinationEndpoint":1,"options":4352,"groupId":0,"sequence":0} header={"frameControl":{"reservedBits":0,"frameType":0,"direction":1,"disableDefaultResponse":true,"manufacturerSpecific":false},"transactionSequenceNumber":13,"commandIdentifier":11}]
``` 

Check the model ID given by your device. It says **WG001-Z01**. My converter should not work as I filled ZGA001 as model ID. From your last log:
```
[2024-09-12 20:23:19] error: 	zh:controller: Interview failed for '0x8c65a3fffe525907 with error 'Error: Interview failed because can not get node descriptor ('0x8c65a3fffe525907')'
[2024-09-12 20:23:19] error: 	z2m: Failed to interview '0x8c65a3fffe525907', device has not successfully been paired
``` 

Your coordinator is not able to talk to your device and get the data required to pair it correctly. In such scenario, my converter can not help. Try to pair at about 50cm from the coordinator and completely reset the device first and follow this procedure

https://aeotec.freshdesk.com/support/solutions/articles/6000248296-aeotec-range-extender-zi-user-guide-

And if your device does not want to pair, let's try my converter again. 

Désolé, plus que ça à distance c'est commpliqué mais pas impossible

flamme-demon, 2024-09-12:
@chris-1243 Thank you, you're absolutely right. After the 4ᵉ unsuccessful attempts, the 5ᵉ succeeded perfectly without the file.
I can use it normally.
Do you know if the blue button would be deactivatable?

I'm curious to know how you can find out what actions are possible on a piece of equipment.

chris-1243, 2024-09-12:
The user manual will help you. Check the link in my last post.

For your light, you may toggle it. Everything you need is in it.

[image]

Tanguille, 2026-01-14:
Just had the same issue and can confirm putting it close to the coordinator and pressing the button once does the job