Koenkk/zigbee2mqtt #23917: [Feature request]: Z2m groups indicate child entities to home assistant

thargy, 2024-09-09:
### Is your feature request related to a problem? Please describe

ZigBee groups are fantastic for improving the efficiency of the ZigBee network and controlling lights simultaneously. However, a ZigBee group is indistinguishable from an individual device as far as a home assistant is concerned. This makes it impossible to automatically determine if controlling a group will change any other devices and vice versa.

### Describe the solution you'd like

Home Assistant groups use an attribute `entity_id` (not to be confused with the entities own ID), which specify a list of child entity IDs. Populating this for Zigbee groups would prove extremely useful.

### Describe alternatives you've considered

It is possible to simulate the solution by customizing the entity in YAML, though this needs maintaining manually and is error-prone, e.g.:

```yaml
homeassistant:
  customize:
    light.bedroom_1_spots:
      entity_id:
        [
          "light.bedroom_1_spot_1",
          "light.bedroom_1_spot_2",
          "light.bedroom_1_spot_3",
          "light.bedroom_1_spot_4",
          "light.bedroom_1_spot_5",
          "light.bedroom_1_spot_6",
          "light.bedroom_1_spot_7",
          "light.bedroom_1_spot_8",
          "light.bedroom_1_spot_9"
        ]
```

Also, making changes requires a HA restart.

### Additional context

Just one example where this is useful. I use a JSON structure to generate my UI, and want to specify lights as a hierarchy. I currently use light groups in HA, particularly where I want to combine Z2m lights with Hue lights. However, wherever possible, I'd prefer to use Zigbee groups. Specifying the entity_id allows the following to work seamlessly:

```yaml
{%- set lights = [ states.light.all_lights ] %}
{%- set l %}
[
  {%- for light in lights recursive %}
{{"  "*loop.depth}}"{{ light.entity_id }}": [
     {%- if light.attributes.entity_id is defined -%}
{{ loop(expand(light.attributes.entity_id)) }}
    {%- endif -%}
]{{',' if not loop.last }}
  {%- endfor %}
]
{%- endset %}
{{ l }}
```

Which produces output like:
```json
[
  "light.all_lights": [
    ...
    "light.bedroom_1_spots": [
      "light.bedroom_1_spot_9": [],
      "light.bedroom_1_spot_8": [],
      "light.bedroom_1_spot_7": [],
      "light.bedroom_1_spot_6": [],
      "light.bedroom_1_spot_5": [],
      "light.bedroom_1_spot_4": [],
      "light.bedroom_1_spot_3": [],
      "light.bedroom_1_spot_2": [],
      "light.bedroom_1_spot_1": []],
    "light.bedroom_2_spots": [],
	...]
]
```

Where I have set the `entity_id` attribute for `bedroom_1_spots` (as above), but not for `bedroom_2_spots`.

thargy, 2024-09-09:
If there are concerns about future compatibility issues with other addons/integrations/extensions, it would be acceptable to use a different attribute than `entity_id`; for example, `z2m_entity_id`, or `z2m_group`.

thargy, 2024-12-11:
I currently run a python script that reads the config and generates a customize section automatically. It has the benefit of being able to create a group hierarchy automatically - which isn't a feature of z2m.

Happy to share.

thargy, 2025-02-11:
Ping

closed 2025-04-22 by github-actions[bot]

shbatm, 2025-07-31:
> I currently run a python script that reads the config and generates a customize section automatically. It has the benefit of being able to create a group hierarchy automatically - which isn't a feature of z2m.
> 
> Happy to share.

Would you mind sharing this script (e.g. via a gist)? Thanks!