Koenkk/zigbee2mqtt #23918: SLZB-06P7 Router Error

Blackduke77, 2024-09-09:
### What happened?

Getting the below errors in my log about SLZB-06P7 routers.

2024-09-09 09:58:43] error:    z2m: Failed to configure 'SLZB-06P7-Router', attempt 3 (TypeError: Cannot read properties of undefined (reading 'bind')
    at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:42:24)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/custom_devices_diy.ts:281:29)
    at Configure.configure (/app/lib/extension/configure.ts:125:37)
    at /app/lib/extension/configure.ts:87:62
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:23)
    at EventEmitter.emit (node:events:529:35)
    at EventBus.emitLastSeenChanged (/app/lib/eventBus.ts:52:22)
    at Controller.<anonymous> (/app/lib/zigbee.ts:76:27)
    at Controller.emit (node:events:517:28)
    at Controller.selfAndDeviceEmit (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:616:14))

### Zigbee2MQTT version

1.40.1

### Adapter firmware version

20240716

### Adapter

ZSTACK

### Setup

HAOS and Docker Z2M

Koenkk, 2024-09-09:
Could you provide the data/database.db entry of this device?

Blackduke77, 2024-09-09:
yes can do... how  :)

Koenkk, 2024-09-10:
You can open this file with a text editor (it's in the z2m data folder)

Blackduke77, 2024-09-10:
As requested

{"id":69,"type":"Router","ieeeAddr":"0x00124b0028f5653d","nwkAddr":28659,"manufId":0,"manufName":"SMLIGHT","powerSource":"Mains (single phase)","modelId":"SLZB-06P7","epList":[8,242],"endpoints":{"8":{"profId":260,"epId":8,"devId":255,"inClusterList":[0,3],"outClusterList":[0],"clusters":{"genBasic":{"attributes":{"modelId":"SLZB-06P7","manufacturerName":"SMLIGHT","powerSource":1,"zclVersion":3,"hwVersion":1,"swBuildId":"20240716"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"hwVersion":1,"swBuildId":"20240716","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1725995534853}

Koenkk, 2024-09-12:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-09-12 by Koenkk