Koenkk/zigbee2mqtt #23922: TS0121 / BlitzWolf BW-SHP13 not reporting energy settings

hotbirdx, 2024-09-09:
### What happened?

I have 4 devices of TS0121 which I use to measure power and worked perfect before. 
After updating to a newer Zigbee2MQTT the Current, Voltage, and Power values don't update. Other powerplug (types) continue to work.
Current Zigbee2MQTT version is 1.40.1 and it worked OK before 17 July 2024.
Old values are reported in HA. I tried removing and adding one of the devices, and the values are reported as: "Null".

### Zigbee2MQTT version

1.40.1

### Adapter firmware version

20210120

### Adapter

ZZH! CC2652R

### Setup

HomeAssistant (OS) on RPi4

### Debug log

2024-09-09 13:33:58z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/[PP11]', payload '{"current":null,"energy":194.15,"indicator_mode":"off/on","last_seen":"2024-09-09T11:33:58.904Z","linkquality":186,"power":null,"power_outage_memory":"on","state":"OFF","voltage":null}'

Friesje, 2024-09-09:
I have the same problem with TS011F.  I'm stuck at Zigbee2MQTT 1.38.0-1.  Every update after that, breaks the power measuring.  So im also stuck at older version.

hotbirdx, 2024-09-09:
I tried to update the firmware on the ZZH! CC2652R adapter to: 20240710, but that didn't help.

Koenkk, 2024-09-10:
- Could you provide a screenshot of the z2m frontend -> device about page?
- Could you provide the debug log of starting z2m + 5 mins after this?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

closed 2024-09-10 by Koenkk