Koenkk/zigbee2mqtt #23929: zigbee2mqtt UI in home assistant displays group that is out of sync with its underlying database

adamawolf, 2024-09-10:
### What happened?

I was working on renaming some of my zigbee devices and groups to have a more consistent naming scheme.

I renamed a group but found that after renaming that group it wasn't getting picked up by the HomeKit Bridge integration (not getting sent to Home.app). So I created another group that included the same zigbee devices, and this group was successful in sending to Home.app via the HomeKit integration.

So far, mostly ok. The real problem started when I went to rename the next set of devices/groups. I renamed the devices, then this time decided I would delete the existing group and create a fresh one (to avoid the possible failure to sync to HomeKit bug I experienced the first time). I created the new group, but then when I attempted to add a device to that group I got the error message: `z2m: Failed to add from group (DatabaseEntry with ID '31' does not exist)`

I tried to delete that group, thinking that something might be wrong with that particular group. I got a similar error (`Failed to remove group 'stairway_chandelier' (force: false) (Error: DatabaseEntry with ID '31' does not exist)`

Ditto if a create a new group.

### What did you expect to happen?

Adding devices, or removing the broken group would be successful.

### How to reproduce it (minimal and precise)

Please see my What Happened? answer for the closest I have to repro steps.

### Zigbee2MQTT version

1.40.1

### Adapter firmware version

20240710

### Adapter

SLZB-06

### Setup

Add-on on HAAS, running on a minix 0db

### Debug log

[z2m_err.log](https://github.com/user-attachments/files/16940565/z2m_err.log)

adamawolf, 2024-09-10:
Could someone point me towards where I can poke around the internal database z2m add-on uses to track the state shown in the Frontend? It seems like for whatever reason z2m has an ID cursor that's out of sync with reality (where "reality" is defined by the actual zigbee protocol groups).

ruderthanyou, 2024-10-25:
Did you ever get this resolved.?  This exact issue just happened to me.

emaasland, 2024-11-06:
I'm running into this as well. I have a fresh install and made 3 groups without issue, then I made a 4th (and renamed it immediately if that's relevant).

Now when I create a new group I cannot add new devices (Error: DatabaseEntry with ID '31' does not exist)). The weird thing is that if I then refresh the page z2m does show devices in the group, however the group is not present in Home Assistant.

When I try to remove the group I get the same error, but removing it a second time immediately after does get rid of the group (since it probably never existed).
It doesn't matter how many new groups are made, any new group made through the frontend does not function at all.

TokoTSP, 2024-12-02:
I have the same issue. 
Few weeks ago I found that one device is missing.
I tried to re-pair without sucess. 
Zigbee2MQTT version: 1.42.0-1
[image]

dschoorisse, 2024-12-21:
I had the same problem, after repairing some of the same lights. I removed the original group, re-created a new group with the same name. 

Then, when trying to add the same lights into the new group I got the error.

After restarting Z2M is was fixed.

Gunni, 2025-01-09:
Dupe of https://github.com/Koenkk/zigbee2mqtt/issues/23368

Gunni, 2025-03-11:
Not stale afaik

Gunni, 2025-05-11:
Afaik nothing has changed

Gunni, 2025-07-12:
Not fixed afaik

adamawolf, 2025-07-13:
@Koenkk I'm returning to this issue after 10 months of first hitting it, now setting up a completely fresh install of homeassistant for a new house. I'm fairly confident that there are straightforward steps to reproduce:

Steps to Reproduce

1. Create a new group.
2. Delete that group.
3. Create a new group with a different name than the first group.
4. Attempt to add an existing device to this group.

Observe that we get an error:

`2025-07-12 20:35:43z2m: Failed to add from group (DatabaseEntry with ID '63' does not exist)`

When hitting this I had already created ~other groups successfully with no issues adding devices. I had never deleted a group.

Koenkk, 2025-07-13:
Could you provide the debug log of the whole procedure?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.