Koenkk/zigbee2mqtt #23930: Trouble upgrading Home Assistant Zigbee2MQTT add-on v1.35.3-1 to 1.40.1-1

mk1blacklimited, 2024-09-10:
### What happened?

Trying to upgrade Home Assistant Zigbee2MQTT add-on v1.35.3-1 to 1.40.1-1. Upgrade seems to go well, but z2m never seems to get ready:

[09:55:23] INFO: Preparing to start...
[09:55:23] INFO: Socat not enabled
[09:55:24] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2024-09-10 09:55:25] info: 	z2m: Logging to console, file (filename: log.log)
[2024-09-10 09:55:25] info: 	z2m: Starting Zigbee2MQTT version 1.40.1 (commit #unknown)
[2024-09-10 09:55:25] info: 	z2m: Starting zigbee-herdsman (0.57.3)
[2024-09-10 09:55:26] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/serial/by-id/usb-Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus_0001-if00-port0","baudRate":115200,"rtscts":false,"autoOpen":false}
[2024-09-10 09:55:26] info: 	zh:zstack:znp: Serialport opened

...and there it stops. WebUI is not accessible (502 bad gateway).

[2024-09-10 09:57:22] error: 	z2m: Not connected to MQTT server!
[2024-09-10 09:57:22] error: 	z2m: Cannot send message: topic: 'zigbee2mqtthjemme/bridge/state', payload: '{"state":"offline"}
[2024-09-10 09:57:22] info: 	z2m: Disconnecting from MQTT server
[2024-09-10 09:57:22] info: 	z2m: Stopping zigbee-herdsman...
[2024-09-10 09:57:30] info: 	zh:controller: Wrote coordinator backup to '/config/zigbee2mqtt/coordinator_backup.json'
[2024-09-10 09:57:30] info: 	zh:zstack:znp: closing
[2024-09-10 09:57:30] info: 	zh:zstack:znp: Port closed
[2024-09-10 09:57:30] info: 	z2m: Stopped zigbee-herdsman
[2024-09-10 09:57:30] info: 	z2m: Stopped Zigbee2MQTT

After trying to restore v1.35.3-1 it fails to start:

[09:59:14] INFO: Preparing to start...
[09:59:14] INFO: Socat not enabled
[09:59:14] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:info  2024-09-10 09:59:16: Logging to console and directory: '/config/zigbee2mqtt/log/2024-09-10.09-59-16' filename: log.txt
Zigbee2MQTT:info  2024-09-10 09:59:16: Starting Zigbee2MQTT version 1.35.3 (commit #unknown)
Zigbee2MQTT:info  2024-09-10 09:59:16: Starting zigbee-herdsman (0.33.8)
Zigbee2MQTT:error 2024-09-10 09:59:16: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2024-09-10 09:59:16: Failed to start zigbee
Zigbee2MQTT:error 2024-09-10 09:59:16: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2024-09-10 09:59:16: Exiting...
Zigbee2MQTT:error 2024-09-10 09:59:16: RangeError: Attempt to access memory outside buffer bounds
    at new NodeError (node:internal/errors:405:5)
    at boundsError (node:internal/buffer:86:11)
    at Buffer.readUInt16BE (node:internal/buffer:334:5)
    at /app/node_modules/zigbee-herdsman/src/utils/backup.ts:102:89
    at Array.map (<anonymous>)
    at Object.fromUnifiedBackup (/app/node_modules/zigbee-herdsman/src/utils/backup.ts:101:33)
    at AdapterBackup.getStoredBackup (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/adapter-backup.ts:50:32)
    at ZnpAdapterManager.determineStrategy (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:134:24)
    at ZnpAdapterManager.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:63:26)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)

**Home Assistant info:**

Core 2024.9.1
Supervisor 2024.08.0
Operating System 13.1
Frontend 20240906.0

### What did you expect to happen?

Successful upgrade.

### How to reproduce it (minimal and precise)

Restore complete HASS with z2m 1.35.3-1. Upgrade z2m to 1.40.1-1.

### Zigbee2MQTT version

1.35.3-1

### Adapter firmware version

20220219

### Adapter

Sonoff Dongle P

### Setup

Home Assistant Add-on, x86-64

Koenkk, 2024-09-12:
Could you provide the complete log? I don't see any error in there.

mk1blacklimited, 2024-09-12:
Well, that’s just the thing, after it says «Serialport opened» there’s nothing in the log.

mk1blacklimited, 2024-09-13:
Is there a way to upgrade to a lesser version of the z2m addon in HA?

Koenkk, 2024-09-13:
I think not, do you see anything in the debug log?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

mk1blacklimited, 2024-09-19:
I can't see anything I can perceive as a clear error. Is it OK to attach the log file here (i.e. is there something I should remove from the file before posting it)?

mk1blacklimited, 2024-09-26:
@Koenkk - ping :)

mk1blacklimited, 2024-10-28:
Just tried 1.40.2-1, still the same problem. Really need help here.

Koenkk, 2024-10-28:
>  I have an older FW version in the stick (20220219), could that be the problem?

Try upgrading and see if it helps

Koenkk, 2024-10-31:
Could you activate the debug logging? You probably need to use the file explorer addon to edit the z2m configuration.yaml.

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

mk1blacklimited, 2024-11-01:
I've done that before, and then I asked: "Is it OK to attach the log file here (i.e. is there something I should remove from the file before posting it)?" which I failed to get a reply to. :) If it's OK to post here then I will ASAP. :)

Koenkk, 2024-11-02:
Please post it here.

mk1blacklimited, 2024-11-04:
OK, here it is.
[log.log](https://github.com/user-attachments/files/17614899/log.log)

mk1blacklimited, 2024-11-07:
OK - updated z2m to 1.41.0-1. Then I got this in the log page:

[image]

Configured MQTT as such:

[image]

Still the same.

Turned on debug log:

[2024-11-07 09:28:16] info: 	z2m: Logging to console, file (filename: log.log)
[2024-11-07 09:28:16] info: 	z2m: Starting Zigbee2MQTT version 1.41.0 (commit #unknown)
[2024-11-07 09:28:16] info: 	z2m: Starting zigbee-herdsman (2.1.7)
[2024-11-07 09:28:16] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/serial/by-id/usb-Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus_0001-if00-port0","baudRate":115200,"rtscts":false,"autoOpen":false}
[2024-11-07 09:28:16] info: 	zh:zstack:znp: Serialport opened
[2024-11-07 09:28:47] error: 	z2m: Failed to call 'Frontend' 'stop' (TypeError: Cannot read properties of undefined (reading 'clients')
    at Frontend.stop (/app/lib/extension/frontend.ts:114:18)
    at Controller.callExtensions (/app/lib/controller.ts:399:17)
    at Controller.stop (/app/lib/controller.ts:265:9)
    at stop (/app/index.js:165:5)
    at process.handleQuit (/app/index.js:173:13))
[2024-11-07 09:28:47] error: 	z2m: Not connected to MQTT server!
[2024-11-07 09:28:47] error: 	z2m: Cannot send message: topic: 'zigbee2mqtthjemme/bridge/state', payload: '{"state":"offline"}
[2024-11-07 09:28:47] info: 	z2m: Disconnecting from MQTT server
[2024-11-07 09:28:47] info: 	z2m: Stopping zigbee-herdsman...
[2024-11-07 09:28:55] info: 	zh:controller: Wrote coordinator backup to '/config/zigbee2mqtt/coordinator_backup.json'
[2024-11-07 09:28:55] info: 	zh:zstack:znp: closing
[2024-11-07 09:28:55] info: 	zh:zstack:znp: Port closed
[2024-11-07 09:28:55] info: 	z2m: Stopped zigbee-herdsman
[2024-11-07 09:28:55] info: 	z2m: Stopped Zigbee2MQTT

MQTT server (addon in HomeAssistant) is definitly running, nothing else has trouble connecting to it.

Koenkk, 2024-11-07:
It seems the error is a bit different now, could you provide the debug log again?

mk1blacklimited, 2025-01-09:
Just to remove stale label. Issue not fixed yet, I haven't had time to research any more. Will continue soon.

mk1blacklimited, 2025-02-04:
Right, finally had the time to test some more. Upgrading from v1.35.3-1 to  2.1.0-1 this time. Still not working unfortunately, but here are a couple of debug logs.

[45df7312_zigbee2mqtt_2025-02-04T07-47-37.427Z.log](https://github.com/user-attachments/files/18653627/45df7312_zigbee2mqtt_2025-02-04T07-47-37.427Z.log)
[45df7312_zigbee2mqtt_2025-02-04T07-53-33.935Z.log](https://github.com/user-attachments/files/18653628/45df7312_zigbee2mqtt_2025-02-04T07-53-33.935Z.log)

mk1blacklimited, 2025-02-04:
To test, uninstalled z2m addon, renamed the aigbee2mqtt folder in /homeassistant and reinstalled the addon. Without configuring anything, just started it up and it started fine:

[08:59:28] INFO: Preparing to start...
[08:59:28] INFO: Socat not enabled
[08:59:28] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-02-04 08:59:30] info: 	z2m: Logging to console, file (filename: log.log)
[2025-02-04 08:59:30] info: 	z2m: Starting Zigbee2MQTT version 2.1.0 (commit #unknown)
[2025-02-04 08:59:30] info: 	z2m: Starting zigbee-herdsman (3.2.5)
[2025-02-04 08:59:30] info: 	zh:adapter:discovery: Matched adapter: {"path":"/dev/ttyUSB0","manufacturer":"Silicon Labs","serialNumber":"0001","pnpId":"usb-Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus_0001-if00-port0","vendorId":"10c4","productId":"ea60"} => zstack: path=/dev/ttyUSB0, score=4
[2025-02-04 08:59:30] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/ttyUSB0","baudRate":115200,"rtscts":false,"autoOpen":false}
[2025-02-04 08:59:30] info: 	zh:zstack:znp: Serialport opened
[2025-02-04 08:59:46] info: 	zh:controller: Wrote coordinator backup to '/config/zigbee2mqtt/coordinator_backup.json'
[2025-02-04 08:59:46] info: 	z2m: zigbee-herdsman started (reset)
[2025-02-04 08:59:46] info: 	z2m: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20240710,"transportrev":2},"type":"zStack3x0"}'
[2025-02-04 08:59:46] info: 	z2m: Currently 0 devices are joined.
[2025-02-04 08:59:46] info: 	z2m: Connecting to MQTT server at mqtt://core-mosquitto:1883
[2025-02-04 08:59:47] info: 	z2m: Connected to MQTT server
[2025-02-04 08:59:47] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
[2025-02-04 08:59:47] info: 	z2m: Started frontend on port 8099

mk1blacklimited, 2025-02-06:
This indicates to me a problem with one (or more) of the configuration files, what do you think, @Koenkk ?

mk1blacklimited, 2025-03-13:
Tought I'd wrap up this topic as it is resolved. :)

I thought I should give a normal update another quick try before removing z2m and reinstalling. Upgrading from 1.35 to 2.1.3-1, and except for an addition of "adapter: zstack" in the config file, it started right up! Amazing, I guess @Koenkk has done some magic behind the scenes. :)

closed 2025-05-21 by github-actions[bot]