Koenkk/zigbee2mqtt #23941: [New device support]: Nous E8 - smoke sensor

Bobsilvio, 2024-09-11:
### Link

https://www.zigbee2mqtt.io/devices/E8.html

### Database entry

{"id":38,"type":"EndDevice","ieeeAddr":"0x8c6fb9fffe2cf889","nwkAddr":2373,"manufId":4098,"manufName":"_TZE284_0zaf1cr8","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184,0,60672],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE284_0zaf1cr8","stackVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1726070066310}

### Zigbee2MQTT version

1.40.1

### Comments

This a new device E8 Nous, is a smoke sensor

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE284_0zaf1cr8',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

nothing work and nothing see

ahorseman, 2024-09-28:
I've tried with the extended definition according to the tutorials for Tuya devices (https://www.zigbee2mqtt.io/advanced/support-new-devices/02_support_new_tuya_devices.html#_4-mapping-the-datapoints), but no luck. I don't get any datapoint:

`const {} = require('zigbee-herdsman-converters/lib/modernExtend');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of Tuya devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE284_0zaf1cr8',
        },
    ],
    zigbeeModel: ['TS0601'],
    model: 'E8',
    vendor: 'Nous',
    description: 'Nous Smoke Sensor E8',
    extend: [],
    meta: {},
    description: 'Nous E8 Smoke Sensor',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
        // Here you should put all functionality that your device exposes
    ],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [],
    },
    extend: [
        // A preferred new way of extending functionality.
    ],
};

module.exports = definition;`

jschwarz88, 2024-09-28:
Hi there,
just got mine today.
According to [https://www.zigbee2mqtt.io/devices/E8.html](https://www.zigbee2mqtt.io/devices/E8.html) support should have been added in sept. '24.

I'll take a look at the documentation for tuya devices/endpoints as well...

kersten-lohmeyer, 2024-09-28:
Just installed Z2M just to find out this device is not supported although it says that on the box and also here: https://www.zigbee2mqtt.io/devices/E8.html

Koenkk, 2024-09-30:
Added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-09-30 by Koenkk

kersten-lohmeyer, 2024-09-30:
Thank you so much! Any ETA when this release is coming to Home Assistant?

Matten-Matten, 2024-10-01:
Hello @Koenkk ,
I manually adjusted the `tuya.js` on my system as you changed, but it didn't work. I had to change it like this.

````
    {
        fingerprint: tuya.fingerprint('TS0601', ['_TZE200_0zaf1cr8', '_TZE204_ntcy3xu1', '_TZE284_0zaf1cr8']),
        model: 'TS0601_smoke_1',
        vendor: 'Tuya',
        description: 'Smoke sensor',
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [tuya.tz.datapoints],
        configure: tuya.configureMagicPacket,
        exposes: [e.smoke(), e.tamper(), e.battery_low()],
        meta: {
            tuyaDatapoints: [
                [1, 'smoke', tuya.valueConverter.trueFalse0],
                [4, 'tamper', tuya.valueConverter.raw],
                [14, 'battery_low', tuya.valueConverter.trueFalse0],
            ],
        },
        whiteLabel: [tuya.whitelabel('Nous', 'E8', 'Smoke sensor', ['_TZE200_0zaf1cr8', '_TZE284_0zaf1cr8'])],
    },
````
I also adjusted the fingerprint.

greeting
Matten Matten

kersten-lohmeyer, 2024-10-01:
> Hello @Koenkk , I manually adjusted the `tuya.js` on my system as you changed, but it didn't work. I had to change it like this.
>
> ```
>     {
>         fingerprint: tuya.fingerprint('TS0601', [...]

Can I adjust this file when running Z2M as an addon in HA? And if yes how?

Euer-Ernst, 2024-10-02:
@Matten-Matten 
Thank you,
I am doing the same and after restarting my Docker Container it works without deleting / reconnecting my device.

Frontend-Version: 0.7.4
Zigbee Herdsman Konverter Version: 20.21.0
Zigbee Herdsman Version: 2.1.3

kersten-lohmeyer, 2024-10-02:
I installed Z2M 1.40.2 in Home Assistant and the device is still unsupported.

markusmauch, 2024-10-02:
Same here, I installed Zigbee2Mqtt v1.40.2-1, restarted Home Assistant, removed and then re-joined the device and it is still 'unsupported'

The log says:
`2024-10-02 17:23:24] warning: 	z2m: Device '0x403059fffe6372cb' with Zigbee model 'TS0601' and manufacturer name '_TZE284_0zaf1cr8' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html`

Koenkk, 2024-10-02:
Made a mistake, added it now

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

rionshin, 2024-10-07:
> Made a mistake, added it now
> 
> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

@Koenkk  I am trying to pull the dev but it said no changes, are converters merged to zigbee2mqtt as well ? :

pi@homebridge:/opt/zigbee2mqtt $ git config remote.origin.fetch "+refs/heads/*:refs/remotes/origin/*"
pi@homebridge:/opt/zigbee2mqtt $ git fetch --all
Fetching origin
remote: Enumerating objects: 123, done.
remote: Counting objects: 100% (120/120), done.
remote: Compressing objects: 100% (56/56), done.
remote: Total 123 (delta 70), reused 102 (delta 64), pack-reused 3 (from 1)
Receiving objects: 100% (123/123), 234.81 KiB | 3.09 MiB/s, done.
Resolving deltas: 100% (70/70), completed with 10 local objects.
From https://github.com/Koenkk/zigbee2mqtt
 * [new branch]        dev                                                                   -> origin/dev
 * [new branch]        feat/ota-update-check-all                                             -> origin/feat/ota-update-check-all
 * [new branch]        feat/the-big-cleanup                                                  -> origin/feat/the-big-cleanup
 * [new branch]        fix/adapter-discovery                                                 -> origin/fix/adapter-discovery
 * [new branch]        fix/ha-permit-join-timeout                                            -> origin/fix/ha-permit-join-timeout
 * [new branch]        release-please--branches--dev--components--zigbee2mqtt                -> origin/release-please--branches--dev--components--zigbee2mqtt
 * [new branch]        release-please--branches--dev--components--zigbee2mqtt--release-notes -> origin/release-please--branches--dev--components--zigbee2mqtt--release-notes
pi@homebridge:/opt/zigbee2mqtt $ git checkout dev
M       data/configuration.example.yaml
M       data/configuration.yaml
Branch 'dev' set up to track remote branch 'dev' from 'origin'.
Switched to a new branch 'dev'
pi@homebridge:/opt/zigbee2mqtt $ git pull
 **Already up to date.**

my git config:

                                                                                                           
[core]
        repositoryformatversion = 0
        filemode = true
        bare = false
        logallrefupdates = true
[remote "origin"]
        url = https://github.com/Koenkk/zigbee2mqtt.git
        fetch = +refs/heads/*:refs/remotes/origin/*
[branch "master"]
        remote = origin
        merge = refs/heads/master
[branch "dev"]
        remote = origin
        merge = refs/heads/dev

rionshin, 2024-10-09:
> make sure to execute "npm ci" after updating

 @Koenkk  unfortunately after change to DEV branch and git pull it said: Already up to date 
i will be waiting next "minor" official release, do you know when it will be out  ?

kersten-lohmeyer, 2024-10-09:
Would be nice anyway to have a minor HA addon update. @Koenkk

daytrax, 2024-10-13:
Same here, system is up to Date. Alle frei, HASS and Zigbee2MQTT-Version
[1.40.2]

Nous Smoke E8 will be known as unsupported Device

??

kersten-lohmeyer, 2024-10-13:
Nous took their E8 off Amazon for a few weeks to "fix" certain things. Now it's available for purchase again. I just wish the Z2M developer would already release the new version on HA. Pretty please! 😚

daytrax, 2024-10-13:
1 have one here and 6 come this week from amazon :(

kersten-lohmeyer, 2024-10-13:
> 1 have one here and 6 come this week from amazon :(

I also have one and need 3 more. But I am not ordering until I know it's working with Z2M.

Koenkk, 2024-10-14:
It's available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) for now (and the 1 November release)

daytrax, 2024-10-21:
Nice, then I wait for November Update and Buy the more if it works

nicoh88, 2024-10-26:
Hi @kersten-lohmeyer, have you found a way to update the zigbee2mqtt docker container. I have customized the tuya.js (`/app/node_modules/zigbee-herdsman-converters/devices/tuya.js`), also tried a `docker commit`. But after a restart of the zigbee2mqtt addon in Home Assistant the changes are unfortunately gone again.

Best regards from Germany

kersten-lohmeyer, 2024-10-26:
> Hi @kersten-lohmeyer, have you found a way to update the zigbee2mqtt docker container. I have customized the `tuya.js` (`/app/node_modules/zigbee-herdsman-converters/devices/tuya.js`), also tried a [...]

No, I didn't. I'm waiting for the next release of the addon now, shouldn't take more than a few days (I hope).

Greetings from southern Germany 😉

PV-Joe, 2025-04-13:
It works for me but the Tamper sensor is unavailable :-(
Any hint to fix this?

[image]