Koenkk/zigbee2mqtt #23946: BLE-YL01 is spamming network.

Passific, 2024-09-12:
### What happened?

Several times a day, BLE-YL01 is spamming with updates (noticeable with the last seen constantly updating).
This cause my Zigbee network to saturate and I get some "Fail to deliver message" errors in the logs when it happens.

Using `debounce: 2` did not change anything (last seen constantly update, and error messages in the logs).

It even made my NODON SIN-4-RS-20 module (where it is connected to) crash.
I removed the BLE-YL01 from the network 3 days ago, and not a single error since then.

Could rebooting Z2M (as part of docker update for example), send the `cluster 0xEF00, attribute 0x03` frame again which cause too many updates? (just guessing)

### Zigbee2MQTT version

1.40.1

### Adapter firmware version

7.4.3

### Adapter

Ember

### Setup

Docker on Intel N100

R3yn4ld, 2024-09-28:
I have the same error.

Also, I don't know if related but I cannot execute the magic trick to get the value updated after the very first pairing:
[image]

The sensor is unusable via Z2M then :(

aperezva, 2024-10-08:
The same error, @Koenkk  could you see what happen, it was after last update

CanuckianOz, 2024-12-03:
@Koenkk this device still doesn't appear to update on the Edge or main Z2M distro. Is it possible that it requires a manual resetting of the backlight_value (to 0, then to 1) similar to the LocalTuya workaround to fetch device updates? If this value were exposed in Z2M, we could set an automation to reset the value as desired. 

https://community.home-assistant.io/t/pool-monitoring-device-yieryi-ble-yl01-zigbee-ph-orp-free-chlorine-salinity-etc/659545/18?u=truemana

Koenkk, 2024-12-04:
Could you provide the debug log when starting z2m + 5 minutes after this?

CanuckianOz, 2024-12-04:
@Koenkk no problem - here you go:

https://pastes.dev/Oc3DJitDRR

Passific, 2024-12-05:
This is almost the same message I got (see my logs in the top message).
But from what I experienced, device does communicate as the « last seen » suggest, and I got some data (not sure about how often though).
I tried to repair it, but each time there was a variable delay in hours before the first meaningful data. So for now it’s in my garage waiting to not crash my Zigbee network…

Judufu, 2025-01-08:
Did you manage to fix yours? I have the same problem with mine :/

CanuckianOz, 2025-01-08:
No I didn’t. I went back to using tuya local custom integration and their wifi hub, then setting an automation to set and reset backlight_value from 0 to 1 and back to 0 every 5 minutes. It’s the only tuya device I have so a bit of a pain.

Carrotpro, 2025-03-07:
Got same problem anyone got a fix yet?

OE5DRO, 2025-04-29:
same problem here

Passific, 2025-04-30:
I personally gave up on the device. It's a shame given the price tag, but it's really not reliable, plus having all sensors on one device, if it fails you have nothing. And above all, it's not starting this spring, it may not have liked the winter or something...
I'm now starting over with independent sensors.
Good luck anyway, I hope you get more results!

CanuckianOz, 2025-04-30:
I haven’t tried to re-pair the device with zigbee, but I have found it to completely reliable running via a tuya gateway and the tuya integration for about 6 months. Takes a little to manually setup but it’s been reporting data consistently. 

However, the data isn’t fully reliable. I have a low salt / low chlorine system with auto acid dosing and it’s consistently between 1-2ppm measured, but the device only occasionally shows 0.1-0.4ppm. pH is always about 0.2 higher than measured, temperature about 1.5C higher. Salt PPM is about 200ppm too high but spikes massively randomly once a day. 

Basically, I use it to ensure they’re aren’t major swings. The temperature is at least very helpful.

juslex, 2025-05-16:
Same problem here

juslex, 2025-05-16:
@Koenkk Could you help us? There are several open issues with the same update problem that does not occur or otherwise the device is spammed

davic80, 2025-05-16:
> [@Koenkk](https://github.com/Koenkk) Could you help us? There are several open issues with the same update problem that does not occur or otherwise the device is spammed

+1

juslex, 2025-05-16:
I believe the problem is that after version 1.39.1 it is no longer possible to send the cluster 0xEF00, attribute 0x03" command in Dev console that causes the device to update the information every 10 minutes, making there no spamming network, according to the issue below the new versions of z2m have not yet fixed the problem. It is also not in the documentation of the device that it is necessary to execute this attribute so that the update is every 10 minutes, nor is there in the Front End any button that allows you to send this attribute easily or some code in the z2m pairing that allows the sending of this attribute without the need for the user to do it manually.

[image]

https://github.com/Koenkk/zigbee2mqtt/issues/18704#issuecomment-2578196818

@Koenkk

juslex, 2025-05-16:
Look at https://www.youtube.com/watch?v=8zGAHyCan0c&t=338s 5m27s

Koenkk, 2025-05-16:
@juslex the command you are referring to is the `dataQuery` command. Z2M automatically sends this command every 5 minutes, so there is no need to send it manually.  If I understand correctly, currently the device is spamming a lot in your network (so reporting way too often). Could you provide the debug of 15 minutes? 

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

juslex, 2025-05-17:
Follow the log. However, after waiting a few minutes to obtain the log for 15 minutes, the spamming ended. I don't remember the last time, but I think it came back after a while and that's why I removed and connected it to tuya's Zigbee hub and started using it with a localtuya fork template.

However, the value on Salinity is still zero, I was having the correct answer from them using localtuya since my pool has salt for chlorine generation. The other parameters presented are in accordance with what I was receiving in localtuya.

[log.txt](https://github.com/user-attachments/files/20260525/log.txt)

juslex, 2025-05-17:
Here is the localtuya template where the salt was being correctly exposed:

- sensor:
    id: '1'
    friendly_name: TDS
    entity_category: diagnostic
    unit_of_measurement: "ppm"
    state_class: measurement
    platform: sensor
- sensor:
    id: '2'
    friendly_name: Temperature
    entity_category: diagnostic
    unit_of_measurement: "°C"
    device_class: temperature
    state_class: measurement
    scaling: 0.1
- sensor:
    id: '7'
    friendly_name: Battery
    entity_category: diagnostic
    unit_of_measurement: "%"
    device_class: battery
    state_class: measurement
    platform: sensor
- sensor:
    id: '10'
    friendly_name: PH
    entity_category: diagnostic
    device_class: ph
    state_class: measurement
    scaling: 0.1
    platform: sensor
- sensor:
    id: '11'
    friendly_name: EC
    entity_category: diagnostic
    unit_of_measurement: "µS/cm"
    state_class: measurement
    platform: sensor
- sensor:
    id: '101'
    friendly_name: ORP
    entity_category: diagnostic
    unit_of_measurement: "mV"
    state_class: measurement
    platform: sensor
- sensor:
    id: '102'
    friendly_name: Chlorine
    entity_category: diagnostic
    unit_of_measurement: "mg/l"
    state_class: measurement
    scaling: 0.1
    platform: sensor
- sensor:
    id: '103'
    friendly_name: PH Standard
    entity_category: diagnostic
    platform: sensor
- number:
    friendly_name: Request
    entity_category: config
    min_value: 0.0
    max_value: 1.0
    step_size: 1.0
    restore_on_reconnect: true
    is_passive_entity: true
    id: '105'
    icon: ""
- sensor:
    id: '117'
    friendly_name: Salt
    entity_category: diagnostic
    unit_of_measurement: "ppm"
    state_class: measurement
    platform: sensor

juslex, 2025-05-17:
[image]

[image]

[image]

juslex, 2025-05-17:
The spamming is back, but after some time it stopped again:

 
Info 2025-05-16 23:12:01z2m:mqtt: MQTT publish: topic 'zigbee2mqtt2/BLE-YL01', payload '{"battery":90,"ec":3486,"ec_max":20000,"ec_min":0,"free_chlorine":0.1,"free_chlorine_max":-0.1,"free_chlorine_min":0,"last_seen":"2025-05-16T23:12:01-03:00","linkquality":69,"orp":258,"orp_max":-1,"orp_min":-1,"ph":6.5,"ph_max":8.9,"ph_min":0,"salinity":0,"tds":1743,"temperature":28.9}'
Info 2025-05-16 23:12:01z2m:mqtt: MQTT publish: topic 'zigbee2mqtt2/BLE-YL01', payload '{"battery":90,"ec":3486,"ec_max":20000,"ec_min":0,"free_chlorine":0.1,"free_chlorine_max":-0.1,"free_chlorine_min":0,"last_seen":"2025-05-16T23:12:01-03:00","linkquality":72,"orp":258,"orp_max":-1,"orp_min":-1,"ph":6.5,"ph_max":8.9,"ph_min":0,"salinity":0,"tds":1743,"temperature":28.9}'
Info 2025-05-16 23:12:06z2m:mqtt: MQTT publish: topic 'zigbee2mqtt2/BLE-YL01', payload '{"battery":90,"ec":3486,"ec_max":20000,"ec_min":0,"free_chlorine":0.1,"free_chlorine_max":-0.1,"free_chlorine_min":0,"last_seen":"2025-05-16T23:12:06-03:00","linkquality":69,"orp":258,"orp_max":-1,"orp_min":-1,"ph":6.5,"ph_max":8.9,"ph_min":0,"salinity":0,"tds":1743,"temperature":28.9}'
Info 2025-05-16 23:12:07z2m:mqtt: MQTT publish: topic 'zigbee2mqtt2/BLE-YL01', payload '{"battery":90,"ec":3486,"ec_max":20000,"ec_min":0,"free_chlorine":0.1,"free_chlorine_max":-0.1,"free_chlorine_min":0,"last_seen":"2025-05-16T23:12:07-03:00","linkquality":69,"orp":258,"orp_max":-1,"orp_min":-1,"ph":6.5,"ph_max":8.9,"ph_min":0,"salinity":0,"tds":1743,"temperature":28.9}'
Info 2025-05-16 23:12:07z2m:mqtt: MQTT publish: topic 'zigbee2mqtt2/BLE-YL01', payload '{"battery":90,"ec":3486,"ec_max":20000,"ec_min":0,"free_chlorine":0.1,"free_chlorine_max":-0.1,"free_chlorine_min":0,"last_seen":"2025-05-16T23:12:07-03:00","linkquality":65,"orp":258,"orp_max":-1,"orp_min":-1,"ph":6.5,"ph_max":8.9,"ph_min":0,"salinity":0,"tds":1743,"temperature":28.9}'
Info 2025-05-16 23:12:07z2m:mqtt: MQTT publish: topic 'zigbee2mqtt2/BLE-YL01', payload '{"battery":90,"ec":3486,"ec_max":20000,"ec_min":0,"free_chlorine":0.1,"free_chlorine_max":-0.1,"free_chlorine_min":0,"last_seen":"2025-05-16T23:12:07-03:00","linkquality":69,"orp":258,"orp_max":-1,"orp_min":-1,"ph":6.5,"ph_max":8.9,"ph_min":0,"salinity":0,"tds":1743,"temperature":28.9}'
Info 2025-05-16 23:12:08z2m:mqtt: MQTT publish: topic 'zigbee2mqtt2/BLE-YL01', payload '{"battery":90,"ec":3486,"ec_max":20000,"ec_min":0,"free_chlorine":0.1,"free_chlorine_max":-0.1,"free_chlorine_min":0,"last_seen":"2025-05-16T23:12:08-03:00","linkquality":69,"orp":258,"orp_max":-1,"orp_min":-1,"ph":6.5,"ph_max":8.9,"ph_min":0,"salinity":0,"tds":1743,"temperature":28.9}'
Info 2025-05-16 23:12:08z2m:mqtt: MQTT publish: topic 'zigbee2mqtt2/BLE-YL01', payload '{"battery":90,"ec":3486,"ec_max":20000,"ec_min":0,"free_chlorine":0.1,"free_chlorine_max":-0.1,"free_chlorine_min":0,"last_seen":"2025-05-16T23:12:08-03:00","linkquality":72,"orp":258,"orp_max":-1,"orp_min":-1,"ph":6.5,"ph_max":8.9,"ph_min":0,"salinity":0,"tds":1743,"temperature":28.9}'
Info 2025-05-16 23:12:08z2m:mqtt: MQTT publish: topic 'zigbee2mqtt2/BLE-YL01', payload '{"battery":90,"ec":3486,"ec_max":20000,"ec_min":0,"free_chlorine":0.1,"free_chlorine_max":-0.1,"free_chlorine_min":0,"last_seen":"2025-05-16T23:12:08-03:00","linkquality":65,"orp":258,"orp_max":-1,"orp_min":-1,"ph":6.5,"ph_max":8.9,"ph_min":0,"salinity":0,"tds":1743,"temperature":28.9}'
Info 2025-05-16 23:12:08z2m:mqtt: MQTT publish: topic 'zigbee2mqtt2/BLE-YL01', payload '{"battery":90,"ec":3486,"ec_max":20000,"ec_min":0,"free_chlorine":0.1,"free_chlorine_max":-0.1,"free_chlorine_min":0,"last_seen":"2025-05-16T23:12:08-03:00","linkquality":72,"orp":258,"orp_max":-1,"orp_min":-1,"ph":6.5,"ph_max":8.9,"ph_min":0,"salinity":0,"tds":1743,"temperature":28.9}'
Info 2025-05-16 23:12:08z2m:mqtt: MQTT publish: topic 'zigbee2mqtt2/BLE-YL01', payload '{"battery":90,"ec":3486,"ec_max":20000,"ec_min":0,"free_chlorine":0.1,"free_chlorine_max":-0.1,"free_chlorine_min":0,"last_seen":"2025-05-16T23:12:08-03:00","linkquality":69,"orp":258,"orp_max":-1,"orp_min":-1,"ph":6.5,"ph_max":8.9,"ph_min":0,"salinity":0,"tds":1743,"temperature":28.9}'
Info 2025-05-16 23:12:09z2m:mqtt: MQTT publish: topic 'zigbee2mqtt2/BLE-YL01', payload '{"battery":90,"ec":3486,"ec_max":20000,"ec_min":0,"free_chlorine":0.1,"free_chlorine_max":-0.1,"free_chlorine_min":0,"last_seen":"2025-05-16T23:12:09-03:00","linkquality":69,"orp":258,"orp_max":-1,"orp_min":-1,"ph":6.5,"ph_max":8.9,"ph_min":0,"salinity":0,"tds":1743,"temperature":28.9}'
Info 2025-05-16 23:12:09z2m:mqtt: MQTT publish: topic 'zigbee2mqtt2/BLE-YL01', payload '{"battery":90,"ec":3486,"ec_max":20000,"ec_min":0,"free_chlorine":0.1,"free_chlorine_max":-0.1,"free_chlorine_min":0,"last_seen":"2025-05-16T23:12:09-03:00","linkquality":72,"orp":258,"orp_max":-1,"orp_min":-1,"ph":6.5,"ph_max":8.9,"ph_min":0,"salinity":0,"tds":1743,"temperature":28.9}'
Info 2025-05-16 23:12:09z2m:mqtt: MQTT publish: topic 'zigbee2mqtt2/BLE-YL01', payload '{"battery":90,"ec":3486,"ec_max":20000,"ec_min":0,"free_chlorine":0.1,"free_chlorine_max":-0.1,"free_chlorine_min":0,"last_seen":"2025-05-16T23:12:09-03:00","linkquality":69,"orp":258,"orp_max":-1,"orp_min":-1,"ph":6.5,"ph_max":8.9,"ph_min":0,"salinity":0,"tds":1743,"temperature":28.9}'
[... cut ...] kquality":0,"orp":258,"orp_max":-1,"orp_min":-1,"ph":6.5,"ph_max":8.9,"ph_min":0,"salinity":0,"tds":1743,"temperature":28.9}'
Info 2025-05-16 23:14:03z2m:mqtt: MQTT publish: topic 'zigbee2mqtt2/BLE-YL01', payload '{"battery":90,"ec":3486,"ec_max":20000,"ec_min":0,"free_chlorine":0.1,"free_chlorine_max":-0.1,"free_chlorine_min":0,"last_seen":"2025-05-16T23:14:03-03:00","linkquality":0,"orp":258,"orp_max":-1,"orp_min":-1,"ph":6.5,"ph_max":8.9,"ph_min":0,"salinity":0,"tds":1743,"temperature":28.9}'

juslex, 2025-05-17:
Another thing I noticed is that during spamming the value of Free chlorine is negative at -0.1, it would be good to prevent this since there is no negative value of Free chlorine.

[image]

juslex, 2025-05-17:
Another error, when I stipulate the parameter for example of orp max or orp min and free chlorine Max or free chlorine min after some time these values are at the minimum. In the app, these values are used to make the number on the display red or in normal color if the minimum or maximum values are met. We could create a binary sensor of Orp, PH, EC and Free chlorine that was on if the reading value was equal to the maximum or minimum marking.

After a few hours, I checked that spam comes and comes back routinely. Analyzing what happens in the application and the way the localtuya integration implemented, the DP 105 that speaks of 105: Background Light Value, this DP is triggered when we open the tuya application and go to the device, during this moment, it is updating all the time and the device light that indicates that the transmission is on. I believe that the 0xEF00 cluster, 0x03 attribute triggers this DP, however, Z2m remains a long time with it activated, causing spam to occur a lot. I think there may be some other attribute for DP 105 to stop working, because it happens when we leave the tuya application and the device goes into standby mode in relation to parameter updates. We could allow it to be activated for some time around 30 seconds every 10 minutes so that the device would update its parameters and then stop spam.

@Koenkk

Koenkk, 2025-05-18:
Could you check if the issue is fixed with the following [external converter](https://gist.github.com/Koenkk/7cb09518bf346adbc94c6f77f372f644): 

- save this as file next to `configuration.yaml` in the `external_converters` directory (create if it doesn't exist yet) as `ext_converter.js` (so `external_converters/ext_converter.js`)
- start z2m, 
- in the frontend, check if the description of this device now ends with `CUSTOM` (this indicates the external converter has been loaded correctly)
- check if issue is fixed

OE5DRO, 2025-05-19:
i have saved the converter file in the described directory. restartet z2m and also restartet HA but i don't find the description with "CUSTOM". 

[image]

so i think the converter is not active?!

jmbinette, 2025-05-20:
Hi to all,

I am following this topic/issue with great interest. I am sniffing zigbee traffic with the Tuya Gateway/Android app and a CC1352P with wireshark. Anyone knows a way to get the encryption key ? 

Packets are encrypted and I am hoping I can help with the driver development.

juslex, 2025-05-20:
> i have saved the converter file in the described directory. restartet z2m and also restartet HA but i don't find the description with "CUSTOM".
> 
> <img alt="Image" width="1088" src="https://private-user-images.githubusercontent.com/39993577/445180249-20484a7c-81ea-44a8-a5e7-224a01d8ac76.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDc3NDQ0ODgsIm5iZiI6MTc0Nzc0NDE4OCwicGF0aCI6Ii8zOTk5MzU3Ny80NDUxODAyNDktMjA0ODRhN2MtODFlYS00NGE4LWE1ZTctMjI0YTAxZDhhYzc2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA1MjAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNTIwVDEyMjk0OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQzMGQ0MmUxOGJiMTZhMzY4OGY5NzIzYjRlNjBlZjFjN2NmNDZjNjEwNGQ0NDg1MjMxNzRhMWRhODRiMTJhNTEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.AXezXO80Wc-YvIWcTT3dfsZyZu90QWDdBWpfTYjx4e4">
> so i think the converter is not active?!
> 
> i also tried the directory zigbee2mqtt/external_converters/ext_converter.js
> 
> ...
> 
> it hast to be at /zigbee2mqtt/data/external_converters/ext_converter.js then i get CUSTOM converter, but it stops sending data after some minutes again

Same problem here, The device stops responding to apps for a few minutes.

OE5DRO, 2025-05-20:
> And what if you change e.g. `// queryIntervalSeconds: 5 * 60,` to `queryIntervalSeconds: 20 * 60,` (make sure to also remove the `//`)

thank you for your time @Koenkk 
i have changed the custom converter to "20 * 60" ... and the spam is still going on:

`info 2025-05-20 18:32:24z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Poolmesser', payload '{"battery":100,"ec":699,"ec_max":20000,"ec_min":0,"free_chlorine":0.2,"free_chlorine_max":-0.1,"free_chlorine_min":0,"last_seen":"2025-05-20T18:32:24+02:00","linkquality":58,"orp":536,"orp_max":-1,"orp_min":-1,"ph":7.61,"ph_max":140,"ph_min":0,"salinity":0,"tds":349,"temperature":19.7}'
info 2025-05-20 18:32:25z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Poolmesser', payload '{"battery":100,"ec":699,"ec_max":20000,"ec_min":0,"free_chlorine":0.2,"free_chlorine_max":-0.1,"free_chlorine_min":0,"last_seen":"2025-05-20T18:32:25+02:00","linkquality":58,"orp":536,"orp_max":-1,"orp_min":-1,"ph":7.61,"ph_max":140,"ph_min":0,"salinity":0,"tds":349,"temperature":19.7}'
info 2025-05-20 18:32:25z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Poolmesser', payload '{"battery":100,"ec":699,"ec_max":20000,"ec_min":0,"free_chlorine":0.2,"free_chlorine_max":-0.1,"free_chlorine_min":0,"last_seen":"2025-05-20T18:32:25+02:00","linkquality":61,"orp":536,"orp_max":-1,"orp_min":-1,"ph":7.61,"ph_max":140,"ph_min":0,"salinity":0,"tds":349,"temperature":19.7}'
info 2025-05-20 18:32:25z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Poolmesser', payload '{"battery":100,"ec":699,"ec_max":20000,"ec_min":0,"free_chlorine":0.2,"free_chlorine_max":-0.1,"free_chlorine_min":0,"last_seen":"2025-05-20T18:32:25+02:00","linkquality":58,"orp":536,"orp_max":-1,"orp_min":-1,"ph":7.61,"ph_max":140,"ph_min":0,"salinity":0,"tds":349,"temperature":19.7}'
info 2025-05-20 18:32:25z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Poolmesser', payload '{"battery":100,"ec":699,"ec_max":20000,"ec_min":0,"free_chlorine":0.2,"free_chlorine_max":-0.1,"free_chlorine_min":0,"last_seen":"2025-05-20T18:32:25+02:00","linkquality":61,"orp":536,"orp_max":-1,"orp_min":-1,"ph":7.61,"ph_max":140,"ph_min":0,"salinity":0,"tds":349,"temperature":19.7}'
info 2025-05-20 18:32:25z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Poolmesser', payload '{"battery":100,"ec":699,"ec_max":20000,"ec_min":0,"free_chlorine":0.2,"free_chlorine_max":-0.1,"free_chlorine_min":0,"last_seen":"2025-05-20T18:32:26+02:00","linkquality":61,"orp":536,"orp_max":-1,"orp_min":-1,"ph":7.61,"ph_max":140,"ph_min":0,"salinity":0,"tds":349,"temperature":19.7}'
info 2025-05-20 18:32:26z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Poolmesser', payload '{"battery":100,"ec":699,"ec_max":20000,"ec_min":0,"free_chlorine":0.2,"free_chlorine_max":-0.1,"free_chlorine_min":0,"last_seen":"2025-05-20T18:32:26+02:00","linkquality":65,"orp":536,"orp_max":-1,"orp_min":-1,"ph":7.61,"ph_max":140,"ph_min":0,"salinity":0,"tds":349,"temperature":19.7}'
info 2025-05-20 18:32:26z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Poolmesser', payload '{"battery":100,"ec":699,"ec_max":20000,"ec_min":0,"free_chlorine":0.2,"free_chlorine_max":-0.1,"free_chlorine_min":0,"last_seen":"2025-05-20T18:32:26+02:00","linkquality":61,"orp":536,"orp_max":-1,"orp_min":-1,"ph":7.61,"ph_max":140,"ph_min":0,"salinity":0,"tds":349,"temperature":19.7}'
info 2025-05-20 18:32:31z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Poolmesser', payload '{"battery":100,"ec":699,"ec_max":20000,"ec_min":0,"free_chlorine":0.2,"free_chlorine_max":-0.1,"free_chlorine_min":0,"last_seen":"2025-05-20T18:32:31+02:00","linkquality":58,"orp":536,"orp_max":-1,"orp_min":-1,"ph":7.61,"ph_max":140,"ph_min":0,"salinity":0,"tds":349,"temperature":19.7}'
info 2025-05-20 18:32:32z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Poolmesser', payload '{"battery":100,"ec":699,"ec_max":20000,"ec_min":0,"free_chlorine":0.2,"free_chlorine_max":-0.1,"free_chlorine_min":0,"last_seen":"2025-05-20T18:32:32+02:00","linkquality":58,"orp":536,"orp_max":-1,"orp_min":-1,"ph":7.61,"ph_max":140,"ph_min":0,"salinity":0,"tds":349,"temperature":19.7}'
info 2025-05-20 18:32:32z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Poolmesser', payload '{"battery":100,"ec":699,"ec_max":20000,"ec_min":0,"free_chlorine":0.2,"free_chlorine_max":-0.1,"free_chlorine_min":0,"last_seen":"2025-05-20T18:32:32+02:00","linkquality":54,"orp":536,"orp_max":-1,"orp_min":-1,"ph":7.61,"ph_max":140,"ph_min":0,"salinity":0,"tds":349,"temperature":19.7}'
info 2025-05-20 18:32:37z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Poolmesser', payload '{"battery":100,"ec":699,"ec_max":20000,"ec_min":0,"free_chlorine":0.2,"free_chlorine_max":-0.1,"free_chlorine_min":0,"last_seen":"2025-05-20T18:32:37+02:00","linkquality":51,"orp":536,"orp_max":-1,"orp_min":-1,"ph":7.61,"ph_max":140,"ph_min":0,"salinity":0,"tds":349,"temperature":19.7}'
info 2025-05-20 18:32:37z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Poolmesser', payload '{"battery":100,"ec":699,"ec_max":20000,"ec_min":0,"free_chlorine":0.2,"free_chlorine_max":-0.1,"free_chlorine_min":0,"last_seen":"2025-05-20T18:32:38+02:00","linkquality":47,"orp":536,"orp_max":-1,"orp_min":-1,"ph":7.61,"ph_max":140,"ph_min":0,"salinity":0,"tds":349,"temperature":19.7}'
info 2025-05-20 18:32:38z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Poolmesser', payload '{"battery":100,"ec":699,"ec_max":20000,"ec_min":0,"free_chlorine":0.2,"free_chlorine_max":-0.1,"free_chlorine_min":0,"last_seen":"2025-05-20T18:32:38+02:00","linkquality":51,"orp":536,"orp_max":-1,"orp_min":-1,"ph":7.61,"ph_max":140,"ph_min":0,"salinit [... cut ...] ture":19.7}'
info 2025-05-20 18:32:46z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Poolmesser', payload '{"battery":100,"ec":699,"ec_max":20000,"ec_min":0,"free_chlorine":0.2,"free_chlorine_max":-0.1,"free_chlorine_min":0,"last_seen":"2025-05-20T18:32:46+02:00","linkquality":54,"orp":536,"orp_max":-1,"orp_min":-1,"ph":7.61,"ph_max":140,"ph_min":0,"salinity":0,"tds":349,"temperature":19.7}'`

but it is still sending updates, but i have to wait, if it is stopping ...
i will give you an update soon ..

OE5DRO, 2025-05-21:
i was wrong with the path for external_converters. 
/homeassistant/zigbee2mqtt/external_converters/ext_converter.js is the working one.
now i have to try again the 20 * 5 setting

pvprodk, 2025-06-04:
> And what if you change e.g. `// queryIntervalSeconds: 5 * 60,` to `queryIntervalSeconds: 20 * 60,` (make sure to also remove the `//`)

I had the same problems with missing updates, and have created the [external converter](https://gist.github.com/Koenkk/7cb09518bf346adbc94c6f77f372f644) and set the queryIntervalSeconds to 20 * 60, and can confirm that now it is working for me. Now I am getting frequent updates from the sensors such as temperature

[image]

jmbinette, 2025-06-04:
Hi,

Thanks to all for the external converter and work on stabilizing the salt readings.
I really wish we could tackle the calibration and be able to use exclusively the zigbee2mqtt converter.

<html>
<body>
<p data-start="319" data-end="472">After extensive tests, I think I figured the calibration out. I would need some help on writing the external converter but willing to test it and report!</p>
<h3 data-start="474" data-end="526">DP Mapping – Tuya Pool Sensor (BLE_YL01 variant)</h3>
<div class="_tableContainer_16hzy_1"><div tabindex="-1" class="_tableWrapper_16hzy_14 group flex w-fit flex-col-reverse">
DP ID | Hex | Type | Meaning / Usage | Scale | Notes / Source
-- | -- | -- | -- | -- | --
1 | 0x01 | value | TDS (ppm) read | 1 | Raw TDS value in ppm
2 | 0x02 | value | Temperature (°C) | 0.1 | Must divide by 10
7 | 0x07 | value | Battery (%) | 1 | Raw integer value
10 | 0x0A | value | pH | 0.01 | Raw pH value
11 | 0x0B | value | EC (µS/cm) | 1 | Raw EC value
66 | 0x42 | — | — | — | Not used in current firmware
67 | 0x43 | value | pH Calibration 1 | 0.01 | First calibration point (write after DP75)
68 | 0x44 | value | Current Calibration Region | — | 0 = EU, 1 = Asia
69 | 0x45 | value | Set Calibration Region | — | Send enum 0 or 1
70 | 0x46 | — | Unknown | — | Never seen in logs
71 | 0x47 | — | Unknown | — | Never seen in logs
72 | 0x48 | value | pH Calibration 2 | 0.01 | Used for step 2 & 3 (e.g., 700, 1000)
73 | 0x49 | value | EC Calibration (set) | 1 | Send calibration µS value
74 | 0x4A | value | ORP Calibration (set) | 1 | Send calibration mV value
75 | 0x4B | value | Calibration Start Trigger | — | Send 0x00000000 to begin calibration
76 | 0x4C | value | Salinity Calibration | — | No calibration button; seen only in passive data
101 | 0x65 | value | ORP (mV) | 1 | Live ORP reading
102 | 0x66 | value | Free Chlorine (mg/L) | 0.01 | Raw value, valid when ORP and pH in range
103 | 0x67 | value | pH Calibration 1 Feedback | 1 | Echo of first point
104 | 0x68 | value | TDS/Salinity Max | 1 | Configuration
105 | 0x69 | value | TDS/Salinity Min | 1 | Configuration
106 | 0x6A | value | pH Max | 0.1 | Divide by 10
107 | 0x6B | value | pH Min | 0.1 | Divide by 10
108 | 0x6C | value | EC Max | 1 | µS configuration
109 | 0x6D | value | EC Min | 1 | µS configuration
110 | 0x6E | value | ORP Max | 1 | mV configuration
111 | 0x6F | value | ORP Min | 1 | mV configuration
112 | 0x70 | value | Free Chlorine Max | 0.1 | Divide by 10
113 | 0x71 | value | Free Chlorine Min | 0.1 | Divide by 10
114 | 0x72 | value | pH Calibration 2 Feedback | 1 | Echo of second point
115 | 0x73 | value | EC Calibration Offset | 1 | Unused (observed empty in all logs)
116 | 0x74 | value | ORP Calibration Reference | 1 | Internal raw reading, likely ADC code
117 | 0x75 | value | Salinity (ppm) | 1 | Live salinity reading

<div class="sticky end-(--thread-content-margin) h-0 self-end select-none"><div class="absolute end-0 flex items-end"><span class="" data-state="closed"><button class="bg-token-bg-primary hover:bg-token-bg-tertiary text-token-text-secondary my-1 rounded-sm p-1 transition-opacity group-[:not(:hover):not(:focus-within)]:pointer-events-none group-[:not(:hover):not(:focus-within)]:opacity-0"><svg width="24" height="24" viewBox="0 0 24 24" fill="none" xmlns="http://www.w3.org/2000/svg" class="icon-md-heavy"><path fill-rule="evenodd" clip-rule="evenodd" d="M7 5C7 3.34315 8.34315 2 10 2H19C20.6569 2 22 3.34315 22 5V14C22 15.6569 20.6569 17 19 17H17V19C17 20.6569 15.6569 22 14 22H5C3.34315 22 2 20.6569 2 19V10C2 8.34315 3.34315 7 5 7H7V5ZM9 7H14C15.6569 7 17 8.34315 17 10V15H19C19.5523 15 20 14.5523 20 14V5C20 4.44772 19.5523 4 19 4H10C9.44772 4 9 4.44772 9 5V7ZM5 9C4.44772 9 4 9.44772 4 10V19C4 19.5523 4.44772 20 5 20H14C14.5523 20 15 19.5523 15 19V10C15 9.44772 14.5523 9 14 9H5Z" fill="currentColor"></path></svg></button></span></div></div></div></div>
<hr data-start="2905" data-end="2908">
<h3 data-start="2910" data-end="2933">Calibration Process</h3>
<ol data-start="2935" data-end="3500">
<li data-start="2935" data-end="3026">
<p data-start="2938" data-end="3026"><strong data-start="2938" data-end="2961">Trigger calibration</strong><br data-start="2961" data-end="2964">
→ Write <code data-start="2972" data-end="2984">0x00000000</code> to <strong data-start="2988" data-end="2996">DP75</strong> to initiate calibration mode.</p>
</li>
<li data-start="3028" data-end="3338">
<p data-start="3031" data-end="3112"><strong data-start="3031" data-end="3057">Send calibration value</strong><br data-start="3057" data-end="3060">
Immediately write the corresponding reference value:</p>
<ul data-start="3116" data-end="3338">
<li data-start="3116" data-end="3168">
<p data-start="3118" data-end="3168"><code data-start="3118" data-end="3124">DP67</code> → pH Calibration 1 (e.g. <code data-start="3150" data-end="3155">400</code> for pH 4.00)</p>
</li>
<li data-start="3172" data-end="3226">
<p data-start="3174" data-end="3226"><code data-start="3174" data-end="3180">DP72</code> → pH Calibration 2 and 3 (e.g. <code data-start="3212" data-end="3217">700</code>, <code data-start="3219" data-end="3225">1000</code>)</p>
</li>
<li data-start="3230" data-end="3284">
<p data-start="3232" data-end="3284"><code data-start="3232" data-end="3238">DP73</code> → EC Calibration (e.g. <code data-start="3262" data-end="3268">1413</code> for 1413 µS/cm)</p>
</li>
<li data-start="3288" data-end="3338">
[... cut ...]
<p data-start="3446" data-end="3500"><code data-start="3446" data-end="3453">DP116</code> → ORP calibration reference (likely ADC value)</p>
</li>
</ul>
</li>
</ol>
<hr data-start="3502" data-end="3505">
<p data-start="3507" data-end="3614">To be confirmed :<br data-start="3518" data-end="3521">
For pH region switching, use <code data-start="3550" data-end="3556">DP69</code> to set EU/Asia mode and check current setting via <code data-start="3607" data-end="3613">DP68</code></p>
</body>
</html>

Koenkk, 2025-06-05:
@pvprodk thanks, I've integrated the fix.

@jmbinette this is a matter of adding them to the datapoints: https://gist.github.com/Koenkk/7cb09518bf346adbc94c6f77f372f644#file-ext-js-L106 and add an expose for it.

E..g.

- `tuyaDatapoints`: add `[75, "trigger_calibration", tuya.valueConverter.raw],`
- `exposes`: add `e.enum("trigger_calibration", ea.SET, ["trigger"])`

jmbinette, 2025-06-06:
> [@pvprodk](https://github.com/pvprodk) thanks, I've integrated the fix.
>
> [@jmbinette](https://github.com/jmbinette) this is a matter of adding them to the datapoints: [...]

Hi @Koenkk 
Thanks a lot. I'm not really a coder, so I'm not fully familiar with what would be required for handling the full calibration process beyond exposing the datapoint. For example, I'm not sure how to properly structure the flow: triggering the calibration, then sending the pH reference values, EC, ORP, etc.

If you have any guidance or examples on how to structure these additional calibration datapoints in the converter, I'd be happy to test and provide feedback.

Thanks again for your help!

pvprodk, 2025-06-11:
Just a quick update
Today it is looking a lot better with regards to the Tuya update frequency.. Still not totally on par with Telldus, but a big improvement. The only problem is that I have not changed anything compared to the screenshot above from two days ago.

[image]

jmbinette, 2025-06-11:
> Same problem here. As I said before, there is something in the implementation of the update when accessing DP 105. When I use it through the tuya site every 5 minutes I trigger the following automation so that there is the update: alias: Medidor de Piscina description: "" triggers:
> 
> * minutes: /15
>   trigger: time_pattern
>   conditions:
> * condition: not
>   conditions:
>   
>   * condition: state
>     entity_id: sensor.medidor_cloro_e_sal_piscina_temperature
>     state: unavailable
>     actions:
> * repeat:
>   count: 24
>   sequence:
>   - data:
>   value: 0
>   target:
>   entity_id: number.medidor_cloro_e_sal_piscina_request
>   action: number.set_value
>   - delay:
>   seconds: 1
>   - data:
>   value: 1
>   target:
>   entity_id: number.medidor_cloro_e_sal_piscina_request
>   action: number.set_value
>   - delay:
>   seconds: 4
>   mode: single
> 
> It repeats 24 times with an interval of 1s for the value 0 and 4s for the value 1, repeats this cycle every 5 minutes, and so I get the updates. Maybe if we implement this similarly in the external converter, we have the update.

Just a quick note for anyone polling DP105 for updates:

Based on my tests, DP105 (TDS/Salinity Min) and related config DPs (104–113) aren’t meant to trigger readings — they’re for setting limits only. Using them to force updates might lead to unstable behavior.

I’ve mapped most known DPs from the sensor and would be happy to collaborate if anyone wants to dig deeper into improving the integration!

juslex, 2025-06-11:
> Just a quick note for anyone polling DP105 for updates:
>
> Based on my tests, DP105 (TDS/Salinity Min) and related config DPs (104–113) aren’t meant to trigger readings — they’re for setting limits [...]

It is quite strange that the DP that refers to TDS/Salinity Min when we put an integer 1 and then 0 allows it to update its status on Tuya. Would it be interesting to sniff the device during the execution of this automation to see some other detail that we didn't get? Unfortunately, I don't have a sniffer here.

OE5DRO, 2025-06-12:
> > as bad as usual:
> 
> What is your LQI ?

when pool is covert with foil about 10, but in daytime 45 and much more when foil is removed

xvolte, 2025-08-08:
Hello,

I think i finally make it work !

Thanks for the work of every one, it needed some small modifications on my side.

Please have a look here: https://community.home-assistant.io/t/pool-monitoring-device-yieryi-ble-yl01-zigbee-ph-orp-free-chlorine-salinity-etc/659545/130?u=xvolte

juslex, 2025-08-11:
> [@xvolte](https://github.com/xvolte) [@juslex](https://github.com/juslex)
> 
> > poolsensor_backlightvalue for determining time?
> 
> At what interval and how much time should there be between the 1 and 0? E.g.: wait 60 minutes, send backlight 1, wait 5 seconds, send backlight 0, repeat forever. Than we can add this to: [Koenkk/zigbee-herdsman-converters#9766](https://github.com/Koenkk/zigbee-herdsman-converters/pull/9766)

Dear @Koenkk 

It could be every 30 minutes repeating the cycle according to the section of automation that I am using via tuya local:

actions:
  - repeat:
      count: 24
      sequence:
        - action: button.press
          target:
            entity_id: button.medidor_cloro_e_sal_piscina_update
          data: {}
        - delay:
            seconds: 1
        - action: button.press
          target:
            entity_id: button.medidor_cloro_e_sal_piscina_update
          data: {}
        - delay:
            seconds: 4

There are 24 repetitions with Number 0 with one second duration and number 1 with 4 seconds. In this way, we have a satisfactory update of the data.

xvolte, 2025-08-13:
@Koenkk  i have an issue with the PH sensor.

If the value is 2 digit, than it should be divided by 10. if the value is 3 digits, then it should be divided by 100.

Is it possible to do so ?

[image]

Koenkk, 2025-08-13:
yes you can do so with a custom `from` converter, example: https://github.com/Koenkk/zigbee-herdsman-converters/blob/8233e5726f7dfb912a46462747eb2eeb723214b5/src/devices/tuya.ts#L15575

`v` is the value here, I guess you need something like this (not tested):

```js
                        from: (v: number) => {
                            if (v > 99) {
                                return v / 10;
                            }
                            return v / 100;
                        },
```

xvolte, 2025-08-14:
hello,

i tried and tried and retried, but i keep having the same error log:

`	z2m: Invalid external converter 'ext_converters.js' was ignored and renamed to prevent interference with Zigbee2MQTT. (Unexpected token ':')`

this is the file i use, but that is not accepted,  can you help me ?

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const color = require('zigbee-herdsman-converters/lib/color');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const lumi = require('zigbee-herdsman-converters/lib/lumi');
const utils = require('zigbee-herdsman-converters/lib/utils');
const {logger} = require('zigbee-herdsman-converters/lib/logger');
const {develcoModernExtend} = require('zigbee-herdsman-converters/lib/develco');
const m = require('zigbee-herdsman-converters/lib/modernExtend');

const NS = 'extconverter';

const definition =  {
    fingerprint: tuya.fingerprint("TS0601", ["_TZE200_v1jqz5cy", "_TZE200_d9mzkhoq"]),
    model: "BLE-YL01",
    vendor: "Tuya",
    description: "Smart WiFi Zigbee chlorine meter CUSTOM7",
    whiteLabel: [tuya.whitelabel("Tuya", "YK-S03", "Smart pH and Chlorine Tester for Swimming Pool", ["_TZE200_d9mzkhoq"])],
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    // Don't query too often. Values are not always updated. https://github.com/Koenkk/zigbee2mqtt/issues/18704
    onEvent: tuya.onEvent({
        queryOnDeviceAnnounce: true,
        queryIntervalSeconds: 10 * 60,
    }),
    configure: tuya.configureMagicPacket,
    exposes: [
        e.numeric("tds", ea.STATE).withUnit("ppm").withDescription("Total Dissolved Solids"),
        e.temperature(),
        e.battery(),
        e
            .numeric("ph", ea.STATE)
            .withUnit("pH")
            .withDescription(
                "pH value, if the pH value is lower than 6.5, it means that the water quality " +
                    "is too acidic and has impurities, and it is necessary to add disinfectant water for disinfection",
            ),
        e.numeric("ec", ea.STATE).withUnit("µS/cm").withDescription("Electrical conductivity"),
        e
            .numeric("orp", ea.STATE)
            .withUnit("mV")
            .withDescription(
                "Oxidation Reduction Potential value. If the ORP value is above 850mv, it " +
                    "means that the disinfectant has been added too much, and it is necessary to add water or change the water for neutralization. " +
                    "If the ORP value is below 487mv, it means that too little disinfectant has been added and the pool needs to be disinfected again",
            ),
        e
            .numeric("free_chlorine", ea.STATE)
            .withUnit("mg/L")
            .withDescription(
                "Free chlorine value. The water in the swimming pool should " +
                    "be between 6.5-8ph and ORP should be between 487-840mv, and the chlorine value will be displayed normally. Chlorine will not " +
                    "be displayed if either value is out of range",
            ),
        e.numeric("ph_max", ea.STATE_SET).withUnit("pH").withDescription("pH maximal value").withValueMin(0).withValueMax(14),
        e.numeric("ph_min", ea.STATE_SET).withUnit("pH").withDescription("pH minimal value").withValueMin(0).withValueMax(14),
        e
[... 59 lines cut ...]
                        {
                            return tuya.valueConverter.divideBy100.from(v)
                        }
                        return tuya.valueConverter.divideBy10.from(v)
                    },
                },
            ],
            [11, "ec", tuya.valueConverter.raw],
            [101, "orp", tuya.valueConverter.raw],
            [102, "free_chlorine", tuya.valueConverter.raw],
            [105, 'backlightvalue', tuya.valueConverter.raw],
            [106, "ph_max", tuya.valueConverter.divideBy10],
            [107, "ph_min", tuya.valueConverter.divideBy10],
            [108, "ec_max", tuya.valueConverter.raw],
            [109, "ec_min", tuya.valueConverter.raw],
            [110, "orp_max", tuya.valueConverter.raw],
            [111, "orp_min", tuya.valueConverter.raw],
            [112, "free_chlorine_max", tuya.valueConverter.raw],
            [113, "free_chlorine_min", tuya.valueConverter.raw],
            [117, "salinity", tuya.valueConverter.raw],
        ],
    },
};

module.exports = definition;`

Koenkk, 2025-08-15:
Ah whoops, `from: (v: number) => ` should be `from: (v) => `

jl1990, 2025-08-20:
I have a chemical kit for measuring the chlorine level, the typical one for the pools. I know my pool is more or less in the correct range because of it.
Also, it would be possible to have 2 mg/l if you add much chlorine in a pool (for example in public swimming pools sometimes they add more chlorine), but 4 is a wild value... which seem to happen in my graph since the upgrade.
I also didn't add any extra chlorine to it for some days, and it's happily changing from 1 to 4, which as mentioned, is a huge difference.
So I would say that the values are wrong for sure.

juslex, 2025-08-20:
> > Have you measured the value with other devices? I'm also measuring with a Poollab 2.0 and the Tuya sensor is always about just a tenth of the Poollab 2.0 value. I do trust in the Poollab measurements which match to some other shake-test results.
> 
> I have a chemical kit for measuring the chlorine level, the typical one for the pools. I know my pool is more or less in the correct range because of it. Also, it would be possible to have 2 mg/l if you add much chlorine in a pool (for example in public swimming pools sometimes they add more chlorine), but 4 is a wild value... which seem to happen in my graph since the upgrade. I also didn't add any extra chlorine to it for some days, and it's happily changing from 1 to 4, which as mentioned, is a huge difference. So I would say that the values are wrong for sure.
> 
> The device was also reporting correct values when I used the tuya gateway some time ago, so it's not a faulty device problem

I believe it's not a problem of zigbee2mqtt, but of its sensor. What is the value of its ORP and PH?

BlueSnake830, 2025-08-20:
I'm using the latest release, not the dev version. So my statement might be misleading, according to your finding.

xvolte, 2025-08-21:
Hello,
You are correct in the way that in my fix, i removed the division by 10, hence you are right to say values seems multiplied by 10.
However, i verified the values with a pool test like this, and values should be between 0 and 10 (if i'm not mistaken), maybe i understood wrongly ?

[image]

If it needs indeed to be divided by 10, i can again do a fix, but i'm really not sure.

jl1990, 2025-08-21:
> Hello, You are correct in the way that in my fix/commit, i removed the division by 10, hence you are right to say values seems multiplied by 10. However, i verified the values with a pool test like the one below, and it says that values should be between 0 and 10 (if i'm not mistaken), maybe i understood wrongly ?
> 
> [image]
> 
> If it needs indeed to be divided by 10, i can again do a fix, but i'm really not sure. Happy to change it anyway if I made a mistake :)

Maybe you are correct and your changes fixed chlorine values, I guess it depends on the resolution of the sensor (if it can measure changes of 0.1 or just 1). It would be nice to compare with tuya gateway connected and see what values are displayed there.
Anyway @BlueSnake830 commented that he's getting readings of a tenth of Poollab 2.0. So maybe with your changes this is fixed now.
The problem I see is that values change very often from 2 to 4 in my case without adding more chlorine to the pool. Maybe is just that the sensor reliability is not good...

juslex, 2025-08-21:
> > > > Have you measured the value with other devices? I'm also measuring with a Poollab 2.0 and the Tuya sensor is always about just a tenth of the Poollab 2.0 value. I do trust in the Poollab measurements which match to some other shake-test results.
> > > 
> > > 
> > > I have a chemical kit for measuring the chlorine level, the typical one for the pools. I know my pool is more or less in the correct range because of it. Also, it would be possible to have 2 mg/l if you add much chlorine in a pool (for example in public swimming pools sometimes they add more chlorine), but 4 is a wild value... which seem to happen in my graph since the upgrade. I also didn't add any extra chlorine to it for some days, and it's happily changing from 1 to 4, which as mentioned, is a huge difference. So I would say that the values are wrong for sure.
> > > The device was also reporting correct values when I used the tuya gateway some time ago, so it's not a faulty device problem
> > 
> > 
> > I believe it's not a problem of zigbee2mqtt, but of its sensor. What is the value of its ORP and PH?
> 
> ORP and PH seem to be more or less consistent in my case:
> 
> <img alt="Image" width="994" height="857" src="https://private-user-images.githubusercontent.com/3340533/480243216-f67d87e9-0ed9-4933-971b-6a5e2977e8c5.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTU3ODE2NDgsIm5iZiI6MTc1NTc4MTM0OCwicGF0aCI6Ii8zMzQwNTMzLzQ4MDI0MzIxNi1mNjdkODdlOS0wZWQ5LTQ5MzMtOTcxYi02YTVlMjk3N2U4YzUucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDgyMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTA4MjFUMTMwMjI4WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NzUxMjNmNzgyYjhjMGQwYWI3NDMyM2JlNzU0ZjMzMmM0NWFjMzE0NGI2ZWE1ZjdiMGE1ZGVmMTRlZmZhMTAyNiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.MpawrwYbL7v3iJr3dFOUVQbJXy2CsXwpjx6oqFY0XOo"> <img alt="Image" width="995" height="873" src="https://private-user-images.githubusercontent.com/3340533/480243236-a3ff8af4-4d9f-4cac-823b-8fd7df7a6750.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTU3ODE2NDgsIm5iZiI6MTc1NTc4MTM0OCwicGF0aCI6Ii8zMzQwNTMzLzQ4MDI0MzIzNi1hM2ZmOGFmNC00ZDlmLTRjYWMtODIzYi04ZmQ3ZGY3YTY3NTAucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDgyMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTA4MjFUMTMwMjI4WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NjRmZWNmZWUyY2NhY2Q1MzcxMGM4MDYyYzQwNWU4YWZlMmEyYmU3ZWEwZjRlNjdiMzhhMzc0NmUzZTFhM2U0YyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.BIkv7HUFOrZd05QC9mc0nHqZLDbwGOdCcuDHCo45MCQ">
> I don't think it's a sensor problem, because as I mentioned previously, it works properly with tuya gateway. Previous values before the update also made more sense... ~0.1 mg/l changes in values are plausible, but changes of 1 mg/l like this, not so much:
> 
> <img alt="Image" width="1002" height="856" src="https://private-user-images.githubusercontent.com/3340533/480244192-192a9e2a-a4d7-45b3-868c-75075cb046c2.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTU3ODE2NDgsIm5iZiI6MTc1NTc4MTM0OCwicGF0aCI6Ii8zMzQwNTMzLzQ4MDI0NDE5Mi0xOTJhOWUyYS1hNGQ3LTQ1YjMtODY4Yy03NTA3NWNiMDQ2YzIucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDgyMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTA4MjFUMTMwMjI4WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ODg4NzNiNWZiZDVhMjM1ZDdhMjI1OTE4OWVhMzdmNmM5ZWU1NDVkZTg2NGEzZjFlMTE1M2Y3ZDQ3ZmI1YTcwNyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.30uEDQfNQvppyGHFwVPMv9JxQT5jVDbACxp1FGXlg1M">
> Because of this, I still believe chlorine should be divided by 10...
> 
> by the way [@BlueSnake830](https://github.com/BlueSnake830) are you using development version? I am referring to the changes done recently on the dev branch (which seem to have multiplied chlorine by 10)

I believe it needs to be divided by 10. I have the Tuya hub and my PH and ORP values are close to yours and I have 0.2 free chlorine. Just remembering that the measurement of free chlorine of this device is indirect through ORP, PH and temperature data. As your ORP and PH data are close to me and 3 are showing, it should actually be 0.3.

xvolte, 2025-08-21:
Before i make the change, just to be 100% sure, can you do a check with a "test stripe", so that we go to the bottom of this ?

thanks in advance,

xvolte, 2025-08-31:
PR made: [Update tuya.ts #9891](xvolte:patch-4)
i updated the min and max value according to @nonameboy007 
i did revert the free chlorine, so it is now divided by 10 as several people mentioned it was incorrect.

I did NOT however, changed the item "Code" as we use the dpid, and it was already working like that.
i don't know if the "Code" is something that needs to be the same, or if the onlything that needs to be the same are the dpids.
@Koenkk 

Thanks to all!

juslex, 2025-09-02:
I was waiting for the stable versio  to come out to be able to test the device after the last PR and now I confirm that everything is working well. My version of the device is the updated version.