Koenkk/zigbee2mqtt #23956: zh:zstack:znp: Error while parsing to ZpiObject 'Error: CommandID '159' from subsystem '5' not found'

harvindhillon, 2024-09-13:
### What happened?

This used to be an issue on startup on earlier versions, however seems like this has returned in [1.40.1]

Old ticket https://github.com/Koenkk/zigbee2mqtt/issues/3363

### What did you expect to happen?

No error on startup

### How to reproduce it (minimal and precise)

Do a clean container startup

### Zigbee2MQTT version

1.40.1

### Adapter firmware version

20240710

### Adapter

SLZB-06

### Setup

Docker Container on x86 based Intel NUC

### Debug log

zh:zstack:znp: Error while parsing to ZpiObject 'Error: CommandID '159' from subsystem '5' not found'

Trashwarez, 2024-09-13:
Nope. It's back since 1.37. I never got rid of it. See https://github.com/Koenkk/zigbee2mqtt/issues/22544

Koenkk, 2024-09-14:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-09-14 by Koenkk