Koenkk/zigbee2mqtt #23959: PTM216Z

chris-1243, 2024-09-13:
### What happened?

While trying to pair a PTM216Z, I get this error:

```
[2024-09-13 17:30:46] error: 	zh:controller:greenpower: AssertionError [ERR_ASSERTION]: Failed to find green power proxy device
    at GreenPower.sendPairingCommand (/app/node_modules/zigbee-herdsman/src/controller/greenPower.ts:95:19)
    at GreenPower.onZclGreenPowerData (/app/node_modules/zigbee-herdsman/src/controller/greenPower.ts:181:25)
    at Controller.onZclPayload (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:690:13)
``` 

It might be related to the device (IKEA - [LED2103G5](https://www.zigbee2mqtt.io/devices/LED2103G5.html#ikea-led2103g5)) to pair it. As the PTM216Z is able to be configured via NFC, I changed its configuration in order to have it recognized as a `Greenpower_2` instead of `Greenpower_7` and it successfully paired. I tried with some HUE devices but not all are able to translate ZGP frames (`Greenpower_7`) correctly. [4034031P7](https://www.zigbee2mqtt.io/devices/4034031P7.html#philips-4034031p7) is able but as `broadcast` only...

The main problem I have is in `greenpower_2`, this device (at least mine) is not able to send any commands, only `press_or_release_all` or identify while sniffing traffic.

I have the same issue while using `ember` if not paired directly to the coordinator.

### Zigbee2MQTT version

1.40.1-dev

### Adapter firmware version

20240710 - 7.4.3

### Adapter

Sonoff Dongle-P - Sonoff Dongle-E

### Setup

Z2M in docker (Pi4)

Koenkk, 2024-09-14:
Could you try with the following change:

```bash
docker exec -it CONTAINER_NAME sh
apk add nano
nano /app/node_modules/zigbee-herdsman/dist/controller/greenPower.js
# Change `if (dataPayload.wasBroadcast) {` to `if (true) {`
# save and exit nano
exit
docker restart CONTAINER_NAME
```

chris-1243, 2024-09-14:
I did multiple tries with the same IKEA - [LED2103G5](https://www.zigbee2mqtt.io/devices/LED2103G5.html#ikea-led2103g5) and an Ubisys [C4](https://www.zigbee2mqtt.io/devices/C4.html#ubisys-c4) and the PTM216Z paired successfully on both devices. It is fully working...

Here a log 
[log_2024.09.14.txt](https://github.com/user-attachments/files/17003707/log_2024.09.14.txt)

Koenkk, 2024-09-15:
- So everything works now? Including all the actions?
- When commissioning, you will see `"gppNwkAddr":1282`, does e.g `1282` in this case match the network address of any of your devices?

Koenkk, 2024-09-16:
Without the fix, does the issue also happen when pairing via the C4? If not, I guess it's due to the LED2103G5?

chris-1243, 2024-09-16:
Without your fix, I have only two routers able to translate the PTM216Z frames and no latency
- [4034031p7](https://www.zigbee2mqtt.io/devices/4034031P7.html#philips-4034031p7)
- [8718696548738](https://www.zigbee2mqtt.io/devices/8718696548738.html#philips-8718696548738)
The main problem with those, it seems there is a bug in the firmware and the PTM never pairs as `unicast`...

The Ubisys C4, S2 nor some HUE, IKEA devices and a Dongle-E as a router are able to do the job without your fix. The drawback is the slight latency which is less than 0.5s

chris-1243, 2025-03-27:
This issue is now definitely solved. May we change the status as completed ?

https://github.com/Koenkk/zigbee-herdsman/pull/1369

Thanks @Nerivec for his hard work

closed 2025-03-27 by Nerivec