Koenkk/zigbee2mqtt #23960: Don't upgrade Raspberry PI 5 kernel if you use the integrated serial interface (/dev/ttyAMA)

mariusmotea, 2024-09-13:
### What happened?

A bug was introduced in the last Raspberry PI OS kernel that will make ZigBee2MQTT to freeze when connecting to the zigbee controller on the integrated Raspberry PI serial port.

### What did you expect to happen?

N/A

### How to reproduce it (minimal and precise)

Update the kernel and start Zigbee2MQTT

### Zigbee2MQTT version

1.40.1

### Adapter firmware version

20240710

### Adapter

TI CC2652P

### Setup

Raspberry PI OS 64 bit

Koenkk, 2024-09-13:
Thanks for notifying, pinned it!

trizmark, 2024-10-14:
Any info as to what's the latest working version? I just finished migrating my whole smart home setup to a RPi5....

mariusmotea, 2024-10-14:
They provided two workarounds that i tested and both are working. See https://github.com/raspberrypi/linux/issues/6357

arnonsha, 2024-12-14:
> I think not, i update yesterday to the last kernel and it is working. I will close this issue.

Which kernel version is that ?

mariusmotea, 2024-12-14:
6.6.62+rpt-rpi-2712

Alexofhiswalnut, 2024-12-27:
Hello, I have the same issue with my Pi 4 - 4GB, which has been running Home Assistant for two years. However, since October, the Sonoff E dongle disconnects six times a day, and Z2Mqtt restarts itself to function normally again.

I’ve migrated everything to a NUC while waiting to see if the Pi 4 will work again or not...

arnonsha, 2024-12-27:
Can anyone say if the latest hassos(14.1) fixes this issue ? 

Thanks

Alexofhiswalnut, 2024-12-27:
I completely reset my Pi 4 from scratch between December 21 and 22, 2024, but the problem was still there.
Version 14.1 was released on December 19.

supersuphot, 2025-02-10:
Same problem here on Raspberry Pi 3

closed 2025-04-19 by github-actions[bot]