Koenkk/zigbee2mqtt #23961: [New device support]: IKEA INSPELNING smart plug with power monitoring

QempZor, 2024-09-13:
### Link

https://www.ikea.com/us/en/p/inspelning-plug-smart-energy-monitor-90569846/

### Database entry

{"id":53,"type":"Router","ieeeAddr":"0xecf64cfffef2b718","nwkAddr":27855,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"INSPELNING Smart plug","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,5,6,8,1794,2820,4096,64636,64645],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"INSPELNING Smart plug","manufacturerName":"IKEA of Sweden","powerSource":1,"zclVersion":8,"appVersion":2,"stackVersion":114,"hwVersion":1,"dateCode":"20240215","swBuildId":"2.4.34"}},"haElectricalMeasurement":{"attributes":{"acPowerDivisor":10,"acPowerMultiplier":1,"acCurrentDivisor":1000,"acCurrentMultiplier":1,"acVoltageDivisor":10,"acVoltageMultiplier":1,"activePower":0,"rmsCurrent":3,"rmsVoltage":236}},"seMetering":{"attributes":{"divisor":1000,"multiplier":1,"currentSummDelivered":[0,77]}},"genLevelCtrl":{"attributes":{"onLevel":255,"currentLevel":254}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0xb43522fffe12b3b4","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0xb43522fffe12b3b4","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1291,"minRepIntval":10,"maxRepIntval":65000,"repChange":50},{"cluster":2820,"attrId":1288,"minRepIntval":10,"maxRepIntval":65000,"repChange":50},{"cluster":2820,"attrId":1285,"minRepIntval":10,"maxRepIntval":65000,"repChange":50},{"cluster":1794,"attrId":0,"minRepIntval":10,"maxRepIntval":65000,"repChange":[0,100]}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":114,"hwVersion":1,"dateCode":"20240215","swBuildId":"2.4.34","zclVersion":8,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1726245436588}

### Zigbee2MQTT version

1.40.1

### Comments

Just picked this plugs up at the Ikea in Amsterdam. They're quite new, no infopage on the Ikea site (NL) 
I tried to follow the "How to add support" page, but.. well,.. to complicated for this noob :)

However I want to help so here is al the information I know.

Out of the box, in Z2M, a lot works already quite well. See the results below.

Extra;
Some Tweakers forum information (Dutch): https://gathering.tweakers.net/forum/list_message/80105532#80105532

### External definition

```shell
const {light, electricityMeter} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['INSPELNING Smart plug'],
    model: 'INSPELNING Smart plug',
    vendor: 'IKEA of Sweden',
    description: 'Automatically generated definition',
    extend: [light(), electricityMeter()],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

Out of the box in:
- ON OFF switch works as expected.
- There is an "Effect" thing in HA/Z2M: Dont think it needs to be there for plug
- There is an "Brightness slider" Dont think it needs to be there for plug
- Energy Sensor is a sum of total consumed energy. Don't know if correct. See next point.
- Power monitoring works kinda. When I charge my iPad it reads around 20W wich seems to be good. However if i plug in my toaster (wich is around 2000w)  it reports around 209w) So there seems like a decimal error thing? 
- Same problem for voltage. It reports 23,3-23,6v (when pluged in to a 230v outlet ofcourse)
- Probably the same thing for the amperage.. Didnt test that.

Configuration page (in HA) shows a Power-on behavior wich works as expected

Rembock, 2024-09-13:
I made this definition:

 ### External definition
 ```shell
const {electricityMeter, onOff, identify} = require('zigbee-herdsman-converters/lib/modernExtend');
const {addCustomClusterManuSpecificIkeaUnknown, ikeaOta} = require('zigbee-herdsman-converters/lib/ikea')

const definition = {
    zigbeeModel: ['INSPELNING Smart plug'],
    model: 'E2206',
    vendor: 'IKEA',
    description: 'INSPELNING Smart plug',
    extend: [addCustomClusterManuSpecificIkeaUnknown(), onOff(), identify(), ikeaOta(), electricityMeter()],
    meta: {},
};

module.exports = definition;
 ```
 
 ### What does this external definition fix?
 
 * Removes "Effect"
 * Removes "Brightness slider"

#### Also:
 * Energy Sensor is a sum of total consumed energy looks correct, I have another energymeter that does the same
 * Power monitoring works like it should, the voltage differs depending of the connected device onto your plug & should never use the full 230v. 
 * Same applies to voltage and amperage

QempZor, 2024-09-14:
clarifying the voltage and wattage.
1500w toaster connected shows this:
[image]

Rembock, 2024-09-16:
You're right, I think I've extended the external definition correctly now. Can you try if this external definition works for you as well?

 ### External definition
 ```shell
const {} = require('zigbee-herdsman-converters/lib/modernExtend');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const {electricityMeter, onOff, identify} = require('zigbee-herdsman-converters/lib/modernExtend');
const {addCustomClusterManuSpecificIkeaUnknown, ikeaOta} = require('zigbee-herdsman-converters/lib/ikea');

const definition = {
    zigbeeModel: ['INSPELNING Smart plug'],
    model: 'E2206',
    vendor: 'IKEA',
    description: 'INSPELNING Smart plug',
    extend: [addCustomClusterManuSpecificIkeaUnknown(), onOff(), identify(), ikeaOta()],      
    fromZigbee: [fz.electrical_measurement, fz.metering],
    toZigbee: [],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['haElectricalMeasurement', 'seMetering']);
        await reporting.activePower(endpoint);
        await reporting.rmsVoltage(endpoint);
        await reporting.rmsCurrent(endpoint); 
        await reporting.currentSummDelivered(endpoint);               
        endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {
            acPowerMultiplier: 1,
            acPowerDivisor: 10,
            acVoltageMultiplier: 1,
            acVoltageDivisor: 1,
            acCurrentMultiplier: 1,
            acCurrentDivisor: 1000,            
        });
        endpoint.saveClusterAttributeKeyValue('seMetering', {divisor: 100, multiplier: 1});        
        device.save();
    },
    exposes: [e.power(), e.voltage(), e.current(), e.energy()],
};

module.exports = definition;
 ```
This is my result:
[image]
I am not sure if the number of kWh is correct yet

QempZor, 2024-09-17:
Voltage is correct now. I think the total sum of Energy already was correct
Power is still wrong when wattage reach >1000w
[image]

ronaldevers, 2024-09-17:
@Rembock the "effect" actually worked on the light on the plug. I could for example make the light "breathe". The brightness did not appear to do anything. So I don't think the "effect" should be removed.

itkama, 2024-09-18:
These are also available in Germany now. Just bought 2. I will take a look if your external controller works for me tomorrow @Rembock. Any clue on how to get these into pairing mode?

basvdploeg, 2024-09-18:
@itkama You can put them in pairing mode by pressing and holding the small button next to the power button. You probably need a pointy object to do so.

itkama, 2024-09-18:
Okay, so I couldn't wait until tomorrow to pair it (thanks @basvdploeg):
[image]
Seems to be kind of working out of the box already? Voltage looks about right. On/Off working. Not sure about the power measurement, as this is the first time I'm measuring my vacuum bot.

Tried the other one with a pretty efficient GAN-USB-Charger to charge a powerbank that shows how much is going in. While the plug reported 30 to 31W, while the powerbank was reporting around 25W of input power. So for low power it may be kind of accurate.

What does the external converter really change now though? The power number atleast on low wattages seems to be looking correct? 🧐

Rembock, 2024-09-18:
Made a new definition, this one fixes the issue's with Energy and Power @QempZor 
@ronaldevers i will try to take a look if there's a dimmer functionality hidden somewere, 

 ### External definition
 ```shell
const {} = require('zigbee-herdsman-converters/lib/modernExtend');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const {electricityMeter, onOff, identify} = require('zigbee-herdsman-converters/lib/modernExtend');
const {addCustomClusterManuSpecificIkeaUnknown, ikeaOta} = require('zigbee-herdsman-converters/lib/ikea');

const definition = {
    zigbeeModel: ['INSPELNING Smart plug'],
    model: 'E2206',
    vendor: 'IKEA',
    description: 'INSPELNING Smart plug',
    extend: [addCustomClusterManuSpecificIkeaUnknown(), onOff(), identify(), ikeaOta()],      
    fromZigbee: [fz.electrical_measurement, fz.metering],
    toZigbee: [],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['haElectricalMeasurement', 'seMetering']);
        await reporting.activePower(endpoint);
        await reporting.rmsVoltage(endpoint);
        await reporting.rmsCurrent(endpoint); 
        await reporting.currentSummDelivered(endpoint);               
        endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {
            acPowerMultiplier: 1,
            acPowerDivisor: 1,
            acVoltageMultiplier: 1,
            acVoltageDivisor: 1,
            acCurrentMultiplier: 1,
            acCurrentDivisor: 1000,            
        });
        endpoint.saveClusterAttributeKeyValue('seMetering', {divisor: 1000, multiplier: 1});        
        device.save();
    },
    exposes: [
        e.power(), 
        e.voltage(), 
        e.current(), 
        e.energy()
    ],    
};

module.exports = definition;
 ```
[image]

Rembock, 2024-09-18:
@ronaldevers I get Data confirm errors in my log when I add light to the extended definition, so I decided not to configure the effect

LeoSum8, 2024-09-19:
This looks very promising! I think I'll need to go out to eat some hotdogs for lunch pretty soon :)
Can anyone share how often the values are updated? Thanks!

QempZor, 2024-09-19:
> Okay, so I couldn't wait until tomorrow to pair it (thanks @basvdploeg): [...]

Power readings below 1000w are fine (out of the box), But above 1000W everything is shown as 100,0w (or f.e. 132,4w)

With the ext. def. from Rembock, everything above 1000W seems to be good; but below 1000w is then show 10x too much (7,2w LED bulb reads 72w power... )

Also; I saw someone on Tweakers (dutch tech forum) who bought 2 of these plugs and didn't show any data. He changed them at another Ikea store and then one of them is showing the correct voltage, and one is shown 10times to high voltage, with the same settings/ext.def.)  So there might also be a (small) problem within the units itself. (<- Thats just speculation)

Rembock, 2024-09-19:
I am almost certain that this must be a firmware issue, I hope Ikea will come up with an update soon.

itkama, 2024-09-19:
@Rembock Firmware versions on both my plugs is 2.4.34 with a Firmware date of 20240215.
@QempZor there also is the TRETAKT plug. It looks really similar - but it has no power readings. Costs 2€ less here in Germany. Maybe he just got the wrong plugs? 😅

basvdploeg, 2024-09-19:
> @Rembock Firmware versions on both my plugs is 2.4.34 with a Firmware date of 20240215. @QempZor there also is the TRETAKT plug. It looks really similar - but it has no power readings. Costs 2€ less here in Germany. Maybe he just got the wrong plugs? 😅

No, they were the INSPELNING. Both showed only 1 (different) reading, and all the other values stayed empty. I tried re-pairing them multiple times, with and without external definition, but the result stayed the same.

When I switched them for two new version (from another IKEA store), I got all the data instantly. So it definitely looked like a problem in the plugs themselves.

technois2, 2024-09-19:
I bought 3 INSPELNING, one of them reports as TRETAKT (2204). I checked the label on the plug, it really says 2206 (INSPELNING).

busene, 2024-09-20:
I bought 3 yesterday and added them to Zigbee2MQTT using the external definition from @Rembock . Somehow 1 shows all the values x10 (so my outlet is now outputting 2340Volts), but the other 2 show correct values. Anybody that knows whats going on?

itkama, 2024-09-20:
@busene do your 3 plugs show different firmware versions?

busene, 2024-09-20:
@itkama nope, all show the same version

Rembock, 2024-09-20:
Instead of using the power sensor from the E2206, I'm now using a sensor with the [template integration](https://www.home-assistant.io/integrations/template/) as a helper from the UI in Home Assistant.

[image]

With this method you can 'tune' each plug to display the correct number of watts until Ikea pushes an firmware update that hopefully fixes the power sensor.
I also saw support voor the Tretakt listed [here](https://ww8.ikea.com/ikeahomesmart/releasenotes/releasenotes.html) since the 3rd of May, I expect the Inspelning to be listed here soon as well.

It also seems like that the latest firmware version is 2.4.45, mine are both 2.4.34.

basvdploeg, 2024-09-20:
> I bought 3 yesterday and added them to Zigbee2MQTT using the external definition from @Rembock . Somehow 1 shows all the values x10 (so my outlet is now outputting 2340Volts), but the other 2 show correct values. Anybody that knows whats going on?

This is the same that I experienced with mine.

busene, 2024-09-20:
@Rembock mine are also 2.4.34, is there a way to update them without the ikea hub?

QempZor, 2024-09-20:
I just bought a Dirigera and indeed there is an update. v2.4.45 is out
With this new update; out of the box the voltage and current are correct in Z2M.

However the Power problem is still there. Above 1000w is shown as 100,0w
I checked the power in the Ikea Home app and there it's shown correctly. Above 1000w and also below 1000w.

basvdploeg, 2024-09-21:
The OTA update for the plug is available here:
https://fw.ota.homesmart.ikea.com/check/update/prod

But I’m not sure how it can be used through Z2M.

notgood, 2024-09-21:
Picked one today, comes with 2.4.34 firmware, would welcome any hints on updating it without Ikea hub.
It seems Zigbee2mqtt only supports the Tradfri OTA feed, not the Dirigera one.

Mine INSPELNING also came with a barely working and unreliable button, only third of keypresses actually did something.

Disassembled it: breakout board with a button and LED is pretty much floating midair, rather unfortunate and sloppy engineering design IMHO.
[image]

Used a piece of plastic zip tie as a wedge between breakout board and relay, with a bit of electric tape to calibrate width.
[image]

folfy, 2024-09-22:
I bought the new INSPELNING plug here in Austria two days ago, and experienced the same issue that power >=1000W is shown as 1/10th of the actual value (on ZHA).

I checked the Measurement Cluster for related attributes, and could figure out that the plug is dynamically adjusting the "ac_power_divisor" between 10 (<1000W) and 1 (>=1000W), which probably is not updated/reflected by ZHA or Z2M in the shown avlue accordingly:
[image]

The off-value for the voltage seems to be fixed for some with the new firmware, see https://github.com/zigpy/zha-device-handlers/issues/3374#issuecomment-2365316168.

I'm currently looking into fixing/expanding the issue/quirk for ZHA, maybe someone here knows how to fix this in Z2M tho ;)

Rembock, 2024-09-22:
@folfy thanks for sharing. This got me thinking, your findings are aligned with wat we're experiencing. 
After some reading this weekend, I also realised that my template sensor is measuring VA (for DC) and not VAR (for AC). To measure VAR I need to know the power factor. Sadly, a power factor cannot be read from the E2206, and after that, I was out of ideas. But with your findings I have another idea.
With the latest firmware update we now know that only the power is off above 1000W. So I'm thinking of creating some sort of conditional statement that changes the power factor based on the power level.

Finally we also need to update the ota file. I'm not sure, but I think I saw an update file override function somewhere in the settings of z2m that can be used as a workaround for now.

If I'm lucky, I will have some time left tonight to take a shot at this. Will share something if everything works correctly.

notgood, 2024-09-22:
I've managed to update firmware to 2.4.45 using z2m, here are the steps.

1. Download https://fw.ota.homesmart.ikea.com/files/inspelning-smart-plug-soc_release_prod_v33816645_02579ff4-6fec-42f6-8957-4048def87def.ota
2. Create inspelning_OTA_index.json with the following contents
~~~
[
    {
        "url": "inspelning-smart-plug-soc_release_prod_v33816645_02579ff4-6fec-42f6-8957-4048def87def.ota"
    }
]
~~~
3. Create external definition file inspelning.js, contents in above post by @Rembock. Temporary comment out lines referencing ikeaOta and add zigbeeOta instead.
~~~
...
// const {addCustomClusterManuSpecificIkeaUnknown, ikeaOta} = require('zigbee-herdsman-converters/lib/ikea');
...
// extend: [addCustomClusterManuSpecificIkeaUnknown(), onOff(), identify(), ikeaOta()],
....
ota: ota.zigbeeOTA,
....
~~~
4. Add references to external definition and OTA override to configuration.yaml
~~~
external_converters:
    - inspelning.js
ota:
  zigbee_ota_override_index_location: inspelning_OTA_index.json
~~~

All files must be located in the same directory as configuration.yaml. Restart z2m, you now should be able to update INSPELNING firmware.

After successful update, rollback OTA related changes in your external definition file and restart z2m again.

folfy, 2024-09-22:
> @folfy thanks for sharing. This got me thinking, your findings are aligned with wat we're experiencing. After some reading this weekend, I also realised that my template sensor is measuring VA (for [...]

Yeah, I forgot to remark on the issue of your previous approach, but you noticed anyway. Still, any conditional check based just on the power or current is gonna be a guess at best. You can't 100% certainly determin if the ~100.0-200.0W that you're reading are actually 1000-2000W or just 100-200W with a really bad power factor. The latter shouldn't really happen _normally_, having such high reactive load with barely any true power being used, but we all know there will be that one guy with a rly bad, old device that shouldn't even rly exist.

It might be a quick and dirty hack to fix it temporarily, but I'd say the only appropiate solution to be pushed upstream should be an actual check/consideration of the scalar/divisor attribute.

Rembock, 2024-09-23:
> You can't 100% certainly determin if the ~100.0-200.0W that you're reading are actually 1000-2000W or just 100-200W with a really bad power factor. The latter shouldn't really happen _normally_, [...]

I hadn't thought about that, with that knowledge I went to see what else could be done to solve it. What I did notice is that after reading acPowerDivisor from the Dev console, the power does show correctly. Maybe refreshing the power divisor more often could be a solution?

itkama, 2024-09-23:
@notgood Update on the first plug worked fine. Second one is running right now. 
First plug now shows 2422V on the outlet - was showing the correct voltage before the update.

basvdploeg, 2024-09-23:
> @notgood Update on the first plug worked fine. Second one is running right now. First plug now shows 2422V on the outlet - was showing the correct voltage before the update.

I found out the same. 

When I use them _**without**_ @Rembock's external definition, they show the voltage and power (up to 1000W) correctly.

basvdploeg, 2024-09-23:
> I hadn't thought about that, with that knowledge I went to see what else could be done to solve it. What I did notice is that after reading acPowerDivisor from the Dev console, the power does show [...]

I probably don't understand enough about this, but... Wouldn't it be possible to read the acPowerDivisor on every power update, and display the right value based on that?

itkama, 2024-09-23:
@basvdploeg Well that's funny now. While it was working with and without the external converter before - on the new firmware I still get the "high voltage" regardless if I use the external converter or not.

Rembock, 2024-09-23:
> @Rembock I probably don't understand enough about this, but... Wouldn't it be possible to read the acPowerDivisor on every power update, and display the right value based on that?

My thoughts exactly! Except I also don't know how (yet)

notgood, 2024-09-23:
> on the new firmware I still get the "high voltage" regardless if I use the external converter or not.

Had "high voltage" right after firmware update. 
If I remember correctly, re-reading acVoltageDivisor and acVoltageMultiplier from Dev console fixed this issue.

itkama, 2024-09-23:
Interesting, reading acVoltageDivisor already did the trick. No need for acVoltageMultiplier on my plugs.

Rembock, 2024-09-23:
just added acPowerDivisor to reporting, this seems to solve the power problem for now
[image]

basvdploeg, 2024-09-24:
> just added acPowerDivisor to reporting, this seems to solve the power issue for now ![Screenshot 2024-09-24 012100](https://private-user-images.githubusercontent.com/18113225/370077462-b65e90bf-d1b3-4bcb-8d60-d28d7fbddc73.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjcxNjQ5ODksIm5iZiI6MTcyNzE2NDY4OSwicGF0aCI6Ii8xODExMzIyNS8zNzAwNzc0NjItYjY1ZTkwYmYtZDFiMy00YmNiLThkNjAtZDI4ZDdmYmRkYzczLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MjQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTI0VDA3NTgwOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWNjNjBhNGE5MmJiMTMyNWZmYjc2MDEzMTA3M2M2NTFmOTMyZmFjMjUwZjEzMWJmYzBmMGJjMTRiOWQ1ZDgzMWEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.3nzKyjeMx40g28bug7N1pght30zTsoKR5qVqOWj3N24)

I can confirm that this is working. 🎉

MaxRower, 2024-09-24:
> I bought the new INSPELNING plug here in Austria

Where did you get it? There is no information online, it's not listed at all, and I don't want to search all stores in Vienna myself.
Did you test the accuracy of low power meterings? Are there decimal digits in the values? In the past I tried Nous A1Z, but that plug is unusable below 10W.

folfy, 2024-09-24:
> > I bought the new INSPELNING plug here in Austria
> 
> @folfy Where did you get it? There is no information online, it's not listed at all, and I don't want to search all stores in Vienna myself. Did you test the accuracy of low power meterings? Are there decimal digits in the values? In the past I tried Nous A1Z, but that plug is unusable below 10W.

Regarding stock, see [my comment in the ZHA issue](https://github.com/zigpy/zha-device-handlers/pull/3379#issuecomment-2366740954).

Below 1000W it reports with .1W resolution, and also is reading from 0.1W already. Checking some .04W load and .2W and .8W of a second INSPELNING in series, it seemed totally fine so far (within rounding error). Power consumption is [a bit higher](https://github.com/bramstroker/homeassistant-powercalc/pull/2516/files#diff-65c2b8fce3e4b5777de5cfa1841d3a056b7f36c3f4016405aa4d1d3ec7f939aeR6-R7) than the amazing Aqara smart plug, but also it's twice the price, and it also has the "IKEA sensors and buttons directly switching the socket on/off, if their zigbee packets are routed through the plug" issue. The physical on/off button on the IKEA plug seems a bit meh, but yeah, overall would still recommend the new IKEA plug, because the additional wasted electricity would take like 10y (@25c/kWh) to offset the higher price of the Aqara, _if_ you'd leave it on 24/7.

notgood, 2024-09-24:
> Did you test the accuracy of low power meterings? Are there decimal digits in the values? In the past I tried Nous A1Z, but that plug is unusable below 10W.

Yup, INSPELNING has a 0.1W resolution at low power values.

Did a quick test, Voltcraft energy logger 4000 was used to compare values.
Plugged in some random gadgets: measurements at 1.7W, 4.4W, 4.9W and 13.8W were pretty much identical, correctly adjusted for power factor etc.

Mains voltage is a bit off, -2V or so.

INSPELNING itself standby power on is 0.8W, standby power off is too low to be measured by Voltcraft.

folfy, 2024-09-24:
Ok, just small update on the off-topic of accuracy:
The ~5V deviation in the Voltage reading (+/-0.3% reference measurement) is reflected even worse in the power measurement unfortunately, so for a ~604W load it read 631W, and for ~1880W it was 196(0)W, so that's a +4.2-4.4% deviation, which is pretty bad.

@notgood Regarding the floppy button pcb, I agree that the button is shit and sometimes not reacting (even though it clicked), but in theory the design definitely ain't as bad, like the pcb isn't left unsupported (if you look at the lid that you removed/opened, you'll see).

QempZor, 2024-09-24:
> just added acPowerDivisor to reporting, this seems to solve the power issue for now ![Screenshot 2024-09-24 [...]

Is this with your latest ext def?

notgood, 2024-09-24:
@folfy 

> so for a ~604W load it read 631W, and for ~1880W it was 196(0)W, so that's a +4.2-4.4% deviation, which is pretty bad.

Tea kettle was the only high powered single-phase device in the general vicinity, so here goes:
2602W according to Voltcraft 4000
2660W according to INSPELNING
So it's ~2% deviation for my piece, but probably different for others, doubt manufacturer calibrates each device at such price point.

I see Z2M supports calibration percentual offsets, probably linear, but better than nothing.

> but in theory the design definitely ain't as bad, like the pcb isn't left unsupported (if you look at the lid that you removed/opened, you'll see).

Didn't inspect the top lid, even if there is some kind of support, it didn't help much. Wedging PCB closer to plastic button solved problem for me, 100% of clicks work now. If this issue is widespread, it will probably cost Ikea quite some returns.

Koenkk, 2024-09-24:
With the latest firmware, does the energy report correctly now? (both < 1000W and >= 1000W) If not, we can hack around this in the converter.

QempZor, 2024-09-24:
> With the latest firmware, does the energy report correctly now? (both < 1000W and >= 1000W) If not, we can apply a hack in the converter.

It works correct if the Cluster: haElectricalMeasurement; Attribute: acPowerDivisor is added with Min. Rep Interval of 1
See https://github.com/Koenkk/zigbee2mqtt/issues/23961#issuecomment-2369761682

Rembock, 2024-09-25:
@Koenkk As @folfy mentioned earlier in https://github.com/Koenkk/zigbee2mqtt/issues/23961#issuecomment-2366733453, the powerdivisor changes dynamically based on the amount of power. Adding acPowerDivisor to reporting from the device solves the power issue. I've done this form within the UI after the successfull pairing based on the latest version.

I've tried to extend the definition with extra reporting (for acPowerDivisor), but that attempt was unsuccessfull. Ideally the reporting of acPowerDivisor would be added when pairing. The Min. Rep interval can also be 5, the 1 was just a guess. I recently tested this and got satisfying results.

Last but not least, updating is also a hassle. See https://github.com/Koenkk/zigbee2mqtt/issues/23961#issuecomment-2366925170 for details. @QempZor has shared the link for the ota in https://github.com/Koenkk/zigbee2mqtt/issues/23961#issuecomment-2365188576

tomried, 2024-09-25:
> @folfy Where did you get it? There is no information online, it's not listed at all, and I don't want to search all stores in Vienna myself. Did you test the accuracy of low power meterings? Are [...]

@MaxRower I just bought the plug from IKEA Westbahnhof, there were still a few left.

gine78, 2024-09-25:
You can use this way to find out if your Ikea has the plug in store.

[https://www.threads.net/@konstantin.m___/post/C__B1nroRtB/](https://www.threads.net/@konstantin.m___/post/C__B1nroRtB/)

QempZor, 2024-09-26:
> I've managed to update firmware to 2.4.45 using z2m, here are the steps.
>
> 1. Download [...]

I can't get my new inspelnings to update... I must do something wrong. But I have no clue what. 

Downloaded the OTA and placed it in the root dir. 
Added the lines to configuration.

Must be somewhere in my external converter file:
`const {electricityMeter, onOff, identify} = require('zigbee-herdsman-converters/lib/modernExtend');
//const {addCustomClusterManuSpecificIkeaUnknown, ikeaOta} = require('zigbee-herdsman-converters/lib/ikea')

const definition = {
   zigbeeModel: ['INSPELNING Smart plug'],
   model: 'E2206',
   vendor: 'IKEA',
   description: 'INSPELNING Smart plug',
//   extend: [addCustomClusterManuSpecificIkeaUnknown(), onOff(), identify(), ikeaOta(), electricityMeter()],
   ota: ota.zigbeeOTA,
   meta: {},
};

module.exports = definition;`

Any smart mind an idea?

Rembock, 2024-09-26:
@QempZor did you also add the ota override index location to the configuration.yaml? Alternatively, you can also add the ota file location from within the UI.

QempZor, 2024-09-26:
> ota override index location to the configuration.ya

Yes, added:
```
ota:
  zigbee_ota_override_index_location: inspelning_OTA_index.json
```

basvdploeg, 2024-09-26:
@QempZor You can check for updates via the OTA tab in Z2M.

[image]

QempZor, 2024-09-26:
> ota: ota.zigbeeOTA,

Wasn't there, but did everything again (for like the 10th time) and used HomeAssistant's MQTT publisher. That worked :)
Thanks, also Thanks to @Rembock  ;)

gine78, 2024-09-27:
I use the external definition above (https://github.com/Koenkk/zigbee2mqtt/issues/23961#issuecomment-2359335397) but i become a false voltage like this. 
[image]

Where can i set the right value?

gllmlbrt, 2024-09-27:
> I use the external definition above ([#23961 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/23961#issuecomment-2359335397)) but i become a false voltage like this. ![image](https://private-user-images.githubusercontent.com/8969123/371479488-84c94123-bfe9-4a81-895f-eec578c0ab95.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjc0NDI2MDcsIm5iZiI6MTcyNzQ0MjMwNywicGF0aCI6Ii84OTY5MTIzLzM3MTQ3OTQ4OC04NGM5NDEyMy1iZmU5LTRhODEtODk1Zi1lZWM1NzhjMGFiOTUucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDkyNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA5MjdUMTMwNTA3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YmU3OTQ2NTY5NWQ5MDRkYWM5MThlYzZhY2MxOWRiYjRiYjdhZDA3YjRlYzJhOGZjYzMxZGMxMTVmMDlmNGMwMCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ._YD6iscuLsliIO0kMJUmisa88d46ARDnI4xYEURSwB4)
> 
> Where can i set the right value?

Do this: https://github.com/Koenkk/zigbee2mqtt/issues/23961#issuecomment-2369102258

gine78, 2024-09-27:
> haElectricalMeasurement

It works, thanks.

MaxRower, 2024-09-27:
> You can use this way to find out if your Ikea has the plug in store.
> 
> https://www.threads.net/@konstantin.m___/post/C__B1nroRtB/

Sadly, it does not work for me, it requires the IKEA App and the first step there is to add your IKEA Gateway. And most people here will not use that.

basvdploeg, 2024-09-27:
@MaxRower To check the availability you need the IKEA Store app, not the IKEA Home Smart app.

[226 more replies]