Koenkk/zigbee2mqtt #23962: SONOFF SWV shows up as not supported

markimarc66, 2024-09-13:
### What happened?

Hello,

I have joined the SWV (multiple times) and it shows up as not supported device despite it is listed as a supported device. As a result I have only battery, switch and linkquality but not the other Exposes listed. The SWV firmware is 1.02 and I can't update via OTA probably because it is not supported. I version 1.40.1 installed. What can I do?

Thanks & Best Regards, Marc

### What did you expect to happen?

That the SWV is supported accoding to the device description with all exposes: flow, battery, switch (state), current_device_status, auto_close_when_water_shortage, cyclic_timed_irrigation, cyclic_quantitative_irrigation, linkquality. Log file says

warning 2024-09-13 22:23:40Device '0x842712fffe3359f2' with Zigbee model 'SWV' and manufacturer name 'SONOFF' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

### How to reproduce it (minimal and precise)

re-joining always creates same result. In the interview process the full list of exposes is not collected and the SWV shows up as unsupported

### Zigbee2MQTT version

1.40.1

### Adapter firmware version

20210708

### Adapter

SONOFF ZigBee 3.0 Plus, ZB-Dongle-P

### Setup

Z2M on Iobroker on Windows 11 on Mini-PC

chris-1243, 2024-09-14:
Could you provide a debug log when pairing it ?

In `configuration.yaml` set:

```
advanced:
  log_level: debug
```

markimarc66, 2024-09-14:
Hello Chris,

this is the debug log when joining:

info 2024-09-14 22:26:34Device '0x842712fffe3359f2' joined
info 2024-09-14 22:26:34MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x842712fffe3359f2","ieee_address":"0x842712fffe3359f2"},"type":"device_joined"}'
info 2024-09-14 22:26:34Starting interview of '0x842712fffe3359f2'
info 2024-09-14 22:26:34MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x842712fffe3359f2","ieee_address":"0x842712fffe3359f2","status":"started"},"type":"device_interview"}'
debug 2024-09-14 22:26:34Device '0x842712fffe3359f2' announced itself
info 2024-09-14 22:26:34MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x842712fffe3359f2","ieee_address":"0x842712fffe3359f2"},"type":"device_announce"}'
debug 2024-09-14 22:26:38Received Zigbee message from '0x842712fffe3359f2', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
debug 2024-09-14 22:26:38Skipping message, still interviewing
debug 2024-09-14 22:26:41Received Zigbee message from '0x842712fffe3359f2', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
debug 2024-09-14 22:26:41Skipping message, still interviewing
debug 2024-09-14 22:26:44Saving state to file c:\zigbee2mqtt\zigbee2mqtt-master\data\state.json
debug 2024-09-14 22:26:45Received Zigbee message from '0x842712fffe3359f2', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
debug 2024-09-14 22:26:45Skipping message, still interviewing
debug 2024-09-14 22:26:46Received Zigbee message from '0x842712fffe3359f2', type 'readResponse', cluster 'genPollCtrl', data '{"checkinInterval":7200}' from endpoint 1 with groupID 0
debug 2024-09-14 22:26:46Skipping message, still interviewing
info 2024-09-14 22:26:46Successfully interviewed '0x842712fffe3359f2', device has successfully been paired
warning 2024-09-14 22:26:46Device '0x842712fffe3359f2' with Zigbee model 'SWV' and manufacturer name 'SONOFF' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
info 2024-09-14 22:26:46MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Automatically generated definition","exposes":[{"access":5,"description":"Remaining battery in %","label":"Battery","name":"battery","property":"battery","type":"numeric","unit":"%"},{"features":[{"access":7,"description":"On/off state of the switch","label":"State","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"access":1,"category":"diagnostic","description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"SWV","options":[{"access":2,"description":"State actions will also be published as 'action' when true (default false).","label":"State action","name":"state_action","property":"state_action","type":"binary","value_off":false,"value_on":true}],"supports_ota":false,"vendor":"SONOFF"},"friendly_name":"0x842712fffe3359f2","ieee_address":"0x842712fffe3359f2","status":"successful","supported":false},"type":"device_interview"}'
info 2024-09-14 22:26:46Configuring '0x842712fffe3359f2'
debug 2024-09-14 22:26:46Configure reporting: true, read: true for 0x842712fffe3359f2/1 genPowerCfg [{"min":"1_HOUR","max":"MAX","change":10,"attribute":"batteryPercentageRemaining"}]
debug 2024-09-14 22:26:48Received Zigbee message from '0x842712fffe3359f2', type 'readResponse', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":200}' from endpoint 1 with groupID 0
info 2024-09-14 22:26:48MQTT publish: topic 'zigbee2mqtt/0x842712fffe3359f2', payload '{"battery":100,"linkquality":54}'
debug 2024-09-14 22:26:48Configure reporting: true, read: true for 0x842712fffe3359f2/1 genOnOff [{"attribute":"onOff","min":0,"max":"MAX","change":1}]
debug 2024-09-14 22:26:49Received Zigbee message from '0x842712fffe3359f2', type 'readResponse', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
info 2024-09-14 22:26:49MQTT publish: topic 'zigbee2mqtt/0x842712fffe3359f2', payload '{"battery":100,"linkquality":51,"state":"OFF"}'
info 2024-09-14 22:26:49Successfully configured '0x842712fffe3359f2'
debug 2024-09-14 22:27:04Received Zigbee message from '0x842712fffe3359f2', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
info 2024-09-14 22:27:04MQTT publish: topic 'zigbee2mqtt/0x842712fffe3359f2', payload '{"battery":100,"linkquality":54,"state":"OFF"}'
debug 2024-09-14 22:27:06Received Zigbee message from '0x842712fffe3359f2', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":200}' from endpoint 1 with groupID 0
info 2024-09-14 22:27:06MQTT publish: topic 'zigbee2mqtt/0x842712fffe3359f2', payload '{"battery":100,"linkquality":54,"state":"OFF"}'
debug 2024-09-14 22:27:10Received Zigbee message from '0x842712fffe3359f2', type 'attributeReport', cluster '64529', data '{"20486":0,"20487":0,"20493":0,"20494":0}' from endpoint 1 with groupID 0
debug 2024-09-14 22:27:10No converter available for 'SWV' with cluster '64529' and type 'attributeReport' and data '{"20486":0,"20487":0,"20493":0,"20494":0}'
debug 2024-09-14 22:27:12Received Zigbee message from '0x842712fffe3359f2', type 'attributeReport', cluster '64529', data '{"20492":0,"20495":0,"20496":0}' from endpoint 1 with groupID 0
debug 2024-09-14 22:27:12No converter available for 'SWV' with cluster '64529' and type 'attributeReport' and data '{"20492":0,"20495":0,"20496":0}'

markimarc66, 2024-09-14:
I have just seen that the SWV also requests OTA update:

debug 2024-09-14 22:29:43Received Zigbee message from '0x842712fffe3359f2', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":4098,"imageType":8202,"manufacturerCode":4742}' from endpoint 1 with groupID 0
debug 2024-09-14 22:29:43Device '0x842712fffe3359f2' requested OTA
debug 2024-09-14 22:29:43Responded to OTA request of '0x842712fffe3359f2' with 'NO_IMAGE_AVAILABLE'

chris-1243, 2024-09-14:
Could you provide an other log with more details. I am looking for the zigbeeModel or modelID of your device.

Would it be possible to remove your device, enable debug log and join it again and send a new copy. I guess your device does not send `SWV` as a modelID and this why it is not recognized by Z2M. It should be supported.

> info 2024-09-14 22:26:34Device '0x842712fffe3359f2' joined

Or you should have other useful data before this line to share maybe

markimarc66, 2024-09-14:
OK, I will join again (I have cleared the log before joining). There is this warning in the log which states correct model and make

warning 2024-09-14 22:26:46Device '0x842712fffe3359f2' with Zigbee model 'SWV' and manufacturer name 'SONOFF' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

markimarc66, 2024-09-14:
this is the new log (has same warning in it)

info 2024-09-14 23:27:38Removing device '0x842712fffe3359f2' (block: false, force: true)
info 2024-09-14 23:27:38MQTT publish: topic 'zigbee2mqtt/0x842712fffe3359f2', payload ''
info 2024-09-14 23:27:38Successfully removed device '0x842712fffe3359f2' (block: false, force: true)
info 2024-09-14 23:27:38MQTT publish: topic 'zigbee2mqtt/bridge/response/device/remove', payload '{"data":{"block":false,"force":true,"id":"0x842712fffe3359f2"},"status":"ok","transaction":"9dmej-4"}'
debug 2024-09-14 23:27:50Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"device":null,"time":254,"transaction":"9dmej-5","value":true}'
info 2024-09-14 23:27:50Zigbee: allowing new devices to join.
info 2024-09-14 23:27:50MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"9dmej-5"}'
info 2024-09-14 23:28:07Device '0x842712fffe3359f2' joined
info 2024-09-14 23:28:07MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x842712fffe3359f2","ieee_address":"0x842712fffe3359f2"},"type":"device_joined"}'
info 2024-09-14 23:28:07Starting interview of '0x842712fffe3359f2'
info 2024-09-14 23:28:07MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x842712fffe3359f2","ieee_address":"0x842712fffe3359f2","status":"started"},"type":"device_interview"}'
debug 2024-09-14 23:28:07Device '0x842712fffe3359f2' announced itself
info 2024-09-14 23:28:07MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x842712fffe3359f2","ieee_address":"0x842712fffe3359f2"},"type":"device_announce"}'
debug 2024-09-14 23:28:11Received Zigbee message from '0x842712fffe3359f2', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
debug 2024-09-14 23:28:11Skipping message, still interviewing
debug 2024-09-14 23:28:14Received Zigbee message from '0x842712fffe3359f2', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
debug 2024-09-14 23:28:14Skipping message, still interviewing
debug 2024-09-14 23:28:19Received Zigbee message from '0x842712fffe3359f2', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
debug 2024-09-14 23:28:19Skipping message, still interviewing
debug 2024-09-14 23:28:20Received Zigbee message from '0x842712fffe3359f2', type 'readResponse', cluster 'genPollCtrl', data '{"checkinInterval":7200}' from endpoint 1 with groupID 0
debug 2024-09-14 23:28:20Skipping message, still interviewing
info 2024-09-14 23:28:20Successfully interviewed '0x842712fffe3359f2', device has successfully been paired
warning 2024-09-14 23:28:20Device '0x842712fffe3359f2' with Zigbee model 'SWV' and manufacturer name 'SONOFF' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
info 2024-09-14 23:28:20MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Automatically generated definition","exposes":[{"access":5,"description":"Remaining battery in %","label":"Battery","name":"battery","property":"battery","type":"numeric","unit":"%"},{"features":[{"access":7,"description":"On/off state of the switch","label":"State","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"access":1,"category":"diagnostic","description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"SWV","options":[{"access":2,"description":"State actions will also be published as 'action' when true (default false).","label":"State action","name":"state_action","property":"state_action","type":"binary","value_off":false,"value_on":true}],"supports_ota":false,"vendor":"SONOFF"},"friendly_name":"0x842712fffe3359f2","ieee_address":"0x842712fffe3359f2","status":"successful","supported":false},"type":"device_interview"}'
info 2024-09-14 23:28:20Configuring '0x842712fffe3359f2'
debug 2024-09-14 23:28:20Configure reporting: true, read: true for 0x842712fffe3359f2/1 genPowerCfg [{"min":"1_HOUR","max":"MAX","change":10,"attribute":"batteryPercentageRemaining"}]
debug 2024-09-14 23:28:21Received Zigbee message from '0x842712fffe3359f2', type 'readResponse', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":200}' from endpoint 1 with groupID 0
info 2024-09-14 23:28:21MQTT publish: topic 'zigbee2mqtt/0x842712fffe3359f2', payload '{"battery":100,"linkquality":57}'
debug 2024-09-14 23:28:21Configure reporting: true, read: true for 0x842712fffe3359f2/1 genOnOff [{"attribute":"onOff","min":0,"max":"MAX","change":1}]
debug 2024-09-14 23:28:23Received Zigbee message from '0x842712fffe3359f2', type 'readResponse', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
info 2024-09-14 23:28:23MQTT publish: topic 'zigbee2mqtt/0x842712fffe3359f2', payload '{"battery":100,"linkquality":57,"state":"OFF"}'
info 2024-09-14 23:28:23Successfully configured '0x842712fffe3359f2'
debug 2024-09-14 23:28:37Received Zigbee message from '0x842712fffe3359f2', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
info 2024-09-14 23:28:37MQTT publish: topic 'zigbee2mqtt/0x842712fffe3359f2', payload '{"battery":100,"linkquality":57,"state":"OFF"}'
[... cut ...]
debug 2024-09-14 23:28:43Received Zigbee message from '0x842712fffe3359f2', type 'attributeReport', cluster '64529', data '{"20486":0,"20487":0,"20493":0,"20494":0}' from endpoint 1 with groupID 0
debug 2024-09-14 23:28:43No converter available for 'SWV' with cluster '64529' and type 'attributeReport' and data '{"20486":0,"20487":0,"20493":0,"20494":0}'
debug 2024-09-14 23:28:45Received Zigbee message from '0x842712fffe3359f2', type 'attributeReport', cluster '64529', data '{"20492":0,"20495":0,"20496":0}' from endpoint 1 with groupID 0
debug 2024-09-14 23:28:45No converter available for 'SWV' with cluster '64529' and type 'attributeReport' and data '{"20492":0,"20495":0,"20496":0}'

chris-1243, 2024-09-14:
May you copy/paste the output of the automatically generated definition as it is suggested by Z2M. For this, open your device page, select `Dev console` and click on `generate_external_definition`.

markimarc66, 2024-09-14:
OK, here is the copy:

const {batteryPercentage, onOff} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['SWV'],
    model: 'SWV',
    vendor: 'SONOFF',
    description: 'Automatically generated definition',
    extend: [batteryPercentage(), onOff({"powerOnBehavior":false})],
    meta: {},
};

module.exports = definition;

chris-1243, 2024-09-14:
> zigbeeModel: ['SWV']

It definitely should be supported....

Could you create an external converter, name it SWV.js and add this in it. Put this file in the same folder as `configuration.yaml`

```
const {
    battery,
    binary,
    bindCluster,
    customTimeResponse,
    deviceAddCustomCluster,
    enumLookup,
    forcePowerSource,
    humidity,
    iasZoneAlarm,
    numeric,
    onOff,
    ota,
    temperature,
} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['SWV'],
        model: 'SWV',
        vendor: 'SONOFF',
        description: 'Zigbee smart water valve',
        fromZigbee: [fz.flow],
        exposes: [e.numeric('flow', ea.STATE).withDescription('Current water flow').withUnit('m³/h')],
        extend: [
            ota(),
            battery(),
            onOff({
                powerOnBehavior: false,
                skipDuplicateTransaction: true,
                configureReporting: true,
            }),
            sonoffExtend.addCustomClusterEwelink(),
            enumLookup({
                name: 'current_device_status',
                lookup: {normal_state: 0, water_shortage: 1, water_leakage: 2, 'water_shortage & water_leakage': 3},
                cluster: 'customClusterEwelink',
                attribute: {ID: 0x500c, type: 0x20},
                description: 'The water valve is in normal state, water shortage or water leakage',
                access: 'STATE_GET',
            }),
            binary({
                name: 'auto_close_when_water_shortage',
                cluster: 'customClusterEwelink',
                attribute: 'lackWaterCloseValveTimeout',
                description: 'Automatically shut down the water valve after the water shortage exceeds 30 minutes.',
                valueOff: ['DISABLE', 0],
                valueOn: ['ENABLE', 30],
            }),
            sonoffExtend.cyclicTimedIrrigation(),
            sonoffExtend.cyclicQuantitativeIrrigation(),
        ],
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'genOnOff']);
            await reporting.bind(endpoint, coordinatorEndpoint, ['msFlowMeasurement']);
            await reporting.onOff(endpoint, {min: 1, max: 1800, change: 0});
            await endpoint.read('customClusterEwelink', [0x500c, 0x5011]);
    },
};

module.exports = definition;
``` 

In `configuration.yaml` add:
```
external_converters:
  - SWV.js
``` 

Restart Z2M and let us know what you get....

markimarc66, 2024-09-15:
Hello Chris,

I have created the SWV.js file with above content, saved it in same folder as the configuration.yaml file and I have added external_converters: -SWV.js at the very end of the configuration.yaml file (see below).
When I restart Z2M I get below error message. After this error messages Z2M starts successfully up with normal messages

Starting Zigbee2MQTT without watchdog.
Zigbee2MQTT:error 2024-09-15 22:22:23: Failed to load external converter file 'SWV.js' (fz is not defined)
Zigbee2MQTT:error 2024-09-15 22:22:23: Probably there is a syntax error in the file or the external converter is not compatible with the current Zigbee2MQTT version
Zigbee2MQTT:error 2024-09-15 22:22:23: Note that external converters are not meant for long term usage, it's meant for local testing after which a pull request should be created to add out-of-the-box support for the device

------------------Content of configuration.yaml file ---------------------------
homeassistant: false
frontend: true
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://192.168.178.61:1885
serial:
  port: \\.\COM3
advanced:
  network_key:
    - 99
    - 53
    - 41
    - 233
    - 146
    - 1
    - 105
    - 117
    - 150
    - 143
    - 119
    - 150
    - 55
    - 106
    - 72
    - 228
  pan_id: 17615
  ext_pan_id:
    - 60
    - 35
    - 242
    - 109
    - 128
    - 96
    - 221
    - 198
  homeassistant_legacy_entity_attributes: false
  legacy_api: false
  legacy_availability_payload: false
  log_level: debug
device_options:
  legacy: false
permit_join: true
devices:
  '0xbc026efffec97952':
    friendly_name: Tradfri Repeater EG
  '0xbc026efffe73cd78':
    friendly_name: Tradfri Repeater UG
  '0x54ef441000aaa419':
    friendly_name: Wassermelder Abstellraum
  '0x54ef441000aaa965':
    friendly_name: Wassermelder Heizung
  '0x54ef441000aaacd7':
    friendly_name: Wassermelder Waschküche
  '0x54ef441000aaaa71':
    friendly_name: Wassermelder Drempel
  '0x048727fffe5c7ffa':
    friendly_name: Fenstermelder EG Ost
  '0x048727fffe5c7e33':
    friendly_name: Fenstermelder DG Bad
  '0x5c0272fffe6d92f8':
    friendly_name: Tradfri Repeater DG
  '0x048727fffe5cc41a':
    friendly_name: Fenstermelder OG Bad
  '0xa4c13861670880a3':
    friendly_name: Temperatursensor OG Bad
    description: Temperatur Luftfeuchte Zeit
  '0x842712fffe3359f2':
    friendly_name: '0x842712fffe3359f2'
ota:
  ikea_ota_use_test_url: false
  disable_automatic_update_check: true
external_converters:
    - SWV.js

markimarc66, 2024-09-15:
I found the fz.flow definition on Github sonoff.ts starting line 1128. It looks like the the SWV device definition is not pproperly found during the interview when pairing

        model: 'SWV',
        vendor: 'SONOFF',
        description: 'Zigbee smart water valve',
        fromZigbee: [fz.flow],

chris-1243, 2024-09-15:
Sorry, I went to fast.... Could you try this one ? It will not be perfect at all (missing options). It is just to see if we can have your device recognized a bit better

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const ea = exposes.access;

const {
    battery,
    binary,
    bindCluster,
    customTimeResponse,
    deviceAddCustomCluster,
    enumLookup,
    forcePowerSource,
    humidity,
    iasZoneAlarm,
    numeric,
    onOff,
    ota,
    temperature,
} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['SWV'],
        model: 'SWV',
        vendor: 'SONOFF',
        description: 'Zigbee smart water valve',
        fromZigbee: [fz.flow],
        exposes: [e.numeric('flow', ea.STATE).withDescription('Current water flow').withUnit('m³/h')],
        extend: [
            ota(),
            battery(),
            onOff({
                powerOnBehavior: false,
                skipDuplicateTransaction: true,
                configureReporting: true,
            }),
        ],
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'genOnOff']);
            await reporting.bind(endpoint, coordinatorEndpoint, ['msFlowMeasurement']);
            await reporting.onOff(endpoint, {min: 1, max: 1800, change: 0});
            //await endpoint.read('customClusterEwelink', [0x500c, 0x5011]);
    },
};

module.exports = definition;
``` 
It should not throw errors at startup

markimarc66, 2024-09-15:
Hello Chris, I think I found the problem. I did run npm ci and afterwards npm run build. Then after starting again with npm start, the SWV shows now properly up as supported with all exposes! I might have forgotten to run npm ci last time I did the upgrade.

Thanks a lot for your support!

ellerar, 2024-10-01:
I hope it’s Ok to add to this, because I think this is a continuation.   I have added a Sonos SWV and it is working however I have some issues getting to operate with home assistant.   Within ZigbeeMQTT it is discovered, correctly sets up and exposes all of the entities, for me the important ones where the cyclic timed irrigation and cyclic quantative irrigation.  Both of these have their entities shown on the expose tab,  allow me to set them using either a slider or an entry box and apply them.  When I do that they are sent to the device and operate correctly.   However,  when I add the device into home assistant these entities do not appear as settable items but rather as sensors so they cannot be set in an automation.  It does however show all of them, just not as controls.   As an alternative I tried to create a scene in ZigbeeMQTT that I could then call with the settings I wanted but when I try to create the scene only the slider is visible and no numeric box.  When I move the slide I  get an error z2m: Publish 'set' 'cyclic_timed_irrigation' to 'Irrigation One' failed: 'Error: ZCL command 0x0ceff6fffeae9c14/1 customClusterEwelink.write({"20488":{"value":[10,0,13,0,0,0,null,0,0,0,null],"type":66}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'INVALID_VALUE')'

Am I missing something?

closed 2024-12-19 by github-actions[bot]