Koenkk/zigbee2mqtt #23963: [New device support]: Sengled E13-A21 Par38 led bulb with motion sensor

clockbrain, 2024-09-13:
### Link

https://www.zigbee2mqtt.io/devices/E13-N11.html

### Database entry

{"id":2,"type":"EndDevice","ieeeAddr":"0xb0ce18140319e015","nwkAddr":27863,"manufId":4448,"manufName":"sengled","powerSource":"Mains (single phase)","modelId":"E13-A21","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,1794,2821,64513],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"E13-A21","manufacturerName":"sengled","powerSource":1,"zclVersion":1,"appVersion":1,"stackVersion":0,"hwVersion":2}},"seMetering":{"attributes":{"divisor":10000,"multiplier":1,"instantaneousDemand":123,"currentSummDelivered":[0,0]}},"manuSpecificSengledMotionSensor":{"attributes":{"enableAutoOnOff":1,"offDelay":45}}},"binds":[{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0xb43522fffe0d15be","endpointID":1}],"configuredReportings":[{"cluster":1794,"attrId":1024,"minRepIntval":10,"maxRepIntval":65000,"repChange":50000,"manufacturerCode":null},{"cluster":1794,"attrId":0,"minRepIntval":10,"maxRepIntval":65000,"repChange":[0,1000],"manufacturerCode":null}],"meta":{}}},"appVersion":1,"stackVersion":0,"hwVersion":2,"zclVersion":1,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1726211849622}

### Zigbee2MQTT version

1.40.1

### Comments

I couldn't find a current product link for this bulb - mostly out of stock. Its similar to the already supported [E13-N11](https://www.zigbee2mqtt.io/devices/E13-N11.html) but this E13-A21 is the Australian 240 volt version.

I do have a Sengled hub though and was able to sniff traffic to find the motion sensor control cluster FC01 and its attributes which are:
- 0 trigger condition 0=dark, 1=weak light
- 1 enable motion triggered light False or True
- 3 motion status 0 or 1
- 4 turn off delay seconds

I've more testing to do with the Sengled hub but am sharing this so I can get some help with the external definition and perhaps help others as disabling motion sensed auto on has been a long requested capability e.g. [https://community.home-assistant.io/t/sengled-motion-sensor-par38-e13-n11/232592/6](https://community.home-assistant.io/t/sengled-motion-sensor-par38-e13-n11/232592/6) 

I've attached the Wireshark sniff of the Sengled hub session which included pairing and playing with the various Sengled app controls. [Sengled E13-A21 join and adjust.zip](https://github.com/user-attachments/files/16999254/Sengled.E13-A21.join.and.adjust.zip)

### External definition

```shell
const {identify, light, electricityMeter, iasZoneAlarm, ota, deviceAddCustomCluster} = require('zigbee-herdsman-converters/lib/modernExtend');

//function sengledLight(args?: LightArgs) {
//    return lightDontUse({effect: false, powerOnBehavior: false, ...args});
//}
const definition = {
    zigbeeModel: ['E13-A21'],
    model: 'E13-A21',
    vendor: 'sengled',
    description: 'Flood light with motion sensor light outdoor',
    extend: [identify(), light(), electricityMeter({"cluster":"metering"}), 
        iasZoneAlarm({zoneType: 'occupancy', zoneAttributes: ['alarm_1']}), ota(),
        deviceAddCustomCluster('manuSpecificSengledMotionSensor', {
            ID: 0xfc01,
            manufacturerCode: 0x1160, // Zcl.ManufacturerCode.SENGLED_CO_LTD
            attributes: {
                triggerCondition: {ID: 0x0000, type: 0x20}, //Zcl.DataType.UINT8
                enableAutoOnOff: {ID: 0x0001, type: 0x10}, //Zcl.DataType.BOOLEAN
                motionStatus: {ID: 0x0003, type: 0x20}, //Zcl.DataType.UINT8
                offDelay: {ID: 0x0004, type: 0x21}, //Zcl.DataType.UINT16
            },
            commands: {},
            commandsResponse: {},
        }),
    ],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

The clusters show in the dev console but I don't have converters for cluster 0xfc01.

enableAutoOnOff disables auto turning on the light when motion is detected but doesn't re-enable it again. I seemed to have the same problem with the Sengled hub and need to do some more testing to confirm that.

Setting the turn off delay works. Minimum value of 30 seconds. Haven't found the max yet but Sengled app has it at 5 minutes.

Yet to properly test the trigger condition of light/dark.

Reporting is only configured for the seMetering cluster. The Sengled hub also sets up reporting for light OnOff, light LevelControl and 0xfc01 attribute 3 (motion status).

I don't know what the function sengledLight in the existing [E13-N11](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/sengled.ts) converter does but I had to disable it in this JS external definition.

clockbrain, 2024-09-18:
I now have an improved external definition which gets the manufacturer specific cluster attributes working. I don't know what attribute 2 saveEnable does though. The Sengled app doesn't seem to have any functionality relating to that.

Here's a better link to the Sengled motion detection capabilities [https://support.sengled.com/hc/en-us/articles/360015375793-Enabling-the-Smart-LED-Bright-White-PAR38-Element-PAR38-bulb-s-motion-detection-feature](https://support.sengled.com/hc/en-us/articles/360015375793-Enabling-the-Smart-LED-Bright-White-PAR38-Element-PAR38-bulb-s-motion-detection-feature)

Still not working is reporting for the motionStatus attribute. Can anyone help me with the definition to setup reporting in the modernExtend style?

I suspect this definition would also work with the similar E13-N11 model but I don't have one of those to test and am not sure if an external definition overrides the included [sengled.ts](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/sengled.ts) definitions.

```

const {identify, light, electricityMeter, iasZoneAlarm, ota, deviceAddCustomCluster, binary, enumLookup, numeric} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['E13-A21'],
    model: 'E13-A21',
    vendor: 'sengled',
    description: 'Flood light with motion sensor light outdoor',
    extend: [
        identify(),
        light(),
        electricityMeter({"cluster":"metering"}),
        iasZoneAlarm({zoneType: 'occupancy', zoneAttributes: ['alarm_1']}),
        ota(),
        deviceAddCustomCluster('manuSpecificSengledMotionSensor', {
            ID: 0xfc01,
            manufacturerCode: 0x1160, //Zcl.ManufacturerCode.SENGLED_CO_LTD
            attributes: {
                triggerCondition: {ID: 0x0000, type: 0x20}, // Zcl.DataType.UINT8
                enableAutoOnOff: {ID: 0x0001, type: 0x10}, // Zcl.DataType.BOOLEAN
                saveEnable: {ID: 0x0002, type: 0x10}, // Zcl.DataType.BOOLEAN
                motionStatus: {ID: 0x0003, type: 0x20}, // Zcl.DataType.UINT8
                offDelay: {ID: 0x0004, type: 0x21}, // Zcl.DataType.UINT16
            },
            commands: {},
            commandsResponse: {},
        }),
        enumLookup({
            name: 'trigger_condition',
            lookup: {dark: 0, 'weak_light': 1},
            cluster: 'manuSpecificSengledMotionSensor',
            attribute: 'triggerCondition',
            description: 'Choose whether the PAR38 bulb comes on when motion is detected under weak light conditions or dark conditions',
            zigbeeCommandOptions: {manufacturerCode: 0x1160},
            access: 'STATE_SET',
        }),
        binary({
            name: 'enable_auto_on_off',
            cluster: 'manuSpecificSengledMotionSensor',
            attribute: 'enableAutoOnOff',
            description: 'Enable the PAR38 bulb to turn on automatically when motion is detected',
            valueOn: [true, 0x01],
            valueOff: [false, 0x00],
            zigbeeCommandOptions: {manufacturerCode: 0x1160},
            access: 'STATE_SET',
        }),
        binary({
            name: 'save_enable',
            cluster: 'manuSpecificSengledMotionSensor',
            attribute: 'saveEnable',
            description: 'Unknown',
            valueOn: [true, 0x01],
            valueOff: [false, 0x00],
            zigbeeCommandOptions: {manufacturerCode: 0x1160},
            access: 'STATE_SET',
        }),
        binary({
            name: 'motion_status',
            cluster: 'manuSpecificSengledMotionSensor',
            attribute: 'motionStatus',
            description: 'Whether the PAR38 bulb has detected motion',
            valueOn: [true, 0x01],
            valueOff: [false, 0x00],
            zigbeeCommandOptions: {manufacturerCode: 0x1160},
            access: 'STATE_GET',
        }),
        numeric({
            name: 'off_delay',
            cluster: 'manuSpecificSengledMotionSensor',
            attribute: 'offDelay',
            description: 'How long the light stays on once the motion sensor is triggered',
            valueMin: 30,
            valueMax: 14400,
            valueStep: 1,
            unit: 'seconds',
            zigbeeCommandOptions: {manufacturerCode: 0x1160},
            access: 'STATE_SET',
        }),
    ],
};

module.exports = definition;
```

Sian-Lee-SA, 2024-09-19:
I replaced mine with your adaption and so far Energy and Occupancy are always null, although Power and Motion Status seems to report. The only issue I have with the motion is the 30 sec min off delay. For a PIR sensor to report presence, it doesn't go to clear til after that time has passed.

clockbrain, 2024-09-20:
Thanks for the feedback @Sian-Lee-SA 

There isn't any occupancy so the iasZoneAlarm part of the definition can be removed. I was misled by some early work on this device using deconz which seemed to suggest it supported that cluster but it doesn't.

The 'saveEnable' attribute can also be removed. That was also something I found documented online but doesn't seem to exist on the device.

The 30 second minimum off delay seems to be a built in limit as far as I can tell.

Sian-Lee-SA, 2024-09-20:
Yh you've done an excellent job, I did research when I bought this model a couple years ago ending up with the assumption that the Australian version didn't even support motion detected and was hardwired so it's good to see that I was wrong in that. 30 Seconds isn't a big deal and will just deal with it, I've just combined it as a group sensor to my other two PIR sensors for the area, Just means security recordings goes a little longer than normal. No biggie... Thanks!

clockbrain, 2024-09-20:
You can also replace the `light()` entry near the top with `light({effect: false, powerOnBehavior: false})` to remove those capabilities this model doesn't have.

Sian-Lee-SA, 2024-09-20:
Yhi already had that from my copy. also have  color: false aswell

clockbrain, 2024-09-21:
Ok, here is my latest nearly working external converter for the E13-A21.

I've added a reporting config for the motion status attribute.

Not working is the default reporting for electricityMeter. It only configures reporting for currentSummDelivered but not for instantaneousDemand. I'd like to be able to set my own reporting values for those two attributes but am stuck without knowing how to specify them. I can't find any good examples to follow. If I configure them manually through the Reporting tab for the device they work perfectly.

```
const {identify, light, electricityMeter, ota, deviceAddCustomCluster, binary, enumLookup, numeric} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['E13-A21'],
    model: 'E13-A21',
    vendor: 'sengled',
    description: 'Flood light with motion sensor light outdoor',
    extend: [
        identify(),
        light({color: false, effect: false, powerOnBehavior: false}),
        electricityMeter({"cluster":"metering"}),
        ota(),
        deviceAddCustomCluster('manuSpecificSengledMotionSensor', {
            ID: 0xfc01,
            manufacturerCode: 0x1160, //Zcl.ManufacturerCode.SENGLED_CO_LTD
            attributes: {
                triggerCondition: {ID: 0x0000, type: 0x20}, // Zcl.DataType.UINT8
                enableAutoOnOff: {ID: 0x0001, type: 0x10}, // Zcl.DataType.BOOLEAN
                motionStatus: {ID: 0x0003, type: 0x20}, // Zcl.DataType.UINT8
                offDelay: {ID: 0x0004, type: 0x21}, // Zcl.DataType.UINT16
            },
            commands: {},
            commandsResponse: {},
        }),
        enumLookup({
            name: 'trigger_condition',
            lookup: {dark: 0, 'weak_light': 1},
            cluster: 'manuSpecificSengledMotionSensor',
            attribute: 'triggerCondition',
            description: 'Choose whether the PAR38 bulb comes on when motion is detected under weak light conditions or dark conditions',
            zigbeeCommandOptions: {manufacturerCode: 0x1160},
            access: 'STATE_SET',
        }),
        binary({
            name: 'enable_auto_on_off',
            cluster: 'manuSpecificSengledMotionSensor',
            attribute: 'enableAutoOnOff',
            description: 'Enable the PAR38 bulb to turn on automatically when motion is detected',
            valueOn: [true, 0x01],
            valueOff: [false, 0x00],
            zigbeeCommandOptions: {manufacturerCode: 0x1160},
            access: 'STATE_SET',
        }),
        binary({
            name: 'motion_status',
            cluster: 'manuSpecificSengledMotionSensor',
            attribute: 'motionStatus',
            reporting: {min: 1, max: 65534, change: 1},
            description: 'Whether the PAR38 bulb has detected motion',
            valueOn: [true, 0x01],
            valueOff: [false, 0x00],
            zigbeeCommandOptions: {manufacturerCode: 0x1160},
            access: 'STATE_GET',
        }),
        numeric({
            name: 'off_delay',
            cluster: 'manuSpecificSengledMotionSensor',
            attribute: 'offDelay',
            description: 'How long the light stays on once the motion sensor is triggered',
            valueMin: 30,
            valueMax: 14400,
            valueStep: 1,
            unit: 'seconds',
            zigbeeCommandOptions: {manufacturerCode: 0x1160},
            access: 'STATE_SET',
        }),
    ],
};

module.exports = definition;
```

Sian-Lee-SA, 2024-09-21:
forgot to mention I did that but from UI. I also did it for power which seemed to update the Watts, just not the kWh

clockbrain, 2024-09-23:
I found there is a known issue with reporting for metering [https://github.com/Koenkk/zigbee-herdsman-converters/issues/7917](https://github.com/Koenkk/zigbee-herdsman-converters/issues/7917).

So that explains why I am having trouble with reporting for Power and Energy.

clockbrain, 2024-10-04:
PR for this device has now been merged https://github.com/Koenkk/zigbee-herdsman-converters/pull/8045

closed 2024-10-04 by clockbrain