Koenkk/zigbee2mqtt #23966: [New device support]: model: 'TS0601', vendor: '_TZE204_bmdsp6bs'

AlexObukhoff, 2024-09-14:
### Link

https://aliexpress.ru/item/1005006332763566.html?sku_id=12000036791926783

### Database entry

{"id":15,"type":"Router","ieeeAddr":"0xa4c138eea0ea935f","nwkAddr":42919,"manufId":4417,"manufName":"_TZE204_bmdsp6bs","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"��x.i��x.i","65506":56,"65508":0,"65534":0,"stackVersion":0,"dateCode":"","appVersion":74,"manufacturerName":"_TZE204_bmdsp6bs","zclVersion":3,"modelId":"TS0601","powerSource":1}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1726327548273}

### Zigbee2MQTT version

1.40.1

### Comments

Just pared the device and Z2M say "not supported"

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_bmdsp6bs',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

Doesn't work presence and luminance sensors. Show only Linkquality.

AlexObukhoff, 2024-09-16:
Data Point's:

```json
{
    "1": "Induction state",
    "101": "状态变化间隔",
    "102": "无人延时",
    "103": "光照度",
    "105": "距离区间开关",
    "110": "灵敏度",
    "111": "开关",
    "114": "触发灵敏度",
    "117": "无人持续时间大于",
    "118": "有人持续时间大于",
    "176": "模式变化",
    "182": "距离",
    "183": "距离区间数据"
}
```

AlexObukhoff, 2024-09-16:
Working converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [
        {modelID: 'TS0601', manufacturerName: '_TZE204_bmdsp6bs',},
    ],
    model: 'TS0601_TZE204_bmdsp6bs',
    vendor: 'Tuya',
    description: 'Smart Human presence sensor 24G',
    whiteLabel: [tuya.whitelabel('Tuya', 'Y1_IN', 'Human presence sensor', ['_TZE204_bmdsp6bs'])],
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
        e.presence(),
        e.illuminance_lux(),
        e
            .numeric('keep_sensitivity', ea.STATE_SET)
            .withValueMin(0)
            .withValueMax(10)
            .withValueStep(1)
            .withDescription('Keep sensitivity'),
        e
            .numeric('trigger_sensitivity', ea.STATE_SET)
            .withValueMin(1)
            .withValueMax(20)
            .withValueStep(1)
            .withDescription('Trigger sensitivity'),
        e
            .numeric('fading_time', ea.STATE_SET)
            .withValueMin(1)
            .withValueMax(60)
            .withValueStep(1)
            .withDescription('Fading time')
            .withUnit('s'),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'presence', tuya.valueConverter.trueFalse1],
            [102, 'fading_time', tuya.valueConverter.raw],
            [103, 'illuminance_lux', tuya.valueConverter.raw],
            [110, 'keep_sensitivity', tuya.valueConverter.raw],
            [114, 'trigger_sensitivity', tuya.valueConverter.raw],
        ],
    },
    extend: [
        // A preferred new way of extending functionality.
    ],
    icon: 'data:image/jpeg;base64,/9j/4AAQSkZJRgABAQEAYABgAAD/2wBDAAgGBgcGBQgHBwcJCQgKDBQNDAsLDBkSEw8UHRofHh0aHBwgJC4nICIsIxwcKDcpLDAxNDQ0Hyc5PTgyPC4zNDL/2wBDAQkJCQwLDBgNDRgyIRwhMjIyMjIyMjIyMjIyMjIyMjIyMjIyMjIyMjIyMjIyMjIyMjIyMjIyMjIyMjIyMjIyMjL/wAARCAEiAT0DASIAAhEBAxEB/8QAHwAAAQUBAQEBAQEAAAAAAAAAAAECAwQFBgcICQoL/8QAtRAAAgEDAwIEAwUFBAQAAAF9AQIDAAQRBRIhMUEGE1FhByJxFDKBkaEII0KxwRVS0fAkM2JyggkKFhcYGRolJicoKSo0NTY3ODk6Q0RFRkdISUpTVFVWV1hZWmNkZWZnaGlqc3R1dnd4eXqDhIWGh4iJipKTlJWWl5iZmqKjpKWmp6ipqrKztLW2t7i5usLDxMXGx8jJytLT1NXW19jZ2uHi4+Tl5ufo6erx8vP09fb3+Pn6/8QAHwEAAwEBAQEBAQEBAQAAAAAAAAECAwQFBgcICQoL/8QAtREAAgECBAQDBAcFBAQAAQJ3AAECAxEEBSExBhJBUQdhcRMiMoEIFEKRobHBCSMzUvAVYnLRChYkNOEl8RcYGRomJygpKjU2Nzg5OkNERUZHSElKU1RVVldYWVpjZGVmZ2hpanN0dXZ3eHl6goOEhYaHiImKkpOUlZaXmJmaoqOkpaanqKmqsrO0tba3uLm6wsPExcbHyMnK0tPU1dbX2Nna4uPk5ebn6Onq8vP09fb3+Pn6/9oADAMBAAIRAxEAPwD2yiiigAooooAKKKKACiiigAooooAKKKKACiiigAooooAKKKKACiiigAooooAKKKKACiiigAooooAKKKKACiiigAooooAKKKKACiiigAooooAKKKKACiiigAooooAKKKKACiiigAooooAKKCQOpppkA96AHUVGZSegppcnvSuBNSZFQ5PqaTmi4E+4DvRuHrUNFFxk24etGR6ioaKLiJ6KhDEUu80XAloqMSetPDA07gLRRRQAUUUUAFFFFABRRRQAUUUUAFFFFABRRRQAUUUUAFFFFABRRRQAUUUUAFFIWCjmo2k9KAJC4FRmQkcUzr1o60gAknrRkVHJPFHwzfN6Dk1F5lzMf3MIVf7znn8qALPbPao3uIo/vSKPxqI2EkvM9yx9hwKkjsLWPB8vJ9TzRYLkLajbg4Us5/2Vphv5CTss5j6EjFaAVE4VcD60vH90flTsBnrd3bDIsiPYsKU3F7/z6j/vqtDOOvFJu9x+dFguUPtV0ODafkaUXzD71s4Hc4q9nPvSbB/dH5UWC5UW/gbqSn1FTJPFJ91wac0UbcFc1XeyiPIGD7HFFgLWKUGqXlTRf6t+PQ04XEi/6xOO7ClYC5kjvTg/rUEc0co+VhUlICUEGlqLkU5W9adwH0UA5opgFFFFABRRRQAUUUUAFFFFABRRRQAUUUUAFFFFABTWcKKR3C8d6hJyc0rgKSWOTSUYpMljtQc/3vSiwA8gjHzdT0A61GFnn7+Wn6mpkhWM7jyx7mnkgKSTgDqadgI47eKLoNzevepC/HXAqLe7/wCrXA/vN0/AU7ylOC5Ln/a7fhTsITzlb7gLn/Z6fnRmVuyp+OTUh/lRQBH5bH70rfhxSmJD1BP1Ymn0UAM8mP8A55p/3yKPKj/55p/3yKfmigRGYYj/AMs0/KjyUHTcPoxFSUUAR+Ww6St9GANITKOqqw/2Tj9DUtFAXI/MQ8Hg+jcUjxBs4GDUpAIwRkehqPydv+rbb7Hkf/W/CgLlVrchs859RTkmkT73zL79an37eJBtJ6HqD+NJJFnpwaVh3HJIsn3Tn2p4qkQUbgYPqKnjn3Ha/wArfzpWGWAcU8HIqLrSgkUAS0UinP1paYBRRRQAUUUUAFFFFABRRRQAUUUUAFNd9o96Vm2jNQE5OTSAQ8mlAopH9KLAIQX4BIGe3epFXaBihRgUx5GL+WmN3cnotUkIV5NrbVG58ZCj+ZpBETzId7dvQfQU5EWNSBk5OST1J9adQIKO1FRyTJFgNks33VHJP4UDJKZJLHCP3jhfQHqfwqLbPNy7eSv91eWP1Pb8KkjgjiJKIAT1Y8k/jQIYLhmGYreVh6n5R+tKPtLDnyk/Nv8ACp6KAuQeXOetxj/djH9aTyZv+flv++FqxRQFyv5dwOk6t/vR/wD16M3K9Y43H+y2P51YooAr/alXAlV4v94cfmOKnVlcblII9RS+vvUDWqbt0ZMTH+JDj8x0NAE9FV/NlhH75Qyf89EHT6j/AAqdWDoGUgqeQQaAAjIIIBB6g1FsaL7g3L/cPX8P8KmooAgO2VCV6e/b61XdNp5z9atvEc74yA/v0b6/40zckynggg4IPUH3oGMhmwQsh+h9asZqi6FG2npU9vLkbG6jp71LGWAcGpQc1COtOBxzSGSUUUVQgooooAKKKKACkZlQZZgB7mlryXxvqFxdeJJ7aV2ENvhY48/L0Bzj1qJy5Vc6sHhXianInY9aBB6GivO/BmvG0uY7C5lbyJziPcfuN2/A9PrivQpDgfWnGXMrk4nDyoVORkbtk+wptFKBzQc4hOFzQo7mmMdz8dBTiwRGZvugZNUgYkrkEIn336egHrTkjEa7QPck9SfU0yFTtLvw78keg7CpRVEi0UlV55XMgt4Th2GWb+4PX60AK8zNIYoMFx95z0T/ABPtT4oEiy3LOfvO3JP+fSliiWGMIgwB6859z71IKACiilpCCiiigYUUUUAFFFFABRRRQAVXaFoiZLfgnlo+zf4GrGRR1pgRwzLMu5cjBwQeoPoakqCeJlb7RCP3o+8v98f4+9SxSLNGskZypHFIB1QSxlT5sYy46j+8PT/Cp6TvQCIG2yxhk5BHBxVXJU8cEHg1ax5U23/lnIcj2bv+dQ3KYbcBQMtRsHQMKfnjFU7aTDhc/K3T61bqRkiHtTqiBwaloQBRRRTAKKKKAA9K8h8bDHiq8YDj5Mn32ivX68b8Wy+d4nviDwJQMe4AH9KyrbHr5N/HfoZ8mfsG9SQykMpzjBr17RLya/0CwubhSJpYFZ89zjr+PWvHLidY2tbcg/6TIIwR2Jr26KNYYY4lAVUUKAOwHFTS2Lzf4loPoJ2qWopsh+VR71sjxRsY5z602T550j7D529/T9f5VMo4qKH5nkkH8T4H0HH+NWImxSUuaO9AiOWUQxNI3IUZx60y2iaNCz/61zu [... cut ...] +goAF61NGrO4VVJJ7AVpWugSsQ1xIEH91eTW1b2kNquIkx79z+NAFCy0nHz3P4J/jWqAFUAAADsKWimAUUUUAFFFFABRRRQAUUUUAFFFFABRRRQAUUUUAFFFFABRRRQAUUUUAFFFFABRRRQAUUUUAGaZJDHMMSRqw9xmn0UAZ8uj2smcKyH/Zb/Gqknh/nKXB+jLW3RQBzjeH7on5Zoj9cj+lMPh29/56Qf8AfR/wrpqKAOaXw1cH788QHtk1Zj8NRDBkndvZRit3jFGRQBRh0axh6QBj6uc1dVVRcKoA9BxTsikJ4oASiiigAooooAKKKKACiiigAooooAKKKKACiiigAooooAKKKKACiiigAooooAKKKKACiiigAooooAKKKKACiiigAooooAKKKKACiiigAooooAKKKKACiiigAooooAKKKKACiiigAooooAKKKKACiiigD//Z',
};

module.exports = definition;
```

Koenkk, 2024-09-18:
Could you make a pull request by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/tuya.ts)?

AlexObukhoff, 2024-09-27:
[image]

AlexObukhoff, 2024-10-05:
> Could you make a pull request by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/tuya.ts)?

Ready: https://github.com/Koenkk/zigbee-herdsman-converters/pull/8082

closed 2024-10-31 by Koenkk