Koenkk/zigbee2mqtt #23967: Turning off dimmer in Z2M sets brightness to 0 and disables the physical switch

RoGu777, 2024-09-14:
### What happened?

I have a TS110E _TZ3210_k1msuvg6 dimmer.

Using the switch toggle works fine and I see all the changes in Z2M interface, but if I turn off the lights via Z2M the brightness is set to 0, which confuses the dimmer and I can't turn on the lights anymore via the hardware toggle.

### What did you expect to happen?

I can turn off the lights via Z2M frontend and then turn on the lights again with the hardware switch. The light brightness should stay the same as before turning off

### How to reproduce it (minimal and precise)

1. turn on the lights with the hardware switch -> the light is on
2. turn off the lights in Z2M -> the lights is off, but brightness is set to 0 in Z2M
3. turn on the lights with the hardware switch -> the light stays off

To "repair" the dimmer, I need to toggle the hardware switch twice and increase the brightness with the hardware switch twice, see logs below

### Zigbee2MQTT version

1.40.1

### Adapter firmware version

20240710

### Adapter

Sonoff Dongle-P zStack3x0

### Setup

Z2M in docker + Home Assistant in docker

### Debug log

Here are the logs in debug mode with the steps I've done

**1. the light is on and I turn it off via the hardware switch, everything is fine so far**

[2024-09-14 18:54:56] debug: 	zh:zstack:unpi:parser: <-- [254,7,69,196,193,44,2,237,56,107,28,203]
[2024-09-14 18:54:56] debug: 	zh:zstack:unpi:parser: --- parseNext [254,7,69,196,193,44,2,237,56,107,28,203]
[2024-09-14 18:54:56] debug: 	zh:zstack:unpi:parser: --> parsed 7 - 2 - 5 - 196 - [193,44,2,237,56,107,28] - 203
[2024-09-14 18:54:56] debug: 	zh:zstack:znp: AREQ: <-- ZDO - srcRtgInd - {"dstaddr":11457,"relaycount":2,"relaylist":[14573,7275]}
[2024-09-14 18:54:56] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-09-14 18:54:56] debug: 	zh:zstack:unpi:parser: <-- [254,27,68,129,0,0,6,0,193,44,1,1,0,25,0,186,168,46,0,0,7,8,20,10,0,0,16,0,107,28,27,125]
[2024-09-14 18:54:56] debug: 	zh:zstack:unpi:parser: --- parseNext [254,27,68,129,0,0,6,0,193,44,1,1,0,25,0,186,168,46,0,0,7,8,20,10,0,0,16,0,107,28,27,125]
[2024-09-14 18:54:56] debug: 	zh:zstack:unpi:parser: --> parsed 27 - 2 - 4 - 129 - [0,0,6,0,193,44,1,1,0,25,0,186,168,46,0,0,7,8,20,10,0,0,16,0,107,28,27] - 125
[2024-09-14 18:54:56] debug: 	zh:zstack:znp: AREQ: <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":11457,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":25,"securityuse":0,"timestamp":3057850,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[8,20,10,0,0,16,0]}}
[2024-09-14 18:54:56] debug: 	zh:controller: Received payload: clusterID=6, address=11457, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=25, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":20,"commandIdentifier":10},"payload":[{"attrId":0,"dataType":16,"attrData":0}],"command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}}
[2024-09-14 18:54:56] debug: 	zh:controller:endpoint: ZCL command 0xa4c138b2beb2b14c/1 genOnOff.defaultRsp({"cmdId":10,"statusCode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"reservedBits":0,"transactionSequenceNumber":20,"writeUndiv":false})
[2024-09-14 18:54:56] debug: 	zh:zstack: sendZclFrameToEndpointInternal 0xa4c138b2beb2b14c:11457/1 (0,0,1)
[2024-09-14 18:54:56] debug: 	zh:zstack:znp: SREQ: --> AF - dataRequest - {"dstaddr":11457,"destendpoint":1,"srcendpoint":1,"clusterid":6,"transid":5,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[24,20,11,10,0]}}
[2024-09-14 18:54:56] debug: 	zh:zstack:unpi:writer: --> frame [254,15,36,1,193,44,1,1,6,0,5,0,30,5,24,20,11,10,0,210]
[2024-09-14 18:54:56] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-09-14 18:54:56] debug: 	z2m: Received Zigbee message from 'EG Eingang Licht', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
[2024-09-14 18:54:56] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/EG Eingang Licht', payload '{"action":"off","brightness":42,"device":{"applicationVersion":64,"dateCode":"","friendlyName":"EG Eingang Licht","hardwareVersion":1,"ieeeAddr":"0xa4c138b2beb2b14c","manufacturerID":4417,"manufacturerName":"_TZ3210_k1msuvg6","model":"TS110E_1gang_1","networkAddress":11457,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"linkquality":25,"max_brightness":140,"min_brightness":15,"power_on_behavior":"previous","state":"OFF","switch_type":"momentary"}'
[2024-09-14 18:54:56] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/EG Eingang Licht', payload '{"action":"","brightness":42,"device":{"applicationVersion":64,"dateCode":"","friendlyName":"EG Eingang Licht","hardwareVersion":1,"ieeeAddr":"0xa4c138b2beb2b14c","manufacturerID":4417,"manufacturerName":"_TZ3210_k1msuvg6","model":"TS110E_1gang_1","networkAddress":11457,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"linkquality":25,"max_brightness":140,"min_brightness":15,"power_on_behavior":"previous","state":"OFF","switch_type":"momentary"}'
[2024-09-14 18:54:56] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/EG Eingang Licht/action', payload 'off'
[2024-09-14 18:54:56] debug: 	zh:zstack:unpi:parser: <-- [254,1,100,1,0,100,254,3,68,128,0,1,5,195]
[2024-09-14 18:54:56] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,100,1,0,100,254,3,68,128,0,1,5,195]
[2024-09-14 18:54:56] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 1 - [0] - 100
[2024-09-14 18:54:56] debug: 	zh:zstack:znp: SRSP: <-- AF - dataRequest - {"status":0}
[2024-09-14 18:54:56] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,68,128,0,1,5,195]
[2024-09-14 18:54:56] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [0,1,5] - 195
[2024-09-14 18:54:56] debug: 	zh:zstack:znp: AREQ: <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":5}
[2024-09-14 18:54:56] debug: 	zh:zstack:unpi:parser: --- parseNext []

**2. I switch the light on again with the hardware switch, everything fine so far**

[... 197 lines cut ...]
[2024-09-14 18:57:19] debug: 	zh:zstack:znp: AREQ: <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":21}
[2024-09-14 18:57:19] debug: 	zh:zstack:unpi:parser: --- parseNext []

**8. I increase the brightness with the hardware switch again, now the light is finally on, the brightness is increased to 66 and the dimmer works fine again**

[2024-09-14 18:57:26] debug: 	zh:zstack:unpi:parser: <-- [254,27,68,129,0,0,8,0,193,44,1,1,0,25,0,202,219,189,0,0,7,8,52,10,0,0,32,66,107,28,27,177]
[2024-09-14 18:57:26] debug: 	zh:zstack:unpi:parser: --- parseNext [254,27,68,129,0,0,8,0,193,44,1,1,0,25,0,202,219,189,0,0,7,8,52,10,0,0,32,66,107,28,27,177]
[2024-09-14 18:57:26] debug: 	zh:zstack:unpi:parser: --> parsed 27 - 2 - 4 - 129 - [0,0,8,0,193,44,1,1,0,25,0,202,219,189,0,0,7,8,52,10,0,0,32,66,107,28,27] - 177
[2024-09-14 18:57:26] debug: 	zh:zstack:znp: AREQ: <-- AF - incomingMsg - {"groupid":0,"clusterid":8,"srcaddr":11457,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":25,"securityuse":0,"timestamp":12442570,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[8,52,10,0,0,32,66]}}
[2024-09-14 18:57:26] debug: 	zh:controller: Received payload: clusterID=8, address=11457, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=25, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":52,"commandIdentifier":10},"payload":[{"attrId":0,"dataType":32,"attrData":66}],"command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}}
[2024-09-14 18:57:26] debug: 	zh:controller:endpoint: ZCL command 0xa4c138b2beb2b14c/1 genLevelCtrl.defaultRsp({"cmdId":10,"statusCode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"reservedBits":0,"transactionSequenceNumber":52,"writeUndiv":false})
[2024-09-14 18:57:26] debug: 	zh:zstack: sendZclFrameToEndpointInternal 0xa4c138b2beb2b14c:11457/1 (0,0,1)
[... cut ...]
[2024-09-14 18:57:26] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 1 - [0] - 100
[2024-09-14 18:57:26] debug: 	zh:zstack:znp: SRSP: <-- AF - dataRequest - {"status":0}
[2024-09-14 18:57:26] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,68,128,0,1,23,209]
[2024-09-14 18:57:26] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [0,1,23] - 209
[2024-09-14 18:57:26] debug: 	zh:zstack:znp: AREQ: <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":23}
[2024-09-14 18:57:26] debug: 	zh:zstack:unpi:parser: --- parseNext []

RoGu777, 2024-09-14:
I've been now looking into the code and found that there are 2 functions:

- tz.TS110E_onoff_brightness
- tz.TS110E_light_onoff_brightness

for _TZ3210_k1msuvg6 TS110E_light_onoff_brightness is used, for some other dimmers TS110E_onoff_brightness. So I've created an external converter with TS110E_onoff_brightness for _TZ3210_k1msuvg6 and after some tests it looks like it solves the problem.

here the converter:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [
        {
            modelID: 'TS110E',
            manufacturerName: '_TZ3210_k1msuvg6',
        },
    ],
    model: 'TS110E_custom',
    vendor: 'Tuya',
    description: '1 channel dimmer customized',

    extend: [modernExtend.light({ powerOnBehavior: false, configureReporting: true })],
    fromZigbee: [tuya.fz.power_on_behavior_1, fz.TS110E_switch_type, fz.TS110E, fz.on_off],
    toZigbee: [tz.TS110E_onoff_brightness, tuya.tz.power_on_behavior_1, tz.TS110E_options],
    exposes: [e.power_on_behavior(), tuya.exposes.switchType(), e.min_brightness(), e.max_brightness()],
    configure: tuya.configureMagicPacket,
};

module.exports = definition;
```

Koenkk, 2024-09-16:
Integrated the fix, thanks!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-09-16 by Koenkk

kiksia95, 2024-09-16:
I am facing the same issue with 2 channel version of this dimmer TS110E _TZ3210_3mpwqzuu. Can the above converter be applied to 2CH version?

Also, I've noticed that the min.&max. brightness thresholds set up is not reflected in HA fronted still allowing to set the brightness below/above the thresholds.

RoGu777, 2024-09-22:
this converter will not work for 2 channel, you will need to do the same adjustment based in the default for 2 channels

kiksia95, 2024-09-22:
I've made the following. I am new to Z2M, but do not see any positive/negative results.  
`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');

const e = exposes.presets;
const ea = exposes.access;

const definition = {
        fingerprint: [{modelID: 'TS110E', manufacturerName: '_TZ3210_wdexaypg'}, {modelID: 'TS110E', manufacturerName: '_TZ3210_3mpwqzuu'}],
        model: 'TS110E_2gang_1',
        vendor: 'TuYa',
        description: '2 channel dimmer',
        extend: [
            deviceEndpoints({endpoints: {l1: 1, l2: 2}}),
            light({powerOnBehavior: false, endpointNames: ['l1', 'l2'], configureReporting: true}),
        ],
        fromZigbee: [tuya.fz.power_on_behavior_1, fzLocal.TS110E_switch_type, fzLocal.TS110E],
        toZigbee: [tz.TS110E_onoff_brightness, tuya.tz.power_on_behavior_1, tzLocal.TS110E_options],
        configure: tuya.configureMagicPacket,
        exposes: [
            e.min_brightness().withEndpoint('l1'), e.max_brightness().withEndpoint('l1'),
            e.min_brightness().withEndpoint('l2'), e.max_brightness().withEndpoint('l2'),
            e.power_on_behavior(),
            tuya.exposes.switchType().withEndpoint('l1'),
            tuya.exposes.switchType().withEndpoint('l2'),
        ]
};

module.exports = definition;`

RoGu777, 2024-10-28:
Yes, I agree.

_TZ3210_k1msuvg6 is now better, but still buggy. E.g. refreshing the brightness or min/max values in Z2M UI shows crazy numbers out of range, e.g. "4375" as brightness.

I have also _TZ3210_3mpwqzuu now for my living room and have the same issue. The fix applied above also does not fix it for me.

Probably a new bug ticket is needed, as this is closed

RoGu777, 2024-10-29:
OK, now I was able to solve the problem for  _TZ3210_3mpwqzuu, see #24528

For _TZ3210_k1msuvg6 I would need more deep analyzing

Aposei, 2024-11-16:
I don't see the brightness changes on z2m, when I modify it with the physical switch

RoGu777, 2024-11-17:
Yes, I also noticed that, but have no idea how to fix that

As far as I remember this worked fine before the change from TS110E_1gang_1 TS110E_1gang_2

Aposei, 2024-11-18:
> Yes, I also noticed that, but have no idea how to fix that
> 
> As far as I remember this worked fine before the change from TS110E_1gang_1 TS110E_1gang_2

Hi. Thanks for the reply. Should I open a new issue?