Koenkk/zigbee2mqtt #23972: BTicino F20T60A paired as Legrand 412015

gianmarco-mameli, 2024-09-14:
### What happened?

Hi, I've bought a BTicino F20T60A, but zigbee2mqtt keeps pairing as Legrand 412015, and I cannot see functions like voltage and current.
Also the OTA fails after some minutes

### What did you expect to happen?

Pair device as BTicino F20T60A

### How to reproduce it (minimal and precise)

Pair a BTicino F20T60A on zigbee2mqtt

### Zigbee2MQTT version

1.40.1

### Adapter firmware version

20240710

### Adapter

Sonoff ZBDongle-P

### Setup

Docker on a Raspberry pi 3b+ Raspberry Pi OS Bookworm 64bit

### Debug log

[log.log](https://github.com/user-attachments/files/17004061/log.log)

gianmarco-mameli, 2024-12-05:
Hi @melettaberde, my Bticino F20T60A, bought on Amazon, is always identified as Legrand device out of the box, so in my case I don't think it's a firmware update. I don't know, maybe some wrong serials or id?

widmnd, 2025-01-08:
The same is happening to me. I've just installed the Bticino today and z2m identifies it as Legrand. No voltage or current reported, only power.

gianmarco-mameli, 2025-03-11:
So apparently only the official hub is capable to install the correct firmware?

closed 2025-05-19 by github-actions[bot]