Koenkk/zigbee2mqtt #23974: [Wrong device]: Mumubiz TYZGTH16A-D1RF mis-reconized as TYZGTH1CH-D1RF and not compeletely working

ProgrammedSoul, 2024-09-15:
### Link

https://www.amazon.com/dp/B0D63CR311

### Model

TYZGTH16A-D1RF

### Description

Relay switch with build in temperature/(or)humidity sensor

### Vendor

Mumubiz

### Picture (link)

https://www.aerial.net/shop/imageslarge/TYTE-D1_relay.jpg

### Database entry

{"id":28,"type":"Router","ieeeAddr":"0xa4c1387e51275ce0","nwkAddr":37252,"manufId":4417,"manufName":"_TZ3218_7fiyo3kv","powerSource":"Mains (single phase)","modelId":"TS000F","epList":[0,1,242],"endpoints":{"0":{"epId":0,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,57345,57344,61184],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":80,"stackVersion":1,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1726368541898}

### Notes

Product link alao available as (the Amazon link is the exact link where I got my device):
https://www.aerial.net/shop/product_info.php?products_id=3194&language=en
This model seems to be sold with different brand name, some don't list brand name at all, "MHCOZY" seems to be one of them, with an related issue: https://github.com/Koenkk/zigbee2mqtt/discussions/23123 (the device in that issue is not the exact one I have, it also went off topic to different devices in its later discussion).
It was mis-reconized as TYZGTH**1CH**-D1RF, as listed here: https://www.zigbee2mqtt.io/devices/TYZGTH1CH-D1RF.html
Dispite mis-reconized, on first pairing out of the box, the device do work as intended for a while, until I have to power cycle the device (both device and z2m server are turned off for about 4 hours). After which the device is only partially working. For example, temperature reading still seems to be updating, while trying to change switch(relay) state would yield error as:

> error 2024-09-14 20:24:39z2m: Publish 'set' 'state' to 'RecirThermostat' failed: 'Error: ZCL command 0xa4c1387e51275ce0/0 genOnOff.off({}, 
{"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Delivery failed for {"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":0,"options":4416,"groupId":0,"sequence":125})'

(some setting seems to be working, some would yield similar error)
None of my attempts of creating custom convertor files were able to restore control of the relay. Resetting device, re-pairing device, power cycling device, restarting Z2M(as an HA addon) were also attempted but none have fixed the error.

technoman63, 2024-09-26:
I have the same problem with TYZGTH16A-D1RF

Tee86, 2024-10-03:
I have both: MHCOZY and MUMUBIZ, and both have the exactly same issue as you're describing. Have you found something new about it?

Birdgage, 2024-10-17:
https://github.com/Koenkk/zigbee2mqtt/issues/23835
Same problem, nobody is interested to give a hint for a solution, Koenkk?? anybody??

ProgrammedSoul, 2024-10-17:
I tried the best I can and cannot get it working. I'm not used to nevagitae large code base and am unable to figure out how to write the convertor file to make it work. If this does not get fixed in a while I'm working on replacing the tuya module by other zigbee hardware with better firmware support.

ProgrammedSoul, 2024-10-26:
So, I got it working, I only need it to behave like a thermostat(not automatic, HA control its on/off) so currently only temperature reading and relay control are implemented:
```
//temperature reading is working so leave it as tuya handling
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');

const {deviceEndpoints, onOff} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS000F'], // The model ID from the device.
    model: 'TS000F',
    vendor: '_TZ3218_7fiyo3kv',
    description: 'Random Amazon Tuya Crap',
    fingerprint: [{modelID: 'TS000F', manufacturerName: '_TZ3218_7fiyo3kv'}], // This helps z2m identify the device
    fromZigbee: [], // Define converters that are used to convert from Zigbee messages to MQTT messages.
    toZigbee: [], // Define converters that are used to convert from MQTT messages to Zigbee messages.
    extend: [
        deviceEndpoints({"endpoints":{"0":0,"1":1}}), 
        onOff({"powerOnBehavior":false,"endpointNames":["1"]}),
        tuya.modernExtend.dpTemperature({dp: 0x66, scale: 10})],
    exposes: [], // Defines what capabilities (like temperature or on/off) are exposed to MQTT
    meta: {
        multiEndpoint: true,
    },

    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await endpoint.bind('genOnOff', coordinatorEndpoint);
        // More configurations can be added here
    },
};

module.exports = definition;
```
What happened:
Z2m cannot configure this device during startup, it errors out 3 times and then nothing happened. I think this leads to z2m not having the correct information to control the relay.
After I got mad enough me and CharGPT went brute force in the debug console, turns out endpoint 1, cluster 0x0006 controls the relay, 0x00 is off, 0x01 is on. We then created a configure file to hard-wire this information to a switch and bypassing power-on auto configuration.
Then it works.

Birdgage, 2024-10-28:
> So, I got it working, I only need it to behave like a thermostat(not automatic, HA control its on/off) so currently only temperature reading and relay control are implemented:
>
> ```
> //temperature [...]

Birdgage, 2024-10-28:
Okay, good work, but I still don't get it what do to, to make it work? Must I create a new config file? Have no idea how do to that.

ProgrammedSoul, 2024-10-29:
https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_2-1-extending-the-external-definition
Create a file called TS000F.js under your z2m config folder, paste in the code, save, enable it in z2m settings external convertors, save, restart z2m. If you use HA get the file editor plugin to do that.

Birdgage, 2024-10-29:
ProgrammedSoul you are my hero, for months and months looking for a solution, thank you so much !!!!

Tee86, 2024-11-17:
Guys, can someone please test with me this one:

```
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const ea = exposes.access;

const {deviceEndpoints, onOff} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS000F'],
    model: 'TS000F',
    vendor: '_TZ3218_7fiyo3kv',
    description: '1G Relay with Temp Probe',
    fingerprint: [{modelID: 'TS000F', manufacturerName: '_TZ3218_7fiyo3kv'}],
    fromZigbee: [
        {
            cluster: 'manuSpecificTuya',
            type: ['attributeReport', 'readResponse'],
            convert: (model, msg, publish, options, meta) => {
                const dpValue = tuya.getDataValue(msg.data.dp);
                if (msg.data.dp === 102) {
                    return {temperature: dpValue};
                }
            }
        },
    ],
    toZigbee: [],
    extend: [
        deviceEndpoints({"endpoints":{"0":0,"1":1}}), 
        onOff({"powerOnBehavior":false,"endpointNames":["1"]}),
        tuya.modernExtend.dpTemperature({dp: 0x66, scale: 10}),
    ],
    exposes: [
        e.temperature(),
        e.offset(),
    ],
    meta: {
        tuyaDatapoints: [
            [102, 'temperature', tuya.valueConverter.raw],
            [108, 'offset', tuya.valueConverter.raw],
        ],
    },

    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await endpoint.bind('genOnOff', coordinatorEndpoint);
    },
};

module.exports = definition;
```
I spent countless hours trying to make this stupid relay work and somehow I've managed with the above.

All of sudden all the exposes are working as they should (not only temp reading and switching) and it survived about 10 z2m restarts till now without any issues.
Let me know if that works for you too...

technoman63, 2024-12-05:
> Guys, can someone please test with me this one:
> 
> ```
> const tuya = require('zigbee-herdsman-converters/lib/tuya');
> const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const e = exposes.presets;
> const ea = exposes.access;
> 
> const {deviceEndpoints, onOff} = require('zigbee-herdsman-converters/lib/modernExtend');
> 
> const definition = {
>     zigbeeModel: ['TS000F'],
>     model: 'TS000F',
>     vendor: '_TZ3218_7fiyo3kv',
>     description: '1G Relay with Temp Probe',
>     fingerprint: [{modelID: 'TS000F', manufacturerName: '_TZ3218_7fiyo3kv'}],
>     fromZigbee: [
>         {
>             cluster: 'manuSpecificTuya',
>             type: ['attributeReport', 'readResponse'],
>             convert: (model, msg, publish, options, meta) => {
>                 const dpValue = tuya.getDataValue(msg.data.dp);
>                 if (msg.data.dp === 102) {
>                     return {temperature: dpValue};
>                 }
>             }
>         },
>     ],
>     toZigbee: [],
>     extend: [
>         deviceEndpoints({"endpoints":{"0":0,"1":1}}), 
>         onOff({"powerOnBehavior":false,"endpointNames":["1"]}),
>         tuya.modernExtend.dpTemperature({dp: 0x66, scale: 10}),
>     ],
>     exposes: [
>         e.temperature(),
>         e.offset(),
>     ],
>     meta: {
>         tuyaDatapoints: [
>             [102, 'temperature', tuya.valueConverter.raw],
>             [108, 'offset', tuya.valueConverter.raw],
>         ],
>     },
> 
>     configure: async (device, coordinatorEndpoint, logger) => {
>         const endpoint = device.getEndpoint(1);
>         await endpoint.bind('genOnOff', coordinatorEndpoint);
>     },
> };
> 
> module.exports = definition;
> ```
> 
> I spent countless hours trying to make this stupid relay work and somehow I've managed with the above.
> 
> All of sudden all the exposes are working as they should (not only temp reading and switching) and it survived about 10 z2m restarts till now without any issues, although it throws me an error in the log. Despite that, it seems like that error is forcing it to work correctly. Really weird, but works 100% for me. Let me know if that works for you too...

Dont' work for me.

- Auto settings don't work

Birdgage, 2024-12-07:
> Dont' work for me.
>
>     * Auto settings don't work

Dont work for me to...keeps installing the orginal software

Tee86, 2024-12-07:
But does it survive restart?

skvarel, 2024-12-25:
Works great but one error:

No image for device model: TS000F found!

How can I fix this?

ProgrammedSoul, 2025-01-06:
I'll post this update here to just to save people's time:
if your custom convertors broke after recent z2m update, they changed how external convertor work. Now you have to put the .js file in /data/external_converters, which will all be automatically loaded during startup, without needing to change configuration.
Except if you install z2m as a plugin in HA, /data does not exist. Its called /zigbee2mqtt inside homeassistant/, which is the root directory if you install file editor plugin. So go inside zigbee2mqtt, and make a directory called external_converters, and put all your custom external convertor files there. That should make your external convertor working again.
A++ programming, changing stuff for the sake of changing stuff, end up breaking stuff and helps nobody

Birdgage, 2025-01-06:
Nothing is working anymore after the last update to 2.0.0.1 Koennk says read this read that, hello, duuhh i,m not a Zigbee programmer, so I have gone back to 1.0.... version. Untill I find the time to find my way in put, put that in you're config to make it work...

Tee86, 2025-01-16:
> Nothing is working anymore after the last update to 2.0.0.1 Koennk says read this read that, hello, duuhh i,m not a Zigbee programmer, so I have gone back to 1.0.... version. Untill I find the time to find my way in put, put that in you're config to make it work...

There is one important change RE external converters. In v2 you don't define them in configuration.yaml, instead you have to mkdir in your data folder named "external_converters" and drop your JS there.

If it wasn't working for you, I believe you missed this step.

closed 2025-03-25 by github-actions[bot]

diver2000, 2026-02-08:
> So, I got it working, I only need it to behave like a thermostat(not automatic, HA control its on/off) so currently only temperature reading and relay control are implemented:
>
> ```
> //temperature [...]

Works for me but i need humidity also, who can help me adding humidity. Would like to use it for my terrarium.

Thx a lot

Marco