Koenkk/zigbee2mqtt #23983: Tuya TH03 temperature and humidity sensor with display not detected correctly, and polling errors

leen67, 2024-09-16:
### What happened?

I got a few of the Tuya TH03 sensors: https://fr.aliexpress.com/item/1005007309529179.html
Z2M detects them as:
   fingerprint: [{modelID: 'TS0201', manufacturerName: '_TZ3000_itnrsufe'}],
   model: 'KCTW1Z',
   vendor: 'Tuya',
   description: 'Temperature & humidity sensor with LCD',
database entry: `{"id":56,"type":"EndDevice","ieeeAddr":"0xa4c138820a7143c8","nwkAddr":24293,"manufId":4742,"manufName":"_TZ3000_itnrsufe","powerSource":"Battery","modelId":"TS0201","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[0,3,1,32,1026,1029],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"TS0201","manufacturerName":"_TZ3000_itnrsufe","powerSource":3,"zclVersion":3,"appVersion":113,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017"}},"msRelativeHumidity":{"attributes":{"measuredValue":5486}},"genPollCtrl":{"attributes":{"checkinInterval":6960}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":50,"batteryVoltage":24}},"msTemperatureMeasurement":{"attributes":{"measuredValue":1987}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c7b32b","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":113,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1726480698937,"checkinInterval":1740}
`
With the default converter no Humidity is reported. This is due to the value being 100x (e.g. 5486 as in the database entry above for actual value of 54.86%), while the default converter in tuya.ts uses fzLocal.humidity10.
For some reason it also completely blocks the Zigbee network after a while, and the Z2M UI stops working without obvious errors in the logs.

I created an external converter:
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
//const ota = require('zigbee-herdsman-converters/lib/ota');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS0201', manufacturerName: '_TZ3000_itnrsufe'}],
	zigbeeModel: ['TS0201'],
    model: 'ZTH05',
    vendor: 'Tuya',
    description: 'ZTH03 temperature & humidity sensor with display',
    extend: [],
	fromZigbee: [fz.temperature, fz.humidity, fz.battery],
	toZigbee: [],
    exposes: [e.temperature(), e.humidity(), e.battery(), e.battery_voltage()],
};

module.exports = definition;
```
Note that I put model ZTH05 because the ZTH03 is not in the device database but they look identical.
I guess there are multiple devices that identify themselves as _TZ3000_itnrsufe, while they behave differently, so the only solution with these devices may be to use an external converter permanently.

With my external converter the device works well and my networks stays stable.
One problem remains:
`zh:controller:device: Handling of poll check-in from 0xa4c13841ca386e49 failed (ZCL command 0xa4c13841ca386e49/1 genPollCtrl.checkinRsp({"startFastPolling":false,"fastPollTimeout":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false,"sendPolicy":"immediate"}) failed (Timeout - 61781 - 1 - 248 - 32 - 11 after 10000ms))`
This error happens about every 30 minutes. The error happens both with my external converter and with the build-in converter for KCTW1Z.

### What did you expect to happen?

No polling errors.

### Zigbee2MQTT version

1.40.1-1

### Adapter firmware version

20240710

### Adapter

Sonoff Zigbee 3.0 USB dongle Plus

### Setup

Add-on on Homeassistant OS on Pi4

### Debug log

[log - Copy.log](https://github.com/user-attachments/files/17011533/log.-.Copy.log)
The ZTH03 devices are: 0xa4c13841ca386e49, 0xa4c138820a7143c8, and 0xa4c138abc2445da0

Koenkk, 2024-09-19:
For the polling, you may want to unbind it, hope that fixes the error. Example: https://github.com/Koenkk/zigbee-herdsman-converters/blob/a687fe3cfc384edd6a3ef70bc959ca5eab9629d1/src/devices/sonoff.ts#L572

leen67, 2024-09-21:
Thanks a lot Koen, that indeed fixes the polling errors.
I've updated the external converter as below. Had to add reporting in the configure section as well to ensure getting updates from the sensor.
I have added the endpoints in the fingerprint as to distinguish from the KCTW1Z fingerprint in tuya.ts.
Would it be possible to add the ZTH03 to the Z2M device database with this more specific fingerprint, or would Z2M still detect it as the more generic KCTW1Z? I.e. which match gets the priority?
If the KCTW1Z gets priority in the detection process, could you add the endpoints to the fingerprint for KCTW1Z in tuya.ts as in my comment in the code below, to allow distinguishing between KCTW1Z and ZTH03 even if both have the same modelID and manufacturerName?

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
//const ota = require('zigbee-herdsman-converters/lib/ota');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [
       {
          modelID: 'TS0201',
          manufacturerName: '_TZ3000_itnrsufe',
	  // added endpoint identifier to distinguish from Tuya model KCTW1Z which uses same modelID and manufacturerName
	  // Note that for KCTW1Z fingerprint in tuya.ts maybe it should include the following line:
	  // endpoints: [{ID: 1, profileID: 260, deviceID: 770, inputClusters: [1,3,1026,1029,57346,0], outputClusters: [3, 25, 10]}],
          endpoints: [{ID: 1, profileID: 260, deviceID: 770, inputClusters: [0,3,1,32,1026,1029], outputClusters: [25]}],
       },
    ],
    zigbeeModel: ['TS0201'],
    //actual model is ZTH03 but for now use ZTH05 that is already in the Z2M device database, as they look the same
    model: 'ZTH05',
    vendor: 'Tuya',
    description: 'ZTH03 temperature & humidity sensor with display',
    extend: [],
    fromZigbee: [fz.temperature, fz.humidity, fz.battery],
    toZigbee: [],
    exposes: [e.temperature(), e.humidity(), e.battery(), e.battery_voltage()],
    configure: async (device, coordinatorEndpoint) => {
       // Unbind genPollCtrl to prevent device from sending checkin message.
       // Zigbee-herdsmans responds to the checkin message which causes the device
       // to behave erratically and cause polling errors
       const endpoint = device.getEndpoint(1);
       if (endpoint.binds.some((b) => b.cluster.name === 'genPollCtrl')) {
           await device.getEndpoint(1).unbind('genPollCtrl', coordinatorEndpoint);
       };
       await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'msTemperatureMeasurement', 'msRelativeHumidity']);
       await reporting.temperature(endpoint, {min: 30, max: 300, change: 10});
       await reporting.humidity(endpoint, {min: 30, max: 300, change: 100});
       await reporting.batteryVoltage(endpoint, {min: 3600, max: 14400});
       await reporting.batteryPercentageRemaining(endpoint, {min: 3600, max: 14400});
    },
};

module.exports = definition;
```

Koenkk, 2024-09-22:
@kirovilya  do you still have this device, if yes could you check this?

atl285, 2024-10-29:
I have bought another ZTH03 sensor from AliExpress (https://de.aliexpress.com/item/1005007625460115.html), which works perfectly with the external converter from @leen67 if I change the vendor to '_TZ3000_v1w2k9dd'. 😄 

This is, what the the **generated_external_definition** on the developer tab shows:
~~~
const {battery, temperature, humidity} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS0201'],
    model: 'TS0201',
    vendor: '_TZ3000_v1w2k9dd',
    description: 'Automatically generated definition',
    extend: [battery(), temperature(), humidity()],
    meta: {},
};

module.exports = definition;
~~~

Maybe it's possible to add the device to the supported devices.

leen67, 2024-10-31:
With the generated external definition it doesn't work for me.

Koenkk, 2024-11-06:
Great! Could you make a PR by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/tuya.ts)?

atl285, 2024-11-17:
> Great! Could you make a PR by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/tuya.ts)?

Because lack of knowledge I'm not able to create a PR. 

And today I found out, that the current external definition breaks the functionality of all my real ZTH05 sensors. The values are not refreshed anymore. Maybe it's related to the using of `model: 'ZTH05'` in the definition.

closed 2025-01-25 by github-actions[bot]