Koenkk/zigbee2mqtt #23985: Zigbee2mqtt not publishing to homeassistant topic even though homeassistant set to true in the config

Eleison667, 2024-09-16:
### What happened?

On the latest version of zigbee2mqtt I am not seeing entities updated in homeassistant. 

### What did you expect to happen?

I expected zigbee2mqtt to keep publishing to the homeassistant topic on the broker

### How to reproduce it (minimal and precise)

Updated to 1.40.1
while troubleshooting i did clear my database after which i saw one message to the topic homeassistant, after restoring my database it didnt work anymore. But I don't understand how those are related
using the config below:
# Home Assistant integration (MQTT discovery)
homeassistant:
    # Optional: Home Assistant discovery topic (default: shown below)
    # Note: should be different from [MQTT base topic](../mqtt.md) to prevent errors in HA software
  discovery_topic: 'homeassistant'
    # Optional: Home Assistant status topic (default: shown below)
    # Note: in addition to the `status_topic`, 'homeassistant/status' will also be used
  status_topic: 'hass/status'
    # Optional: Home Assistant legacy entity attributes, (default: shown below), when enabled:
    # Zigbee2MQTT will send additional states as attributes with each entity. For example,
    # A temperature & humidity sensor will have 2 entities for the temperature and
    # humidity, with this setting enabled both entities will also have
    # an temperature and humidity attribute.
    # Note: Disabling this option, requires a Home Assistant restart
  legacy_entity_attributes: true
    # Optional: Home Assistant legacy triggers (default: shown below), when enabled:
    # - Zigbee2mqt will send an empty 'action' or 'click' after one has been send
    # - A 'sensor_action' and 'sensor_click' will be discoverd
  legacy_triggers: true
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://192.168.179.12:1883'
serial:
  port: /dev/ttyACM0
advanced:
  # Optional: state caching, MQTT message payload will contain all attributes, not only changed ones.
  # Has to be true when integrating via Home Assistant (default: true)
  cache_state: true
  # Optional: persist cached state, only used when cache_state: true (default: true)
  cache_state_persistent: true
  # Optional: send cached state on startup, only used when cache_state_persistent: true (default: true)
  cache_state_send_on_startup: true
  report: true
  legacy_api: true
device_options:
  legacy: false
#frontend:
#  port: 8099
devices:
  '0x90fd9ffffe888a64':
    friendly_name: studybutton
  '0x90fd9ffffe6fa8a5':
    friendly_name: '0x90fd9ffffe6fa8a5'
  '0x000b57fffe13c943':
    friendly_name: vision
  '0x000d6ffffe46f610':
    friendly_name: '0x000d6ffffe46f610'
  '0x14b457fffe4e195a':
    friendly_name: '0x14b457fffe4e195a'
  '0x000d6ffffe151cd5':
    friendly_name: '0x000d6ffffe151cd5'
  '0x14b457fffe34c0f3':
    friendly_name: '0x14b457fffe34c0f3'
etc

### Zigbee2MQTT version

1.40.1

### Adapter firmware version

20240710

### Adapter

aliexpress sonoff knockoff

### Setup

Pi

### Debug log

[log.log](https://github.com/user-attachments/files/17012776/log.log)

Eleison667, 2024-09-17:
another round of trial and error mostly error yielded some desired results!
zigbee2mqtt/bridge/request/options with payload {"options": {"homeassistant": true}} instead. got z2m publishing on the homeassistant topic again. Completely in the dark why the config didn't work.

Eleison667, 2024-10-14:
Decided to reinstall homeassistant in docker so i could get the latest version and now its working again. Still in the dark why the debug setting didnt work from the config file but oh well.

closed 2024-10-14 by Eleison667

rubin110, 2024-12-07:
@Eleison667 I'm hitting a similar issue. Occasionally (too often during the day), MQTT will push new topic updates that is received from Z2M over to HA, however HA never shows them received in its logs. This lasts anywhere between 20-60 seconds. Sometimes the updates are dropped, other times they come rolling in after the delay. The annoying thing is updates from other non-Z2M devices come right through MQTT and HA gets them fine while the Z2M get stuck whenever that symptom is appearing.

Did you ever hit a regression of your issue again? Thanks!

Eleison667, 2024-12-10:
I'm not sure, I noticed yesterday some devices were gone and updated zigbee2mqtt - now its not working at all and I haven't figured out what is going wrong now. I see nothing in the logs of homeassistant - which is communicating to mqtt and I can publish directly to devices which were still visible before I updated and control them that way. But all devices are unavailable in HA.

Eleison667, 2024-12-11:
No sorry my problem was different. somehow my database got reset and all my devices have a new ID and need to be accessed via friendly name in home assistant. After fixing that I got most of it working again. except a few devices which are no longer connected but I guess thats just IKEA glitchyness x)