Koenkk/zigbee2mqtt #23987: [Wrong device]: Tuya TS0043

david-alexandru-v, 2024-09-16:
### Link

https://www.aliexpress.com/item/1005001298730723.html

### Model

SS600ZB

### Description

Wireless Switch 3 Channel Remote Control

### Vendor

Tuya

### Picture (link)

https://ae-pic-a1.aliexpress-media.com/kf/H00e3b95410954b58abc3d97b7f47b026p.png_120x120.png_.webp

### Database entry

{"id":2,"type":"EndDevice", "ieeeAddr":"0x5c0272fffec887b9","nwkAddr": 53965, "manufId" : 4098, "manufName":"_TZ3000_bi6lpsew", "powerSource" :"Battery", "modelId":"TS0043", "epList": [1,2, 3], "endpoints": {"1": {"profId" : 260, "epId": 1, "devId":0, "inClusterList": [0, 10, 1,6], "outClusterList": [25], "clusters": {"genBasic": {"attributes": {"65503" :"f#{.\u0012","65534" :0, "modelId":"TS0043", "manufacturerName":"_TZ3000_bi6lpsew","stackVersion":0, "dateCode":"", "zclVersion":3,"appVersion":66, "powerSource": 3}}, "genPowerCfg": {"attributes": {"batteryPercentageRemaining" : 200, "batteryVoltage": 30}}}, "binds" : [], "configuredReportings" : [], "meta" : {}}, "2": {"profId" : 260, "epId" : 2, "devId": 0, "inClusterList": [1,6], "outClusterList": [], "clusters": {}, "binds": [], "configuredReportings" : [], "meta": {}},"3": {"profId": 260, "epId":3, "devId": 0, "inClusterList": [1,6], "outClusterList": [], "clusters": {}, "binds": [], "configuredReportings" : [], "meta": {}}}, "appVersion":66, "stackVersion":0, "hwVersion": 1, "dateCode":"", "zclVersion" : 3, "interviewCompleted": true, "meta": {"configured" : 217189513}, "lastSeen" : 1726499638131}

### Notes

SS600ZB seems to be the whitelabel device for TS0043 and the device with that whitelabel is seen as defined in tuya.yaml. But the actions are not recognized as defined in tuya.yaml. Also it is really not the same product. I pasted above where I got it from
[image]

Aresitoo, 2024-11-23:
got the same problem here, thus cannot use this remote, no actions available

david-alexandru-v, 2025-01-23:
I'm not sure how this goes with the tickets opened by users, do I need to mark them as anything so that they get picked up for development? Or will this device not receive support and the ticket gets closed automatically?

david-alexandru-v, 2025-03-24:
Ok, so if anyone is looking for a solution, I found it. I tried to open a Pull Request to the repo, but I was not able to make it happen ultimately. The local solution is to generate an external converter file, configure the addon to use it: here's the code file

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{
        modelID: 'TS0043',
        manufacturerName: '_TZ3000_bi6lpsew',
    }],
    model: 'TS0043',
    vendor: 'Tuya',
    description: 'Wireless switch with 3 buttons',
    fromZigbee: [fz.battery, {
        cluster: 'genOnOff',
        type: 'commandTuyaAction',
        convert: (model, msg, publish, options, meta) => {
            const button = msg.endpoint.ID;
            const value = msg.data.value;
            let action;
            if (value === 0) action = 'single';
            else if (value === 1) action = 'double';
            else if (value === 2) action = 'hold';
            return {action: `${button}_${action}`};
        },
    }],
    toZigbee: [],
    exposes: [e.battery(), e.battery_voltage(), e.action(['1_single', '1_double', '1_hold',
                        '2_single', '2_double', '2_hold',
                        '3_single', '3_double', '3_hold'])],
    meta: {
        multiEndpoint: true,
    },
};

module.exports = definition;
`

Place it under this folder (pic 1):

[image]

And have your zigbee2mqtt config file look like this: (pic 2):

[image]

david-alexandru-v, 2025-03-24:
@Koenkk sorry I was not able to get through the Github PAT Tokens and push to the repo, but hope you can pick up this form of a contribution

Koenkk, 2025-03-25:
Could you make a PR by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/tuya.ts)?

david-alexandru-v, 2025-03-26:
> Could you make a PR by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/tuya.ts)?

It does look like I can from the WebUI of Github, but not sure: should I simply replace the entire existing definition for "TS0043" in the tuya.ts with what I have in the external converter file that works?

If that is the case, https://github.com/Koenkk/zigbee-herdsman-converters/pull/9067

closed 2025-06-02 by github-actions[bot]