Koenkk/zigbee2mqtt #23991: Tuya SPI LED controller

Ashwoodie, 2024-09-16:
### Link

https://www.iskydance.com/index.php?c=product_show&a=index&id=2433

### Database entry

{"id":4,"type":"Router","ieeeAddr":"0xa4c138cf01f4c3b4","nwkAddr":21421,"manufId":4417,"manufName":"_TZE204_8fffc3kb>

### Zigbee2MQTT version

1.40.1

### Comments

Hi. I’ve used a few Tuya devices with zigbee2MQTT and I naively thought that this device would be compatible. As it turns out I seem to be struggling a little bit and hoped that someone would lend a hand. The device is the Tuya WZ-SPI LED controller. Anyone with any experience with this device? Many thanks in advance.

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_8fffc3kb',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

The devices page still stated unsupported and its not showing on the nodered MQTT device list.

Ashwoodie, 2024-09-17:
Hi all. Does anyone have any suggestions (ie. 'support new tuya devices') or is it just wise waiting for it to be integrated into an update.?

GPSPerth, 2024-10-06:
Is this any help?

{
  "result": {
    "properties": [
      {
        "code": "switch_led",
        "custom_name": "",
        "dp_id": 1,
        "time": 1728210802688,
        "type": "bool",
        "value": true
      },
      {
        "code": "work_mode",
        "custom_name": "",
        "dp_id": 2,
        "time": 1728210827829,
        "type": "enum",
        "value": "white"
      },
      {
        "code": "bright_value",
        "custom_name": "",
        "dp_id": 3,
        "time": 1728210802554,
        "type": "value",
        "value": 255
      },
      {
        "code": "temp_value",
        "custom_name": "",
        "dp_id": 4,
        "time": 1728210801596,
        "type": "value",
        "value": 0
      },
      {
        "code": "colour_data",
        "custom_name": "",
        "dp_id": 5,
        "time": 1728210801596,
        "type": "string",
        "value": ""
      },
      {
        "code": "countdown",
        "custom_name": "",
        "dp_id": 7,
        "time": 1728210801596,
        "type": "value",
        "value": 0
      },
      {
        "code": "music_data",
        "custom_name": "",
        "dp_id": 8,
        "time": 1728210801596,
        "type": "string",
        "value": ""
      },
      {
        "code": "light_length",
        "custom_name": "",
        "dp_id": 46,
        "time": 1728210802343,
        "type": "value",
        "value": 1000
      },
      {
        "code": "light_pixel",
        "custom_name": "",
        "dp_id": 47,
        "time": 1728210802370,
        "type": "value",
        "value": 1000
      },
      {
        "code": "dreamlight_scene_mode",
        "custom_name": "",
        "dp_id": 51,
        "time": 1728210801596,
        "type": "raw"
      },
      {
        "code": "dreamlightmic_music_data",
        "custom_name": "",
        "dp_id": 52,
        "time": 1728210801596,
        "type": "raw"
      },
      {
        "code": "lightpixel_number_set",
        "custom_name": "",
        "dp_id": 53,
        "time": 1728210824718,
        "type": "value",
        "value": 1000
      },
      {
        "code": "paint_colour_data",
        "custom_name": "",
        "dp_id": 61,
        "time": 1728210801596,
        "type": "raw"
      },
      {
        "code": "chip_type_data",
        "custom_name": "",
        "dp_id": 101,
        "time": 1728210802754,
        "type": "enum",
        "value": "TM1809"
      },
      {
        "code": "rgb_order_data",
        "custom_name": "",
        "dp_id": 102,
        "time": 1728210802798,
        "type": "enum",
        "value": "BRG"
      }
    ]
  },
  "success": true,
  "t": 1728211934991,
  "tid": "0c4bdca783d111efa55d025acf19914b"

chrishue, 2024-10-12:
Hi, I understand that this device is planned to be added soon. Do you know when it will be integrated? I can't find any other device to control an LED strip using the SPI protocol for WS2812B LEDs. I'm wondering if I should consider returning it and switching to using WLED home assistant based on ESP32.

dav972, 2024-10-23:
hi, We can't know the dates for the updates?

B4Bcoin, 2024-11-03:
Any chance we can get an example config for this device till it's integrated?

blackberry2nd, 2024-11-27:
waiting for the full working config.
Until then I can do basic actions (on/off, brightness) via a local config file. Feel free to further improve...

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const {light} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition =  {
        fingerprint: tuya.fingerprint('TS0601', ['_TZB210_3zfp8mki', '_TZB210_gj0ccsar', '_TZE204_8fffc3kb']),
        model: 'TS0601',
        vendor: 'Tuya',
        description: 'Zigbee RGB+CCT light',
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [tuya.tz.datapoints],
//        fromZigbee: [],
//        toZigbee: [tz.light],
        onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
        configure: tuya.configureMagicPacket,
        whiteLabel: [
            tuya.whitelabel('Skydance', 'WZ5_dim_2', 'Zigbee & RF 5 in 1 LED controller (DIM mode)', ['_TZB210_3zfp8mki']),
            tuya.whitelabel('Skydance', 'WZ-SPI', 'Zigbee & RF 5 in 1 LED controller (DIM mode)', ['_TZE204_8fffc3kb']),
            tuya.whitelabel('QA', 'QADZC5', '5 in 1 LED controller', ['_TZB210_gj0ccsar']),
        ],

        exposes: [e.light(),
        // Here you should put all functionality that your device exposes
        ],
        meta: {
            tuyaDatapoints: [
                [1, 'state', tuya.valueConverter.onOff], // <- mapped the datapoint
                [3, 'brightness', tuya.valueConverter.scale0_254to0_1000], // <- mapped the datapoint
                [4, 'color_temp', tuya.valueConverter.scale250_454to0_1000], // <- mapped the datapoint
                [14, 'power_on_behavior', tuya.valueConverter.powerOnBehaviorEnum],
        ],
        },

        extend: [
            light({colorTemp: {range: [153, 500]}, color: {modes: ['hs'], applyRedFix: true, enhancedHue: false}}),
        ],
        configure: async (device, coordinatorEndpoint) => {
            device.getEndpoint(1).saveClusterAttributeKeyValue('lightingColorCtrl', {colorCapabilities: 29});
        },
//    },
//    meta: {
//        // All datapoints go in here
//        tuyaDatapoints: [],
//    },
//    extend: [
        // A preferred new way of extending functionality.
//    ],
};

module.exports = definition;

dav972, 2024-11-27:
I'm not good at mqtt, can you give us the steps to follow to enter the code?

dav972, 2024-11-28:
I created a file to put the configuration (I don't know if it's the right method) and there are still errors

abertolin, 2024-11-28:
> waiting for the full working config. Until then I can do basic actions (on/off, brightness) via a local config file. Feel free to further improve...
>
> const fz = [...]

Thank you
I'm not so good, where I have to put this "code" ?

blackberry2nd, 2024-11-28:
Create local file: in home assistant folder: /homeassistant/zigbee2mqtt/wz-spi.js  (or whatever filename you prefer). put the line items as mentioned above in that file.

In the same folder there is a configuration.yaml file. Add to the end of the file:
external_converters:
  - wz-spi.js

Then restart zigbee2mqtt. Then the device should be regognized.

abertolin, 2024-11-29:
> Create local file: in home assistant folder: /homeassistant/zigbee2mqtt/wz-spi.js (or whatever filename you prefer). put the line items as mentioned above in that file.
> 
> In the same folder there is a configuration.yaml file. Add to the end of the file:
> 
> external_converters:
> 
>     * wz-spi.js
> 
> 
> Note: before wz-spi.js should be a '-' but I notice that it shows as a dot due to the markup Then restart zigbee2mqtt. Then the device should be regognized.

Thank you!

wawan822, 2024-11-30:
Hi
I tested your code, the on/off and brightness adjustment work very well
On the other hand, the color adjustment does not work, it tells me: "'Error: No datapoint defined for 'color''"

diahellfire, 2024-11-30:
dp_id: 102
rgb_order_data:
RGB, RBG, GRB, GBR, BRG, BGR, RGBW, RBGW, GRBW, GBRW, BRGW, BGRW, WRGB, WRBG, WGRB, WGBR, WBRG, WBGR

dp_id: 101
chip_type_data:
TM1803, TM1809, TM1829, TLS3001, GW6205, MBI6120, TM1814B, SK6812, UCS8904B, LPD6803, LPD8806, WS2801, P9813, SK9822, MBI6020, TM1914A, GS8206, UCS2904, SM16904, SM16825, SM16714, UCS5603, UCS2603, SM16714D, UCS7604, UCS7804

Values for chip select and color mode.

diahellfire, 2024-11-30:
work_mode
dp_id: 2
Values: colour, white, scene

dav972, 2024-11-30:
I can't get it to work on my side. Can you give me all the stuff to put in the new file and what to put in the configuration.yaml of ZigBee2mqtt

diahellfire, 2024-11-30:
Put file to serve directory .../homeassistant/zigbee2mqtt/wz_spi.js

In Z2M - Open Settings page -> External Converters -> click "+", write "wz_spi.js", click Submit! Done!

[image]

> I can't get it to work on my side. Can you give me all the stuff to put in the new file and what to put in the configuration.yaml of ZigBee2mqtt

dav972, 2024-11-30:
Ok thanks But I can't control the module 🤔 I did something wrong 
[image]

diahellfire, 2024-11-30:
> Ok thanks But I can't control the module 🤔 I did something wrong ![Screenshot_20241130_102304_io_homeassistant_companion_android_WebViewActivity](https://private-user-images.githubusercontent.com/84111189/391242529-9a01b9d0-0ba5-42f4-874c-988c94b95f6d.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzI5OTg1ODQsIm5iZiI6MTczMjk5ODI4NCwicGF0aCI6Ii84NDExMTE4OS8zOTEyNDI1MjktOWEwMWI5ZDAtMGJhNS00MmY0LTg3NGMtOTg4Yzk0Yjk1ZjZkLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMzAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTMwVDIwMjQ0NFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTAwYzY4Mjk3ZTA5MTc3N2Y3ZGRiMzYzMWI3MTFkYjcyNGVjNWNjMDkxNmNjOGMzMDcyYzI1YjM2OGFlOGI5M2QmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.sRgsganjhAT_ir2f5QYm3hDJ2e0tGEARxnfbp6MJ7IE)

Try to restart Z2M, check again.

dav972, 2024-11-30:
It still doesn't work. I must have done something wrong.

marcin-gacek, 2024-12-25:
@dev972 
It looks your file was not correctly read.
Check the formatting in the file (the *.js one)

especially these last lines:

};

module.exports = definition;

Mskera, 2025-01-16:
Hi there!
Did anyone found the solution?
I am with same issue :(

Mskera, 2025-01-16:
I just added this code into a new file under zigbee2mqtt folder called tuya_converter.js

I then added this below code to the main configuration.yaml
external_converters:
  - tuya_converter.js (with slash)

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const {light} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    fingerprint: tuya.fingerprint [{modelID: 'TS0601', manufacturerName: '_TZE204_8fffc3kb'}],
    model: 'TS0601',
    vendor: 'Tuya',
    description: 'SPI LED controller',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    exposes: [
        e.switch(), 
        e.brightness(), 
        e.color()
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'state', tuya.valueConverter.onOff], // <- mapped the datapoint
            [3, 'brightness', tuya.valueConverter.scale0_254to0_1000], // <- mapped the datapoint
            [4, 'color_temp', tuya.valueConverter.scale250_454to0_1000], // <- mapped the datapoint
            [14, 'power_on_behavior', tuya.valueConverter.powerOnBehaviorEnum],
    ],
    },
    extend: [
        light({colorTemp: {range: [153, 500]}, color: {modes: ['hs'], applyRedFix: true, enhancedHue: false}}),
    ],
    configure: async (device, coordinatorEndpoint) => {
        device.getEndpoint(1).saveClusterAttributeKeyValue('lightingColorCtrl', {colorCapabilities: 29});
    },
    onEvent: tuya.onEventSetTime,
    configure: tuya.configureMagicPacket,
    whiteLabel: [
    tuya.whitelabel('Skydance', 'WZ5_dim_2', 'Zigbee & RF 5 in 1 LED controller (DIM mode)', ['_TZB210_3zfp8mki']),
    tuya.whitelabel('Skydance', 'WZ-SPI', 'Zigbee & RF 5 in 1 LED controller (DIM mode)', ['_TZE204_8fffc3kb']),
    tuya.whitelabel('QA', 'QADZC5', '5 in 1 LED controller', ['_TZB210_gj0ccsar']),
],
};

module.exports = definition;`

No way it works, still unsupported and can't act on it :(

happy-daisy, 2025-02-06:
Got the basic interface working with the external definition above.
I have a 7.5m rgbw strip attached to it. Only white with dimming is currently functioning.

The post by diahellfire above indicates the following settings below. Really interested in figuring out how to integrate them into the external definition file.

Any ideas?

> work_mode
> dp_id: 2
> Values: colour, white, scene
> 
> 
> 
> Lightpixel Number Set (strip length)
> dp_id: 53
> value from 1.00 to 78.70
> 
> dp_id: 102
> rgb_order_data:
> RGB, RBG, GRB, GBR, BRG, BGR, RGBW, RBGW, GRBW, GBRW, BRGW, BGRW, WRGB, WRBG, WGRB, WGBR, WBRG, WBGR
> 
> dp_id: 101
> chip_type_data:
> TM1803, TM1809, TM1829, TLS3001, GW6205, MBI6120, TM1814B, SK6812, UCS8904B, LPD6803, LPD8806, WS2801, P9813, SK9822, MBI6020, TM1914A, GS8206, UCS2904, SM16904, SM16825, SM16714, UCS5603, UCS2603, SM16714D, UCS7604, UCS7804
> 
> Values for chip select and color mode.

KrolikROJER, 2025-03-01:
> I just added this code into a new file under zigbee2mqtt folder called tuya_converter.js
>
> I then added this below code to the main configuration.yaml external_converters:
>
> * tuya_converter.js (with [...]

i try this file to submit and see error's that functions not supported:
 * `e.brightness()`
 * `e.color()`

try to replays :
```
    exposes: [
		e.light(),
		// e.switch(), duplicate in e.light
        // e.brightness(), not supported
        // e.color(), not supported
        // Here you should put all functionality that your device exposes
    ],

```

THK-artjom, 2025-03-22:
I have exported a full list of the datapoints maybe that will help someone to add that to the script. I will investiage the enpoints further and share what i can find. Probably will try to extend the provided script myself. But I am not fammilar with the dev environment yet.

```json
{"1":"Switch Led",
"2":"Work Mode",
"3":"Bright Value",
"4":"Temp Value",
"5":"Colour Data",
"7":"Countdown",
"8":"Music Data",
"46":"Light Length",
"47":"Light Pixel",
"51":"Dreamlight Scene Mode",
"52":"Dreamlightmic Music Data",
"53":"Lightpixel Number Set",
"61":"Paint Colour",
"101":"Light strip driver chip settings",
"102":"RGB color order"}

```

THK-artjom, 2025-03-23:
I had a chat in the discord and it seems that implementing adressable rgb strips is currently not possible as the front end is missing the necessary functionalities. A comparable already implemented controller can be found here:
https://www.zigbee2mqtt.io/devices/GL-C-103P.html
but it does also only support the basic functionalities.

THK-artjom, 2025-03-23:
Here is what I got working so far. Still can not set the color but some basic settings are working:

https://codefile.io/f/QhRDzd6SL3

THK-artjom, 2025-04-06:
I got myself now a zigbee sniffer and I analysed the traffic. Here is what I got so far. Not sure how to integrate that into zigbee2mqtt yet. But if anyone can start using this data please reache me out:

On - DP: 257 - Payload: 01
Off - DP: 257 - Payload: 00
Set Predefined Scene - DP:51 - Payload: (Work In Progress)
Music Mode - DP: 52 - Payload: (Work In Progress)
Strip Length - DP: 565 - Payload: (Work In Progress uint16 length in cm e.g. 1,99m -> 0000 00c7
Set Chip Type - DP:1125 - Payload: 'TM1803' = 00 'TM1809' = 01 'TM1829' = 01 ...
Set Color Order  - DP: 1126 - Payload: 'GBR' = 03, 'GRB' = 02

... but most important Color Setup DP: 61

-----------------------------------------------------------
For Non Segmented White Control:

Mode = No Color
0000 

Blend Colors = Off
00 

Static Field(function unkown)
14 

Segmented = Off
00 

Brightness uint4 10-1000: 
e.g. 03e8 = 1000 

Color Temp uint4 0-1000:
e.g. 03e8 = 1000 
------------------------------------------------------
Color Control:

Mode 1 or 2 uint4:
0001 or 0002

Blend Colors of Segment Boundaries:
00 - No
01 - Yes

Const Value fuinction unkown:
14

Segmentation On/Off
00 - Off - color is used on whole length
01 - On - only set color of segment defined at the end

Color 0-330 uint4
0000 - Red (min)
014a - Purple (max)

Color Saturation 0-1000 uint4
Brightness 0-1000 uint4
Another constant with unkown function
81

Segment Number Only if Segmentation is active uint2 00-14
00 - set 1st segment of led strip
14 - set last segment of led strip

THK-artjom, 2025-04-15:
Here is my implementation of the Color Chanel 61, please give me feedback for improvement. On my device I can control it like in the App.

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const types = require("zigbee-herdsman-converters/lib/types");

const fzLocal = {
    tuya_color_hs_segmented: {
        cluster: 'manuSpecificTuya',
        type: ['commandDataResponse', 'commandDataReport'],
        convert: (model, msg, publish, options, meta) => {
            const dp = msg.data.dp;
            if (dp === 61) {
                //white:            0a 00 00 00 14 00 03e8 03e8
                //color no segment: 0c 00 01 01 14 00 014a 03e8 03e8 
                //color and segment:0e 00 01 01 14 01 014a 03e8 03e8 81 01
                const data = msg.data.data;

                const paint_mode = data[1];
                let paint_mode_enum = paint_mode === 0x00 ? "White" :
                                      paint_mode === 0x01 ? "Color" :
                                      paint_mode === 0x02 ? "Color_" : "Unknown";
                
                const gradient_on = data[2];
                const segmentCount = data[1];
                const set_segment_on = data[4];
                const color_value = data.readUInt16BE(5);
                const saturation_value = data.readUInt16BE(7);
                let brightness_value = 0;
                if (paint_mode > 0x00) {
                    brightness_value = data.readUInt16BE(9);
                }
                let segment_id = 0;
                if (set_segment_on) {
                    segment_id = data[12];
                }

                return {
                    paint_mode_enum: paint_mode_enum,
                    gradient_on: gradient_on,
                    set_segment_on: set_segment_on,
                    color_value: color_value,
                    color_saturation: saturation_value,
                    brightness_value: brightness_value,
                    segment_id: segment_id,
                };
            }
        },
    },
};

const tzLocal = {
    tuya_color_hs_segmented: {
        key: ['color_hs_segmented'],
        convertSet: async (entity, key, value, meta) => {
            const state = meta.state || {};
            const newState = {...state};

            newState.paint_mode = value.paint_mode_enum === "White" ? 0x00 :
                                value.paint_mode_enum === "Color" ? 0x01 :
[... 134 lines cut ...]
                .withValueStep(1))
            .withFeature(e.numeric("segment_id", ea.ALL)
                .withValueMin(0)
                .withValueMax(20)
                .withValueStep(1)
                .withDescription("Segment Nr. of Color to Set")),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'state', tuya.valueConverter.onOff],
            [3, 'bright_value', tuya.valueConverter.raw],
            [4, 'temp_value', tuya.valueConverter.raw],
            [7, 'countdown', tuya.valueConverter.countdown],
            //[51, 'dreamlightmic_scene_mode', tuya.valueConverter.raw],
            //[52, 'dreamlightmic_music_data', tuya.valueConverter.raw],
            [53, 'lightpixel_number_set', tuya.valueConverter.raw],
            [61, 'color_hs_segmented', tuya.valueConverter.raw],
            [101, 'chip_type_data', tuya.valueConverterBasic.lookup({"TM1803":0, "TM1809":1, "TM1829":2, "TLS3001":3, "GW6205":4, "MBI6120":5, "TM1814B":6, "SK6812":7, "UCS8904B":8, "LPD6803":9, "LPD8806":10, "WS2801":11, "P9813":12, "SK9822":13, "MBI6020":14, "TM1914A":15, "GS8206":16, "UCS2904":17, "SM16904":18, "SM16825":19, "SM16714":20, "UCS5603":21, "UCS2603":22, "SM16714D":23, "UCS7604":24, "UCS7804":25})], //TM1803, TM1809, TM1829, TLS3001, GW6205, MBI6120, TM1814B, SK6812, UCS8904B, LPD6803, LPD8806, WS2801, P9813, SK9822, MBI6020, TM1914A, GS8206, UCS2904, SM16904, SM16825, SM16714, UCS5603, UCS2603, SM16714D, UCS7604, UCS7804
            [102, 'rgb_order_data', tuya.valueConverterBasic.lookup({"RGB":0, "RBG":1, "GRB":2, "GBR":3, "BRG":4, "BGR":5, "RGBW":6, "RBGW":7, "GRBW":8, "GBRW":9, "BRGW":10, "BGRW":11, "WRGB":12, "WRBG":13, "WGRB":14, "WGBR":15, "WBRG":16, "WBGR":17})]
        ],
    },
};

module.exports = definition;
```

artomka, 2025-04-17:
> Here is my implementation of the Color Chanel 61, please give me feedback for improvement. On my device, I can control it like in the App.
> 
> ```
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const reporting = require('zigbee-herdsman-converters/lib/reporting');
> const ota = require('zigbee-herdsman-converters/lib/ota');
> const utils = require('zigbee-herdsman-converters/lib/utils');
> const globalStore = require('zigbee-herdsman-converters/lib/store');
> const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
> const e = exposes.presets;
> const ea = exposes.access;
> const tuya = require('zigbee-herdsman-converters/lib/tuya');
> const types = require("zigbee-herdsman-converters/lib/types");
> 
> const fzLocal = {
>     tuya_color_hs_segmented: {
>         cluster: 'manuSpecificTuya',
>         type: ['commandDataResponse', 'commandDataReport'],
>         convert: (model, msg, publish, options, meta) => {
>             const dp = msg.data.dp;
>             if (dp === 61) {
>                 const data = msg.data.data;
> 
>                 const paint_mode = data[1];
>                 let paint_mode_enum = paint_mode === 0x00 ? "White" :
>                                     paint_mode === 0x01 ? "Colour" :
>                                     paint_mode === 0x02 ? "Color_" : "Unknown";
>                 
>                 const gradient_on = data[2];
>                 const segmentCount = data[1];
>                 const set_segment_on = data[4];
>                 const color_value = data.readUInt16BE(5);
>                 const saturation_value = data.readUInt16BE(7);
>                 let brightness_value = 0;
>                 if (paint_mode > 0x00) {
>                     brightness_value = data.readUInt16BE(9);
>                 }
>                 let segment_id = 0;
>                 if (set_segment_on) {
>                     segment_id = data[12];
>                 }
> 
>                 return {
>                     paint_mode_enum: paint_mode_enum,
>                     gradient_on: gradient_on,
>                     set_segment_on: set_segment_on,
>                     color_value: color_value,
>                     color_saturation: saturation_value,
>                     brightness_value: brightness_value,
>                     segment_id: segment_id,
>                 };
>             }
>         },
>     },
> };
> 
> const tzLocal = {
>     tuya_color_hs_segmented: {
>         key: ['color_hs_segmented'],
>         convertSet: async (entity, key, value, meta) => {
>             const state = meta.state || {};
>             const newState = {...state};
> 
>             newState.paint_mode = value.paint_mode_enum === "White" ? 0x00 :
>                                 value.paint_mode_enum === "Color" ? 0x01 :
>                                 value.paint_mode_enum === "Color_" ? 0x02 : 0x00;
>             newState.paint_mode_enum = value.paint_mode_enum;
>             newState.gradient_on = value.gradient_on === "Yes" ? 0x01 : 0x00;
[... 148 lines cut ...]
>         tuyaDatapoints: [
>             [1, 'state', tuya.valueConverter.onOff],
>             [2, 'work_mode', tuya.valueConverterBasic.lookup({"Color" : tuya.enum(0), "White" : tuya.enum(1), "Scene" : tuya.enum(2), "Music" : tuya.enum(3)})],
>             [3, 'bright_value', tuya.valueConverter.raw],
>             [4, 'temp_value', tuya.valueConverter.raw],
>             //[5, 'color_data', tuya.valueConverter.raw],
>             [7, 'countdown', tuya.valueConverter.countdown],
>             //[51, 'dreamlightmic_scene_mode', tuya.valueConverter.raw],
>             //[52, 'dreamlightmic_music_data', tuya.valueConverter.raw],
>             [53, 'lightpixel_number_set', tuya.valueConverter.raw],
>             [61, 'color_hs_segmented', tuya.valueConverter.raw],
>             [101, 'chip_type_data', tuya.valueConverterBasic.lookup({"TM1803" : tuya.enum(0), "TM1809" : tuya.enum(1), "TM1829" : tuya.enum(2), "TLS3001" : tuya.enum(3), "GW6205" : tuya.enum(4), "MBI6120" : tuya.enum(5), "TM1814B" : tuya.enum(6), "SK6812" : tuya.enum(7), "UCS8904B" : tuya.enum(8), "LPD6803" : tuya.enum(9), "LPD8806" : tuya.enum(10), "WS2801" : tuya.enum(11), "P9813" : tuya.enum(12), "SK9822" : tuya.enum(13), "MBI6020" : tuya.enum(14), "TM1914A" : tuya.enum(15), "GS8206" : tuya.enum(16), "UCS2904" : tuya.enum(17), "SM16904" : tuya.enum(18), "SM16825" : tuya.enum(19), "SM16714" : tuya.enum(20), "UCS5603" : tuya.enum(21), "UCS2603" : tuya.enum(22), "SM16714D" : tuya.enum(23), "UCS7604" : tuya.enum(24), "UCS7804" : tuya.enum(25)})],
>             [102, 'rgb_order_data', tuya.valueConverterBasic.lookup({"RGB" : tuya.enum(0), "RBG" : tuya.enum(1), "GRB" : tuya.enum(2), "GBR" : tuya.enum(3), "BRG" : tuya.enum(4), "BGR" : tuya.enum(5), "RGBW" : tuya.enum(6), "RBGW" : tuya.enum(7), "GRBW" : tuya.enum(8), "GBRW" : tuya.enum(9), "BRGW" : tuya.enum(10), "BGRW" : tuya.enum(11), "WRGB" : tuya.enum(12), "WRBG" : tuya.enum(13), "WGRB" : tuya.enum(14), "WGBR" : tuya.enum(15), "WBRG" : tuya.enum(16), "WBGR" : tuya.enum(17)})],
>         ],
>     },
> };
> 
> module.exports = definition;
> ```

Hi Artjom, can you please help me manage this device's integration? I have created the file wz-spi.js and copied your last configuration (Colour channel 61), added it into the config file: 
```external_converters:
  - wz-spi.js
```
and restarted. I still see Unsupported device in the ZigBee2MQTT. What am I doing incorrectly?

artomka, 2025-04-21:
> Here is my implementation of the Color Chanel 61, please give me feedback for improvement. On my device I can control it like in the App.
> 
> ```
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const reporting = require('zigbee-herdsman-converters/lib/reporting');
> const ota = require('zigbee-herdsman-converters/lib/ota');
> const utils = require('zigbee-herdsman-converters/lib/utils');
> const globalStore = require('zigbee-herdsman-converters/lib/store');
> const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
> const e = exposes.presets;
> const ea = exposes.access;
> const tuya = require('zigbee-herdsman-converters/lib/tuya');
> const types = require("zigbee-herdsman-converters/lib/types");
> 
> const fzLocal = {
>     tuya_color_hs_segmented: {
>         cluster: 'manuSpecificTuya',
>         type: ['commandDataResponse', 'commandDataReport'],
>         convert: (model, msg, publish, options, meta) => {
>             const dp = msg.data.dp;
>             if (dp === 61) {
>                 const data = msg.data.data;
> 
>                 const paint_mode = data[1];
>                 let paint_mode_enum = paint_mode === 0x00 ? "White" :
>                                     paint_mode === 0x01 ? "Color" :
>                                     paint_mode === 0x02 ? "Color_" : "Unknown";
>                 
>                 const gradient_on = data[2];
>                 const segmentCount = data[1];
>                 const set_segment_on = data[4];
>                 const color_value = data.readUInt16BE(5);
>                 const saturation_value = data.readUInt16BE(7);
>                 let brightness_value = 0;
>                 if (paint_mode > 0x00) {
>                     brightness_value = data.readUInt16BE(9);
>                 }
>                 let segment_id = 0;
>                 if (set_segment_on) {
>                     segment_id = data[12];
>                 }
> 
>                 return {
>                     paint_mode_enum: paint_mode_enum,
>                     gradient_on: gradient_on,
>                     set_segment_on: set_segment_on,
>                     color_value: color_value,
>                     color_saturation: saturation_value,
>                     brightness_value: brightness_value,
>                     segment_id: segment_id,
>                 };
>             }
>         },
>     },
> };
> 
> const tzLocal = {
>     tuya_color_hs_segmented: {
>         key: ['color_hs_segmented'],
>         convertSet: async (entity, key, value, meta) => {
>             const state = meta.state || {};
>             const newState = {...state};
> 
>             newState.paint_mode = value.paint_mode_enum === "White" ? 0x00 :
>                                 value.paint_mode_enum === "Color" ? 0x01 :
>                                 value.paint_mode_enum === "Color_" ? 0x02 : 0x00;
>             newState.paint_mode_enum = value.paint_mode_enum;
>             newState.gradient_on = value.gradient_on === "Yes" ? 0x01 : 0x00;
[... 388 lines cut ...]
            'BRGW', 'BGRW', 'WRGB', 'WRBG', 'WGRB', 'WGBR', 'WBRG', 'WBGR', 'UNKNOWN',
        ]).withDescription('RGB Order'),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'state', tuya.valueConverter.onOff],
            [2, 'work_mode', tuya.valueConverterBasic.lookup({
                'white': 0,
                'color': 1,
                'scene': 2,
                'music': 3,
            })],
            [3, 'bright_value', tuya.valueConverter.raw],
            [4, 'temp_value', tuya.valueConverter.raw],
            [46, 'light_length', tuya.valueConverter.raw],
            [47, 'light_pixel', tuya.valueConverter.raw],
            [53, 'lightpixel_number_set', tuya.valueConverter.raw],
            [101, 'chip_type_data', tuya.valueConverter.raw],
            [102, 'rgb_order_data', tuya.valueConverter.raw],
            [103, 'color_hs', tuya.valueConverter.raw],
        ],
    },
};

module.exports = definition;`

THK-artjom, 2025-04-21:
Here is additonally the Datapoint 51 to control the Scenes

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const types = require("zigbee-herdsman-converters/lib/types");

const fzLocal = {
    tuya_color_hs_segmented: {
        cluster: 'manuSpecificTuya',
        type: ['commandDataResponse', 'commandDataReport'],
        convert: (model, msg, publish, options, meta) => {
            const dp = msg.data.dp;
            if (dp === 61) {
                const data = msg.data.data;

                const paint_mode = data[1];
                let paint_mode_enum = paint_mode === 0x00 ? "White" :
                                    paint_mode === 0x01 ? "Color" :
                                    paint_mode === 0x02 ? "Color_" : "Unknown";
                
                const gradient_on = data[2];
                const segmentCount = data[1];
                const set_segment_on = data[4];
                const color_value = data.readUInt16BE(5);
                const saturation_value = data.readUInt16BE(7);
                let brightness_value = 0;
                if (paint_mode > 0x00) {
                    brightness_value = data.readUInt16BE(9);
                }
                let segment_id = 0;
                if (set_segment_on) {
                    segment_id = data[12];
                }

                return {
                    paint_mode_enum: paint_mode_enum,
                    gradient_on: gradient_on,
                    set_segment_on: set_segment_on,
                    color_value: color_value,
                    color_saturation: saturation_value,
                    brightness_value: brightness_value,
                    segment_id: segment_id,
                };
            }
        },
    },
};

const tzLocal = {
    tuya_color_hs_segmented: {
        key: ['color_hs_segmented'],
        convertSet: async (entity, key, value, meta) => {
            const state = meta.state || {};
            const newState = {...state};

            newState.paint_mode = value.paint_mode_enum === "White" ? 0x00 :
                                value.paint_mode_enum === "Color" ? 0x01 :
                                value.paint_mode_enum === "Color_" ? 0x02 : 0x00;
            newState.paint_mode_enum = value.paint_mode_enum;
            newState.gradient_on = value.gradient_on === "Yes" ? 0x01 : 0x00;
[... 406 lines cut ...]
                    .withValueMax(100)
                    .withValueStep(1)
                    .withUnit("%")
                    .withDescription("Color 6 Saturation"))
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'state', tuya.valueConverter.onOff],
            [3, 'bright_value', tuya.valueConverter.raw],
            [4, 'temp_value', tuya.valueConverter.raw],
            [7, 'countdown', tuya.valueConverter.countdown],
            [51, 'dreamlightmic_scene_mode', tuya.valueConverter.raw],
            [46, 'state_unknown1', tuya.valueConverter.raw], //probably max_brightness
            [47, 'state_unknown2', tuya.valueConverter.raw], //probably max_color_temperature
            //[52, 'dreamlightmic_music_data', tuya.valueConverter.raw],
            [53, 'lightpixel_number_set', tuya.valueConverter.raw],
            [61, 'color_hs_segmented', tuya.valueConverter.raw],
            [101, 'chip_type_data', tuya.valueConverterBasic.lookup({"TM1803" : tuya.enum(0), "TM1809" : tuya.enum(1), "TM1829" : tuya.enum(2), "TLS3001" : tuya.enum(3), "GW6205" : tuya.enum(4), "MBI6120" : tuya.enum(5), "TM1814B" : tuya.enum(6), "SK6812" : tuya.enum(7), "UCS8904B" : tuya.enum(8), "LPD6803" : tuya.enum(9), "LPD8806" : tuya.enum(10), "WS2801" : tuya.enum(11), "P9813" : tuya.enum(12), "SK9822" : tuya.enum(13), "MBI6020" : tuya.enum(14), "TM1914A" : tuya.enum(15), "GS8206" : tuya.enum(16), "UCS2904" : tuya.enum(17), "SM16904" : tuya.enum(18), "SM16825" : tuya.enum(19), "SM16714" : tuya.enum(20), "UCS5603" : tuya.enum(21), "UCS2603" : tuya.enum(22), "SM16714D" : tuya.enum(23), "UCS7604" : tuya.enum(24), "UCS7804" : tuya.enum(25)})],
            [102, 'rgb_order_data', tuya.valueConverterBasic.lookup({"RGB" : tuya.enum(0), "RBG" : tuya.enum(1), "GRB" : tuya.enum(2), "GBR" : tuya.enum(3), "BRG" : tuya.enum(4), "BGR" : tuya.enum(5), "RGBW" : tuya.enum(6), "RBGW" : tuya.enum(7), "GRBW" : tuya.enum(8), "GBRW" : tuya.enum(9), "BRGW" : tuya.enum(10), "BGRW" : tuya.enum(11), "WRGB" : tuya.enum(12), "WRBG" : tuya.enum(13), "WGRB" : tuya.enum(14), "WGBR" : tuya.enum(15), "WBRG" : tuya.enum(16), "WBGR" : tuya.enum(17)})],
        ],
    },
};

module.exports = definition;
```

THK-artjom, 2025-04-21:
> Can you check a bit modified version from you?
> 
> ```
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const tuya = require('zigbee-herdsman-converters/lib/tuya');
> const e = exposes.presets;
> const ea = exposes.access;
> 
> const fzLocal = {
>     tuya_color_hs_segmented: {
>         cluster: 'manuSpecificTuya',
>         type: ['commandDataResponse', 'commandDataReport'],
>         convert: (model, msg, publish, options, meta) => {
>             const dp = msg.data.dp;
>             const data = msg.data.data;
> 
>             switch (dp) {
>                 case 1: // switch_led
>                     return { state: data[0] === 1 ? 'ON' : 'OFF' };
> 
>                 case 2: // work_mode
>                     return { work_mode: data.toString() };
> 
>                 case 3: // bright_value
>                     return { bright_value: data.readUInt16BE(0) };
> 
>                 case 4: // temp_value
>                     return { temp_value: data.readUInt16BE(0) };
> 
>                 case 46: // light_length
>                     return {
>                         light_length: data.readUInt16BE(0),
>                     };
> 
>                 case 47: // light_pixel
>                     return {
>                         light_pixel: data.readUInt16BE(0),
>                     };
> 
>                 case 53: // lightpixel_number_set
>                     return {
>                         lightpixel_number_set: data.readUInt16BE(0),
>                     };
> 
>                 case 101: // chip_type_data
>                     const chipTypes = {
>                         1: 'TM1803', 2: 'TM1809', 3: 'TM1829', 4: 'TLS3001',
>                         5: 'GW6205', 6: 'MBI6120', 7: 'TM1814B', 8: 'SK6812',
>                         9: 'UCS8904B', 10: 'LPD6803', 11: 'LPD8806', 12: 'WS2801',
>                         13: 'P9813', 14: 'SK9822', 15: 'MBI6020', 16: 'TM1914A',
>                         17: 'GS8206', 18: 'UCS2904', 19: 'SM16904', 20: 'SM16825',
>                         21: 'SM16714', 22: 'UCS5603', 23: 'UCS2603', 24: 'SM16714D',
>                         25: 'UCS7604', 26: 'UCS7804',
>                     };
>                     return { chip_type_data: chipTypes[data[0]] || 'UNKNOWN' };
> 
>                 case 102: // rgb_order_data
>                     const rgbOrders = {
>                         1: 'RGB', 2: 'RBG', 3: 'GRB', 4: 'GBR', 5: 'BRG', 6: 'BGR',
>                         7: 'RGBW', 8: 'RBGW', 9: 'GRBW', 10: 'GBRW', 11: 'BRGW', 12: 'BGRW',
>                         13: 'WRGB', 14: 'WRBG', 15: 'WGRB', 16: 'WGBR', 17: 'WBRG', 18: 'WBGR',
>                         19: 'UNKNOWN',
>                     };
>                     return { rgb_order_data: rgbOrders[data[0]] || 'UNKNOWN' };
> 
>                 case 103: // color_hs_segmented
>                     return { color_hs: { hue: data[0], saturation: data[1] } };
> 
[... 162 lines cut ...]
>                 'scene': 2,
>                 'music': 3,
>             })],
>             [3, 'bright_value', tuya.valueConverter.raw],
>             [4, 'temp_value', tuya.valueConverter.raw],
>             [46, 'light_length', tuya.valueConverter.raw],
>             [47, 'light_pixel', tuya.valueConverter.raw],
>             [53, 'lightpixel_number_set', tuya.valueConverter.raw],
>             [101, 'chip_type_data', tuya.valueConverter.raw],
>             [102, 'rgb_order_data', tuya.valueConverter.raw],
>             [103, 'color_hs', tuya.valueConverter.raw],
>         ],
>     },
> };
> 
> module.exports = definition;
> ```

I've checked your script and it has unfortunatelly some issues:

datapoint 2 (work_mode) seems to not have any impact . I suppose it is readonly and sends data to the tuya app.
dataponit 46,47 seems to be both readonly
color setting does not have any impact as you probably to not use datapoint 61 in your script

You could adapt the script i postet above to use the color picker. This seems to me a valid improvement. Thank you.

artomka, 2025-04-21:
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const exposes = [...]

Great job! Thanx

artomka, 2025-04-21:
@THK-artjom Do you have well-functioning MQTT Light code for HA?

I cannot make it better :/ 
```

mqtt:
  light:
    - name: "WZ-SPI LED Controller MQTT"
      unique_id: wz_spi_led_controller_light

      # Basic ON/OFF Control
      command_topic: "zigbee2mqtt/WZ-SPI LED Controller/set"
      state_topic: "zigbee2mqtt/WZ-SPI LED Controller"
      state_value_template: "{{ 'ON' if value_json.state == 'ON' else 'OFF' }}"
      payload_on: "ON"
      payload_off: "OFF"

      # Brightness Control (0-1000 scale as per the script)
      brightness_command_topic: "zigbee2mqtt/WZ-SPI LED Controller/set/bright_value"
      brightness_state_topic: "zigbee2mqtt/WZ-SPI LED Controller"
      brightness_value_template: "{{ value_json.bright_value }}"
      brightness_scale: 1000

      # Colour Temperature Control with Automatic Mode Switching
      color_temp_command_topic: "zigbee2mqtt/WZ-SPI LED Controller/set"
      color_temp_value_template: "{{ value_json.temp_value }}"
      color_temp_command_template: >
        {%- set mode = 'white' -%}
        {"work_mode": "{{ mode }}", "temp_value": {{ value }}}

      # HS Color Control with Automatic Mode Switching
      hs_command_topic: "zigbee2mqtt/WZ-SPI LED Controller/set"
      hs_value_template: >
        {% if value_json.work_mode == 'Color' %}
          [
            {{ (value_json.color_hs_segmented["0"] | float * 360 / 330) | round(0) }},
            {{ (value_json.color_hs_segmented["1"] | float / 10) | round(0) }}
          ]
        {% else %}
          null
        {% endif %}
      hs_command_template: >
        {%- set mode = 'color' -%}
        {"work_mode": "{{ mode }}", "color_hs_segmented": {"hue": {{ h }}, "saturation": {{ s }}}}

      # Effect Control (Dreamlight Effects)
      effect_command_topic: "zigbee2mqtt/WZ-SPI LED Controller/set"
      effect_state_topic: "zigbee2mqtt/WZ-SPI LED Controller"
      effect_value_template: "{{ value_json.scene_type }}"
      effect_list:
        - "Gradient"
        - "Jump"
        - "Breathe"
        - "Flashing"
        - "Flowing"
        - "Rainbow"
        - "Meteor"
        - "PileUp"
        - "Falling"
        - "Follow"
        - "Flutter"
        - "Flash"
        - "Rebound"
        - "Shuttle"
        - "Random"
        - "Switch"

```

THK-artjom, 2025-04-23:
I am not that familar yet with the HA Stuff. This is my first device I have tried to reverse egineer by package analysis. As I got now the effects changel working. I will do some improvements and simplifaction to the exposes stuff and check where to go from there. I basically try to synchronise my light colors in one place.

rejoe2, 2025-06-03:
Hi there, tried to use this converter on recent zigbee2mqtt versions, latest is
```
Zigbee2MQTT-Version
    [2.4.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/2.4.0) commit: [bdb94da46e0461337f4a61b4f2a6bfa5172f608f ](https://github.com/Koenkk/zigbee2mqtt/commit/bdb94da46e0461337f4a61b4f2a6bfa5172f608f)

Coordinator-Typ
    ZStack3x0

Coordinator-Version
    20240710

Coordinator IEEE Adresse
    0x00124b00257bfe5f

Frontend-Version
    0.9.13

Zigbee Herdsman Konverter Version
    23.53.0

Zigbee Herdsman Version
    4.1.0
```
But unfortunately, zigbee2mqtt refuses to load the versions provided in https://github.com/Koenkk/zigbee2mqtt/issues/23991#issuecomment-2819656347 and the post before that.
Tried upload via frontend ("mod is not a constructor") and direct edit of the js file in external_converters directory.

Any ideas on this?

galaxyfar, 2025-08-22:
Following the long discussion trail it seems folks have had luck getting the integration to work for device in question. 

What is the hold up to make it official?

wawan822, 2025-10-07:
> ml88nl left a comment [ https://github.com/Koenkk/zigbee2mqtt/issues/23991#issuecomment-3373060780 | (Koenkk/zigbee2mqtt#23991) ]
>
> I've got an update that creates a working solution. I've converted the TypeScript code from the gledopto.ts (to a external converter file javascript) and replaced some values related to the device definition to work with the _TZE204_8fffc3kb:

Hi ML88NL 

It's works! 
Very nice job! 
Thank you so much 

Sincerely 

Erwann 

const libColor = require('zigbee-herdsman-converters/lib/color'); 
const NS = "zhc:tuya"; 

const SCENE_DATA = { 
ice_land_blue: [0x01, 0x15, 0x0a, 0x52, 0x52, 0xe0, 0x00, 0x00, 0x64, 0x00, 0xc1, 0x61, 0x00, 0xb4, 0x30, 0x00, 0xb5, 0x52, 0x00, 0xc4, 0x63], 
glacier_express: [0x01, 0x16, 0x0a, 0x64, 0x64, 0x60, 0x00, 0x00, 0x64, 0x00, 0x92, 0x5f, 0x00, 0xc6, 0x60], 
sea_of_clouds: [0x01, 0x17, 0x03, 0x5e, 0x5e, 0x60, 0x00, 0x00, 0x64, 0x00, 0x38, 0x2f, 0x00, 0x1e, 0x5c, 0x00, 0xd5, 0x45, 0x01, 0x1a, 0x64], 
fireworks_at_sea: [0x01, 0x18, 0x02, 0x64, 0x64, 0xe0, 0x00, 0x00, 0x64, 0x00, 0xb2, 0x39, 0x01, 0x0a, 0x64, 0x01, 0x2d, 0x64, 0x01, 0x3f, 0x64], 
firefly_night: [0x01, 0x1a, 0x03, 0x4b, 0x4b, 0xe0, 0x00, 0x00, 0x64, 0x00, 0xe0, 0x39, 0x01, 0x09, 0x53], 
grass_land: [0x01, 0x1c, 0x0a, 0x5a, 0x5a, 0xe0, 0x00, 0x00, 0x52, 0x00, 0x9d, 0x64, 0x00, 0x8e, 0x64], 
northern_lights: [0x01, 0x1d, 0x03, 0x52, 0x52, 0xe0, 0x00, 0x00, 0x64, 0x00, 0xae, 0x64, 0x00, 0xa6, 0x64, 0x00, 0xc1, 0x64, 0x00, 0xcc, 0x64], 
late_autumn: [0x01, 0x1e, 0x0a, 0x52, 0x52, 0xe0, 0x00, 0x00, 0x64, 0x00, 0x19, 0x64, 0x00, 0x22, 0x5e, 0x00, 0x2c, 0x5b, 0x00, 0x14, 0x64, 0x00, 0x0c, 0x64], 
game: [0x01, 0x1f, 0x02, 0x5f, 0x5f, 0x60, 0x00, 0x00, 0x64, 0x01, 0x10, 0x64, 0x00, 0xd2, 0x64, 0x00, 0xad, 0x64, 0x00, 0x8b, 0x64], 
holiday: [0x01, 0x20, 0x0a, 0x55, 0x55, 0x60, 0x00, 0x00, 0x64, 0x00, 0xc2, 0x58, 0x01, 0x3e, 0x33, 0x00, 0xff, 0x46, 0x01, 0x1d, 0x64], 
party: [0x01, 0x22, 0x04, 0x64, 0x64, 0x60, 0x00, 0x00, 0x64, 0x00, 0xd7, 0x5c, 0x00, 0xbc, 0x53, 0x00, 0x37, 0x1e, 0x00, 0x2c, 0x3f, 0x01, 0x61, 0x3f], 
trend: [0x01, 0x23, 0x02, 0x64, 0x64, 0x60, 0x00, 0x00, 0x64, 0x01, 0x08, 0x4b, 0x00, 0xb1, 0x2f, 0x00, 0xcd, 0x57], 
meditation: [0x01, 0x25, 0x03, 0x43, 0x43, 0x60, 0x00, 0x00, 0x64, 0x00, 0xb7, 0x35, 0x00, 0x9b, 0x54, 0x00, 0xcd, 0x61], 
dating: [0x01, 0x26, 0x01, 0x59, 0x59, 0xe0, 0x00, 0x00, 0x64, 0x01, 0x19, 0x47, 0x01, 0x49, 0x3d, 0x00, 0xcd, 0x61, 0x00, 0x26, 0x64], 
valentines_day: [0x01, 0x2a, 0x01, 0x64, 0x64, 0x60, 0x00, 0x00, 0x64, 0x01, 0x15, 0x64, 0x01, 0x05, 0x64, 0x01, 0x45, 0x64, 0x01, 0x2f, 0x64], 
neon_world: [0x01, 0x37, 0x0a, 0x5a, 0x5a, 0x60, 0x00, 0x00, 0x64, 0x00, 0x33, 0x58, 0x00, 0x18, 0x64, 0x01, 0x00, 0x45, 0x00, 0xe3, 0x5e, 0x00, 0xac, 0x30], 
}; 

const MUSIC_DATA = { 
rock: [ 
0x01, 0x01, 0x00, 0x03, 0x64, 0x32, 0x01, 0x00, 0x00, 0x64, 0x00, 0x00, 0x64, 0x00, 0x78, 0x64, 0x00, 0xf0, 0x64, 0x00, 0x3c, 0x64, 0x00, 0xb4, 0x64, 0x01, 0x2c, 0x64, 0x00, 0x00, 0x00, 
], 
jazz: [ 
0x01, 0x01, 0x00, 0x02, 0x64, 0x32, 0x01, 0x00, 0x00, 0x64, 0x00, 0x00, 0x50, 0x00, 0x78, 0x50, 0x00, 0xf0, 0x50, 0x00, 0x3c, 0x50, 0x00, 0xb4, 0x50, 0x01, 0x2c, 0x50, 0x00, 0x00, 0x00, 
], 
classic: [ 
0x01, 0x01, 0x00, 0x12, 0x64, 0x32, 0x01, 0x00, 0x00, 0x64, 0x00, 0x00, 0x64, 0x00, 0x78, 0x64, 0x00, 0xf0, 0x64, 0x00, 0x3c, 0x64, 0x00, 0xb4, 0x64, 0x01, 0x2c, 0x64, 0x00, 0x00, 0x00, 
], 
rolling: [ 
0x01, 0x01, 0x01, 0x12, 0x64, 0x32, 0x01, 0x00, 0x00, 0x64, 0x00, 0x00, 0x64, 0x00, 0x78, 0x64, 0x00, 0xf0, 0x64, 0x00, 0x3c, 0x64, 0x00, 0xb4, 0x64, 0x01, 0x2c, 0x64, 0x00, 0x00, 0x00, 
], 
energy: [ 
0x01, 0x01, 0x02, 0x12, 0x64, 0x32, 0x01, 0x00, 0x00, 0x64, 0x00, 0x00, 0x64, 0x00, 0x78, 0x64, 0x00, 0xf0, 0x64, 0x00, 0x3c, 0x64, 0x00, 0xb4, 0x64, 0x01, 0x2c, 0x64, 0x00, 0x00, 0x00, 
], 
spectrum: [ 
0x01, 0x01, 0x03, 0x12, 0x64, 0x32, 0x01, 0x00, 0x00, 0x64, 0x00, 0x00, 0x64, 0x00, 0x78, 0x64, 0x00, 0xf0, 0x64, 0x00, 0x3c, 0x64, 0x00, 0xb4, 0x64, 0x01, 0x2c, 0x64, 0x00, 0x00, 0x00, 
], 
}; 

const localValueConverter = { 
scene_data_converter: { 
to: (v) => { 
const sceneData = SCENE_DATA[v]; 
if (!sceneData) { 
throw new Error( Unknown scene: ${v} ); 
} 
return sceneData; 
}, 
from: (v) => { 
const data = Buffer.isBuffer(v) ? Array.from(v) : v; 
for (const [sceneName, sceneData] of Object.entries(SCENE_DATA)) { 
if (JSON.stringify(data) === JSON.stringify(sceneData)) { 
return sceneName; 
} 
} 
return null; 
[... 330 lines cut ...]
white: tuya.enum(0), 
colour: tuya.enum(1), 
[4, "color_temp", tuya.valueConverter.raw], 
[51, "scene", localValueConverter.scene_data_converter], 
[52, "music_mode", localValueConverter.music_data_converter], 
102, 
"light_bead_sequence", 
WS2801: tuya.enum(0), 
LPD6803: tuya.enum(1), 
LPD8803: tuya.enum(2), 
WS2811: tuya.enum(3), 
TM1814B: tuya.enum(4), 
TM1934A: tuya.enum(5), 
SK6812: tuya.enum(6), 
SK9822: tuya.enum(7), 
UCS8904B: tuya.enum(8), 
WS2805: tuya.enum(9), 
}), 
], 
[103, "do_not_disturb", tuya.valueConverter.onOff], 
], 
}, 
}; 

This provides me with full control including Scenes

Christini, 2025-10-21:
Hi,
Nice job guys, especially to ml88nl 

@ abertolin : some typographical errors in the code, probably during copy/paste.

line 60 : "throw new Error(`Unknown scene: ${v}` );"
line 215 : throw new Error(`Unknown scene: ${sceneName}`);
line 241 : throw new Error(`Unknown music mode: ${musicMode}`);
line 270 : const mapped = Math.round(utils.mapNumberRange(utils.toNumber(brightness, "brightness"), [ 0, 254, 10, 1000 | 0, 254, 10, 1000 ] ));

Once corrected, the converter works perfectly. Enjoy! :)

abertolin, 2025-10-21:
Something is strange, except for line 60, the other lines you mentioned are different that mine :(

marcin-gacek, 2025-10-28:
> Hi,
> this is the code that work for me:
>
> {
> "description": "Zigbee WZ-SPI RGBW Controller V1",
> "exposes": [
> {
> "features": [
> {
> "access": 7,
> "description": "On/off state of this light",
> "label": "State",
> "name": "state",
> "property": "state",
> "type": "binary",
> "value_off": "OFF",
> "value_on": "ON",
> "value_toggle": "TOGGLE"
> }
> ],
> "type": "light"
> },
> {
> "access": 2,
> "description": "Brightness",
> [... 480 lines cut ...]
> "options": [],
> "source": "external",
> "supports_ota": false,
> "vendor": "_TZE204_8fffc3kb"
> }

@abertolin
Can you show your file?
I mean your *.js file from external-converters.

Below you have shown external definition but I do not see option to modify
mine definition, it is read-only.

I was able to fix the one js file posted here but my definition has less
number of led-chips and my led strip is not working correctly.

Best regards
Marcin

Caspero83, 2025-12-12:
Hey I got it working.
This is how i solved it.
The wz-spi.js file mentioned should be in:
external_converters folder not directly in the zigbee2mqtt folder like mentioned above.
/homeassistant/zigbee2mqtt/external_converters

[image]

Now i only need suport for the WS2811 led chip....

[image]

abertolin, 2025-12-12:
Hi,
I have the IC SM16703 that is similar to yours and seems that GS8206 "profile" works for mine, maybe it will be fine also for yours.
It's important that you put the right number of LED.

Caspero83, 2025-12-16:
> Hi @bigvag 
> 
> I also have WS2811 lights, and for me the profile that works is TM1809. I have 3m of LED strip and with 120 pixels I cover it entirely (you can see the image I posted 5 posts above).
> 
> My problem is that I can't make automations with parameters like light effects.

Hi @bigvag 

Yes, I actually found the device data sheet and it says that WS2811 should use TM1809.

[image]

Maybe this could help @abertolin as well? Maybe it could improve your led a little bit :) Let me know if it helps, would be nice to actually being able to give something back for once.
I think I have somewhat figured out what is the biggest issue with WZ-SPI 
Just so everyone know i am an armature so this is just my understanding i could very well be wrong.  
But i think we can get it to change colors and minor changes. But if you want havier data driven comands like running water and such  The Zigbee net is not enough. If you use WZ-SPIs specific app and Gateway (is it Tuya? I do knot remember) It should work because the changes occurs in the device it self. But when we use HA the data is sent from HA and then it is to heavy for Zigbee signals much like CCTW. As i wrote do not take my word for it.