Koenkk/zigbee2mqtt #23993: IKEA Starkvind reporting Error: Cluster with name 'manuSpecificIkeaAirPurifier' does not exist

AquaXV, 2024-09-16:
### What happened?

```
Error z2m: Publish 'get' 'fan_state' to 'air_purifier' failed: 'Error: Cluster with name 'manuSpecificIkeaAirPurifier' does not exist' 
```

when trying to refresh "Exposes" entries

----
```
z2m: Failed to configure 'air_purifier', attempt 3 (Error: Cluster with name 'manuSpecificIkeaAirPurifier' does not exist 
at getClusterDefinition (/app/node_modules/zigbee-herdsman/src/zspec/zcl/utils.ts:166:15) 
at Object.getCluster (/app/node_modules/zigbee-herdsman/src/zspec/zcl/utils.ts:285:29) 
at Endpoint.getCluster (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:849:26) 
at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:518:30) 
at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:42:24) 
at configure (/app/node_modules/zigbee-herdsman-converters/src/lib/ikea.ts:414:29) 
at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:193:27) 
at Configure.configure (/app/lib/extension/configure.ts:125:37) 
at /app/lib/extension/configure.ts:87:62 
at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:23))
``` 
when trying to Interview or Reconfigure the device.

### How to reproduce it (minimal and precise)

N/A 
Started happening this weekend after updating z2m to 1.40.1. I don't know the version I ran prior, it had been a while since I updated 😅, my guess is around 1.36.0

### Zigbee2MQTT version

1.40.1

### Adapter firmware version

20240710

### Adapter

SONOFF ITead Zigbee 3.0 Dongle (CC2652P)

### Setup

In Docker with MQTT, and Home Assistant Integration

### Debug log

Fetching Fan state: 
```
[2024-09-16 19:49:16] debug:    z2m:mqtt: Received MQTT message on 'zigbee2mqtt/air_purifier/get' with data '{"fan_state":""}'
[2024-09-16 19:49:16] debug:    z2m: Publishing get 'get' 'fan_state' to 'air_purifier'
[2024-09-16 19:49:16] error:    z2m: Publish 'get' 'fan_state' to 'air_purifier' failed: 'Error: Cluster with name 'manuSpecificIkeaAirPurifier' does not exist'
[2024-09-16 19:49:16] debug:    z2m: Error: Cluster with name 'manuSpecificIkeaAirPurifier' does not exist
    at getClusterDefinition (/app/node_modules/zigbee-herdsman/src/zspec/zcl/utils.ts:166:15)
    at Object.getCluster (/app/node_modules/zigbee-herdsman/src/zspec/zcl/utils.ts:285:29)
    at Endpoint.getCluster (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:849:26)
    at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:429:30)
    at Object.convertGet (/app/node_modules/zigbee-herdsman-converters/src/lib/ikea.ts:360:30)
    at Publish.onMQTTMessage (/app/lib/extension/publish.ts:305:37)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:23)
    at EventEmitter.emit (node:events:529:35)
    at EventBus.emitMQTTMessage (/app/lib/eventBus.ts:115:22)
    at MQTT.onMessage (/app/lib/mqtt.ts:147:27)
```

Reconfiguring:
```
[2024-09-16 20:03:01] debug:    z2m:mqtt: Received MQTT message on 'zigbee2mqtt/bridge/request/device/configure' with data '{"id":"air_purifier","transaction":"sli1q-8"}'
[2024-09-16 20:03:01] info:     z2m: Configuring 'air_purifier'
[2024-09-16 20:03:01] error:    z2m: Failed to configure 'air_purifier', attempt 5 (Error: Cluster with name 'manuSpecificIkeaAirPurifier' does not exist
    at getClusterDefinition (/app/node_modules/zigbee-herdsman/src/zspec/zcl/utils.ts:166:15)
    at Object.getCluster (/app/node_modules/zigbee-herdsman/src/zspec/zcl/utils.ts:285:29)
    at Endpoint.getCluster (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:849:26)
    at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:518:30)
    at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:42:24)
    at configure (/app/node_modules/zigbee-herdsman-converters/src/lib/ikea.ts:414:29)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:193:27)
    at Configure.configure (/app/lib/extension/configure.ts:125:37)
    at Configure.onMQTTMessage (/app/lib/extension/configure.ts:56:32)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:23))
[2024-09-16 20:03:01] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"air_purifier"},"error":"Failed to configure (Cluster with name 'manuSpecificIkeaAirPurifier' does not exist)","status":"error","transaction":"sli1q-8"}'
```

Performing interview is a bit larger; log included.
[interview.txt](https://github.com/user-attachments/files/17019291/interview.txt)

sjorge, 2024-09-21:
@Koenkk I can't replicate this with mine, I wonder if there is still some sort of race with adding the custom clusters from zhc.

Koenkk, 2024-09-22:
Think I found it, pushed a possible fix.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

AquaXV, 2024-09-23:
Appears to be fixed now!

Thanks!