Koenkk/zigbee2mqtt #23996: Cannot pair IKEA Badring E2202

Ar-eh, 2024-09-17:
### What happened?

Badring device fails to be configured or interviewed upon entering pairing mode.

### What did you expect to happen?

Expected device to be paired successfully

### How to reproduce it (minimal and precise)

Enable permit join in zigbee2mqtt front end, press pairing button on Badring 4x. Light on Badring flashes as if paired successfully, but device does not appear in zigbee2mqtt.

### Zigbee2MQTT version

1.40.1

### Adapter firmware version

0x26530900

### Adapter

Conbee3

### Setup

Docker on Synology DS718+

### Debug log

[2024-09-16 21:33:28] debug: 	zh:deconz: PERMIT_JOIN - 254 seconds
[2024-09-16 21:33:28] debug: 	zh:deconz: zclFrame to all - zclFrame.payload:
[2024-09-16 21:33:28] debug: 	zh:deconz: [object Object]
[2024-09-16 21:33:28] debug: 	zh:deconz: sendZclFrameToAll - message send
[2024-09-16 21:33:28] debug: 	zh:deconz:driver: DATA_REQUEST - destAddr: 0xfffd EP:242 SeqNr. 144 request id: 117
[2024-09-16 21:33:28] debug: 	zh:deconz:frameparser: DATA_REQUEST RESPONSE - request id: 117 status: 0
[2024-09-16 21:33:28] debug: 	zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2024-09-16 21:33:29] debug: 	zh:deconz:frameparser: DEVICE_STATE changed: 100110
[2024-09-16 21:33:29] debug: 	zh:deconz:driver: networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2024-09-16 21:33:29] debug: 	zh:deconz:driver: query aps data confirm
[2024-09-16 21:33:29] debug: 	zh:deconz:driver: DATA_CONFIRM - sending data state request - SeqNr. 145
[2024-09-16 21:33:29] debug: 	zh:deconz:frameparser: DATA_CONFIRM RESPONSE - destAddr: 0xfffd request id: 117 confirm status: 0
[2024-09-16 21:33:29] debug: 	zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2024-09-16 21:33:29] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"zrved-1"}'
[2024-09-16 21:33:30] debug: 	zh:deconz:driver: send read device state from queue. seqNr: 146
[2024-09-16 21:33:30] debug: 	zh:deconz:frameparser: device state: 100010
[2024-09-16 21:33:30] debug: 	zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2024-09-16 21:33:34] debug: 	zh:deconz:frameparser: DEVICE_STATE changed: 101010
[2024-09-16 21:33:34] debug: 	zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
[2024-09-16 21:33:34] debug: 	zh:deconz:driver: query aps data indication
[2024-09-16 21:33:34] debug: 	zh:deconz:driver: DATA_INDICATION - sending read data request - SeqNr. 147
[2024-09-16 21:33:34] debug: 	zh:deconz:frameparser: DATA_INDICATION RESPONSE - seqNr. 147 srcAddr: 0xa9b9 destAddr: 0x0 profile id: 0x104 cluster id: 0xfc7d lqi: 218
[2024-09-16 21:33:34] debug: 	zh:deconz:frameparser: response payload: 12,124,17,112,10,4,0,33,15,0
[2024-09-16 21:33:34] debug: 	zh:controller: Received payload: clusterID=64637, address=43449, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=218, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":true,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"manufacturerCode":4476,"transactionSequenceNumber":112,"commandIdentifier":10},"payload":[{"attrId":4,"dataType":33,"attrData":15}],"command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}}
[2024-09-16 21:33:34] debug: 	zh:controller:endpoint: ZCL command 0xdc8e95fffe5625e2/1 manuSpecificIkeaAirPurifier.defaultRsp({"cmdId":10,"statusCode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"reservedBits":0,"manufacturerCode":4476,"transactionSequenceNumber":112,"writeUndiv":false})
[2024-09-16 21:33:34] debug: 	zh:deconz: no response expected (112)
[2024-09-16 21:33:34] debug: 	zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2024-09-16 21:33:34] debug: 	z2m: Received Zigbee message from 'Air Purifier North (Living Room)', type 'attributeReport', cluster 'manuSpecificIkeaAirPurifier', data '{"particulateMatter25Measurement":15}' from endpoint 1 with groupID 0
[2024-09-16 21:33:35] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Air Purifier North (Living Room)', payload '{"air_quality":"good","child_lock":"UNLOCK","device_age":null,"fan_mode":"auto","fan_speed":1,"fan_state":"ON","filter_age":188355,"led_enable":true,"linkquality":218,"pm25":15,"replace_filter":false,"update":{"installed_version":69633,"latest_version":69633,"state":"idle"},"update_available":false}'
[2024-09-16 21:33:35] debug: 	zh:deconz:driver: DATA_REQUEST - destAddr: 0xa9b9 EP:1 SeqNr. 148 request id: 118
[2024-09-16 21:33:35] debug: 	zh:deconz:frameparser: DATA_REQUEST RESPONSE - request id: 118 status: 0
[2024-09-16 21:33:35] debug: 	zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2024-09-16 21:33:35] debug: 	zh:deconz:frameparser: DEVICE_STATE changed: 100110
[2024-09-16 21:33:35] debug: 	zh:deconz:driver: networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2024-09-16 21:33:35] debug: 	zh:deconz:driver: query aps data confirm
[2024-09-16 21:33:35] debug: 	zh:deconz:driver: DATA_CONFIRM - sending data state request - SeqNr. 149
[2024-09-16 21:33:35] debug: 	zh:deconz:frameparser: DATA_CONFIRM RESPONSE - destAddr: 0xa9b9 request id: 118 confirm status: 0
[2024-09-16 21:33:35] debug: 	zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2024-09-16 21:33:35] debug: 	zh:deconz: sendZclFrameToEndpoint - message send with transSeq Nr.: 112
[2024-09-16 21:33:35] debug: 	zh:deconz: false, true, false, 10000
[... 268 lines cut ...]
[2024-09-16 21:34:22] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Philips Sensor (Downstairs Hall)', payload '{"battery":9.5,"illuminance":2464,"illuminance_lux":2,"led_indication":true,"linkquality":203,"motion_sensitivity":"high","occupancy":false,"occupancy_timeout":0,"temperature":22.68,"update":{"installed_version":1107323831,"latest_version":1107323831,"state":"idle"},"update_available":false}'
[2024-09-16 21:34:22] debug: 	zh:controller:endpoint: CommandResponse 0x0017880106f73d5c/2 genOta.queryNextImageResponse({"status":152}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"reservedBits":0,"writeUndiv":false})
[2024-09-16 21:34:22] debug: 	zh:deconz: no response expected (103)
[2024-09-16 21:34:22] debug: 	z2m: Responded to OTA request of 'Philips Sensor (Downstairs Hall)' with 'NO_IMAGE_AVAILABLE'
[2024-09-16 21:34:22] debug: 	zh:deconz:driver: DATA_REQUEST - destAddr: 0xcc8a EP:2 SeqNr. 184 request id: 126
[2024-09-16 21:34:22] debug: 	zh:deconz:frameparser: DATA_REQUEST RESPONSE - request id: 126 status: 0
[2024-09-16 21:34:22] debug: 	zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2024-09-16 21:34:22] debug: 	zh:deconz:frameparser: DEVICE_STATE changed: 100110
[2024-09-16 21:34:22] debug: 	zh:deconz:driver: networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2024-09-16 21:34:22] debug: 	zh:deconz:driver: query aps data confirm
[2024-09-16 21:34:22] debug: 	zh:deconz:driver: DATA_CONFIRM - sending data state request - SeqNr. 185
[2024-09-16 21:34:22] debug: 	zh:deconz:frameparser: DATA_CONFIRM RESPONSE - destAddr: 0xcc8a request id: 126 confirm status: 0
[2024-09-16 21:34:22] debug: 	zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2024-09-16 21:34:22] debug: 	zh:deconz: sendZclFrameToEndpoint - message send with transSeq Nr.: 103
[2024-09-16 21:34:22] debug: 	zh:deconz: false, true, false, 10000
[2024-09-16 21:34:22] debug: 	zh:deconz: resolve request (103)
[2024-09-16 21:34:28] debug: 	zh:deconz:frameparser: DEVICE_STATE changed: 101010
[2024-09-16 21:34:28] debug: 	zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
[... cut ...]
[2024-09-16 21:34:28] debug: 	zh:deconz:frameparser: DATA_INDICATION RESPONSE - seqNr. 186 srcAddr: 0x6aa1 destAddr: 0x0 profile id: 0x0 cluster id: 0x5 lqi: 184
[2024-09-16 21:34:28] debug: 	zh:deconz:frameparser: response payload: 144,0,0
[2024-09-16 21:34:28] debug: 	zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2024-09-16 21:34:30] debug: 	zh:deconz:driver: send read device state from queue. seqNr: 187
[2024-09-16 21:34:30] debug: 	zh:deconz:frameparser: device state: 100010

Ar-eh, 2024-09-17:
I discovered that I could not pair any devices on 1.40.1 or 1.40.0. Rolling back to 1.39.1 solved the issue and allowed me to pair Badring as well as other devices. So, it is a bug introduced in 1.40.X, perhaps only for deconz coordinators?

Ar-eh, 2024-09-17:
> I can confirm same issue with E2002 remote + z2m 1.40-1 and conbee 3 stick

I edited the title of the issue and description to more accurately reflect the problem.

Koenkk, 2024-09-19:
Could you provide the debug log of joining a device both with 1.39.1 and 1.40.1?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

Ar-eh, 2024-09-19:
> Could you provide the debug log of joining a device both with 1.39.1 and 1.40.1?
> 
> See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

The debug log for attempting to join a device in 1.40.1 is in the original post. I’ll post the debug log for joining a device on 1.39.1 later today unless someone else beats me to it.

markxroberts, 2024-09-20:
I don't know whether this is related...will open another issue if not.
I have Conbee II running latest firmware.  With Z2M 1.40.1 but not Z2M 1.40.0, I get repeated MQTT timeout errors, and many devices fail to check in or fail to receive messages.  I've downgraded to 1.40.0 and this behaviour disappears.  Logs can follow if these would be helpful.

SpiGAndromeda, 2024-09-20:
I can confirm the issue with Z2M 1.40.

I migrated my Conbee II to a Conbee III today. The network was instantly unstable (migrated the network with the Phoscon App). I dont have that much devices so resetting the whole network isn't that much of a problem. It didn't work. I couldn't pair any new devices (thats only party true).

So I switched from Z2M to ZHA and I could pair all devices effortless. Unfortunatly my thermostats are not well supported by ZHA so I wanted to switch back.

I wanted to pair the devices with constant power first but it didn't work with a single device. So I just paired a sensor and it worked instantly. Restoring a backup with Z2M 1.39 resolved the problems. Everything is working fine.

The problem seems to affect only routing devices for me. There were no pairing issues with errors. Z2M just did ... nothing. It didn't show anything and there wasn't anything in the logs.

Koenkk, 2024-09-21:
@SpiGAndromeda are you able to pair device to z2m 1.40.1 + Conbee II? (is it specific to the Conbee III)?

markxroberts, 2024-09-21:
I can pair devices to Conbee II with 1.40.1

SpiGAndromeda, 2024-09-22:
@Koenkk I can pair devices with Conbee II.

Koenkk, 2024-09-22:
I see, did pairing with Conbee 3 work with older versions of z2m?

SpiGAndromeda, 2024-09-22:
> I see, did pairing with Conbee 3 work with older versions of z2m?

Yes. I am currently using 1.39.1-1 as Addon in HA. I could pair all my devices.

Koenkk, 2024-09-23:
Great, then I need this: https://github.com/Koenkk/zigbee2mqtt/issues/23996#issuecomment-2361841895

Please also try with the latest [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html), this might fix it.

SpiGAndromeda, 2024-09-23:
> Great, then I need this: [#23996 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/23996#issuecomment-2361841895)
> 
> Please also try with the latest [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html), this might fix it.

For 1.39 with Conbee II, for 1.40 with Conbee II?
For Conbee III I could only provide logs for a pairing attempt. I don't want to reset the network because the Conbee III is in productive use. So Conbee III with 1.39 and/or 1.40?

Koenkk, 2024-09-23:
Both Conbee III with 1.39.1 (working) and 1.40.1 (not working)

notjosh, 2024-09-26:
fwiw, I'm on 1.40.1 with Conbee III and can add _some_ devices (coincidentally had to re-pair a LED driver - worked fine), but not others (Tretakt - doesn't interview).

I had the Tretakt paired yesterday, but wanted to re-pair, so I can confirm it _did_ work.

The relevant log lines referring to the device/failure look like (from the first post, but same for me locally):

```
[2024-09-16 21:33:50] debug: zh:deconz: Try to find network address of 94b216fffeb75d16
[2024-09-16 21:33:50] debug: zh:deconz:frameparser: DATA_INDICATION RESPONSE - AssertionError [ERR_ASSERTION]: Failed to find srcAddr of message
```

Koenkk, 2024-09-27:
@schrluka looks like it was introduced in https://github.com/Koenkk/zigbee-herdsman/commit/14b7b537f15f8dfbeb6b615fec55cbd2805ddff3, could you take a look at this?

fjumeaux, 2024-09-28:
Hello. I don't know if my problem is the same but it doesn't work with the-dev commit [15d845b] and my conbee3. 

Attached, a attemps of join a sonoff zbmini2 from the beginning of permit of join until the end. 

[log.log](https://github.com/user-attachments/files/17173355/log.log)

To test, I also tried to join a plug OSRAM AB3257001NJ : work perfectly.

But nothing with : 
Sonoff zbmini 2 (extreme)
Sonoff SNZP-06P
Ikea VALLHORN

fjumeaux, 2024-09-28:
forgot to say that it worked perfectly before... But I don't know before which commit

schrluka, 2024-09-28:
Hi Everyone

@Koenkk asked me to look at this. Looks like some restructuring might have broken something in heardsmean. My suggestions is in this PR: https://github.com/Koenkk/zigbee-herdsman/pull/1207

It's untested so far as I don't have access to the appropriate hardware ATM. I will try to check this next week, but I can't promise.

BR,
Lukas

Ar-eh, 2024-10-02:
@Koenkk @schrluka I was able to successfully join an IKEA LED1545G12 after updating to 1.40.2. :)

Koenkk, 2024-10-04:
Great, I believe this is fixed now.

closed 2024-12-19 by github-actions[bot]