Koenkk/zigbee2mqtt #23998: TS0601 _TZE204_lzriup1j wrong order in 'Preset'

arkadi222, 2024-09-17:
### What happened?

the 'Preset' setting is displayed incorrectly. when I select auto mode on the thermostat, manual mode is displayed in the GUI. and vice versa.

### What did you expect to happen?

change order in https://github.com/Koenkk/zigbee-herdsman-converters/blob/34de76fe3d1ffb2aa6dbcd712ceb5178f38712f9/src/devices/tuya.ts#L4352 ?

### Zigbee2MQTT version

1.40.1-dev commit: d3de77d

### Adapter firmware version

7.4.2

### Adapter

ember/skyconnect

### Setup

unraid docker

Koenkk, 2024-09-18:
Could you provide a screenshot of the device about tab?

arkadi222, 2024-09-19:
[image]

Koenkk, 2024-09-20:
@vobelic, since you also have this device, could you confirm this issue?

arkadi222, 2024-09-30:
@tispokes - maybe you can confirm also ?

tispokes, 2024-09-30:
@arkadi222 @Koenkk may can check it this week. I'm moving so I'm not on my old flat

vmrk00, 2024-09-30:
@arkadi222 idk, I don't use that, I created new thermostat to my boiler with ESPhome via OpenTherm protocol. I high recommended If your boiler can communicate in OpenTherm: https://github.com/arthurrump/esphome-opentherm

vobelic, 2024-09-30:
@Koenkk I'm running 1.39.0, and can't confirm an issue. No time to upgrade now.
Changing the preset on the thermostat shows the correct mode in Z2M.

arkadi222, 2024-09-30:
@vobelic are you sure that your device presents itself as: _TZE204_lzriup1j (not _TZE200_viy9ihs7) ?
because I see that both models share the same configuration - maybe it is a reason ?
I can confirm my issue for each of the 5 thermostats I have :(

vobelic, 2024-10-02:
@arkadi222 yes it's the Zigbee Manufacturer: _TZE204_lzriup1j
[image]

Once I do an update I'll let you know.

Btw what do the logs in Z2M look like when you change from auto to manual and v/s?
e.g.
`2024-10-02 20:24:58z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Living Room Thermostat', payload '{"backlight":3,"backlight_mode":"high","child_lock":"UNLOCK","current_heating_setpoint":18,"deadzone_temperature":1,"factory_reset":"OFF","faultalarm":0,"frost_protection":"ON","holiday_mode":"no_rest","last_seen":"2024-10-02T20:24:58+02:00","linkquality":142,"local_temperature":23.2,"local_temperature_calibration":-1,"max_temperature":6,"max_temperature_limit":60,"preset":"manual","running_state":"idle","schedule_holiday":"08:00/22.0°C 23:00/16.0°C","schedule_weekday":"06:00/20.0°C 08:00/16.0°C 11:30/16.0°C 12:30/16.0°C 17:00/22.0°C 22:00/16.0°C","sensor":"internal","sensor_mode":"in","system_mode":"heat","temperature_difference_start":1,"working_day":"6-1"}'`

arkadi222, 2024-10-04:
when I change from auto to manual:
`
2024-10-02 20:24:58z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Living Room Thermostat', payload '{"backlight":3,"backlight_mode":"high","child_lock":"UNLOCK","current_heating_setpoint":18,"deadzone_temperature":1,"factory_reset":"OFF","faultalarm":0,"frost_protection":"ON","holiday_mode":"no_rest","last_seen":"2024-10-02T20:24:58+02:00","linkquality":142,"local_temperature":23.2,"local_temperature_calibration":-1,"max_temperature":6,"max_temperature_limit":60,"preset":"manual","running_state":"idle","schedule_holiday":"08:00/22.0°C 23:00/16.0°C","schedule_weekday":"06:00/20.0°C 08:00/16.0°C 11:30/16.0°C 12:30/16.0°C 17:00/22.0°C 22:00/16.0°C","sensor":"internal","sensor_mode":"in","system_mode":"heat","temperature_difference_start":1,"working_day":"6-1"}'`
from manual to auto:
`
2024-10-04 07:53:15z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Sypialnia termostat', payload '{"backlight_mode":"off","child_lock":"UNLOCK","current_heating_setpoint":22.5,"deadzone_temperature":0.5,"factory_reset":"OFF","faultalarm":0,"frost_protection":"OFF","inverse":"OFF","key_sound":"ON","last_seen":"2024-10-04T07:52:57+02:00","linkquality":152,"local_temperature":22.4,"local_temperature_calibration":-2.5,"max_temperature_limit":26,"preset":"auto","running_state":"idle","schedule_holiday":"08:00/23.0°C 23:00/20.0°C","schedule_weekday":"06:00/24.0°C 06:55/21.0°C 13:05/23.5°C 15:55/22.0°C 21:05/23.0°C 23:30/21.0°C","sensor":"both","sensor_choice":"inside","sensor_limit":60,"system_mode":"heat","working_day":"5-2"}'
2024-10-04 07:53:17z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Sypialnia termostat', payload '{"backlight_mode":"off","child_lock":"UNLOCK","current_heating_setpoint":22.5,"deadzone_temperature":0.5,"factory_reset":"OFF","faultalarm":0,"frost_protection":"OFF","inverse":"OFF","key_sound":"ON","last_seen":"2024-10-04T07:52:58+02:00","linkquality":160,"local_temperature":22.4,"local_temperature_calibration":-2.5,"max_temperature_limit":26,"preset":"manual","running_state":"idle","schedule_holiday":"08:00/23.0°C 23:00/20.0°C","schedule_weekday":"06:00/24.0°C 06:55/21.0°C 13:05/23.5°C 15:55/22.0°C 21:05/23.0°C 23:30/21.0°C","sensor":"both","sensor_choice":"inside","sensor_limit":60,"system_mode":"heat","working_day":"5-2"}'`
it doesn't matter whether I do it with home assistant or from the thermostat.

arkadi222, 2024-10-08:
I've just used https://gist.github.com/Koenkk/4b4c582f8a01fd71108acce6a5e2db15 :)
and changed line: 85 in my ext_converter file:
`if (device.manufacturerName === '_TZE204_lzriup1j') {`
now it works perfect for me :D

closed 2024-12-19 by github-actions[bot]