Koenkk/zigbee2mqtt #24001: Bit weird implementation of: TS0601/_TZE204_aoclfnxz

Dukobpa3, 2024-09-17:
### What happened?

Device is recognized as https://www.zigbee2mqtt.io/devices/BHT-002_BHT-006.html#moes-bht-002%252Fbht-006 but actually looks like it is not.

It works in general, but:
- in UI supported settings degrees step as 0.1, but looks like really works only integers. (but device itself reports local_temperature with step 0.5)
- doesn't supports `deadzone_temperature` and `max_temperature_limit`
- it spamming state each 8 seconds (with debounce:1) and not supports any settings about it

### What did you expect to happen?

I saw in converters it is implemented as alias to some older termostat, but looks like it is needed proper new one implementation...

Yes, I've checked other similar issues about this device. But nothing changed in general. Just conversations in comments and that's all... Looks like needs to be fixed anyway :)

### How to reproduce it (minimal and precise)

Just play with it? ))

[image]

[This one](https://www.aliexpress.com/item/1005005202931454.html?spm=a2g0o.order_list.order_list_main.69.577f1802FsJ53P)

### Zigbee2MQTT version

1.40.1

### Adapter firmware version

7.4.1 [GA]

### Adapter

SLZB-06m

### Setup

Separate Docker container

Dukobpa3, 2024-09-18:
Ok I debugged device a little (had to buy tuya hub for it lol)

I hope it will help:

[image]

**I want to highlight couple points:**
– As I understand they've introduced two different codes for upper/lover temps. That's why deadzone and upper temps not working
– _DeadZones_ : `lover_temp` have limit **1-5** but `lover_temp_f` **1-5** (in the app 1-5)
– _Max Temperature Limit_ : both `upper_temp` – limits: 45-70 (in app also)
– ALL values is integers! Not floats.

```
dead zone temp : 20 // lover_temp_f
dead zone temp : 26 // lover_temp

set temperature upper limit F : 18 // upper_temp_f
high limit temperature ceiling : 19 // upper_temp
```

```
**Standard Status Set**

switch	Boolean	
"{true,false}"
work_state	Enum	
{
  "range": [
    "hot"
  ]
}
temp_set	Integer	
{
  "unit": "℃",
  "min": 0,
  "max": 45,
  "scale": 0,
  "step": 1
}
upper_temp_f	Integer	
{
  "unit": "℃",
  "min": 45,
  "max": 70,
  "scale": 0,
  "step": 1
}
upper_temp	Integer	
{
  "unit": "℃",
  "min": 45,
  "max": 70,
  "scale": 0,
  "step": 1
}
lower_temp_f	Integer	
{
  "unit": "℃",
  "min": 0,
  "max": 5,
  "scale": 0,
  "step": 1
}
temp_current	Integer	
{
  "unit": "℃",
  "min": 0,
  "max": 700,
  "scale": 1,
  "step": 1
}
lower_temp	Integer	
{
  "unit": "℃",
  "min": 1,
  "max": 5,
  "scale": 0,
  "step": 1
}
temp_correction	Integer	
{
  "unit": "",
  "min": -9,
  "max": 9,
  "scale": 0,
  "step": 1
}
valve_state	Enum	
{
  "range": [
    "open",
    "close"
  ]
}
child_lock	Boolean	
"{true,false}"
sensor_choose	Enum	
{
  "range": [
    "in"
  ]
}

```

```

**Standard Instructions set**

switch	Boolean	
"{true,false}"
temp_set	Integer	
{
  "unit": "℃",
  "min": 0,
  "max": 45,
  "scale": 0,
  "step": 1
}
upper_temp_f	Integer	
{
  "unit": "℃",
  "min": 45,
  "max": 70,
  "scale": 0,
  "step": 1
}
upper_temp	Integer	
{
  "unit": "℃",
  "min": 45,
  "max": 70,
  "scale": 0,
  "step": 1
}
lower_temp_f	Integer	
{
  "unit": "℃",
  "min": 0,
  "max": 5,
  "scale": 0,
  "step": 1
}
lower_temp	Integer	
{
  "unit": "℃",
  "min": 1,
  "max": 5,
  "scale": 0,
  "step": 1
}
temp_correction	Integer	
{
  "unit": "",
  "min": -9,
  "max": 9,
  "scale": 0,
  "step": 1
}
child_lock	Boolean	
"{true,false}"
sensor_choose	Enum	
{
  "range": [
    "in"
  ]
}
```

Dukobpa3, 2024-09-18:
**High limit temperature ceiling** – from app changes `upper_temp`, with code 19
**DeadZone temp** – from app changes `lower_temp`, with code 26

Qweua, 2024-09-18:
Yes, recognition incorrect.

Max temperature limit and Deadzone temperature look incorrect too :(

artemenko2a, 2024-09-18:
Have the same problems with this device. Hope, this will be fixed. Count me in

Dukobpa3, 2024-09-18:
https://github.com/Koenkk/zigbee-herdsman-converters/pull/8005

stenvix, 2024-09-21:
Hi, I'll add my 5 cents too. If I set debounce = 5 I can get the default settings set for program, but after 5 seconds they disappear
[image]
After 5 seconds 
[image]

Dukobpa3, 2024-10-13:
ok I've made custom converter

https://github.com/Dukobpa3/z2m

stenvix, 2024-12-12:
Hello, @dukobpa3 please share the link to the hub, I also want to debug

closed 2025-02-19 by github-actions[bot]