Koenkk/zigbee2mqtt #24003: YAMLException: duplicated mapping key (47:1)

pascalbcnn, 2024-09-17:
### What happened?

My Home Assistant is not able to open Zigbee2mqtt. I´ve setted it up yesterday and it worked fine. Today i wanted to add an device and now its not starting anymore. Does anyone has an idea why?

Thats the information out of the protocol.

```
`YAMLException: duplicated mapping key (47:1)

 44 |   homeassistant:
 45 |     last_seen:
 46 |       enabled_by_default: true
 47 | device_options:
------^
 48 |   legacy: false
 49 | devices:
    at generateError (/app/node_modules/js-yaml/lib/loader.js:183:10)
    at throwError (/app/node_modules/js-yaml/lib/loader.js:187:9)
    at storeMappingPair (/app/node_modules/js-yaml/lib/loader.js:358:7)
    at readBlockMapping (/app/node_modules/js-yaml/lib/loader.js:1173:9)
    at composeNode (/app/node_modules/js-yaml/lib/loader.js:1441:12)
    at readDocument (/app/node_modules/js-yaml/lib/loader.js:1625:3)
    at loadDocuments (/app/node_modules/js-yaml/lib/loader.js:1688:5)
    at Object.load (/app/node_modules/js-yaml/lib/loader.js:1714:19)
    at Object.read (/app/lib/util/yaml.ts:7:29)
    at read (/app/lib/util/settings.ts:367:20)
[14:38:00] INFO: Preparing to start...
[14:38:01] INFO: Socat not enabled
[14:38:01] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.

/app/node_modules/js-yaml/lib/loader.js:183
  return new YAMLException(message, mark);
         ^
YAMLException: duplicated mapping key (47:1)

 44 |   homeassistant:
 45 |     last_seen:
 46 |       enabled_by_default: true
 47 | device_options:
------^
 48 |   legacy: false
 49 | devices:
    at generateError (/app/node_modules/js-yaml/lib/loader.js:183:10)
    at throwError (/app/node_modules/js-yaml/lib/loader.js:187:9)
    at storeMappingPair (/app/node_modules/js-yaml/lib/loader.js:358:7)
    at readBlockMapping (/app/node_modules/js-yaml/lib/loader.js:1173:9)
    at composeNode (/app/node_modules/js-yaml/lib/loader.js:1441:12)
    at readDocument (/app/node_modules/js-yaml/lib/loader.js:1625:3)
    at loadDocuments (/app/node_modules/js-yaml/lib/loader.js:1688:5)
    at Object.load (/app/node_modules/js-yaml/lib/loader.js:1714:19)
    at Object.read (/app/lib/util/yaml.ts:7:29)
    at read (/app/lib/util/settings.ts:367:20)
[14:38:02] INFO: Preparing to start...
[14:38:02] INFO: Socat not enabled
[14:38:03] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.

/app/node_modules/js-yaml/lib/loader.js:183
  return new YAMLException(message, mark);
         ^
YAMLException: duplicated mapping key (47:1)

 44 |   homeassistant:
 45 |     last_seen:
 46 |       enabled_by_default: true
 47 | device_options:
------^
 48 |   legacy: false
 49 | devices:
    at generateError (/app/node_modules/js-yaml/lib/loader.js:183:10)
    at throwError (/app/node_modules/js-yaml/lib/loader.js:187:9)
    at storeMappingPair (/app/node_modules/js-yaml/lib/loader.js:358:7)
    at readBlockMapping (/app/node_modules/js-yaml/lib/loader.js:1173:9)
    at composeNode (/app/node_modules/js-yaml/lib/loader.js:1441:12)
    at readDocument (/app/node_modules/js-yaml/lib/loader.js:1625:3)
    at loadDocuments (/app/node_modules/js-yaml/lib/loader.js:1688:5)
    at Object.load (/app/node_modules/js-yaml/lib/loader.js:1714:19)
    at Object.read (/app/lib/util/yaml.ts:7:29)
    at read (/app/lib/util/settings.ts:367:20)
[14:38:03] INFO: Preparing to start...
[14:38:04] INFO: Socat not enabled
[14:38:04] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.

/app/node_modules/js-yaml/lib/loader.js:183
  return new YAMLException(message, mark);
         ^
YAMLException: duplicated mapping key (47:1)

 44 |   homeassistant:
 45 |     last_seen:
 46 |       enabled_by_default: true
 47 | device_options:
------^
 48 |   legacy: false
 49 | devices:
    at generateError (/app/node_modules/js-yaml/lib/loader.js:183:10)
    at throwError (/app/node_modules/js-yaml/lib/loader.js:187:9)
    at storeMappingPair (/app/node_modules/js-yaml/lib/loader.js:358:7)
    at readBlockMapping (/app/node_modules/js-yaml/lib/loader.js:1173:9)
    at composeNode (/app/node_modules/js-yaml/lib/loader.js:1441:12)
    at readDocument (/app/node_modules/js-yaml/lib/loader.js:1625:3)
    at loadDocuments (/app/node_modules/js-yaml/lib/loader.js:1688:5)
    at Object.load (/app/node_modules/js-yaml/lib/loader.js:1714:19)
    at Object.read (/app/lib/util/yaml.ts:7:29)
    at read (/app/lib/util/settings.ts:367:20)`
```

### Zigbee2MQTT version

1.40.1-1

### Adapter firmware version

2024.9.2

### Adapter

ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus

### Setup

Proxmox

chris-1243, 2024-09-17:
Could you provide your `configuration.yaml` ?

pascalbcnn, 2024-09-17:
[image]

Sure. please see attached picture.

chris-1243, 2024-09-17:
Zigbee2mqtt `configuration.yaml`. You did provide homeassistant

pascalbcnn, 2024-09-17:
oh sure, sorry.

```
homeassistant: true
advanced:
  network_key:
    - 84
    - 68
    - 172
    - 175
    - 175
    - 188
    - 152
    - 8
    - 200
    - 140
    - 63
    - 138
    - 211
    - 255
    - 177
    - 238
  pan_id: 2162
  ext_pan_id:
    - 11
    - 18
    - 113
    - 20
    - 121
    - 223
    - 3
    - 151
  homeassistant_legacy_entity_attributes: false
  legacy_api: false
  legacy_availability_payload: false
mqtt:
  server: mqtt://core-mosquitto:1883
  user: addons
  password: eeji6aidoonae2ahr8aingu5aeku6Aeme0Jah4Eimeeshig7cie0shouga4Sei9a
serial:
  port: >-
    /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_423e5f9c16a6ed11b1fceba32981d5c7-if00-port0
frontend:
  port: 8099
device_options:
  legacy: false
  homeassistant:
    last_seen:
      enabled_by_default: true
device_options:
  legacy: false
devices:
  '0x0ceff6fffe3e796e':
    friendly_name: Thermometer Wohnzimmer
  '0xa4c138c05c430a36':
    friendly_name: Steckdose Sofa
  '0xa4c1380998c1e0df':
    friendly_name: Waschmaschiene
```

pascalbcnn, 2024-09-17:
[image]

chris-1243, 2024-09-17:
You have two times `device_options`. Remove the second and it should be fine.

Check line 42 and line 47. I would remove lines 47, 48

pascalbcnn, 2024-09-17:
okay, thanks. I´ll deleted it and now i restart home assistant.

How can that be happend? I did not change anything in there or added something.

closed 2024-09-17 by pascalbcnn