Koenkk/zigbee2mqtt #24005: Vibration Sensor Model TS0210 vendor: '_TZ3000_lzdjjfss'

Beer17HWAM, 2024-09-17:
### Link

https://nl.aliexpress.com/item/1005006286554833.html?spm=a2g0o.order_list.order_list_main.43.618a79d2n37PHI&gatewayAdapt=glo2nld

### Database entry

{"id":18,"type":"EndDevice","ieeeAddr":"0xa4c1381d43b23081","nwkAddr":54384,"manufId":4417,"manufName":"_TZ3000_lzdjjfss","powerSource":"Battery","modelId":"TS0210","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[1,1280,0],"outClusterList":[25,10],"clusters":{"ssIasZone":{"attributes":{"currentZoneSensitivityLevel":0,"iasCieAddr":"0xe0798dfffebb52b8","zoneState":1}},"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1726565988983}

### Zigbee2MQTT version

1.40.1

### Comments

It's working but the battery status is not correct:

{
    "battery_low": false,
    "last_seen": "2024-09-17T15:41:58+02:00",
    "linkquality": 236,
    "sensitivity": 0,
    "tamper": false,
    "vibration": false,
    "battery": null,
    "voltage": null
}

### External definition

```shell
const {battery, iasZoneAlarm} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS0210'],
    model: 'TS0210',
    vendor: '_TZ3000_lzdjjfss',
    description: 'Automatically generated definition',
    extend: [battery(), iasZoneAlarm({"zoneType":"generic","zoneAttributes":["alarm_1","alarm_2","tamper","battery_low"]})],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

It's working but the battery status is not correct:

{
    "battery_low": false,
    "last_seen": "2024-09-17T15:41:58+02:00",
    "linkquality": 236,
    "sensitivity": 0,
    "tamper": false,
    "vibration": false,
    "battery": null,
    "voltage": null
}

closed 2024-12-19 by github-actions[bot]

Beer17HWAM, 2024-12-19:
Can you please fix this?

GuyFromMars365, 2025-07-27:
Why is it not planned to correct battery status when the z2m website lists states this device lists batt & volage?
Reference: https://www.zigbee2mqtt.io/devices/TZ3000_lzdjjfss.html

inigortega, 2025-11-15:
Any upgrades on getting battery status working on zha? Thanks in advance.