Koenkk/zigbee2mqtt #24007: Button SonOff SNZB-01P power icon is still a question mark

almirus, 2024-09-17:
### What happened?

[image]
[image]
[image]

### Zigbee2MQTT version

1.40.1

### Adapter firmware version

20240710

### Adapter

egony cp2652p

### Setup

x86-64 HA os

Koenkk, 2024-09-19:
Reconfiguring the device via the frontend (yellow refresh icon) should fix this.

almirus, 2024-10-07:
> Reconfiguring the device via the frontend (yellow refresh icon) should fix this.

`error 2024-10-07 21:14:06z2m: Failed to configure 'BUTTON', attempt 1 (Error: Bind 0x1c34f1fffed6742a/1 genOnOff from '0x00124b0022810ef5/1' failed (AREQ - ZDO - bindRsp after 10000ms) at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:59:23) at /app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:403:47 at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20) at ZStackAdapter.sendZdo (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:311:16) at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:563:30) at configure (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:178:17) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:205:21) at Configure.configure (/app/lib/extension/configure.ts:130:13) at Configure.onMQTTMessage (/app/lib/extension/configure.ts:57:21) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:206:17))`

mithomas, 2024-10-07:
Thanks for quick response and for hinting at the previous comment - I actually tried that before (apologies for not mentioning it).

Regardless, I tried again:
```
info 2024-10-07 21:29:02z2m: Configuring 'WZ Sofaecke Klima'
info 2024-10-07 21:29:02z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/WZ Sofaecke Klima', payload '{"battery":60,"humidity":63.8,"linkquality":12,"temperature":19.5}'
info 2024-10-07 21:29:04z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/WZ Sofaecke Klima', payload '{"battery":60,"humidity":63.8,"linkquality":42,"temperature":19.5}'
info 2024-10-07 21:29:06z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/WZ Sofaecke Klima', payload '{"battery":60,"humidity":63.8,"linkquality":18,"temperature":19.5}'
info 2024-10-07 21:29:08z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/WZ Sofaecke Klima', payload '{"battery":60,"humidity":64.8,"linkquality":15,"temperature":19.5}'
info 2024-10-07 21:29:09z2m: Successfully configured 'WZ Sofaecke Klima'
```

But this does not help:
[image]

Koenkk, 2024-10-09:
@mithomas its still not OK after restarting z2m?

xdmore70, 2024-10-10:
it's very funny

I have same device and for me it's ok
[image]

Insted i have the same issue for SNZB-03P
Just try to reconfigure , join , etc , but nothing
[image]

I use
Zigbee2MQTT version
[1.40.2](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.40.2) commit: [e06848d](https://github.com/Koenkk/zigbee2mqtt/commit/e06848d)
Coordinator type
EmberZNet
Coordinator revision
7.4.4 [GA]
Coordinator IEEE Address
0xb43522fffec19b2e
Frontend version
0.7.4
zigbee-herdsman-converters version
20.21.0
zigbee-herdsman version
2.1.3

piokopl, 2024-10-10:
Can you check why my button is not supported?

[image]

2024-10-10 20:34:39Received Zigbee message from 'button_sonoff1', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0

mithomas, 2024-10-28:
> @mithomas its still not OK after restarting z2m?

Hey @Koenkk, unfortunately restarting Z2M also does not fix this.

DaN66O, 2024-11-01:
Zigbee2MQTT version
[1.41.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.41.0) commit: [cc31cea](https://github.com/Koenkk/zigbee2mqtt/commit/cc31cea)
Coordinator type
zStack3x0
Coordinator revision
20240710
Coordinator IEEE Address
0x00245b222a4ea464
Frontend version
0.7.4
zigbee-herdsman-converters version
20.38.0
zigbee-herdsman version
2.1.7

[image]

alexsotoaguilera, 2024-11-08:
same problem with SNZB-03P.

another problem with this sensor is recurrent error when try to change detection timeout:

[2024-11-07 19:20:40] error: 	z2m: Failed to configure 'Motion Sensor - Main Corridor', attempt 1 (Error: Bind 0x0ceff6fffe5b07f1/1 msOccupancySensing from '0x6c5cb1fffefac19f/1' failed (Delivery failed for '22216'.)
    at EmberOneWaitress.deliveryFailedFor (/usr/local/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:100:31)
    at EmberAdapter.onMessageSent (/usr/local/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:469:42)
    at Ezsp.emit (node:events:519:28)
    at Ezsp.ezspMessageSentHandler (/usr/local/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:5032:14)
    at Ezsp.callbackDispatch (/usr/local/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:735:26)
    at Ezsp.onAshFrame (/usr/local/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:395:22)
    at UartAsh.emit (node:events:519:28)
    at Immediate.<anonymous> (/usr/local/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ember/uart/ash.ts:1169:45)
    at processImmediate (node:internal/timers:483:21))

fibu-freak, 2024-11-10:
<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/peter/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/peter/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

Zigbee2MQTT-Version | 1.41.0 commit: cc31ceab
-- | --
Coordinator-Typ | EmberZNet
Coordinator-Version | 7.4.4 [GA]
Coordinator IEEE Adresse | iknowitbutidontsay
Frontend-Version | 0.7.4
Zigbee Herdsman Konverter Version | 20.38.0
Zigbee Herdsman Version | 2.1.7

</body>

</html>
I have the same problem with several devices (SNZB-02D, SNZB-01P), when pairing this weekend. Others (SNZB-03, SNZB-04) had been ok.
Although I have the same type of devices in my setup (earlier paired) which are working quite fine.

[image]

[image]

I tried the above steps to clear the problem, but doesn't help.

All items seem to work and JSON-Strings (state) showing the correct values
`{
    "battery": 100,
    "humidity": 55.6,
    "last_seen": "2024-11-10T17:35:52+01:00",
    "linkquality": 184,
    "temperature": 17.8
}`
Does anyone has a solution for this ?

fibu-freak, 2024-11-19:
@Koenkk   There's really no one to explain me this phenomenon ?

[image]

Koenkk, 2024-11-20:
With z2m 1.41.0, try to re-configure the device by pressing the yellow refresh button (makes sure to wakeup the device right before doing so) on the frontend and restart z2m.

fibu-freak, 2024-11-21:
@Koenkk Thx for reply
What I did:
1. Pressed knob of a Switch SNZB-01P to be shure that it is waked up.
2. Pressed yellow refresh Button in the frontend (I got an error message)
3. I restarted Z2M (sudo systemctl restart zigbee2mqtt)

After that the question mark still was there

The Error-Message was:
`error 2024-11-21 11:10:02z2m: Failed to configure 'snzb_01p_02', attempt 1 (Error: Bind 0x048727fffeac2794/1 genOnOff from '0x90395efffe3ff12c/1' failed (Delivery failed for '36885'.) at EmberOneWaitress.deliveryFailedFor (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:100:31) at EmberAdapter.onMessageSent (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:469:42) at Ezsp.emit (node:events:517:28) at Ezsp.ezspMessageSentHandler (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:5032:14) at Ezsp.callbackDispatch (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:735:26) at Ezsp.onAshFrame (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:395:22) at UartAsh.emit (node:events:517:28) at Immediate.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ember/uart/ash.ts:1169:45) at processImmediate (node:internal/timers:476:21))`

Just a stupid question from a beginner: How do I know if an item is sleeping or awake? 
In the frontend I can see if an item is online or offline in the Devices-Tab, and I also can see in the "Last seen"-Column when it was refreshed. 
Does it mean the it's awake when it's online ?

fibu-freak, 2024-11-21:
Just pressed yellow button with a "normal" device and got an error-message too. ??!!
`error 2024-11-21 11:28:45z2m: Failed to configure 'snzb_03_04', attempt 1 (Error: Bind 0x00124b002faa8b7b/1 genPowerCfg from '0x90395efffe3ff12c/1' failed (Delivery failed for '51191'.) at EmberOneWaitress.deliveryFailedFor (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:100:31) at EmberAdapter.onMessageSent (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:469:42) at Ezsp.emit (node:events:517:28) at Ezsp.ezspMessageSentHandler (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:5032:14) at Ezsp.callbackDispatch (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:735:26) at Ezsp.onAshFrame (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:395:22) at UartAsh.emit (node:events:517:28) at Immediate.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ember/uart/ash.ts:1169:45) at processImmediate (node:internal/timers:476:21))`

fibu-freak, 2024-11-21:
I have a look in my setup and found out that there's no directory "/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src"

Directory-Tree looks:

> hab4@raspi58:~ $ ls -alF /opt/zigbee2mqtt/node_modules/zigbee-herdsman
insgesamt 160
drwxr-xr-x   5 hab4 openhab  4096 11. Nov 12:10 ./
drwxrwxr-x 410 hab4 openhab 16384 11. Nov 12:10 ../
-rw-r--r--   1 hab4 openhab    72 11. Nov 12:10 .babelrc.js
-rw-r--r--   1 hab4 openhab 81935 11. Nov 12:10 CHANGELOG.md
drwxr-xr-x   8 hab4 openhab  4096 11. Nov 12:10 dist/
-rw-r--r--   1 hab4 openhab  1069 11. Nov 12:10 eslint.config.mjs
drwxr-xr-x   2 hab4 openhab  4096 11. Nov 12:10 examples/
drwxr-xr-x   4 hab4 openhab  4096 11. Nov 12:10 .github/
-rw-r--r--   1 hab4 openhab  1204 11. Nov 12:10 LICENSE
-rw-r--r--   1 hab4 openhab  3121 11. Nov 12:10 package.json
-rw-r--r--   1 hab4 openhab    27 11. Nov 12:10 .prettierignore
-rw-r--r--   1 hab4 openhab   501 11. Nov 12:10 .prettierrc
-rw-r--r--   1 hab4 openhab  2378 11. Nov 12:10 README.md
-rw-r--r--   1 hab4 openhab   238 11. Nov 12:10 release-please-config.json
-rw-r--r--   1 hab4 openhab    21 11. Nov 12:10 .release-please-manifest.json
-rw-r--r--   1 hab4 openhab   607 11. Nov 12:10 tsconfig.json
-rw-r--r--   1 hab4 openhab  1187 11. Nov 12:10 typedoc-tsconfig.json
hab4@raspi58:~ $ ls -alF /opt/zigbee2mqtt/node_modules/zigbee-herdsman/src
ls: Zugriff auf '/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src' nicht möglich: Datei oder Verzeichnis nicht gefunden
hab4@raspi58:~ $

Maybe this can help ?

stickpin, 2024-11-21:
> With z2m 1.41.0, try to re-configure the device by pressing the yellow refresh button (makes sure to wakeup the device right before doing so) on the frontend and restart z2m.

Hi @Koenkk,

I have 7 SNZB-02D and all of them with Unknown power, all of them got paired on Z2M v1.41.0.
[image]

Let me know if you have any idea what you want me to test.
Not sure if it has anything to do with the coordinator. I am on SLZB-07mg24 with darkxst 8.0.2 firmware.
I don't see any errors during pairing/re-configuration.

xdmore70, 2024-11-21:
Hi Koenkk, i followed your tips for SNZB-01P and it work.
Now i have battery indicator.
Thank you very much
David

alexsotoaguilera, 2024-11-21:
doesn't works for sonoff SNZB-03P

xdmore70, 2024-11-21:
Same result for SNZB-03P sonoff infrared sensor. Followed your imstructions
and now all ok... i see battery indicator in frontend. Thanks

fibu-freak, 2024-11-21:
As said the problem is still pending for item-type **SNZB-03P** (Button) and **SNZB-02D** (Temp/Hum-Measurement).

I paired a new Motion-Sensor (SNZB-03) and a new Window Sensor (SNZB-04) at the moment without any problems. No question mark but a battery icon.
[image]

Koenkk, 2024-11-21:
If reconfigure doesn't help, try to force remove the device through the z2m frontend, **restart z2m** and re-pair the device.

stickpin, 2024-11-21:
@Koenkk I've tried it and unfortunately it's not solving the issue for SNZB-02D.

fibu-freak, 2024-11-22:
If reconfigure doesn't help, try to force remove the device through the z2m frontend, **restart z2m** and re-pair the device.

_Originally posted by @Koenkk in https://github.com/Koenkk/zigbee2mqtt/issues/24007#issuecomment-2492309965_
@Koenkk 
I did as you wrote, but doesn't help.
 I repeated the procedure by including a step to de-energize the device (SNZB-01P)  by pulling out the battery and wait a minute before reinstall.
But the behavior is still the same. There's still a question mark shown in the Web-UI, although the sent JSON-String shows actual values 

> {
    "battery": 100,
    "last_seen": "2024-11-22T10:57:32+01:00",
    "linkquality": 248,
    "update": {
        "installed_version": 8704,
        "latest_version": 8704,
        "state": "idle"
    },
    "voltage": 3100,
    "action": "single"
}

[image]

All values a published in a correct manner.

As said in a post above, the correct icon is shown for such devices in earlier pairings and only those the two device-types are involved in my setup.
Cheers - Peter

stickpin, 2024-11-22:
I've played dirty with Z2M. I just modified the db directly.
changed from `"powerSource": "Unknown"` to `"powerSource": "Battery"`
[image]

It's a hacky operation, so be careful, if you decide to do it.
Do backup first, then turn off Z2M, modify the db, and turn it back on again.

fibu-freak, 2024-11-22:
@stickpin 
Even it's a "quick and dirty" hack, as it worked for me too.
But at the very end this is not the solution of the problem. Maybe @Koenkk has another idea !? Maybe in the frontend ?

fibu-freak, 2024-11-23:
Today I installed another router (No. 3) (flashing it before with _EZNet6.10.3_V1.0.0), which works quite fine and with no problems at all. I like this network (adapter) and the possibility to connect to other smart homes (for me it's openHAB)with no (or better said with nearly no) problems. !!!!

closed 2025-01-30 by github-actions[bot]