Koenkk/zigbee2mqtt #24010: Z2M cannot connect to adapter

dam57950, 2024-09-17:
### What happened?

I installed HA, Mosquitto and Z2M with docker compose, at first it worked great but after leaving it alone for a few hour, Z2M frontend didn't worked anymore 

I tried restarting from scratch deploying a new stack with new configuration files but that doesn't help.
I tried using eszp instead of ember but no change

Looks like the adapter cannot be found by the Z2M instance but... it was just before and is still online at the same IP on the same network
I also tried to restart the host

Only thing I saw was the hour which is -2h in the logs, is it important?

### What did you expect to happen?

For Z2M to connect to my SLZB-06M

### How to reproduce it (minimal and precise)

Instal Z2M with adapter tcp address in the yaml file

### Zigbee2MQTT version

koenkk/zigbee2mqtt:latest

### Adapter firmware version

20240510

### Adapter

SLZB-06M

### Setup

Docker 

### Debug log

[log-4.log](https://github.com/user-attachments/files/17035117/log-4.log)

dam57950, 2024-09-18:
I found the problem, when tailscale is connected, z2m cannot find the coordinator, but why? I'm still searching, normally tailscale don't "disconnect" local network

troylynch, 2024-09-19:
Following having the same problem after the update yesterday

closed 2024-12-19 by github-actions[bot]