Koenkk/zigbee2mqtt #24012: Ikea button: the light turns off automatically

Fafitude, 2024-09-18:
### What happened?

I use the button toggle. When I press this the lamp should turn on or off depending on the starting state.
Since the last update 1.40.1, when I press it the lamp turns on and off immediately (or vice versa), as if pressed twice in a row.
I came back 1.40 and I no longer have the problem.

### What did you expect to happen?

I use the button toggle. When I press this the lamp should turn on or off depending on the starting state.
Since the last update 1.40.1, when I press it the lamp turns on and off immediately (or vice versa), as if pressed twice in a row.
I came back 1.40 and I no longer have the problem.

### Zigbee2MQTT version

1.40.1

### Adapter firmware version

0x26500900

### Adapter

Conbee2

### Setup

HA sur NUC

vwegert, 2024-10-06:
I may have run into the same problem, see discussion #24242 - do you see the same MQTT message situation?

vwegert, 2024-10-07:
I can confirm that downgrading from 1.40.2 to 1.40.0 solves the issue in my case.

closed 2024-12-19 by github-actions[bot]