Koenkk/zigbee2mqtt #24013: Errors during automatic polling Loginovo ZG-101ZL

codinv, 2024-09-18:
### What happened?

When the device automatically updates data, there are constant errors in the log.

### Zigbee2MQTT version

1.40.1 commit: 403d3c0

### Adapter firmware version

7.4.4 [GA]

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus-E (EFR32MG21)

### Setup

Docker

### Debug log

`[2024-09-18 09:01:32] info: 	z2m: Logging to console (silenced), file (filename: log.log)
[2024-09-18 09:01:32] info: 	z2m: Starting Zigbee2MQTT version 1.40.1 (commit #403d3c0)
[2024-09-18 09:01:32] info: 	z2m: Starting zigbee-herdsman (0.57.3)
[2024-09-18 09:01:32] info: 	zh:ember: Using default stack config.
[2024-09-18 09:01:32] info: 	zh:ember: ======== Ember Adapter Starting ========
[2024-09-18 09:01:32] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2024-09-18 09:01:32] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-09-18 09:01:32] info: 	zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
[2024-09-18 09:01:32] info: 	zh:ember:uart:ash: Serial port opened
[2024-09-18 09:01:32] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-09-18 09:01:33] info: 	zh:ember:uart:ash: ======== ASH connected ========
[2024-09-18 09:01:33] info: 	zh:ember:uart:ash: ======== ASH started ========
[2024-09-18 09:01:33] info: 	zh:ember:ezsp: ======== EZSP started ========
[2024-09-18 09:01:33] info: 	zh:ember: Adapter EZSP protocol version (13) lower than Host. Switched.
[2024-09-18 09:01:33] info: 	zh:ember: Adapter version info: {"ezsp":13,"revision":"7.4.4 [GA]","build":0,"major":7,"minor":4,"patch":4,"special":0,"type":170}
[2024-09-18 09:01:33] info: 	zh:ember: [STACK STATUS] Network up.
[2024-09-18 09:01:33] info: 	zh:ember: [INIT TC] Adapter network matches config.
[2024-09-18 09:01:33] info: 	zh:ember: [CONCENTRATOR] Started source route discovery. 1249ms until next broadcast.
[2024-09-18 09:01:33] info: 	z2m: zigbee-herdsman started (resumed)
[2024-09-18 09:01:33] info: 	z2m: Coordinator firmware version: '{"meta":{"build":0,"ezsp":13,"major":7,"minor":4,"patch":4,"revision":"7.4.4 [GA]","special":0,"type":170},"type":"EmberZNet"}'

`[2024-09-18 05:00:29] error: 	z2m: Failed to configure 'Кнопка ванная', attempt 3 (Error: ZCL command 0xa4c138748807dbcc/1 genBasic.read([4,0,1,5,7,65534], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:347:28)
    at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:916:26)
    at runNextTicks (node:internal/process/task_queues:60:5)
    at processImmediate (node:internal/timers:447:9)
    at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:443:29)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/moes.ts:548:13)
    at Configure.configure (/app/lib/extension/configure.ts:125:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))`

`
[2024-09-18 13:00:32] error: 	z2m: Failed to configure 'Bathroom button', attempt 1 (Error: ZCL command 0xa4c138748807dbcc/1 genBasic.read([4,0,1,5,7,65534], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:347:28)
    at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:916:26)
    at runNextTicks (node:internal/process/task_queues:60:5)
    at processImmediate (node:internal/timers:447:9)
    at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:443:29)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/moes.ts:548:13)
    at Configure.configure (/app/lib/extension/configure.ts:125:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))
`

rursache, 2024-09-19:
having the same issue with the same device

Koenkk, 2024-09-20:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-09-20 by Koenkk

codinv, 2024-10-02:
steel have errors after today update
Zigbee2MQTT version
 [1.40.2] commit: [e06848d]
Frontend version 0.7.4
zigbee-herdsman-converters version 20.21.0
zigbee-herdsman version 2.1.3

```
[2024-10-02 04:51:35] error: 	z2m: Failed to configure 'Кнопка ванная', attempt 1 (Error: ZCL command 0xa4c138748807dbcc/1 genOnOff.write({"tuyaOperationMode":1}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:350:28)
    at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:956:26)
    at processTicksAndRejections (node:internal/process/task_queues:95:5)
    at runNextTicks (node:internal/process/task_queues:64:3)
    at processImmediate (node:internal/timers:447:9)
    at Endpoint.write (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:393:9)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/moes.ts:565:13)
    at Configure.configure (/app/lib/extension/configure.ts:130:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:206:17))
[2024-10-02 08:51:24] error: 	z2m: Failed to configure 'Кнопка ванная', attempt 2 (Error: ZCL command 0xa4c138748807dbcc/1 genOnOff.write({"tuyaOperationMode":1}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:350:28)
    at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:956:26)
    at processTicksAndRejections (node:internal/process/task_queues:95:5)
    at runNextTicks (node:internal/process/task_queues:64:3)
    at processImmediate (node:internal/timers:447:9)
    at Endpoint.write (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:393:9)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/moes.ts:565:13)
    at Configure.configure (/app/lib/extension/configure.ts:130:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:206:17))
[2024-10-02 12:51:10] error: 	z2m: Failed to configure 'Кнопка ванная', attempt 3 (Error: ZCL command 0xa4c138748807dbcc/1 genOnOff.write({"tuyaOperationMode":1}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:350:28)
    at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:956:26)
    at processTicksAndRejections (node:internal/process/task_queues:95:5)
    at runNextTicks (node:internal/process/task_queues:64:3)
    at processImmediate (node:internal/timers:447:9)
    at Endpoint.write (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:393:9)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/moes.ts:565:13)
    at Configure.configure (/app/lib/extension/configure.ts:130:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:206:17))

```

originalcom, 2025-08-21:
I get error  UNSUPPORTED_ATTRIBUTE on version 2.6.0. . It's still periodically throwing this error even when the buttons haven't been used. Tried resetting but still happens. 

Z2M Server: z2m: Failed to

configure 'Porch Zigbee Porch

Button', attempt 1 (Error: ZCL

command 0xa4c138957bbfbef6/1

genOnOff.write({"tuyaOperationMode" :1},

{"timeout":10000, "disableResponse": false, "disableRecovery": false, "disa bleDefaultResponse": true, "direction ":0,"reservedBits":0, "writeUndiv":f alse}) failed (Status

'UNSUPPORTED_ATTRIBUTE')