Koenkk/zigbee2mqtt #24016: [New device support]: Tuya 213E Ultrasonic water meter (_TZE200_zlwr0raf)

Takacsk76, 2024-09-18:
### Link

https://www.aliexpress.com/item/1005007308058989.html?spm=a2g0o.order_list.order_list_main.5.79851802N0tQyH

### Database entry

{"id":56,"type":"EndDevice","ieeeAddr":"0xa4c1389cbaf936cd","nwkAddr":807,"manufId":4417,"manufName":"_TZE200_zlwr0raf","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"��|.i��|.i��|.i\u0004�|.f\u0005�|.\u0012","65506":56,"65508":0,"stackVersion":0,"dateCode":"","appVersion":72,"modelId":"TS0601","manufacturerName":"_TZE200_zlwr0raf","powerSource":3,"zclVersion":3,"hwVersion":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1726666275707}

### Zigbee2MQTT version

1.40.1

### Comments

I bought a Zigbee water meter.
I tried to make the configuration based on another type:
[https://github.com/Koenkk/zigbee2mqtt/issues/21255](url)

Unfortunately, it didn't work, and then I did a little simplification, but it also failed.
[image]
[image]
[image]
[image]

I don't really understand how Zigbee and Tuya clusters work.

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE200_zlwr0raf',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

const {} = require('zigbee-herdsman-converters/lib/modernExtend');
// Add the lines below
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE200_zlwr0raf',
    description: 'Tuya water meter with valve',
    extend: [],
    fromZigbee: [],
    toZigbee: [],
    exposes: [
        e.battery_voltage('battery_voltage', ea.STATE).withDescription('Battery voltage'),
        e.temperature().withDescription('Water Temperature'),
    ],
    meta: {
        tuyaDatapoints: [
            [22,'temperature', tuya.valueConverter.divideBy100],
            [26,'battery_voltage', tuya.valueConverter.raw],
        ],
    },
};

module.exports = definition;

atanesmarc, 2024-10-10:
Thank you very much!
after a couple of months your post gave me light!
I'm also struggling with decoding the JSON in month_consumption and daily_consumption but with no success.
AFAIK the last position in the buffer is the current state but I had no success implementing it on the external converter.

Takacsk76, 2024-10-14:
> Thank you very much! after a couple of months your post gave me light! I'm also struggling with decoding the JSON in month_consumption and daily_consumption but with no success. AFAIK the last [...]

I have not yet installed the device in its final location, but it is expected to be ready this week. I didn't know what to do with the JSON data yet, because there is nothing to compare it to. In my opinion, the data could be statistics for the last 7 days and the last 7 months, but this is just a guess (but why only 7 months?).

danielewyss, 2024-11-01:
Is there something already available to add manually ?

Takacsk76, 2024-11-02:
> Is there something already available to add manually ?
As far as I know, there isn't.

Takacsk76, 2024-11-02:
As far as I know, there isn't.

senrabyar, 2024-11-13:
I have just acquired a 214C and have just managed to get it paired (the supplied manual described using a Hall Effect button with a magnet - it doesn't exist - you have to long-press when the date is shown)    I can see the device OK in Tuya, but unsupported in Z2M.   Not yet plumbed in.  But hopefully I can contribute or at least test anything others find.

Takacsk76, 2024-11-13:
Hi, I received a video from the seller for zigbee pairing: 

User Manual===https://drive.google.com/file/d/1BZ7b_sgXukM0FRVaEws1VgDZbckWVEoa/view?usp=sharing

213E===https://drive.google.com/file/d/1q-v2mhXkvjOclElcICHZBHT3yqfJ9d_G/view?usp=sharing

214C===https://drive.google.com/file/d/1gM6k_ndeKYNl9mXyKixQ_uLA6TiDvyAa/view?usp=sharing

senrabyar, 2024-11-13:
Ah thanks, now I have the full manual - after question to the supplier they had sent me just the two relevant pages.    The manual in the box is totally different.     Your info will certainly help anyone else who discovers this.
But that's really outside of this thread on Z2M !

Takacsk76, 2024-11-13:
Very welcome.
I would also be happy to help implement the integration into Z2M, but no such request has been received so far.

Takacsk76, 2024-11-15:
I just need to figure out how to calculate the battery percentage. I'm trying, but I can't.

danielewyss, 2025-01-15:
The device is not supported, therefore I think that we would like to keep it open ;-)

senrabyar, 2025-01-15:
I've no idea how many devices are out there needing device to work with MQTT - in my case a 214C - through to HomeAssistant or direct to NodeRed.   But I will have to dabble further into this soon - I've already spent £££s buying the device, and will have to spend ££ to get it installed into my house plumbing.   I just don't have the time to get my head round external definitions yet, so was hoping it would become a supported device.

Takacsk76, 2025-01-16:
I had to fix the processing of template sensors because the binary processing became invalid.
      {
        "created_at": "2024-10-02T10:34:55.948720+00:00",
        "data": {},
        "disabled_by": null,
        "domain": "template",
        "entry_id": "01J96CR52CE0F3MK6QW02PAX75",
        "minor_version": 1,
        "modified_at": "2024-10-02T10:34:55.948723+00:00",
        "options": {
          "template_type": "binary_sensor",
          "name": "Water meter battery alarm",
          "state": "{{ ### '{:013b}'.format(states('sensor.water_meter_warning') | int(0))[-1] == '1' }}",
          "device_class": "problem",
          "device_id": "e88e4ca5742993c7b66d24a10fa3fdee"
        },

Takacsk76, 2025-01-16:
> I've no idea how many devices are out there needing device to work with MQTT - in my case a 214C - through to HomeAssistant or direct to NodeRed. But I will have to dabble further into this soon - [...]

I don't know either, but I hope it will be added to the list of supported devices one day and maybe they will fix it. I offer to help with testing, but so far no one has contacted me about it.

almostlunatic, 2025-02-07:
I got 413E hoping it could be added by ZHA but nothing.  I'll send it back until there will be a solution, if there will...
I need a real time precise water meter to discover leaks in the old pipework and toilets and bathrooms, then close a valve automatically and get an alarm. It seemed really promising. So sad!

Takacsk76, 2025-02-07:
> I got 413E 2w ago hoping it could be added by ZHA but nothing. I cannot pair it. I'll send it back until there will be a solution, if there will... I need a real time precise water meter to discover leaks in the old pipework and toilets and bathrooms, then close a valve automatically and get an alarm. It seemed really promising. So sad!
> 
> I could use it with Zigbee2MQTT if it would work, but I have no single device through it, I need to learn how to use it along with ZHA.
> 
> One question - can it transmit real-time or every hour? In the user manual it says real time. I need a real-time device. Can anybody suggest anything?

I also started with ZHA. I switched to Z2M with 30 devices. Both platforms have problems and cannot follow newly appearing devices. I think you should try it too. Temporarily disable ZHA and install Z2M. If you don't like it, you can quickly restore it.
Zigbee is a communication channel. Unfortunately, Tuya data is handled very freely by developers. Many device manufacturers just buy the same software components from a third party. The device would be perfect, but the software side is poor.
The 213E device sends the clock cyclically. The sending can be set in an interval of 1-24 hours. By default, it sends every 12 hours. I have a 1-hour cycle set.
In case of error detection, it sends "Warning" (fault) data immediately. The "Water meter empty pipe alarm" is easy to test.

Sthev61800, 2025-04-09:
[image]
I managed to partially integrate the meter

bertrandgressier, 2025-04-20:
> [image]
> I managed to partially integrate the meter

Super! Could you share your work, maybe I can help you. 
I just installed a 213e and disappointed to not have a z2m support.

closed 2025-06-28 by github-actions[bot]

bouboun59, 2025-07-21:
> [image] I managed to partially integrate the meter

Hi French friands

Could you hello ys and post your file configuration thanks a lot

RobyMango, 2025-10-15:
Hello, I also have a 213E Water Meter and would like to connect it to HA via Z2M. I have paired it but it doesn't expose anything. Can anyone help me please?