Koenkk/zigbee2mqtt #24018: Z2M crash while running OTA update of Philips Hue Devices

iaincaradoc, 2024-09-18:
### What happened?

While running OTA update of any Philips Hue device on my network, Z2M crashes and restarts.

### What did you expect to happen?

I expected the OTA update to proceed and complete.

### How to reproduce it (minimal and precise)

Select any Hue device and attempt to run an OTA update. Attached log shows a Hue Strip named "BookshelfStrip," Model # 8718699703424 

https://www.zigbee2mqtt.io/devices/8718699703424.html#philips-8718699703424

### Zigbee2MQTT version

1.40.1

### Adapter firmware version

7.4.3 [GA]

### Adapter

EmberZNET / Home Assistant SkyConnect USB

### Setup

Add-on on Home Assistant Yellow - built-in SkyConnect is configured for Matter

### Debug log

[log.log](https://github.com/user-attachments/files/17045578/log.log)

lux73, 2024-09-19:
No Problem here - have updated 10 Hue Bulbs E14, P14 and E27 without any Problem

plain z2m 1.40.1 / CC2652RB

iaincaradoc, 2024-09-19:
> plain z2m 1.40.1 / CC2652RB

I suspect EmberZNet is the root, but I'm not certain yet.

Koenkk, 2024-09-19:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/24018, possible solution: https://github.com/Koenkk/zigbee2mqtt/issues/23852#issuecomment-2332472942

closed 2024-09-19 by Koenkk