Koenkk/zigbee2mqtt #24024: Niko 552-721X1 state not reported in decoupled mode

FHannes, 2024-09-18:
### What happened?

When in decoupled mode, my Niko 552-721X1 does not report state changes when you interact with the physical button. Seeing as the Niko hub can use this decoupled mode without issue, there must be some way to get these states to report correctly. I dont see any messages come into the logs either though.

### Zigbee2MQTT version

1.40.1

### Adapter firmware version

20211217

### Adapter

Sonoff ZBDongle-E

### Setup

Running in docker container

vdkeybus, 2024-12-01:
I observe similar behaviour. I used it straight out of the box, and I also tried running it with the Niko bridge first.
My log is in debug mode, and nothing happens when I push a 'decoupled' switch.

vdkeybus-ni, 2024-12-11:
@rorie1982 Did you mean to say that in 'decoupled' mode nothing happens?

Interesting to know that it apparently worked before. I also see nothing happening in any log when I push a button on a decoupled switch.

rorie1982, 2024-12-11:
In decoupled mode there is a extra attribute added to the JSON message, so for example when you press the right switch action is added with the following value ""action": "single_right"". When you release the switch it changes to release_right. All other values in the JSON are not updated. When I change the "Operation mode" to relay_controled and press one of the physical switches no message is send by zigbee2mqtt, so no update. This worked in previous releases.

vdkeybus, 2024-12-11:
I also don't see those actions in relay_controlled mode. But I also don't
see them in decoupled mode. How do you switch between the modes?

rorie1982, 2024-12-11:
I switch these mode using the UI of zigbee2mqtt
[image]

vdkeybus, 2024-12-28:
@rorie1982 Thanks. I found that already, but wanted to know if you did it in another way.

svenjochems, 2025-01-18:
Related to https://github.com/Koenkk/zigbee2mqtt/issues/13737

closed 2025-03-27 by github-actions[bot]