Koenkk/zigbee2mqtt #24025: Sonoff SNZB-04P doesn't report any power source

willdurand, 2024-09-19:
### What happened?

The Sonoff [SNZB-04P](https://www.zigbee2mqtt.io/devices/SNZB-04P.html#sonoff-snzb-04p) device doesn't report the power source (which should be battery). Instead, it shows `?`. This looks fairly similar to https://github.com/Koenkk/zigbee-herdsman-converters/pull/8003

[image]

### What did you expect to happen?

The power source should be "battery".

### Zigbee2MQTT version

1.40.1 commit: 403d3c0

### Adapter firmware version

7.4.3 [GA]

### Adapter

Sonoff dongle with EmberZNet

### Setup

Z2M + Mosquitto + HA (dockerized)

Koenkk, 2024-09-19:
Should be fixed now, after updating make sure to re-configure the device by pressing the yellow configure button.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-09-19 by Koenkk

maximvd, 2024-10-20:
> Should be fixed now, after updating make sure to re-configure the device by pressing the yellow configure button.
> 
> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

I'm still having this issue on version [1.40.2](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.40.2)

Koenkk, 2024-10-21:
Did you re-configure the device via the z2m frontend, if it fails make sure to wake it up before configuring and restart z2m afterwards

Koenkk, 2024-10-22:
Found the issue and fixed it!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

vip-klaus, 2024-11-01:
Still the same problem with Z2M version 1.41.0 for all my 8 Sonoff SNZB-02P devices: unknown power source. Home assistant and Z2M restarted of course.

Regards
Klaus

claudiucismaru, 2024-11-02:
Same here, for SNZB-02D. I have reconfigured the device, after upgrade to 1.41.0 and still shows unknown power source.

vip-klaus, 2024-11-02:
I also still have the problem that the frient smoke detectors report "Table full". Although this problem is explicitly mentioned as being solved.

camlin, 2024-11-02:
Same here. I tried the refresh and also did redo a full pairing but no dice.

What is also interesting is that Sonoff devices I paired a long time ago (with 1.35.1) are still displaying the battery symbol fine (for example my SNZB-02D is ok). It is only the device I am trying to pair now (SNZB-04P) that is having problems (and I can't roll back to 1.35.1):

[image]

But there is definitely progress. With 1.40.2 I was getting big errors in the log after the pairing (because it was grumpy about some of the device attributes). But these errors are now gone and the logs looks a lot cleaner:

```
[2024-11-02 10:02:51] info:     zh:ember: [STACK STATUS] Network opened.
[2024-11-02 10:02:52] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"9nn9i-10"}'
[2024-11-02 10:03:24] info:     zh:controller: Interview for '0x0ceffBLANKED' started
[2024-11-02 10:03:24] info:     z2m: Device '0x0ceffBLANKED' joined
[2024-11-02 10:03:24] info:     z2m: Starting interview of '0x0ceffBLANKED'
[2024-11-02 10:03:24] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0ceffBLANKED","ieee_address":"0x0ceffBLANKED"},"type":"device_joined"}'
[2024-11-02 10:03:24] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0ceffBLANKED","ieee_address":"0x0ceffBLANKED","status":"started"},"type":"d
evice_interview"}'
[2024-11-02 10:03:24] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0ceffBLANKED","ieee_address":"0x0ceffBLANKED"},"type":"device_announce"}'
[2024-11-02 10:03:24] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Bibliotheque', payload '{"battery":100,"humidity":71.3,"linkquality":124,"temperature":17.5,"update":{"installed_version":8448,"latest_
version":-1,"state":"idle"}}'
[2024-11-02 10:03:27] info:     zh:controller: Succesfully interviewed '0x0ceffBLANKED'
[2024-11-02 10:03:27] info:     z2m: Successfully interviewed '0x0ceffBLANKED', device has successfully been paired
[2024-11-02 10:03:27] info:     z2m: Device '0x0ceffBLANKED' is supported, identified as: SONOFF Contact sensor (SNZB-04P)
[2024-11-02 10:03:28] info:     z2m: Configuring '0x0ceffBLANKED'
[2024-11-02 10:03:28] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Contact sensor","exposes":[{"access":1,"description":"Indicates whether t
he device is opened or closed","label":"Contact","name":"contact","property":"contact","type":"binary","value_off":true,"value_on":false},{"access":1,"category":"diagnostic","description":"Indicates whether the
 battery of the device is almost empty","label":"Battery low","name":"battery_low","property":"battery_low","type":"binary","value_off":false,"value_on":true},{"access":5,"description":"Tamper-proof status","la
bel":"Tamper","name":"tamper","property":"tamper","type":"binary","value_off":false,"value_on":true},{"access":5,"category":"diagnostic","description":"Remaining battery in %","label":"Battery","name":"battery"
,"property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":5,"category":"diagnostic","description":"Reported battery voltage in millivolts","label":"Voltage","name":"voltage","pr
operty":"voltage","type":"numeric","unit":"mV"},{"access":1,"category":"diagnostic","description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"num
eric","unit":"lqi","value_max":255,"value_min":0}],"model":"SNZB-04P","options":[],"supports_ota":true,"vendor":"SONOFF"},"friendly_name":"0x0ceffBLANKED","ieee_address":"0x0ceffBLANKED","status":"succe
ssful","supported":true},"type":"device_interview"}'
[2024-11-02 10:03:28] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x0ceffBLANKED', payload '{"battery":100,"linkquality":144}'
[2024-11-02 10:03:29] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x0ceffBLANKED', payload '{"battery":100,"linkquality":148,"voltage":3000}'
[2024-11-02 10:03:29] info:     z2m: Successfully configured '0x0ceffBLANKED'
[2024-11-02 10:03:57] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x0ceffBLANKED', payload ''
[2024-11-02 10:03:57] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x0ceffBLANKED/availability', payload ''
[2024-11-02 10:03:57] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/MainDoor', payload '{"battery":100,"linkquality":148,"voltage":3000}'
[2024-11-02 10:03:57] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/rename', payload '{"data":{"from":"0x0ceffBLANKED","homeassistant_rename":false,"to":"MainDoor"},"status":"o
k","transaction":"9nn9i-11"}'
[2024-11-02 10:03:57] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/MainDoor/availability', payload '{"state":"online"}'
[2024-11-02 10:04:46] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Cuisine', payload '{"battery":100,"humidity":67.1,"linkquality":144,"temperature":18.3,"update":{"installed_version":8448,"latest_versi
on":-1,"state":"idle"}}'
```

So I think it is now processing the battery attributes correctly.

Could it now be an issue with cleaning-up / updating some states in the DB?

xdmore70, 2024-11-06:
I have the same issue ;

if it can be any help, the SNZB-02D+ SNZB-01P devices before  OS+HomeHassistant+Zigbee2mqtt  upgrade that I did on my RaspBerry, was working correctly.

[image]

[image]

alainvdu69, 2024-12-03:
Hi !
Same issue with 4 SNZB-04P and 2 SNZB-02D : bad report of the powerSource
"powerSource": "Unknown"

Drac0fly, 2025-01-05:
Up - I have same problem with "SNZB-04P" but with the TAMPER.

It's already installed on my window, so the tamper is push normally...  and the sensor value is "N/A" on one, and "NONE" on the other one

help ^^

alainvdu69, 2025-01-06:
i've used this "trick" to solve.
stop Z2M.
make a backup of database.db
edit (with VI or NANO) the file database.db
find the devices with bad powersource and change it.
start Z2M.
that's all.

good or bad solution, i don't know, but it's fine now.
i think that if i lost the device and repair, i've to edit again.

camlin, 2025-05-21:
Me I did it by going to the UI and doing a force delete of the device (on Z2M 2.3.0), that cleaned the database.db from the record and I just had to re-pair it. And this time it did show the correct data.

One thing to note is that the UI/MQTT reports the initial state "as is". So I had no "Contact" info until I open and then closed the door (so it now shows "Closed).

@Drac0fly I suspect the issue is just the same for the Tamper: Unless you played with the tamper after pairing it will keep displaying "N/A" until you have a tamper event.

So basically having tamper showing "N/A" or whatever is the good tamper state (i.e. "NONE"?) means it is all good.