Koenkk/zigbee2mqtt #24026: Problem with TS0601 Dimmer handling brightness from Home Assistant

eatoff8, 2024-09-19:
### What happened?

The light turns on and off fine. but when turning on with a brightness command, the light ignores the brightness and just toggles on at whatever the last brightness was set to.

An example is I have the light 100% during the day, and only 5% brightness between midnight and 5am. This works fine with my philips zigbee bulbs. What happens with this dimmer switch is that it will turn on at 100% brightness at 1am. or if i manually set it to 50% brightness and then run the automation to turn on at 5% (or 100%) it will turn on again at 50%.

A workaround that i am currently using is the template light found here in this issue - https://github.com/Koenkk/zigbee2mqtt/issues/19513

### What did you expect to happen?

I expect the light to turn on with the correct brightness set

### How to reproduce it (minimal and precise)

Set a brightness and then turn off the light. Then use a service call to turn the light on at a different brightness. that brightness will be ignored.

### Zigbee2MQTT version

1.40.1

### Adapter firmware version

20240716

### Adapter

SLZB-06P

### Setup

Add-on to Home Assistant OS on a mini PC

SoldierFace, 2024-09-19:
This might help:This file might fix it
https://mega.co.nz/#!9vE0jIrD!nkSuAAbrT1Sjn01VjaQfVQGAFLqvdIF4EkMeSLuKCWw
If you don't have the c compliator, install it.(gcc or clang)

eatoff8, 2024-09-19:
A small section of the debu log when turning it on and adjusting the brightness if that is of any use (its the Light - Ensuite):

`info 2024-09-19 18:31:13z2m: Successfully changed options
info 2024-09-19 18:31:13z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"8r0th-1"}'
info 2024-09-19 18:31:16z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Plug - Olivers Lamp', payload '{"child_lock":"UNLOCK","countdown":0,"current":0,"energy":22.1,"indicator_mode":"off/on","last_seen":"2024-09-19T18:31:20+08:00","linkquality":58,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":null,"voltage":245}'
info 2024-09-19 18:31:18z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Light - Lounge 3', payload '{"brightness":202,"color":{"h":40,"hue":40,"s":49,"saturation":49,"x":0.3804,"y":0.3767},"color_mode":"color_temp","color_power_on_behavior":null,"color_temp":250,"do_not_disturb":null,"last_seen":"2024-09-19T18:31:23+08:00","linkquality":108,"state":"OFF"}'
info 2024-09-19 18:31:19z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Sensor - Network Cabinet', payload '{"battery":100,"humidity":40.7,"last_seen":"2024-09-19T18:31:23+08:00","linkquality":177,"temperature":24.9,"update":{"installed_version":8448,"latest_version":8448,"state":"idle"}}'
info 2024-09-19 18:31:21z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Plug - Olivers Lamp', payload '{"child_lock":"UNLOCK","countdown":0,"current":0,"energy":22.1,"indicator_mode":"off/on","last_seen":"2024-09-19T18:31:25+08:00","linkquality":58,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":null,"voltage":245}'
info 2024-09-19 18:31:25z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Light - Lounge 4', payload '{"brightness":3,"color":{"h":null,"hue":null},"color_mode":"color_temp","color_temp":250,"do_not_disturb":true,"last_seen":"2024-09-19T18:31:29+08:00","linkquality":177,"state":"OFF"}'
info 2024-09-19 18:31:25z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Light - Ensuite', payload '{"backlight_mode":"normal","brightness":51,"countdown":0,"last_seen":"2024-09-19T18:31:12+08:00","light_type":"led","linkquality":108,"max_brightness":100,"min_brightness":3,"power_on_behavior":"previous","state":"OFF"}'
info 2024-09-19 18:31:25z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Light - Ensuite/type', payload '{"light":"led"}'
info 2024-09-19 18:31:25z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Light - Ensuite', payload '{"backlight_mode":"normal","brightness":51,"countdown":0,"last_seen":"2024-09-19T18:31:30+08:00","light_type":"led","linkquality":108,"max_brightness":100,"min_brightness":3,"power_on_behavior":"previous","state":"ON"}'
info 2024-09-19 18:31:25z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Light - Ensuite/type', payload '{"light":"led"}'
info 2024-09-19 18:31:26z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Light - Ensuite', payload '{"backlight_mode":"normal","brightness":51,"countdown":0,"last_seen":"2024-09-19T18:31:30+08:00","light_type":"led","linkquality":108,"max_brightness":100,"min_brightness":3,"power_on_behavior":"previous","state":"ON"}'
info 2024-09-19 18:31:26z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Light - Ensuite/type', payload '{"light":"led"}'
info 2024-09-19 18:31:26z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Plug - Olivers Lamp', payload '{"child_lock":"UNLOCK","countdown":0,"current":0,"energy":22.1,"indicator_mode":"off/on","last_seen":"2024-09-19T18:31:30+08:00","linkquality":63,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":null,"voltage":245}'
info 2024-09-19 18:31:26z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Plug - Olivers Lamp', payload '{"child_lock":"UNLOCK","countdown":0,"current":0,"energy":22.1,"indicator_mode":"off/on","last_seen":"2024-09-19T18:31:30+08:00","linkquality":58,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":null,"voltage":245}'
info 2024-09-19 18:31:28z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Light - Ensuite', payload '{"backlight_mode":"normal","brightness":129,"countdown":0,"last_seen":"2024-09-19T18:31:30+08:00","light_type":"led","linkquality":108,"max_brightness":100,"min_brightness":3,"power_on_behavior":"previous","state":"ON"}'
info 2024-09-19 18:31:28z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Light - Ensuite/type', payload '{"light":"led"}'
info 2024-09-19 18:31:28z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Light - Living Room Downlight 2', payload '{"brightness":254,"color":{"x":0.3644,"y":0.3661},"color_mode":"color_temp","color_temp":227,"last_seen":"2024-09-19T18:31:32+08:00","linkquality":90,"state":"OFF","update":{"installed_version":570503169,"latest_version":570503169,"state":"idle"},"update_available":null}'
info 2024-09-19 18:31:28z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Light - Ensuite', payload '{"backlight_mode":"normal","brightness":129,"countdown":0,"last_seen":"2024-09-19T18:31:32+08:00","light_type":"led","linkquality":108,"max_brightness":100,"min_brightness":3,"power_on_behavior":"previous","state":"ON"}'
info 2024-09-19 18:31:28z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Light - Ensuite/type', payload '{"light":"led"}'
[... cut ...]
info 2024-09-19 18:31:37z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Light - Hallway 2', payload '{"brightness":3,"color":{"h":null,"hue":null},"color_mode":"color_temp","color_temp":243,"do_not_disturb":true,"last_seen":"2024-09-19T18:31:41+08:00","linkquality":177,"state":"OFF"}'
info 2024-09-19 18:31:37z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Plug - Olivers Lamp', payload '{"child_lock":"UNLOCK","countdown":0,"current":0,"energy":22.1,"indicator_mode":"off/on","last_seen":"2024-09-19T18:31:41+08:00","linkquality":63,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":null,"voltage":245}'`

closed 2024-12-19 by github-actions[bot]

eatoff8, 2025-01-10:
> Looks like I just missed the window on this one but for what it’s worth i’m experiencing exactly the same behavior described. Everything was working fine, and now suddenly (not sure of the timing or whether an update triggered it) I see exactly the same issue as @eatoff8

Sorry, I don't actually have any kind of solution. I have resorted to a different brand of smart dimmers for some other lights, but my 3 of these are still having this same issue