Koenkk/zigbee2mqtt #24035: LYWSD03MMC-z Failed to configure

pniewiadowski, 2024-09-20:
### What happened?

When I want to pair a LYWSD03MMC-z device it failed to configure and reports/expose only temperature.

[image]
[image]

### What did you expect to happen?

I expect to finish the configuration

### How to reproduce it (minimal and precise)

Try to pair the device

### Zigbee2MQTT version

1.40.1

### Adapter firmware version

20210211

### Adapter

CC2652P

### Setup

docker

### Debug log

error 2024-09-20 11:03:25z2m: Failed to configure '0xa4c1386ec1174330', attempt 1 (Error: Bind 0xa4c1386ec1174330/1 msTemperatureMeasurement from '0x00124b0022811afd/1' failed (ZDO error: bind failed with status 'TABLE_FULL' (140)) at /app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:1126:36)
error 2024-09-20 11:03:55z2m: Failed to configure '0xa4c1386ec1174330', attempt 2 (Error: Bind 0xa4c1386ec1174330/1 msTemperatureMeasurement from '0x00124b0022811afd/1' failed (ZDO error: bind failed with status 'TABLE_FULL' (140)) at /app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:1126:36)
error 2024-09-20 11:04:44z2m: Request 'zigbee2mqtt/bridge/request/device/configure_reporting' failed with error: 'Bind 0xa4c1386ec1174330/1 msTemperatureMeasurement from '0x00124b0022811afd/1' failed (ZDO error: bind failed with status 'TABLE_FULL' (140))'
error 2024-09-20 11:04:45z2m: Failed to configure '0xa4c1386ec1174330', attempt 3 (Error: Bind 0xa4c1386ec1174330/1 msTemperatureMeasurement from '0x00124b0022811afd/1' failed (ZDO error: bind failed with status 'TABLE_FULL' (140)) at /app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:1126:36)
error 2024-09-20 11:08:04z2m: Request 'zigbee2mqtt/bridge/request/device/configure_reporting' failed with error: 'Bind 0xa4c1386ec1174330/1 msRelativeHumidity from '0x00124b0022811afd/1' failed (ZDO error: bind failed with status 'TABLE_FULL'

nomc2, 2024-10-03:
Same issue here, It still showed when rolling back, unfortunately I was not able to rollback further than 1.40.0

closed 2024-12-19 by github-actions[bot]

DittmannCz, 2025-01-07:
any solution please?

ArXa1L, 2025-01-07:
Hi, I had the same issue and this comment helped me
https://github.com/Koenkk/zigbee2mqtt/issues/23858#issuecomment-2333376393

ArXa1L, 2025-01-07:
> how did You proceed? I switched unbind -what now?

I followed these steps:
- %your_thermometer% -> Tab `Bind`
- Clicked `Unbind`, waited for it to complete
- Selected source endpoint: `1`, Destination: `Coordinator`, Destination endpoint: `1`, Clusters: `genPollCtrl`, `PowerCfg`, `Humidity`, `Temperature` -> Clicked `Bind`

But it didn't help in long run and the issue came back after a while. After that I tried to remove one of my thermometers from z2m and bind it again - so far no error, but I noticed that only the `genPollCtrl` cluster is bound