Koenkk/zigbee2mqtt #24039: [Wrong device]: Tuya Zigbee USB Power Switch shown as  TS0001

KrX3D, 2024-09-20:
### Link

https://www.aliexpress.com/item/1005007444117700.html

### Model

XMSJ

### Description

Zigbee USB plug

### Vendor

Tuya

### Picture (link)

https://ae-pic-a1.aliexpress-media.com/kf/S5b232c22748a41e5a7d9fcaccb4d842dM.jpg_640x640.jpg_.webp

### Database entry

{"id":83,"type":"Router","ieeeAddr":"0xa4c1382e261cf40a","nwkAddr":53829,"manufId":4417,"manufName":"_TZ3000_8n7lqbm0","powerSource":"Mains (single phase)","modelId":"TS0001","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"57344":{"attributes":{"53251":"AAAA"}},"genBasic":{"attributes":{"65506":56,"65508":0,"65534":0,"modelId":"TS0001","manufacturerName":"_TZ3000_8n7lqbm0","powerSource":1,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"onOff":1,"onTime":0,"offWaitTime":0,"moesStartUpOnOff":2}},"manuSpecificTuya_3":{"attributes":{"53248":0,"53249":0,"53250":0,"53251":0,"53252":0,"53253":0,"switchType":2}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c0afbf","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1726842924303}

### Notes

Device is shown as "[Tuya TS0001](https://www.zigbee2mqtt.io/devices/TS0001.html#tuya-ts0001)" 

the switch is working so far.

Ther is also 2 more Versions for 2 and 3 Ports - as well this zigbee and a wifi type

Koenkk, 2024-09-20:
Added!

Could you also submit a picture for the docs? This can be done by:
1. Create a fork by clicking [here](https://github.com/Koenkk/zigbee2mqtt.io/fork)
2. Go to the `public/images/devices` directory, *Add file* -> *Upload files*
3. Upload the files and press *Commit changes*
4. Press *Contribute* -> *Open pull request* -> update title/description -> *Create pull request*

Make sure that:
- The filename is `XMSJ.png`
- The size is 512x512
- The background is transparent (use e.g. [Adobe remove background](https://new.express.adobe.com/tools/remove-background))

closed 2024-09-20 by Koenkk

KrX3D, 2024-09-21:
Ok thanks.

i created the pull request here:

https://github.com/Koenkk/zigbee2mqtt.io/pull/3032