Koenkk/zigbee2mqtt #24041: Error ZCL command

JericoPablo, 2024-09-20:
### What happened?

After update Bosch Thermostat II in livingroom stopped working, losing connection and needs to be paired again. Loses again after a while the connection.

Konfig:
port: /dev/ttyACM0
adapter: ember
baudrate: 115200

Mosquitto broker
Current version: 6.4.1

### What did you expect to happen?

i expect that updates which have major issues are getting fixed ASAP

### How to reproduce it (minimal and precise)

Install ncp-uart-hw-v7.4.4.0-zbdonglee-115200.gbl - and Zigbee2MQTT Current version: 1.40.1-1  - and Mosquitto broker (MQTT) Current version: 6.4.1
Pair Bosch Thermostat II via Installation Code in Zigbee2MQTT Addon

### Zigbee2MQTT version

1.40.1-1

### Adapter firmware version

ncp-uart-hw-v7.4.4.0-zbdonglee-115200.gbl

### Adapter

Sonoff ZigBee 3,0 USB Dongle Plus-E with ember

### Setup

Add-on in HomeAssistant OS on a Acer Aspire V5-573G - connected via 1m USB Cable 

### Debug log

```
[2024-09-20 20:03:43] error: 	zh:ember: Delivery of BROADCAST failed for '65535' [apsFrame={"profileId":260,"clusterId":10,"sourceEndpoint":1,"destinationEndpoint":1,"options":0,"groupId":0,"sequence":69} messageTag=255]
[2024-09-20 20:04:53] error: 	zh:ember: Delivery of BROADCAST failed for '65535' [apsFrame={"profileId":260,"clusterId":10,"sourceEndpoint":1,"destinationEndpoint":1,"options":0,"groupId":0,"sequence":76} messageTag=255]
[2024-09-20 20:11:43] error: 	zh:ember: Delivery of BROADCAST failed for '65535' [apsFrame={"profileId":260,"clusterId":10,"sourceEndpoint":1,"destinationEndpoint":1,"options":0,"groupId":0,"sequence":117} messageTag=255]
[2024-09-20 20:21:13] error: 	zh:ember: Delivery of BROADCAST failed for '65535' [apsFrame={"profileId":260,"clusterId":10,"sourceEndpoint":1,"destinationEndpoint":1,"options":0,"groupId":0,"sequence":175} messageTag=255]
[2024-09-20 20:27:24] error: 	zh:ember: Delivery of BROADCAST failed for '65535' [apsFrame={"profileId":260,"clusterId":10,"sourceEndpoint":1,"destinationEndpoint":1,"options":0,"groupId":0,"sequence":212} messageTag=255]
[2024-09-20 20:35:40] error: 	z2m: Publish 'set' 'occupied_heating_setpoint' to 'Wohnzimmer Heizung' failed: 'Error: ZCL command 0x18fc26000008245d/1 hvacThermostat.write({"occupiedHeatingSetpoint":2400}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Delivery failed for {"profileId":260,"clusterId":513,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":172})'
[2024-09-20 20:35:40] error: 	z2m: Publish 'set' 'occupied_heating_setpoint' to 'Wohnzimmer Heizung' failed: 'Error: ZCL command 0x18fc26000008245d/1 hvacThermostat.write({"occupiedHeatingSetpoint":2500}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Delivery failed for {"profileId":260,"clusterId":513,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":174})'
[2024-09-20 20:35:59] error: 	z2m: Publish 'set' 'occupied_heating_setpoint' to 'Wohnzimmer Heizung' failed: 'Error: ZCL command 0x18fc26000008245d/1 hvacThermostat.write({"occupiedHeatingSetpoint":2350}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Delivery failed for {"profileId":260,"clusterId":513,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":183})'
```

JericoPablo, 2024-09-20:
also i always need to delete that entity and readd it again - pressing configure or interview button does not help - ends in error

b177y, 2024-09-22:
Similar issue is occurring with Hive receiver and thermostat:

```
 z2m: Publish 'set' 'occupied_heating_setpoint' to '0x001e5e09027d7641' failed: 'Error: ZCL command 0x001e5e09027d7641/6 hvacThermostat.write({"occupiedHeatingSetpoint":2400}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'INVALID_VALUE')'
```

Tried to reset and re-pair both devices, receiver is pairing but still getting above error and thermostat is failing to re-pair to receiver with:
```
zh:controller: Interview failed for '0x001e5e09027bc61f with error 'Error: Interview failed because can not get active endpoints ('0x001e5e09027bc61f')'
```

JericoPablo, 2024-09-30:
now i cannot connect my Bosch Thermostat II again. Its insane...

reinhardsiegel79, 2024-10-02:
same here, anything I can do to help nail this down?

allbercik, 2024-10-03:
Similar problem 
z2m: Publish 'set' 'remote_temperature' to 'termostat sypialnia' failed: 'Error: ZCL command 0x18fc2600000914b8/1 hvac Thermostat.write({"remote Temperatur e":2349}, {"timeout":10000,"disableResponse":false,"di sableRecovery":false,"disableDefaultRespon se":true,"direction":0,"reservedBits":0,"manuf acturerCode":4617,"writeUndiv":false}) failed (Timeout - 42515-1-95-513-4 after 10000ms)'

JericoPablo, 2024-12-12:
i rolled back to 1.38.0-1 since then its working, but there is already 1.42.0-2 out. Did anyone here with same problems tried it out yet?

tommyguuuun, 2024-12-23:
same problem here…

tommyguuuun, 2024-12-31:
> i rolled back to 1.38.0-1 since then its working, but there is already 1.42.0-2 out. Did anyone here with same problems tried it out yet?

How did you roll back? I have the dame problems with all my aqara radiators and it doesn‘t seem there will be a fix for this soon…

JericoPablo, 2025-01-21:
> > i rolled back to 1.38.0-1 since then its working, but there is already 1.42.0-2 out. Did anyone here with same problems tried it out yet?
> 
> How did you roll back? I have the dame problems with all my aqara radiators and it doesn‘t seem there will be a fix for this soon…

sorry didnt see ur msg, Settings, Backups, then choose the backup before u updated.

wtkchampion, 2025-02-21:
2.1.1 this is still a problem

error 2025-02-21 11:24:05z2m: Publish 'set' 'occupied_heating_setpoint' to 'Bedroom Radiator' failed: 'Error: ZCL command 0x44e2f8fffe1a5ba7/1 hvacThermostat.write({"occupiedHeatingSetpoint":2000}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (ZCL command 0x44e2f8fffe1a5ba7/1 hvacThermostat.write({"occupiedHeatingSetpoint":2000}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Delivery failed for '9312'.))'

Dangersohn, 2025-03-04:
I get the exect same error:

`error 2025-03-04 13:50:49z2m: Publish 'set' 'occupied_heating_setpoint' to 'Schlafzimmer Heizung' failed: 'Error: ZCL command 0x74057c000000c772/1 hvacThermostat.write({"occupiedHeatingSetpoint":600}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (ZCL command 0x74057c000000c772/1 hvacThermostat.write({"occupiedHeatingSetpoint":600}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Delivery failed for '47369'.))'`

Zigbee2MQTT version :2.1.3
Coordinator type: EmberZNet
Coordinator revision: 7.4.4 [GA]

ondrejmo, 2025-04-25:
Similar issue, but with Z2M `v2.2.1`, firmware ember 7.4.4 and Sonoff Zigbee dongle plus v2.

closed 2025-07-03 by github-actions[bot]